morrownr/USB-WiFi #324: ALFA AWUS036ACHM Range & LED Problems

2goese, 2023-10-15:
So You're telling about first hand experience with this adapter, yes it's true **BUT** only on supported OS listed in [here](https://docs.alfa.com.tw/Support/Linux/MT7610U/).

I use Arch linux, unfortunately it's not listed there, I've already explain about the issues [here](https://bbs.archlinux.org/viewtopic.php?id=289557).

I've updated the firmware using [this](https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md) guide step-by-step even though there's already `mt7610u.bin.zst` in `/lib/firmware/mediatek` directory.

Am i missing something?

bjlockie, 2023-10-15:
Does the led work and does it have good range on the listed distros?

Are there any module parameters to enable the led?
I think the modinfo xxx will show you.

Maybe the default txpower is low and that affects the range.
I don't know the command to check it.

2goese, 2023-10-15:
> Does the led work and does it have good range on the listed distros?

Yep no issues at all

Tried `modinfo mt76x0u ` with output: 
```
filename:       /lib/modules/6.5.7-arch1-1/kernel/drivers/net/wireless/mediatek/mt76/mt76x0/mt76x0u.ko.zst
license:        GPL
firmware:       mediatek/mt7610u.bin
firmware:       mediatek/mt7610e.bin
srcversion:     9BE8EDA0752A3110CF9AB7E
alias:          usb:v0E8Dp7650d*dc*dsc*dp*icFFisc02ipFFin*
alias:          usb:v0E8Dp7630d*dc*dsc*dp*icFFisc02ipFFin*
alias:          usb:v2357p0105d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0123d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0079d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pC711d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p806Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v293Cp5702d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v057Cp8502d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0951d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p7610d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0586p3425d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3D02d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB31d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0075d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p17DBd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p17D1d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp760Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp761Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pB711d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA711d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E8Dp7610d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v13B1p003Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp7610d*dc*dsc*dp*ic*isc*ip*in*
depends:        mt76,mt76x02-lib,mt76-usb,mt76x0-common,mac80211,mt76x02-usb
retpoline:      Y
intree:         Y
name:           mt76x0u
vermagic:       6.5.7-arch1-1 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        20:28:E7:B1:AE:78:54:88:41:13:D6:4A:5B:7C:50:6D:79:3E:96:6E
sig_hashalgo:   sha512
signature:      30:66:02:31:00:DB:2E:53:13:9F:A3:56:B4:50:9F:65:43:F0:10:EA:
                FD:97:8E:48:4D:04:72:7B:8A:F0:7B:6C:6F:6D:2C:40:1D:44:FB:4E:
                B7:21:76:7B:EB:1F:4E:53:34:57:B1:B7:E2:02:31:00:DB:4F:6E:09:
                AA:68:FB:28:F9:BC:BE:72:1D:74:F3:87:22:60:3E:98:BA:A4:26:ED:
                A9:F4:6C:B7:EC:37:C5:03:9B:56:B2:06:51:BB:C5:96:C8:AA:AC:48:
                2E:09:10:AC
``` 
But still no clue about LED

> I don't know the command to check it.

maybe
```
iwconfig wlan0 | grep Tx-Power
          Bit Rate=72.2 Mb/s   Tx-Power=2 dBm
```

bjlockie, 2023-10-15:
Try modinfo on the other modules.
Look for any that have parameters.
Look for parm.

$ modinfo mt76_usb
filename:       
/lib/modules/6.1.21-v8+/kernel/drivers/net/wireless/mediatek/mt76/mt76-usb.ko.xz
license:        Dual BSD/GPL
author:         Lorenzo Bianconi ***@***.***>
srcversion:     AF367BA27120C1ECD0D8DFD
depends:        mt76
intree:         Y
name:           mt76_usb
vermagic:       6.1.21-v8+ SMP preempt mod_unload modversions aarch64
parm:           disable_usb_sg:Disable usb scatter-gather support (bool)

2goese, 2023-10-15:
I found it on the same module you just listed, now what should i do with the parm?

morrownr, 2023-10-15:
Hi @2goese 

I'll talk about the range first. I made a range test using a few adapters a couple of years ago. The document is not on the Main Menu currently as it dated so I need to do a new test. Going into the test I did not realize that the ACHM had the range that it does so the data caught my eye. The test is here:

https://github.com/morrownr/USB-WiFi/blob/main/home/Performance_Comparison.md

This is a relative test in that the specific numbers only mean something when compared to other adapters that were tested using the same conditions. The data that I keyed in on for range is Link Quality and Signal Level. I looked at iperf3 for same class chipset. I need to put a plan together to do an updated. Maybe I can knock that out by the end of the year.

About the LED: I need to ask the devs but I do not think LED support is in the modern Linux wifi stack. I can't think of any wifi 5 and later usb wifi adapters where the in-kernel driver is supporting an LED, if the adapter has one. Let me do some testing and checking. There is a manual way to start LED support for the mt7612u chipset via debugging. Alfa, with its WiFi 6e adapters that it released this year has LEDs that work but the LEDs are not driver controlled. The LEDs are actually cool. If there is power, the LED is on but they slightly recessed the LEDs and made the light level very low. You can see it but it is not bright enough to bother you if you are trying to sleep.

My experience with the ACHM is a good one. It has proved to be a very dependable adapter that has no thermal problems and huge uptime. If the LED worked, I'd be looking for a way to turn it off so that but if it is important to you, maybe a solution can be found... maybe something like for the mt7612u but we would need to talk to the devs.

Nick

bjlockie, 2023-10-15:
I am looking for a parm related to the led, not the scatter gather.
It sounds like there isn't one from what Nick said so I don't know how 
other distros control the led.

Maybe run this command on archlinux and a distro that has better range.

$ iw wlp0s20f3 info
Interface wlp0s20f3
         txpower 22.00 dBm

See if the default txpower is different.

Also check the country between the one that works and archlinux:
$ iw reg get

2goese, 2023-10-16:
@morrownr 

Thanks for the information.

About the range test. Does it still valid with the non supported OS?

Regarding the LED. I would appreciate if you could find a solution, I like it blinking.

@bjlockie 

> $ iw wlp0s20f3 info

uhhh... nope wrong command. I think it's `iwconfig`

> $ iw reg get

This one i got a bunch of help command spreading all over my screen

Thanks for the suggestions

morrownr, 2023-10-16:
@2goese 

> About the range test. Does it still valid with the non supported OS?

Should still be valid. Non-supported in many cases just means resellers don't want to be bothered with paying for support to learn the details of several different distros and it is a valid issue. I have built up a small lab with test systems over the years, I have several distros on the systems so I can test the drivers I work on with multiple distros. Different ranges with different distros is not something that has caught my attention.

> For me range seems abnormal here in Arch.

Tell me what you are looking at that is telling me the range in different in Arch.

> Uhhh... nope wrong command. I think it's iwconfig

iwconfig has been depreciated for years, Please use iw. What are you trying to do?

> $ iw reg get

Here is what I get:

$ iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

bjlockie, 2023-10-16:
`$ uname -a`

One of mine is:
```
$ iw reg get
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

tugoese, 2023-10-16:
Replying with my 2nd account 

@morrownr 

For the range, I just feels something different on Windows graph bar than in Linux `nmcli` graph bar.
For example my neighbours network who lived approximately 7m away from my house got 3 bars on Windows, 1 on Linux. The connection is unreliable though, sometimes it got disconnected too. Should i trust the display bar?

About `iwconfig` and `iw` (I'm not really familiar with it)
I usually like to view status of my adapter and also not forgetting `ifconfig` for IP and MAC. Also lot of wifi hacking tutorial still use those commands.

```
~ iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)
```
Is there something wrong?

morrownr, 2023-10-16:
> For example my neighbours network who lived approximately 7m away from my house got 3 bars on Windows, 1 on Linux.

This is a test where you are at your neighbors house with your laptop connecting to your wifi router? I'm trying to figure out the exact situation here, Did you check to see what channel was in use with Windows and what channel was in use with Linux? Did you do any throughput testing.

A handy utility that I use to help when testing is wavemon:

On Debian the installation would be:

$ sudo apt install wavemon

To run:

$ wavemon

It would also be handy if you did a speed test with both Windows and Linux to see where we are. Something like:

https://testmy.net/

> I usually like to view status of my adapter and also not forgetting ifconfig for IP and MAC. Also lot of wifi hacking tutorial still use those commands.

I know that a lot of tutorials use iwconfig but many tutorials just recycle old stuff to save effort. Here us the linux.org informational link:

https://wireless.wiki.kernel.org/en/users/documentation/iw

There are a lot of good tutorials about 'iw' available. The most often use is probably the following command:

$ iw dev

which shows the wifi interfaces.

> Is there something wrong?

Possibly so. You appear to be running global defaults which is usually not desirable. Were you connected to the internet when you ran $ iw reg get ?

I may need to round you up a tutorial regarding the regdb. It is possible that Arch is messing this up.

morrownr, 2023-10-16:
Also, what is your country if you don't mind telling us?

morrownr, 2023-10-17:
> Link Quality=45/70 Signal level=-65 dBm.

This is a marginal situation. You may be able to get a usable connection and maybe not. It won't be a fast connection.

> Link Quality=70/70 Signal level=-31 dBm.

This is an outstanding situation where you should have no problem with the connection or speed as long as congestion is not too bad.

> Really bad for 5Ghz maybe because my router is not that good or maybe my ISP doesn't give me enough bandwidth.

This speed check on the internet gives me some ideas but internet speed checks are not optimal for what we are going. If you have a RasPi4B, we could setup a iperf3 server, it is easy, and we could check via your lan.

Speaking of your router: Are you familiar with how to access the settings and how to change the settings?

> Same thing happens on Windows. Only the range seems better, for example on Linux 5Ghz network I got only 2-3 bars on Windows the bar maxed out.

We need to determine if this is a perceived difference or a real difference. Hopefully we can do that.

> About iw I'm still trying to get used to it.

You will get used to it.

Not showing a country code with $ iw reg get is a little concerning and could be a problem. Let's mess with it:

$ iw reg set ID

Then run:

$ iw reg get

...and post the results.

I need to look in the regdb to see what is allowed in ID.

What Linux distro and version are you using?

2goese, 2023-10-17:
> If you have a RasPi4B, we could setup a iperf3 server, it is easy, and we could check via your lan.

I don't have a RasPI :(
I also don't have any experience with LAN :(
My interest is mostly wireless, but maybe in the future.

> Speaking of your router: Are you familiar with how to access the settings and how to change the settings?

A little bit only some that I’m already familiar with, some router have different settings.

> We need to determine if this is a perceived difference or a real difference. Hopefully we can do that.

I'll wait then, because it's the main issue aside from the LED.

> You will get used to it.

Hopefully.

` ~ sudo iw reg set ID`
```
~ iw reg get   
global
country 00: DFS-UNSET
    	(2402 - 2472 @ 40), (6, 20), (N/A)
    	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
    	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
    	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
    	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
    	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
    	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
    	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```
Nope nothing's changed.

> What Linux distro and version are you using?

```
~ uname -r
6.5.7-arch1-1
```

morrownr, 2023-10-17:
> My interest is mostly on wireless, but maybe in the future.

I'm not trying to turn you into a lan administrator, just trying to get more accurate test results but we are good.

Something does seem messed up but it is not clear what.

2goese, 2023-10-18:
No offense.

I manage to change the country code after installing `wireless-regdb` then reboot, here's the results:

```
~ iw reg get           
global
country CN: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5350 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5725 - 5850 @ 80), (N/A, 33), (N/A)
        (57240 - 59400 @ 2160), (N/A, 28), (N/A)
        (59400 - 63720 @ 2160), (N/A, 44), (N/A)
        (63720 - 65880 @ 2160), (N/A, 28), (N/A)

phy#1
country CN: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5350 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5725 - 5850 @ 80), (N/A, 33), (N/A)
        (57240 - 59400 @ 2160), (N/A, 28), (N/A)
        (59400 - 63720 @ 2160), (N/A, 44), (N/A)
        (63720 - 65880 @ 2160), (N/A, 28), (N/A)
```

```
~  iw reg set ID 
 
~ iw reg get        
global
country 98: DFS-UNSET (ALFA AWUS036ACHM)
        (2400 - 2483 @ 40), (N/A, 20), (N/A), NO-OUTDOOR
        (5150 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5725 - 5825 @ 80), (N/A, 23), (N/A), NO-OUTDOOR

phy#1
country CN: DFS-FCC (TP-LINK WN722N V1)
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5350 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5725 - 5850 @ 80), (N/A, 33), (N/A)
        (57240 - 59400 @ 2160), (N/A, 28), (N/A)
        (59400 - 63720 @ 2160), (N/A, 44), (N/A)
        (63720 - 65880 @ 2160), (N/A, 28), (N/A)
```

morrownr, 2023-10-18:
> country CN: DFS-FCC (TP-LINK WN722N V1)

Canada?

> global
country 98: DFS-UNSET (ALFA AWUS036ACHM)

So the ACHM is set to global default and the &22 is set to Canada. Yup. Something is messed up.

I'm pondering this. In this day in time, this setup should be automatic but something is not right. The 722 may be hardcoded in eeprom to Canada and we cannot do much about that. The ACHM does not appear to be hardcoded, which is good, but it is not getting the proper info from the kernel.

Are you locked into using Arch or would you consider an alternative distro?

2goese, 2023-10-19:
It's China probably set to default, I could use ID though but it won't last until reboot.

> So the ACHM is set to global default

This I had different result than the other one:

```
~ iw reg get
global
country ID: DFS-JP (ALFA AWUS036ACHM)
   	 (2400 - 2483 @ 40), (N/A, 26), (N/A), NO-OUTDOOR
   	 (5150 - 5350 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
   	 (5725 - 5825 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
```

> Are you locked into using Arch

Probably, I just recently migrated from Windows and I'am still learning the way Arch works.

> or would you consider an alternative distro?

Maybe if using a virtual machine?

morrownr, 2023-10-19:
> It's China probably set to default, I could use ID though but it won't last until reboot.

Yes. Canada is CA so I had that wrong. Why we are seeing CN is not clear but it may be a that CN is set in eeprom. A lot of things have been messed up over the years due to a lot of floundering over the years regarding exactly how to handle the laws regarding wifi over the world. 

> country ID: DFS-JP (ALFA AWUS036ACHM)

Your ACHM appears to be working the way it should.

You can download and check the regdb to make sure what you are seeing is the correct info for ID:

https://kernel.googlesource.com/pub/scm/linux/kernel/git/sforshee/wireless-regdb/

You not have needed to download the wireless file you downloaded to set the country code because things don't work that way anymore and have not for a few years. It seems the ACHM is working as it should and the 722 is locked into an incorrect country code.

> I'am still learning the way Arch works...

I'm probably going to catch the wrath of Arch users when I say this but learning Linux with Arch is not what I would suggest to the those starting out on Linux. I have numerous distros installed in my lab and to test "Arch" I use Manjaro which is based on Arch. If you were to ask me what you would suggestion as a distro to start with:

https://linuxmint.com/download_lmde.php

LMDE 6 is based on Debian 12. It is more than just a good starter distro... I use it on my main dev box. It is extremely stable and easy to use.

Sorry for the extra discussion. To get back on track, where are we? What problem are we trying to solve?

morrownr, 2023-10-19:
> A lot of things have been messed up over the years due to a lot of floundering over the years regarding exactly how to handle the laws regarding wifi over the world.

Let me clarify: This comment is not intended just for Linux. It is intended for Wireless in general and applies to Windows and all other operating systems. A lot of modernization has had to take place with wireless over the last several years and there are products out there that will cause problems.

2goese, 2023-10-19:
Why should I care about the regulatory domain thing, it shouldn't affect the perfomance right?

You're getting offtopic here

morrownr, 2023-10-19:
> Why should I care about the regulatory domain thing, it shouldn't affect the perfomance right?

It can have everything to do with performance. With the wrong regulatory information or simply the worldwide default, you almost certainly get will get less than optimal performance. Without that sorted out, we will have a hard time working toward good performance. 

> You're getting off-topic here

I'm trying to help. I know a lot about many distros but I am not an expert on Arch. It might be a good idea for you to post on an Arch forum.

> Please focus on the Range and LED. I need solution

Try a different distro. This performance could be an Arch problem. Arch needs to be eliminated as a source of the problem.

I can't solve the LED problem as the driver simply does not support the LED. I can ask the devs why this is and if it may change.

2goese, 2023-10-19:
The optimal choice for me are either CN or ID, but sometimes ID do get a buffering when I streaming on 1080p, I haven't got any problem with CN, gonna try Singapore this time because it's the closest to my country.

> It might be a good idea for you to post on an Arch forum.

I think you should scroll to the top, already posted there.

> Arch needs to be eliminated as a source of the problem.

This should be at a last resort.

Based on previous test I did on Windows, it might show different range on Arch but in fact the speed test are exactly the same on both OS, don't know why.

> I can't solve the LED problem as the driver simply does not support the LED. I can ask the devs why this is and if it may change.

Well it's good to know that, I'm waiting for the fix.

2goese, 2025-05-20:
The problems still persist confirmed in Kali