morrownr/USB-WiFi #326: Problems with the EDUP MS8551 USB WiFi Adapter for PC - High Gain 6dBi Antenna

DealBreacker, 2023-10-19:
The drivers for this adapter seems to not be working.. (Working with OpenWRT O.S). Go the adapter in today, plugged it in, installed the firmware and mt7601u.bin, proberly in /lib/firmware/mediatek, and the firmware refuses to load. Continuously receiving a "Direct firmware load for mt7601u.bin failed with error -2" followed with "probe of 1-1.1:1.0 failed with error -12". Did some extensive research, seems to just be a faulty driver + hardware issue known for this adapter. If anyone has run into this issue and found out how to fix is please let me know.

morrownr, 2023-10-19:
@DealBreacker 

>  it is driving me crazy and I might just have to return the adapter :(.

Don't let it driver you crazy.

I have not tested my Edup with mt7901u chipset with OpenWRT. Why? Because the mt7601u driver does not support AP mode. For reference:

https://github.com/morrownr/USB-WiFi

See Main Menu item 5. USB WiFi chipset information for Linux.
See Main Menu item 2. Plug and Play List. Go to mt7601u chipset section.

With that said, I think your little Edup adapter is almost the perfect little plug and play adapter for a wireless toolkit as it does not cost much and sometimes things happen and you need something that just works in managed mode to help get things going again.

I have used several adapters with OpenWRT and can suggest some that work well. Keep in mind, you should not need to manually install the firmware for these adapters as it should install when you install the driver. The best chipsets are mt7921au, mt7612u and mt7610u.

I'll throw a couple of ideas out:

https://www.amazon.com/dp/B08NPX2X4Z/?tag=pandaw-20

https://www.aliexpress.us/item/3256804283254522.html?gatewayAdapt=glo2usa

Let me know if you need further advise.

Nick

DealBreacker, 2023-10-19:
Ahhhh, I made a mistake when I was buying the edup adapter then! :P I'm going to keep it anyway, since it was only $8. Thank you for the clarification!

P.S: How long normally does AliExpress take to ship things? Do they overestimate or underestimate their shipping time? If you don't know then just ignore me!

morrownr, 2023-10-20:
> Do they overestimate or underestimate their shipping time?

Based on the comments I have seen over the last year, I'd say most of the time the adapter arrives on time or early.

dubhater, 2023-10-22:
Error -2 appears to be „No such file or directory”. On my (Arch Linux) system, the firmware is in ``/lib/firmware/mediatek``. You wrote ``/lib/firmware/MediaTek``, which is not the same. The case matters.

morrownr, 2023-10-22:
@dubhater 

Can you be specific about where you see '/lib/firmware/MediaTek'?

I am well aware that case matters but sometimes mistakes are made.

Nick

dubhater, 2023-10-22:
It's in @DealBreacker 's original post:

> The drivers for this adapter seems to not be working.. (Working with OpenWrt O.S). Go the adapter in today, plugged it in, installed the firmware and mt7601u.bin, properly in /lib/firmware/MediaTek, and the firmware refuses to load.

morrownr, 2023-10-22:
@dubhater 

I see. I missed that. Good eyes.

@DealBreacker 

Do you understand what @dubhater is saying?

DealBreacker, 2023-10-23:
Yeah I get what he is saying. I will check when I get home if I wrote that down wrong or not. Thanks for the info! When I figure it out I'm going to send it here to let y'all know.

shyguyCreate, 2024-09-14:
@DealBreacker was the error caused by `/lib/firmware/MediaTek` in uppercase?

morrownr, 2024-09-14:
@shyguyCreate 

When I saw your msg a few minutes ago and reread this thread, I said to myself, something is amiss here. I seem to have overlooked the probable problem last year when this thread started:

Not all Mediatek firmware goes in `/lib/firmware/mediatek". you can look in the driver as a last resort as it will have code that shows where it will look for the firmware. The proper place for mt7601u firmware is:

`/lib/firmware`

as in...

`/lib/firmware/mt7601u.bin`

In fact, and for future reference for anyone that happen to read this thread, the firmware for the new WiFi 7 generation mt7925 chips is:

`/lib/firmware/mediatek/mt7925`

This is noted in the Firmware menu item on the Main Menu. However, I do not address the mt7601u so I probably should add it to the Firmware guide.

Hope this helps and let me also add that the mt7601u driver does not support AP mode. I do not know why but wish it did. It is however a really good managed mode driver. Very solid and stable.

morrownr, 2024-09-14:
FYI: I have now updated the Firmware guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

If anyone wants to check my work, please do and let me know.

shyguyCreate, 2024-09-15:
> https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

@morrownr  how did you found out the path the driver was looking for the firmware?

morrownr, 2024-09-15:
@shyguyCreate 

> how did you found out the path the driver was looking for the firmware?

The primary reason I created this site was not to discover where firmware files should be place, it was to improve the information that is available to Linux users regarding USB WiFi. However, in doing that you learn a lot of things. I have been maintaining some Realtek out-of-kernel drivers here and I keep an eye on the kernel wifi programming. 

What I did in this case to save time was I went and found where the firmware file was in a known good distro (Debian 12). I can dig into the kernel code and show you if you wish.

shyguyCreate, 2024-09-15:
That would be great, I tried to find it but not good at C.

shyguyCreate, 2024-09-16:
@morrownr I was looking at the README code and it seems strange that you remove the number 3 from the list. https://github.com/morrownr/USB-WiFi/commit/317d5d5ad9b7846d355a055128d82a067b53233a

morrownr, 2024-09-16:
@shyguyCreate 

Hmmm... in looking at the current readme, menu item 3 is there. One of the problems with using github for essentially what is a editor issues is that the merged patches don't always make sense.

shyguyCreate, 2024-09-17:
When github renders the readme the item 3 is there, but in the source code there is no item 3, it does not affect anything but for consistency sake.

shyguyCreate, 2024-09-28:
> Not all Mediatek firmware goes in `/lib/firmware/mediatek". you can look in the driver as a last resort as it will have code that shows where it will look for the firmware. The proper place for mt7601u firmware is:
> 
> `/lib/firmware`
> 
> as in...
> 
> `/lib/firmware/mt7601u.bin`

I can confirm this information, in **Arch** there is a symlink in the `linux-firmware` package `/lib/firmware/mt7601u.bin.zst -> mediatek/mt7601u.bin.zst`.

To prove the **location of the firmware** first I did:

```
rm /lib/firmware/mt7601u.bin.zst
mv /lib/firmware/mediatek/mt7601u.bin.zst /lib/firmware/mediatek/mt7602u.bin.zst
```

I connected the USB and got an error.

<details>
<summary>No firmware nor symlink</summary>

```
[  617.214370] usb 2-3: new high-speed USB device number 7 using xhci_hcd
[  617.483767] usb 2-3: New USB device found, idVendor=148f, idProduct=7601, bcdDevice= 0.00
[  617.483775] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  617.483779] usb 2-3: Product: 802.11 n WLAN
[  617.483781] usb 2-3: SerialNumber: 1.0
[  617.864579] usb 2-3: reset high-speed USB device number 7 using xhci_hcd
[  618.007862] mt7601u 2-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  618.008944] mt7601u 2-3:1.0: Direct firmware load for mt7601u.bin failed with error -2
[  618.009372] mt7601u 2-3:1.0: probe with driver mt7601u failed with error -2
[  618.009453] usbcore: registered new interface driver mt7601u
```

</details>

After that I move back the firmware `mv /lib/firmware/mediatek/mt7602u.bin.zst /lib/firmware/mediatek/mt7601u.bin.zst` and still got an error after replugging.

<details>
<summary>Firmware but no symlink</summary>

```
[  850.614337] usb 2-3: USB disconnect, device number 7
[  866.380983] usb 2-3: new high-speed USB device number 8 using xhci_hcd
[  866.650719] usb 2-3: New USB device found, idVendor=148f, idProduct=7601, bcdDevice= 0.00
[  866.650735] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  866.650742] usb 2-3: Product: 802.11 n WLAN
[  866.650746] usb 2-3: SerialNumber: 1.0
[  866.774608] usb 2-3: reset high-speed USB device number 8 using xhci_hcd
[  866.917307] mt7601u 2-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  866.917832] mt7601u 2-3:1.0: Direct firmware load for mt7601u.bin failed with error -2
[  866.918087] mt7601u 2-3:1.0: probe with driver mt7601u failed with error -2
```

</details>

At last I remade the symlink `ln -s mediatek/mt7601u.bin.zst /lib/firmware/mt7601u.bin.zst` and got the USB working after reconnecting.

<details>
<summary>Firmware and symlink</summary>

```
[ 1035.223402] usb 2-3: USB disconnect, device number 8
[ 1035.517642] usb 2-3: new full-speed USB device number 9 using xhci_hcd
[ 1050.900870] usb 2-3: new high-speed USB device number 10 using xhci_hcd
[ 1051.170957] usb 2-3: New USB device found, idVendor=148f, idProduct=7601, bcdDevice= 0.00
[ 1051.170973] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1051.170980] usb 2-3: Product: 802.11 n WLAN
[ 1051.170984] usb 2-3: SerialNumber: 1.0
[ 1051.294697] usb 2-3: reset high-speed USB device number 10 using xhci_hcd
[ 1051.437038] mt7601u 2-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[ 1051.463115] mt7601u 2-3:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[ 1051.857694] mt7601u 2-3:1.0: EEPROM ver:0c fae:00
[ 1052.056932] ieee80211 phy3: Selected rate control algorithm 'minstrel_ht'
[ 1052.074747] mt7601u 2-3:1.0 wlp0s20u3: renamed from wlan0
```

</details>

morrownr, 2024-09-29:
@shyguyCreate 

> I can confirm this information, in Arch there is a symlink in the linux-firmware package /lib/firmware/mt7601u.bin.zst -> mediatek/mt7601u.bin.zst.

My work maintaining some of the out-of-kernel drivers over the last several years causes me to look with a questioning eye at distros that do this type of thing... and things like this are common and cause inconsistencies which can lead to problems and troubleshooting issues. I would prefer distros contact driver maintainers to sort out any of their concerns instead of doing things like this. We need more consistency amount distros, not less.

shyguyCreate, 2024-09-30:
Although it does not follow a standard, I am glad that it is plug-and-play.