morrownr/USB-WiFi #329: Can anyone help to add a new device ID (VID/PID) of a nineplus mt7921au wifi adapter to official  kernel?

pcslide, 2023-11-06:
Under Windows 10/11, it's a plug and play device, and is identified as "Nineplus Wi-Fi 6/6E Wireless USB Lan Card".
Its device ID is  USB\VID_363E&PID_7961&REV_0100
I manually added device ID and it worked on kernel 6.5

morrownr, 2023-11-06:
Hi @pcslide 

I'll try to get you some help. I'm busy this week and i don't have a guide up for this but I wish I did. Maybe someone will read this and create one. In the meantime, read the following to get an idea:

https://wireless.wiki.kernel.org/en/developers/documentation/submittingpatches

It does not include everything you need to know so will need me or someone to help this first time.

Nick

morrownr, 2023-11-06:
Also, can you provide a link to the product? to the place where you bought it and to the company product site?

bjlockie, 2023-11-06:
> Under Windows 10/11, it's a plug and play device, and is identified as "Nineplus Wi-Fi 6/6E Wireless USB Lan Card". Its device ID is USB\VID_363E&PID_7961&REV_0100 I manually added device ID and it worked on kernel 6.5

Anyone know how Windows knows which driver goes with which device?
they don't do have to add a device id.

morrownr, 2023-11-06:
> Anyone know how Windows knows which driver goes with which device?

Windows... what is this Windows? Is it a Linux distro that I have not heard about?

Now back to reality.

@bjlockie 

Is that you volunteering to help the OP submit a patch to linux-wireless?

bjlockie, 2023-11-06:
> Windows... what is this Windows? Is it a Linux distro that I have not heard about?

I wondered why Linux needs patches to support new devices and Windows doesn't.
Linux is usually way better but Windows seems to have some smarts for new USB devices that I wish Linux had.

pcslide, 2023-11-07:
> Also, can you provide a link to the product? to the place where you bought it and to the company product site?

As it sells on Taobao's Workshop( you could see it as Alibaba's Amazon essential equivalent), and the item bears no brand on itself and on it's pack. The idea is to let the

pcslide, 2023-11-07:
I'm afraid Taobao's Workshop is not accessible without an account.  Here is the product summary( it was originally in Chinese).
[image]

pcslide, 2023-11-07:
Does it work with HUB?  I have three hub here is the results.
| Brand    | Chips |  Ports  |  Result  |  
| -------- | ------- | ------- | ------- |
| SSK  | GL3520    |  USB 3.0 |  can not detect hardware properly |
| ICSI |  GL3520 + GL3523    | USB 3.0 |  works properly |
| $1 item with no brand   |  unknown    | USB 3.0 + USB 2.0 | works properly on USB 2.0 ports,  can not connect to AP when adaptor is on USB 3.0 port|

morrownr, 2023-11-10:
@pcslide 

> I don't know if it is against community rules to give a direct link to buy this wireless card. If it is not, I will add a link.

Adding links is actually encouraged.

My opinion based on what you have posted is that this adapter is being made by some manufacturer as a "white box" product where sellers can buy it and sell it as a unbranded product or brand it as their own. This is fairly command in the market. This can make it a challenge for us to narrow the product down something we can recommend.

I see in the picture above the words "driver free". That usually means the adapter is a multi-state adapter which rules out me putting it in the Plug and Play list.

Have you made any progress toward how to submit a patch to include the vid/pid? If you used the manual method to include it and it is working, there is not need to be a hurry because a properly done patch can be taken back as far as kernel 6.1 in a case like this.

@morrownr

pcslide, 2023-11-11:
@morrownr 
It is **NOT** a multi-state adapter.  I suppose it is "driver free" in a sense that windows 10/11 will automatically download and install the driver for you.  However, it seems a little absurd, as you need a internet connection for it to be "driver free".

morrownr, 2023-11-11:
@pcslide 

> It is NOT a multi-state adapter.

That is good to hear. Have you deactivated usb_modeswitch to double check?

How to deactivate usb_modeswitch:

$ sudo nano /etc/usb_modeswitch.conf

Change

DisableSwitching=0

to

DisableSwitching=1

Then reboot

Hey @russeree 

@pcslide has a new adapter that has a company vid/pid that needs to go into the kernel. I'm neck deep on a couple of things right now and don't have a guide for this and the posted guide on the kernel site is less than complete. Would you mind giving him a helping hand?

@morrownr

russeree, 2023-11-11:
I would love to submit a patch! 

With that said with certainty the VID 363E PID 7961 belongs to a mt7921au chipset? I would hate to submit a bogus patch. If time is acceptable I can order an adapter to do proper testing?

pcslide, 2023-11-12:
@russeree  Hi, from the driver automatically downloaded and installed by windows 11 when I plug the adapter in, I am pretty sure the adaptor belongs to MediaTek mt7921 series.  
[image]

It works flawlessly in station mode with this driver so far.  (It lacks Soft AP and WIFI-direct support under Windows)
 As for how well it behaves with Linux kernel driver, allow me more time to test it.

russeree, 2023-11-13:
@morrownr This response from Nineplus doesn't confirm the findings of members here. I however don't trust the support agent fully because they didn't disclose which other chipset that adapter uses. 

[image]

pcslide, 2023-11-13:
@russeree It looks the same as the one I bought,  but I don't think anyone can be 100 percent sure they use the same chip inside from the look🤔