morrownr/USB-WiFi #331: comfast cf953 random time crash on openwrt 23.05

ritech, 2023-11-09:
just speedtest or much client connect to ap  
kernel 5.15.127 x86/64

Fri Nov 10 04:11:31 2023 daemon.info hostapd: phy1-ap0: STA 72:bd:41:7c:12:7d IEEE 802.11: authenticated
Fri Nov 10 04:11:31 2023 daemon.info hostapd: phy1-ap0: STA 72:bd:41:7c:12:7d IEEE 802.11: associated (aid 2)
Fri Nov 10 04:11:31 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 72:bd:41:7c:12:7d auth_alg=open
Fri Nov 10 04:12:11 2023 kern.err kernel: [  417.365011] mt7921u 2-1:1.0: Message 00020002 (seq 6) timeout
Fri Nov 10 04:12:12 2023 kern.err kernel: [  417.594940] mt7921u 2-1:1.0: timed out waiting for pending tx
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618122] ------------[ cut here ]------------
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618124] WARNING: CPU: 0 PID: 10152 at kthread_park+0x87/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618161] Modules linked in: xt_FULLCONENAT pppoe ppp_async zstd pppox ppp_mppe ppp_generic mt7921u mt7921_common mt76_usb mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 bnx2x amdgpu zstd_decompress zstd_compress xxhash xt_time xt_tcpudp xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_helper xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_FLOWOFFLOAD xt_CT ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip tcp_bbr slhc r8168 r8125 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_syslog nf_flow_table nf_conntrack_netlink nf_conncount mlx5_core mlx4_en mlx4_core macvlan libcrc32c iptable_raw iptable_nat iptable_mangle iptable_filter ip_tables igc iavf i915 i40e gpu_sched forcedeth e1000e crc_ccitt compat bnx2 alx 8139too 8139cp ntfs3 evdev i2c_dev cryptodev xt_set
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618239]  ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ixgbe igbvf e1000 amd_xgbe mdio ip6_tunnel tunnel6 raid10 raid1 raid0 md_mod nls_utf8 nls_iso8859_1 nls_cp437 zram zsmalloc ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_ssse3 sha512_generic sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg hmac cmac authencesn authenc arc4 crypto_acompress uas sdhci_pltfm xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl igb vfat fat exfat button_hotplug tg3 realtek broadcom bcm_phy_lib mii
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618409] CPU: 0 PID: 10152 Comm: kworker/u2:0 Not tainted 5.15.127 #0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618414] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 11/12/2020
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618417] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618445] RIP: 0010:kthread_park+0x87/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618449] Code: 74 31 31 c0 5b 41 5c 5d c3 cc cc cc cc 0f 0b 4c 8b a3 90 05 00 00 a8 04 74 a8 0f 0b b8 da ff ff ff 5b 41 5c 5d c3 cc cc cc cc <0f> 0b b8 f0 ff ff ff eb d1 0f 0b eb cb 66 66 2e 0f 1f 84 00 00 00
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618451] RSP: 0018:ffffc90000fcfd78 EFLAGS: 00010202
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618454] RAX: 0000000000000004 RBX: ffff888004ea1d00 RCX: ffff888009807ba8
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618455] RDX: 0000000000000001 RSI: 0000000000000202 RDI: ffff888004ea1d00
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618457] RBP: ffffc90000fcfd88 R08: 0000000000003c00 R09: ffff888003f7a268
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618458] R10: 0000000000000000 R11: ffffffff822ae508 R12: ffff888009403000
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618460] R13: 0000000000000000 R14: 0000000000000020 R15: 0000000000000100
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618463] FS:  0000000000000000(0000) GS:ffff88803ec00000(0000) knlGS:0000000000000000
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618465] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618468] CR2: 00007f13341f3b7c CR3: 00000000035e2006 CR4: 00000000003706f0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618503] Call Trace:
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618510]  <TASK>
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618518]  ? show_regs.part.0+0x1e/0x24
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618549]  ? show_regs.cold+0x8/0xd
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618551]  ? __warn+0x74/0xf0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618569]  ? kthread_park+0x87/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618572]  ? report_bug+0x86/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618593]  ? handle_bug+0x38/0x90
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618606]  ? exc_invalid_op+0x18/0x70
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618608]  ? asm_exc_invalid_op+0x1b/0x20
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618615]  ? kthread_park+0x87/0xa0
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618618]  mt76u_resume_rx+0x301/0xd60 [mt76_usb]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618626]  ? wait_woken+0x60/0x60
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618633]  mt7921u_mac_reset+0x69/0x1c0 [mt7921u]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618639]  mt7921_mac_reset_work+0x85/0x140 [mt7921_common]
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618644]  process_one_work+0x1f8/0x360
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618654]  worker_thread+0x4e/0x410
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618657]  ? process_one_work+0x360/0x360
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618660]  kthread+0x128/0x150
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618662]  ? set_kthread_struct+0x50/0x50
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618665]  ret_from_fork+0x1f/0x30
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618676]  </TASK>
Fri Nov 10 04:12:12 2023 kern.warn kernel: [  417.618678] ---[ end trace 56a2ce3888ddc6b9 ]---
Fri Nov 10 04:12:12 2023 kern.info kernel: [  417.759857] mt7921u 2-1:1.0: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
Fri Nov 10 04:12:12 2023 kern.info kernel: [  417.759857]
Fri Nov 10 04:12:12 2023 kern.info kernel: [  417.770990] mt7921u 2-1:1.0: WM Firmware Version: ____010000, Build Time: 20230526130958

morrownr, 2023-11-09:
@ritech 

Is the adapter plugged directly into a usb port or are you using an extention cable or powered hub?

Have you tested on both USB2 and USB3 ports?

Details of the setup? 5 Ghz or 2 Ghz?

ritech, 2023-11-09:
directly plugged
usb3.0 port
5ghz on ax mode

ritech, 2023-11-09:
same problem on openwrt forum
https://forum.openwrt.org/t/mt7921u-ap-mode-crashes-on-23-05-0/175472

morrownr, 2023-11-09:
OpenWRT maintains a downstream of mt76 at this location:

https://github.com/openwrt/mt76

It might be better to report it there. First read the Issues and the Pull Requests, both open and closed to see what you can see.

Have you deactivated scatter-gather?

Have you tested in a USB2 port?

morrownr, 2023-11-09:
For what it is worth: I run a Alfa AXML (same chipset and driver as your adapter on my Zyxel wifi router and I am not seeing this. It could be hardware specific. Contact the other guy on the OpenWRT forum to see if you can find out what is common between your setups.

ritech, 2023-11-10:
echo 1 > /sys/module/mt76_usb/parameters/disable_usb_sg
kernel crash 
change to usb2.0 port
kernel crash

morrownr, 2023-11-10:
> VMware, Inc. VMware Virtual Platform/

On closer inspection of your original post, I see the above. Are you running OpenWRT in a VM?

If so, that is likely the source of your problems.

ritech, 2023-11-10:
The original fault was on my physical machine. I tested multiple kernel versions using a virtual machine to rule out kernel issues, but the results were not good.

morrownr, 2023-11-10:
> The original fault was on my physical machine. 

Tell me about your physical machine and the os?

ritech, 2023-11-10:
pc : j4125 / 4G RAM / Msata 64G / 4 x 2.5G i226
os : openwrt 23.5 kernel 5.15.137

morrownr, 2023-11-10:
If your os is OpenWRT, why are you running a VM?

> pc : j4125 / 4G RAM / Msata 64G / 4 x 2.5G i226

That tells me nothing. Is the cpu based on amd64, arm64 or something else?

It is hard to help you because I am having to dig the info needed one piece at a time.

If you are inexperienced, that is fine, learning is good but learning with an easier project is better at the beginning. Running OpenWRT in a VM on what may be an amd64 box is an advanced project. I might be able to help you decide on an easier project if you tell me what you are trying to do.

ritech, 2023-11-10:
This pc is currently in the company, and the virtual machine is my testing environment at home. I took the USB wifi adapter home and connected it to the virtual machine for testing. I ran openwrt 22.03.5, openwrt 23.05.0 and openwrt snapshot on the virtual machine. Compile and add kmod-mt7921u. All systems are x86/64 based.

ritech, 2023-11-10:
I found that it may be caused by insufficient USB power supply. I used a voltmeter to test 5v. It would drop during large data transmission, and then the USB device was lost.

morrownr, 2023-11-11:
Try this: (it turns off scatter-gather support)

Open a terminal (Ctrl + Alt + t)

sudo nano /etc/modprobe.d/mt76_usb.conf

add:

options mt76_usb disable_usb_sg=1

Save the file: Ctrl + X, Y, Enter,

ritech, 2023-11-11:
No improvement, I have switched to rt8812au, very stable

morrownr, 2023-11-11:
> I found that it may be caused by insufficient USB power supply. I used a voltmeter to test 5v. It would drop during large data transmission, and then the USB device was lost.

After pondering this issue, I have to wonder if your cf953 is faulty. Manufacturing errors happen. Have you tested it on a basic PC without a VM or OpenWRT?

I have a cf951 and it has what I'll call faults that may be manufacturing as nobody else seems to be reporting the errors. I'm not that comfortable with Comfast as they seem interested in making many adapters at a low price. That compares with Alfa that makes a few well done adapters with high quality components.

> I have switched to rt8812au, very stable...

If this is a hobby project, fine but if this is a project to bring a product online for your comany, be careful. The rtl8812au chipset is a generation 1 AC1200 chipset from 2013. There is no in-kernel driver and Realtek terminated support for the out-of-kernel driver in 2021. I have a repo for the driver up here. It is a very good chipset and the out-of-kernel driver is probably the best one Realtek has ever made but remember that the Realtek out-of-kernel drivers are not Linux Standards compliant drivers so as capabilities need to be added or tweaked, you may be stuck. Not a good thing. It might be wise to investigate another solution with an in-kernel driver.

If you want to tell me more about the project, I will try to advise you. What adapter capabilities do you need? IS AC1200 good enough?

@morrownr

ritech, 2023-11-11:
This is not a company project. It mainly provides routing and Internet access functions for 20 people. Considering that the Internet bandwidth is 200mbps, I think rtl8812au is fully capable. After checking the openwrt forum, there are various problems with mediatek's usb wifi chip. In addition, rtl8812bu rtl8812cu uses usb3.0 interface, and there is no ipk driver package on openwrt.

morrownr, 2023-11-11:
> After checking the openwrt forum, there are various problems with mediatek's usb wifi chip.

I'd be interested to know what those problems are. Links?

I've been using adapters with Mediatek chipsets for several years. 

> rtl8812bu rtl8812cu uses usb3.0 interface, and there is no ipk driver package on openwrt.

The rtl8812au uses a USB3 interface as well. I'm aware that there is no driver support in OpenWRT for numerous Realtek chipsets. Some of the OpenWRT devs are also Mediatek kernel devs so Mediatek is well supported. I'm not aware of any Realtek presence at OpenWRT.

 >  It mainly provides routing and Internet access functions for 20 people. 

You may not need OpenWRT for this project which would open up what drivers are available for use. Do you have an ethernet cable running from another router into the G30W? Is the G30W capable of running another os such as Debian 12?

I have an AP guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

I also have an example WiFi 6 hostapd.conf that is not included in the document but is a separate link in the AP section of the Main Menu.

This guide uses systemd-networkd and hostapd (which is the core of OpenWRT). The guide says Raspberry Pi OS but it works on other distros like Debian 12, just ignore the RasPi specific stuff.

Keep in mind that you need 2 adapters if you want to support 2 bands. Some of your users may want 2.4 GHz if they are located a fair distance from the AP. The guide supports dual band or even tri-band. All of the 6 Realtek drivers I have up here will work with this guide so you have a lot more flexibility. My RasPi4B is using the setup in this guide currently. My 2 adapters are Alfa ACM for 2.4 GHz but a Alfa ACHM would work as well. The other adapter for 5 Ghz (WiFi 6 and WiFi 5) is an Alfa AXML. Rock solid.

ritech, 2023-11-11:
I don't have any specific links in my collection. Most of them are about the MCU of the Mediatek chip not responding, as well as errors such as vendor. There are also quite a few bug reports in github.com/openwrt/mt76.
At present, I can only use OpenWrt because the device is in China and needs to jump over GFW. OpenWrt can provide many third-party plugins, and if I use Ubuntu or Debian, more operations are needed. Additionally, I used ubuntu 22.04 yesterday to directly drive mt7921au under kernel 6. x, but the same error was reported, so I increasingly suspect it is a hardware issue.

morrownr, 2023-11-11:
> Most of them are about the MCU of the Mediatek chip not responding, as well as errors such as vendor.

Keep in mind that Mediatek produces numerous wifi products and given the mutual support that OpenWRT and Mediatek have, it would be normal to see problem reports. I run a wifi router with a Mediatek mt7981soc. It is WiFi 6 and is impressive for a low cost router.

> There are also quite a few bug reports in github.com/openwrt/mt76.

I am aware. Keep in mind that mt76 supports the wireless for numerous chipsets, not just usb. It is supporting many soc's for routers. If there was a rtl88 repo at OpenWRT, we could compare... but there isn't. mt76 is downstream repo of the Linux kernel. It is a testing location. Patches flow back and forth between to the Linux kernel mt76.

> OpenWrt can provide many third-party plugins

Debian 12 would allow you to install basically anything.

> and if I use Ubuntu or Debian, more operations are needed. 

With Debian 12, it would be less than Ubuntu depending on the desktop of lack thereof but that only matters if it matters. Is the G30 a really low processor capability kit?

> I used ubuntu 22.04 yesterday to directly drive mt7921au under kernel 6. x, but the same error was reported, so I increasingly suspect it is a hardware issue.

You very well may have a bad adapter. It happens. Which kernel is in use in Ubuntu 22.04? Did you upgrade the firmware files for the adapter?

ritech, 2023-11-11:
Linux Ubuntu 6.2.0-36-generic #37~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Oct  9 15:34:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
WM Firmware Version: ____010000, Build Time: 20230526130958

morrownr, 2023-11-12:
You have the latest firmware for the mt7921au chipset so my bet at this point is that you have a faulty product. I've seen a lot of users here with cf-953ax adapters and most talk highly about the little adapter. My own opinion is that a project for work where you will have ~20 people using it as an AP, I would probably look at an adapter by a company that has a history of using good quality parts and has a record of products that work well and last a long time. A concern that I have with the 953ax in AP mode to several people is does it have the range. It is not designed to be a long range adapter. That has nothing to do with the chipset as the makers decide on the amp and antennas to be used.

In another issue here us a possible problem with use in AP mode so I'll look extra hard to see if I see any issues. If you want to follow along, it is the stickied issue and you will need to go to the bottom of the thread.

ritech, 2023-11-12:
I bought another comfast 953 and tested it. The fault was the same. A friend reminded me that although it was all mt7921 chips, the content of each manufacturer's eeprom settings was different. Maybe the problem was that I didn't find the mt7921 eeprom datasheet.

ritech, 2023-11-12:
But I found the eeprom manual for mt7628, https://mangopi.org/_media/mt7628_eeprom_guideline_v2_00.pdf

morrownr, 2023-11-12:
> A friend reminded me that although it was all mt7921 chips, the content of each manufacturer's eeprom settings was different.

A couple of years ago, a user reported a problem with a little adapter that uses the mt7612u chipset. It took us a couple of months to see what was happening. In the mt7610u section of the Plug and Play list is an adapter that uses the mt7610u chipset but under the favorable review for that adapter is a WARNING for users not to buy the version of this adapter that uses the mt7612u chipset due to a flaw in the company design. If the adapter happened to use usb 3, port 1, the results were bad. If using other ports it worked fine. The mt7612u chipset is not the cause of the flaw as many Linux users have adapters with the mt7612u chipset and the firmware/driver has been stable for years. In fact, the Alfa ACM is a really good adapter for AP mode. The Alfa ACHM is also a really good adapter for AP mode. Neither are as cheap as the cf-953ax but they are really solid and both work well in 24/7/365 situations.

bjlockie, 2023-12-04:
If it is a raspberry pi, disable scatter_gather

MEL1H, 2023-12-04:
Already applied but same result.

pi@md-ap:~ $ grep [[:alnum:]] /sys/module/mt76_usb/parameters/*
Y

bjlockie, 2023-12-04:
Try a different USB port.

morrownr, 2023-12-05:
> I have a AWUS036AXML which has mt7921u chipset. I have same issue with same message "mt7921u 2-1:1.0: timed out waiting for pending tx". That happens if I do some speedtests. It works on ax mode with good speed performance (I was using plug and play and it was 10mbps, I have installed driver from @morrownr 's guide and speed increased to 600 mbps). I also disabled logging usb which was a suggestion in AWUS036ACH topic but still it crashes. I hope it is a driver issue and they will patch it soon..

@MEL1H 

I've been very sick over the last week so have not been around. When I read the above report, and you may not realize this, it provides almost nothing that is of use in determining the problem. I'm saying this so as to provide information, not to throw any shade in your general direction. So, what was useful? The chipset and adapter. What needed to be provided?

- What distro are you using and what version?

- What kernel?

- What mode is your adapter in? Managed, Monitor, AP, P2P? This is really important in the the different modes have different code in the driver that is running.

What hardware are using? Is it ARM,ARM64, X86, amd64? Is to a RasPi4b or a Dell Desktop?

What were you doing that caused this issue to come to your attention? and what is the issue? Slow speed?

> I have installed driver from @morrownr 's guide ...

Well, @morrownr , does not have a driver for the mt7921au chipset anywhere because the only driver is in the Linux kernel. What guide are you talking about?

> but still it crashes.

Can you elaborate?

I also have an Alfa AXML. I am will to help you by trying to duplicate the problem but I do not have the information needed to know what setup I need to duplicate the problem.

Rule number 1 of troubleshooting, whether it is a jet engine or computer code, is first, duplicate the problem in a reliable manner. If that does not happen, the chances of a fix drop considerably.

Sir, please go forth and provide the information so that I can duplicate the problem. It will likely be best if you start a new issue that is specific to your case so please do so.

Regards,

@morrownr

morrownr, 2023-12-05:
@ritech 

Been sick but am trying to catch up...

> You have the latest firmware for the mt7921au chipset 

When I said that, it was true. It is no longer true as a Nov release is now available. My firmware guide gives specific instructions for OpenWRT.

Can you update me on your status?

@morrownr

Simon566, 2024-03-22:
Hi,

i do have the same issues , also on intel similar HW, with the ALFA stick !
https://github.com/morrownr/USB-WiFi/issues/410

im using an ODROID H3 , that s a N5105 intel Atom

:-(

its not the stick manufacturor or the power supply, we have a bug here

regards,
Simon

flipreverse, 2024-05-04:
I've just discovered a patch for this issue [fa3fbe6403](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/fa3fbe64037839f448dc569212bafc5a495d8219q?id=fa3fbe64037839f448dc569212bafc5a495d8219).
That patch was just applied to the mt7921**e** driver.