morrownr/USB-WiFi #334: ALFA AWUS036AXML Driver

SmallLamSam, 2023-11-16:
Hello everyone, did anyone here succeded to make the ALFA AWUS036AXML work on the kali linux 2023.3?

morrownr, 2023-11-17:
Hi @SmallLamSam 

When you say `work`, can you elaboate?

SmallLamSam, 2023-11-17:
by work I mean it was detected and I can see the card of the network manager and connect to wireless networks.

but the card was slow when turning on monitor mode, doing mdk4 with it was working sometimes and sometimes not, also the strenght of signal was not correct for many networks, that's all I have noticed until now.

ZerBea, 2023-11-20:
The driver is broken and the problem is reported here:
https://github.com/openwrt/mt76/issues/839

A test is here:
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-7567045

ZerBea, 2023-11-20:
@SmallLamSam 
it might be helpful to the developer if you confirm that the driver doesn't work for you and commit your configuration on upstream issue report, too:
https://github.com/openwrt/mt76/issues/839
The more information, the earlier we can expect a fix.

ZerBea, 2023-11-20:
@SmallLamSam 
Are you sure that monitor mode and packet injection has been working?
I went back
to kernel 6.5.1 (Debian kernel config) -> neither monitor mode nor packet injection is working
to kernel 6.1.21 (Raspbian kernel config) -> neither monitor mode nor packet injection is working

SmallLamSam, 2023-11-20:
> @SmallLamSam Are you sure that monitor mode and packet injection has been working? I went back to kernel 6.5.1 (Debian kernel config) -> neither monitor mode nor packet injection is working to kernel 6.1.21 (Raspbian kernel config) -> neither monitor mode nor packet injection is working

Yes they worked, I downloaded kali 2022.4
just don't install any driver connect the adapter and wait, use normal usb port not type C

I did returned the adapter to the seller, so I can't help from now on
this adapter is so overrated, many are having issues with wifi 6, so if wifi 6 doesn't work then it's useless.

ZerBea, 2023-11-20:
@SmallLamSam 
Thanks. Your information is helpful.
KALI 2022.4 is running kernel 6.0.0 which I haven't tested.