morrownr/USB-WiFi #335: Kernel headers work-around?

froginperil, 2023-11-17:
New computer.
I installed Fedora 39.
Kernels: 6.5.6 from installation medium - 6.5.11 in updates
The only kernel-headers available are 6.5.4 and the module installer doesn't work.

Making matters worse, Asus is shipping high end machines without an ethernet card and, of course, the internal wifi doesn't work in Linux. I can send the toy back to Amazon but would prefer to work out a solution.

Any ideas?

I have another machine running f38. Can I copy the module over and then run depmod?

bjlockie, 2023-11-17:
> Asus is shipping high end machines without an ethernet card and, of course, the internal wifi doesn't work in Linux.

What chipset?
I thought Linux supports most pci wifi.

morrownr, 2023-11-17:
> Kernels: 6.5.6 from installation medium - 6.5.11 in updates
The only kernel-headers available are 6.5.4 and the module installer doesn't work.

There is not really a work around. That is a fedora screw up.

I'm like bjlockie , what chipset are we talking about?

I use the following in a desktop (plug and Play, mt7922 chipset):

https://www.amazon.com/gp/product/B0C388Z522/ref=ppx_yo_dt_b_asin_title_o09_s02?ie=UTF8&th=1