morrownr/USB-WiFi #337: Run start-mon.sh with switches from command line

pigsfoot, 2023-11-19:
Is it possible to run the start-mon via command line using switches so I can enter it into a script, i need automate the process on startup

morrownr, 2023-11-20:
Hi @pigsfoot 

I designed the start-mon.sh script to be such that you can pull what you need and include it in your own script. I would have to check but there may be a few bash only items but that could be changed if needed. 

I've been sick recently and am still recovering but I can work with you on this project as I feel better and am caught up.

@morrownr

pigsfoot, 2023-11-20:
That would be great if you can help. Happy to test anything you come up with.

Ultimately i am looking to launch monitor mode via another bash file i currently run which also sets the interface to channel hop and scan the most popular 2.4 & 5Ghz channels