morrownr/USB-WiFi #338: News: New Firmware available for mt7921 and mt7922 chipsets

morrownr, 2023-11-27:
The guide for upgrading firmware is:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

Most will need to use section 2 as that section is about upgrading the firmware for mt7921 based usb wifi adapters.

```
$ ethtool -i wlx000000000000
driver: mt7921u
version: 6.5.0-0.deb12.1-amd64
firmware-version: ____010000-20231109190959
```

I haven't had time to note any additional features or fixes. If you upgrade and notice new features or fixes, please post in this thread to let us know.

@morrownr

kasinjsh, 2023-12-04:
As I cant compile newer kernel to my SBC it made no difference to stability whit kismet etc.

gordboy, 2024-02-01:
Finally got myself a proper USB wifi with MT7921au chipset, the Comfast CF-953AX, and a pair of right angle USB adaptors.

Updated my firmware, and turned off power saving in NetworkManager config. Works nicely, getting ~650 Mbps transfers on my LAN with sftp and such like. The chipset in the laptop gets a miserable 300 Mbps, so this is a great upgrade For A Few Dollars More (TM).

Running Ubuntu 22.04.3 LTS with HWE 6.5 kernel and I made a deb package for this with dpkg diversions, so I can revert to vanilla firmware and NM simply by uninstalling my package.

Good work Nick, and thanks for all your help and support over the years. I remember the bad old days of the RTL8812au and begging Realtek (or anyone else) for a login to get their driver, and then laboriously hammering it into shape,

Mediatek are a breath of fresh air. More power to them, and to yourself Good Sir.