morrownr/USB-WiFi #341: Lomubue M-1200M link goes to a possible scam

Pointedstick, 2023-11-29:
On the recommendation of this website, I purchased a Lomubue M-1200M antenna from [the provided link to Walmart](https://www.walmart.com/ip/Lomubue-M-1200M-Wireless-Network-Card-High-speed-Anti-interference-Driver-free-USB3-0-MT7612U-Dual-Band-WiFi-Transceiver-for-Router/731709450?athbdg=L1400).

Unfortunately I received a different device with a Realtek RTW8822CU chip, not the advertised Mediatek MT7612U. The seller reviews are all very bad, so it appears this product listing is not reliable and trustworthy, and I would recommend removing it.

morrownr, 2023-11-29:
Hi @Pointedstick 

Thank you. I removed the entire Walmart section. I have little toleration for bs.

Sorry for your experience. Hopefully you can get your money back.

Pointedstick, 2023-11-29:
Thanks. I'm trying again with an Alfa AWUS036ACM.

closed 2023-11-29 by Pointedstick

morrownr, 2023-11-29:
I have an Alfa ACM. It does very well in managed, monitor and AP modes. It is a good quality adapter. Right now Rokland has a sale on it:

https://store.rokland.com/pages/weekenddeals-sale

I can vouch for Rokland being a good outlet and they know Linux.

Pointedstick, 2023-11-29:
Thank you, that's super helpful! Just placed an order from them.

morrownr, 2023-11-29:
Let me know how it goes.

@morrownr

morrownr, 2023-12-03:
> JFYI I in no way expect you to help me troubleshoot this, as it's clearly not your fault,

No problem. I enjoy it. I have the FLU right now so expect snappy replies.

One app I use in situations like this is "wavemon". Once installed, start it by typing "wavemon" in a command line interface (terminal). It would be good to know what signal level and link quality you are seeing at different locations.

>  it has a clear line of sight through two windows 

Some glass works well with RF but some does not.

I recommend you use the laptop and wavemon to move around while watching the readings. It may be possible to figure out what is the primary cause of the poor transfer rates. There is also aluminum foil or a beer can.

Pointedstick, 2023-12-04:
Absolutely fascinating. It was my office window! When I relocated the AWUS036ACM with the directional antenna to *under* the window so that the signal had to go through a wall, speeds rose to 130/20! That's better than the repeater, and easily enough for my purposes. Thanks so much for your help!

Pointedstick, 2023-12-04:
I'm having difficulty determining whether it does under OpenWRT or not.

morrownr, 2023-12-05:
> I'm having difficulty determining whether it does under OpenWRT or not.

If you are in the US, look at the 5 Ghz channels and if any channels in the range of 100-148 are available, those are the DFS channels. DFS channels are usually not used at all or at that is my experience so you can have all of the clean air and speed you want. Start with channels 100 and 80 MHz channel width and see what happens. Remember that DFS channels do not become active as to soon as boot is complete as a mandatory scan for interference can take about 60 seconds.

Pointedstick, 2023-12-13:
Ah, in that case, yes, they are available. I'll experiment. Thanks for your help.