morrownr/USB-WiFi #342: MT7662u seems to use USB3 Interface

petter1, 2023-11-29:
I searched for a dongle using MT7662u chipset and found the specs of the chipset at [MediaTek website.](https://www.mediatek.com/products/broadband-wifi/mt7662u) There, it is written, that the chipset has a USB3 interface. Is it tested to work on USB2 as well, or is it an error in the list?

morrownr, 2023-11-29:
Hi @petter1 

> Is it tested to work on USB2 as well...

Yes

> or is it an error in the list?

The mt7612u variant of the mt7662u chipset is the chipset that will support USB3. The mt7612u is the AC1200 class chipset that I recommend for Linux users.

USB WiFi adapters that also support Bluetooth need to be limited to USB2 operation. This is true whether adapters use Mediatek or Realtek chipsets. The reason has to do with USB3 cables and connections causing interference on the frequency that Bluetooth uses.

Here is a link that explains a little and it provides a link to the Intel White Paper from 2012 that explains the problem:

https://www.bluetoothandusb3.com/the-explanation#:~:text=Unfortunately%20the%20range%20of%20frequencies,are%20using%20in%20your%20computer.

You have to select "Give me all of the facts" to see the Intel link.

We have had wifi speeds that exceed what USB2 can do for years but adapter makers continue to make AC1200 and later adapters that also include Bluetooth but it is a BAD idea. The adapter makers need to tell people in clear text that buying a USB WiFi adapter that includes Bluetooth is a BAD idea but they don't. There is no way to fix the issue. If you really want combined WiFi and Bluetooth, get a card... PCIe or M.2 as this is not a problem with cards as they don't use USB for WiFi.

Regards,

@morrownr

petter1, 2023-12-06:
Thank you very much for the detailed explanation! Never would I have thought, that USB3 interfere with Bluetooth 😮. My case is a device, which only has USB 2.0 so this should not apply to me, and I can use USB 3.0 WiFi adapter on USB 2.0, but with less speed than they are marketed at, which would be OK for me, I just need 5 GHz (WiFi 5 +) on my embedded linux device on 5.19.x. I will just backport rtw88 using a patch file, so I can get the dongles Working there.

closed 2026-10-31 by petter1