morrownr/USB-WiFi #344: I bought NETGEAR A6210 to use mt7612u

Ekiben542, 2023-12-04:
Hello. I purchased NETGEAR's A6210-100JPS to use the mt7612u in Japan (because it passed the technical standards and was sold in Japan).
However, when I investigated, I found out that
・txpower has limitations
- openwrt is not supported
I found something like.
but,
I think that txpower and openwrt are only restricted in the A6210-10000 series such as A6210-10000S, and it does not affect the A6210-100 series, but can it be used in this series?
If I cannot use it, please let me know what technical-compliant products I can purchase in Japan (which can be purchased on Amazon).
thank you.

Ekiben542, 2023-12-07:
@morrownr

morrownr, 2023-12-07:
@Ekiben542 

I've been sick over last week. It has slowed me down.

I would like to help but you will need to educate me on the specifics of the requirements in Japan. 

My review of the NETGEAR's A6210 in the Plug and Play list recommends against AP mode purchase unless your requirement is for same room only due to the limited range of the product. The design evidently limits its range as limited range is not a problem on other mt7612u based adapters.

For an AC1200 class adapter that has proven itself to me over time with OpenWRT, I recommend the Alfa ACM but do not know if it is available to you. If you want to send a link to Amazon Japan, I can take a look around.

@morrownr

bjlockie, 2023-12-07:
www.amazon.co.jp I think.

Ekiben542, 2023-12-08:
@morrownr 
If it has the mark in the picture below, it means that you can use it in Japan.
And as a way to check if the product actually has the technical certification,
Go to "https://www.tele.soumu.go.jp/giteki/SearchServlet?pageID=js01"
If you go to the above link, you will see a place to enter the "型式又は名称(or Model or Name)". Basically, do not enter the company name, just the model number, and it will come up (if something comes up, it has technical certification
[image]

morrownr, 2023-12-09:
@Ekiben542 

It may be better if I list the adapters by model that I know first hand work well in AP mode and work well with OpenWRT. Then you can input the model number:

ALFA AWUS036AXML (AXE3000)
ALFA AWUS036ACM (AC1200)
ALFA AWUS036ACHM (AC600)

Adapters that I do not have personal experience with but will likely work well:

PANDA - PAU0B (AC600)
ALFA AWUS036AXM (AXE3000)

OpenWRT has packages for the mt7921ac, mt7612u and mt7610u chipsets. The drivers are called mt7921u, mt7612u and mt7610u. The mt7921u driver handles WiFi 6 very well. It can also handle 6 GHz but whether it works well in your country really depends on the the rules in the country and how far along things are. 6 GHz seems to do well in AP mode in the EU but is hit and miss in many parts of the world plus 6 GHz has much less range than 5 GHz.

Ekiben542, 2023-12-10:
https://www.amazon.co.jp/WiFi-%E7%84%A1%E7%B7%9ALAN-1300Mbps-WIFI%E3%82%A2%E3%83%80%E3%83%97%E3 %82%BF%E3%83%BC-210-131768%E3%80%8D/dp/B0CJ6Z5VNV/ref=sr_1_2_sspa?keywords=wifi+usb+linux&qid=1702194827&sr=8-2-spons&sp_csd=d2lkZ2V0TmFtZT1zcF 9hdGY&psc=1
This product seems to work on Linux.
Is this rtl8812bu? Or is it mt7612u?

morrownr, 2023-12-10:
You could contact Alfa directly so as to find a dealer you can use:

https://www.alfa.com.tw/

Ekiben542, 2023-12-11:
To test the NETGEAR A6210 100 series, I tried to see if it works with txpower and openwrt.
In conclusion, both txpower and openwrt worked.
When he set txpower to 1dbm he got 3dbm, and when he set it to 10dbm he got a solid 10dbm.
However, the reception condition is quite poor.
I'll try using it just in case. If that doesn't seem possible, I'll contact ALFA.

morrownr, 2023-12-11:
> However, the reception condition is quite poor.

Understand. As I said, I have noted the range is limited with this adapter so I recommend using it for AP mode unless it is a same room situation. It is not a problem with the chipset as other adapters with the mt7612u chipset have good range.

Good luck.

morrownr, 2023-12-13:
> (I'm considering rtl8812au as an alternative, but what do you think?)

The rtl8812au chipset is a good old chipset but I would not recommend it for a new purchase as the Linux out-of-kernel drivers ceased updates in 2021 and it is unlikely we will ever see an in-kernel driver for this chipset. I consider it a dead end.

Have you looked over the adapters shown in the Plug and Play list? The Netgear A8000 is listed. Based on the reports I have seen, it is a much better adapter than the A6210. It is based on the mt7921au chipset. It is not cheap but those who have bought it seem to have good things to say about it. Maybe it is available?

morrownr, 2023-12-14:
@Ekiben542 

> I have a question I would like to ask, is it possible to change txpower?

I do not know the answer to this question but possibly @russeree will know as he is the one who wrote the review for the Netgear A8000.

@morrownr

morrownr, 2023-12-15:
> As I was doing my usual research, I found a USB dongle made by COMFAST. Do you recommend this?

I don't know. You did not say what adapter that it is? Do you have a link?

morrownr, 2023-12-15:
> It is CF-953AX.

If memory serves me correctly, that was the first adapter on the market with the mt7921au chipset. I think it came on the market about 1 year and 4 months ago. A lot of Linux users have purchased that adapter. The reports seem to be mostly positive. You might do a search on 953AX as I remember several conversations. For the price, it is hard to go wrong unless the adapter is junk and I don't remember any reports to that effect. It won't be the quality of Alfa adapters but I know of nothing that would cause me to talk you out of buying it.

> I would like it to arrive in about a week after I buy it, but with Aliexpress it's 1 month.)

It seems like most buyers get it in less than a month but right now with XMas gifts clogging the shipping channels, who knows.

That adapter is in the Plug and Play list it would not be there if I have been getting negative reports. There are many adapters that could be in the Plug and Play list that are not because of one problem or another.

russeree, 2023-12-15:
> I have a question I would like to ask, is it possible to change txpower?

I have attempted to change the txpower and always get the same response back from iwconfig - ```3 dBm```

```
sudo ifconfig wlx9418653d6a3e down
sudo iw wlx9418653d6a3e set txpower fixed 6dBm
sudo ifconfig wlx9418653d6a3e up
```

[image]

morrownr, 2023-12-16:
@Ekiben542 

For what it is worth: Many modern usb wifi adapters appear to have locked power settings. I have many adapters, it kinda goes with supporting this site. Some adapters seller as well as some adapter makers send me adapters either for pre-production testing or for me to review in the hopes that their adapter will make it into the Plug and Play list. Most don't make it into the list as I am very picky and have rules about multi-state and multi-function adapters as I want the list to only include adapters that give a very high probability of user satifaction.

So, my point being, I know first hand that many adapters have locked power settings. I don't find it to be a problem.

Did you check on the site where you posted the ad for the Comfast 953AX? I'm pretty sure that the owner of that store is also the owner of the Comfast brand.

morrownr, 2023-12-17:
> I have another question: what are the limitations of txpower?

It depends on what is in the regdb for your country. 

https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb

morrownr, 2023-12-18:
> (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW

5 GHz channels 36-48 txpwr max 20

> (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW

5 GHz channels 52-64 txpwr max 20 - DFS

> (5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS

5 GHz channels 100-140 txpwr max 23 - DFS

morrownr, 2023-12-18:
> Is that limitation only on the surface and can you change the dbm by actually operating it?

Exactly how txpwr has evolved over the years is a long story but things have gone from relatively simple back in the WiFi 4 and earlier days to very complicated today. With WiFi 5 and later, there is little standardization among countries around the world. Laws vary widely. There have been a lot of things that have been done by makers to try to stay in accordance with the laws so what you can get when buying an adapter can vary by when the adapter was designed and built. Many older designs are still on the market. Some adapters are locked either fully or somewhat. Most modern adapters can vary txpwr but it is controlled by the Linux kernel, not us.

The basic idea in this day in time, if my understanding is correct, is that adapters are built to vary txpwr depending on that country you are in. There has been a realization that people travel so the basic idea is that our computers and phones should get a signal from the AP/router that tells them what country they are in so that the appropriate information can be used by the Linux kernel to adjust the txpwr and other details so that your device works with the laws of the country that you are in.

More or less, modern hardware should adjust the settings for you if everything, to include the AP/router, is modern enough to send the country location information. What it seems that we see is that normally txpwr will be set to what the regdb max is for where you are. When people try to increase txpwr above the max, it will not change... however, if they reduce txpwr to something below max txpwr, it will change and there are situations where a reduction in txpwr works well.

Generally speaking, I ignore txpwr as it should be set for me. I test range and throughput to compare adapters without caring about txpwr. Good range and throughput comes from high quality antennas and amp's.

morrownr, 2023-12-23:
That was quick.

morrownr, 2024-01-01:
> As it stands, I would like you to list the problems with rtl 8812 au.

With managed and master modes, there are not many problems at all. With monitor mode, there are some missing features but that only matters if it matters. I don't keep things memorized as far as specific problems go so if you want to ask about specific things you need to work, I'll check. The 8812au driver is, in my opinion, the best out of kernel driver Realtek has ever released. I am currently working on a newer version of the driver in a private repo. I am looking for testers and would like for you to test if you are interested.

> Is the chipset 802.11 ac wave 2?

No

morrownr, 2024-01-01:
I read the news about the strong earthquake in Japan. I hope you and your family are okay. Wishing good to the people of Japan.

> Even used ones cost 13000 yen (about 92 dollars).

The USB WiFi market is interesting in that only 2 chipset makers supply chipset to the market. This along with the perception that USB WiFi is a home market causes there to be more limited competition is some ways. There are a lot of adapter makers in the market but most makers want to be in the low cost end of the market. since it is the home market I guess ?, and not a lot of makers provide good quality adapters. The lack of competition in the higher end of the market makes for non-compeitive prices. 

I can tell you from the analytical data at this site that a lot of usb wifi adapter users are not home users. I see a lot of information that tells me many corporate IT types and a lot of other organizations involved in product development are buyers. I say that with a lot of data to back me up. This site gets around 22,500 hits per week. I think we need a better assortment of high quality adapters at fair prices and different designs of adapters. Most adapter makers just keep making the same designs while going for the low cost market.

bjlockie, 2024-01-02:
> The 8812au driver is, in my opinion, the best out of kernel driver Realtek has ever released. I am currently working on a newer version of the driver in a private repo.

Is that version going to work with new kernels?

morrownr, 2024-01-02:
> Is that version going to work with new kernels?

The new version currently works with kernel 6.7. It will like continue working with new kernels as long as someone like me is making the necessary updates to the driver code. However, I can't guarantee it will work out that way as Realtek has terminated its support for the 8812au chipset. The chipset is about 11 years old. How long a maker is going to support their project is up to them but in-kernel drivers tend to be supported for a LONG time. The 8812au has no in-kernel support. I use many peripherals in my lab here that work fine that are 10-20 years old and never had Windows 7 drivers, let alone anything newer. It pays to seek out good computer hardware that is supported with good in-kernel drivers. My main dev box is 11 years old but is snappy and work well.

bjlockie, 2024-01-05:
I was referring to the discontinuing of wireless extensions.
Does the driver use cfg80211?

morrownr, 2024-01-05:
@bjlockie 

If my understanding of the situation is correct, WEXT is not being pulled out. The code that went into the kernel to stop WEXT use with WiFi 7 drivers is designed to detect WiFi 7 drivers and if it then sees WEXT use, it will not allow the driver to run. That should mean that older drivers that do use WEXT should continue to work. I think it is a way to put a stop to WEXT use going forward so that it is easier to totally remove WEXT at some point in the future. 

Forcing certain devs, I'm looking at Realtek when I say this, to discontinue WEXT use was needed for several reasons. This is not a new issue and Realtek has know about it for years. The Realtek devs that do the drivers for the cards is already on board with this and has RTW88 and RTW89 in the kernel and they are working well in the Linux WiFi dev community. The USB support for some of the drivers in RTW88 was only possible because of RTW88. I hope that Realtek wakes up and puts USB devs to work on the RTW88, RTW89 and future in-kernel drivers. That would be best for all concerned. It does take time to learn and Realtek has really gotten to the point that they are getting really behind Mediatek with their USB WiFi support and it seems to be getting worse. Mediatek has been training Linux WiFi devs for several years now and you see a lot of names with mediatek.com email addresses. Mediatek has to have a lot of devs because they are in several markets. Many of the Mediatek WiSoCs (Wireless System on a Chip) products ONLY need Linux drivers because the products are built to go inside other products such as WiFi routers and TV's and the like and the ONLY OS being used in those products is Linux.

You may have seen the NEWS that I posted earlier this week in Issues about the firmware for the mt7925 chipset? It is now available. The driver went into kernel 6.7. This is a WiFi 7 driver for a WiFi 7 chipset. It takes makers a while to develop new products but I suspect we will see new WiFi 7 USB adapters this year, possibly by mid-year depending on when chips were shipped to makers.

Where is Realtek? I have no idea. Usually you can get an idea about new products if you snoop around but I see nothing.

fakemanhk, 2024-01-07:
> I can tell you from the analytical data at this site that a lot of usb wifi adapter users are not home users. I see a lot of information that tells me many corporate IT types and a lot of other [...]

The way you view it is different from the way how manufacturer viewing it.
First of all, why do people use USB WiFi? I can only think of a few reasons:

1.  PC doesn't have WiFi, people wants to add WiFi support but not willing to (or not possible to) open the case and add PCI WiFi
2. PC/laptop is using old WiFi standard and people want to upgrade with minimal effort.
3. Need to capture multi-stream WiFi data (e.g. to use with Kali Linux) for analysis

For the above reasons, the number of people that really needs USB WiFi aren't really that many, so manufacturers only want to make their product "just works", rather than allowing extra features/tuning. One good example is AP mode, some chipsets have AP mode support but the USB one simply....sucks.....I have the COMFAST CF953AX with OpenWrt in AP mode however it looks like there is a bug that when it's running high traffic the dongle will hang and reset itself after a few minutes, but it has no issue at all when I am using it as client.

morrownr, 2024-01-07:
> The way you view it is different from the way how manufacturer viewing it.

Exactly.

> First of all, why do people use USB WiFi? I can only think of a few reasons:

I am aware of more....

4. Many product makers use small boards like RasPi's in their products these days. They often look to have one or two usb wifi adapters connected as internal wifi usually does not cut it for one reason or another.
5. Numrous companies have discovered that wifi can take care of use cases that it might not have been able to handle many years ago and it increases the flexibility of installations.

I am aware of more use cases, In robotics, there are many use cases. I do some consulting on the side.

> some chipsets have AP mode support but the USB one simply....sucks...

This could make for an interesting discussion.

> I have the COMFAST CF953AX with OpenWrt in AP mode however it looks like there is a bug that when it's running high traffic the dongle will hang and reset itself after a few minutes, 

Well, then, let's find the bug. Can I get you to post a new issue with a title something like `AP mode bug using OpenWRT and CF-953AX (mt7921au based) adapter`?

I run AP mode a lot. I also use OpenWRT. The new issue should include the details of the hardware, software and configuration you are using as well a detailed description of what the problem is.

@morrownr

morrownr, 2024-01-08:
@Ekiben542 

> I have a question, does rtl 8812 bu currently not support MU-MIMO in AP mode ...?

This answer is for the out-of-kernel driver as I have not tested the in-kernel driver for this function.

No.

The below is from file 88x2bu.conf, which is where the module parameter docs are located:

```
# Beamforming options ( rtw_beamform_cap )
#
# 1 = SU Beamformer (recommended for AP mode)
# 2 = SU Beamformee
# 3 = SU Beamformer and SU Beamformee
# 10= SU Beamformee and MU Beamformee (default)
# 11= SU Beamformer and SU Beamformee and MU Beamformee
#
# Note: MU Beamformer is not supported.
```

Regards

@morrownr

morrownr, 2024-01-08:
@Ekiben542 

> I updated the kernel version of my Raspi to 6.6.9, but rtl 8812 bu doesn't work (it should work with 6.2 +), what is the cause ...?

There is a lot of info missing but my first guess would be missing firmware.

My firmware guide is located at:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

@morrownr

morrownr, 2024-01-08:
@morrownr 

> I installed the rtw 8822 b one, didn't have rtw 88...

That indicates no firmware was installed in your distro.

> but it doesn't work ...

Have you installed the out-of-kernel driver? If so, remove it with `remove-driver.sh`. install-driver.sh blacklists the in-kernel driver.

If that does not work, tell me where you got the 6.6.9 kernel that you compiled?

morrownr, 2024-01-08:
Okay, it sounds like you did not compile the kernel. You are likely using a kernel that does not have the driver compiled.

Kernel 101: Just because a new driver shows up in Mainline does not mean it automatically ends up in your distro. The distro maintainers have to change their kernel compilation setup to include compiling the driver which the RasPi folks have not done yet.

Here are the instructions for compiling 6.6 for the RasPiOS.

Note: You have now entered the DEV Zone so if you have no experience, go slow and ask questions. You may find that you have to wait for my replies in some cases and keep in mind you will have to dig to find and turn on the driver. Good luck.

-----

Step 2: Install git and the build dependencies

$ sudo apt install git bc bison flex libssl-dev make libncurses5-dev

Note: libncurses5-dev is required for menuconfig (not used in this guide yet.)

Note: git, bc and make will show as already installed for the RasPiOS version
that I am using in this guide but that is okay.

-----

Step 3: Download the sources

Note: change "rpi-6.6.y" to the branch you want in the command below.

$ git clone --depth=1 --branch rpi-6.6.y https://github.com/raspberrypi/linux

Note: Omitting --depth=1 will download the entire repository, including the
full history of all branches, but this takes much longer and occupies much
more storage.

Note: Refer to the original GitHub repository for information about the
available branches.

https://github.com/raspberrypi/linux

-----

Step 4: Prepare the kernel configuration

Prepare the default configuration by running the following commands.

Warning: Once you have moved into the linux subdirectory below, you
need to stay there until complete.

$ cd linux
$ KERNEL=kernel8
$ make bcm2711_defconfig

In addition to your kernel configuration changes, you may wish to
adjust the LOCALVERSION to ensure your new kernel does not receive
the same version string as the upstream kernel. This both clarifies
that you are running your own kernel in the output of uname and
ensures existing modules in /lib/modules are not overwritten.

To do so, change the following line in .config:

CONFIG_LOCALVERSION="-v8-MY_CUSTOM_KERNEL"

$ geany .config

Note: This is the point where you can modify or patch the kernel
source before compiling.

-----

Step 5: Build the Kernel

$ make -j4 Image.gz modules dtbs

-----

Step 6: Install the kernel, modules, and Device Tree blobs

$ sudo make modules_install

$ sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/
$ sudo cp arch/arm64/boot/dts/overlays/*.dtb* /boot/overlays/
$ sudo cp arch/arm64/boot/dts/overlays/README /boot/overlays/

$ sudo cp arch/arm64/boot/Image.gz /boot/kernel8.img

Note: If you have not already deactivated the internal wifi, now would
be a good time to do it before you reboot.

Note: It would also be a good time to recheck and update the firmware
files.

Reboot. Your Raspberry Pi should be running your freshly-compiled kernel!

You should be finished at this point.

-----

morrownr, 2024-01-08:
@Ekiben542 

Rats!

I'll try to find time to burn a fresh RasPiOS card and test my guide.

morrownr, 2024-01-08:
@Ekiben542 

Opps... my guide did not show you how to edit ,config to turn on the RTW88 support for this adapter. My bad. Mine is compiling now which will probably take most of the night so let me check things and get a new guide to you.

By the way, the default .config was not compiling the support you need. RasPiOS devs generally only do with upgrades with LTS kernels and RTW88 came one kernel after the last LTS so the timing was not good.

morrownr, 2024-01-09:
> I see. You don't have to rush.

What should have been easy is turning out not to be easy. The RasPi devs have made some changes. I keep telling them not to make changes without asking me first but that has not worked. Ha ha.

I need to start over.

morrownr, 2024-01-10:
> I'm sorry to hear that....

It is not a problem. I regularly do clean installs because of all of the work and testing that I do. It just slows things down.

morrownr, 2024-01-15:
/home/morrow/AppDev/Raspberry Pi/Upgrade_RasPiOS_Kernel/Compile_and_ Install New_RasPiOS_Kernel.tar.gz

@Ekiben542 

Here is the updated guide. Please let me know how well it works. If it works for you, I think I will add it to the Main Menu as others may want this guide.

@morrownr

morrownr, 2024-01-15:
[Compile_and_ Install New_RasPiOS_Kernel.tar.gz](https://github.com/morrownr/USB-WiFi/files/13935170/Compile_and_.Install.New_RasPiOS_Kernel.tar.gz)

Opps. here is the file.

bjlockie, 2024-01-15:
Are you trying to boot a raspberry pi from usb?
remove the SD card from the pi.

morrownr, 2024-01-15:
> What should I do?

I am skeptical of booting with USB on the RasPi4B. I tried it at one time but was not happy with the results. The USB subsystem on the RasPi4B is problematic for a lot of reasons. The sd slot is fast enough for what the RasPi4B is capable of doing well and it is more stable. I recommend you grab a sd card and burn it.

Is it possible that I made a mistake on the guide? Yes. I did test it on a freshly burned sd and the results were good. Is it possible that my writing was not clear and you were unable to follow what I intended for you to do? This very well could be the case. It is a detailed process.

Let back up for a moment. I answer a lot of questions in different places on this site everyday that I am able so it can be hard for me to remember all of the details and this issue has drifted a lot since the beginning.

You have a RasPi4B running the 64 bit latest RasPiOS? Correct? You want to get a rtl8812bu based usb wifi adapter working with the in-kernel driver? Are you planning to use managed mode or master mode?

@morrownr

morrownr, 2024-01-15:
> Trying to get it to work with AP (Could it work?)

It should work. I am going to leave my PI4B as it is until you are ready to press on. Then we can set the AP up together

morrownr, 2024-01-16:
Let's go over things:

I use:

https://www.raspberrypi.com/software/operating-systems/

Raspberry Pi OS (64-bit)
Raspberry Pi OS with desktop
Release date: December 5th 2023

I burn it to a 32 GB sd card with the Raspberry Pi Imager.

I use the version with the gui desktop because the gui desktop that RasPi uses is so light that it uses very little in terms of resources and some things are easier to do with gui apps and utilities.

I have updated and upgraded it and told it the WLAN country as well as some other things per the guide I posted. After running the guide, I see:

```
 $ uname -r
6.6.11-v8+
```

``` $ iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 00:c0:ca:ad:de:9e
		type managed
		txpower 30.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	0

```

That is from an 8812bu based adapter using the in-kernel 8822BU driver included in RTW88.

It is ready for me to start setting up AP mode for testing when you are ready.

Do you have any idea what is going on? Is this a clean installation? Are you using a sd card? Is the card or usb flash in good shape? Cards and drives go bad. I use really high quality cards to reduce problems.

@morrownr

morrownr, 2024-01-16:
> I tried it a few times, and it took effect even after restarting.

I'll take this to mean that you are on kernel 6.6 and the adapter is showing a wifi interface?

> However, when I try to use the wifi signal, the message "Cannot connect to this network" appears.

I need to know how you are setup. My Pi4B is headless. The only two wires going into my Pi is power and cat6 cable How about yours?

What are you trying to connect to? I assume you are using the Pi in client mode?

morrownr, 2024-01-16:
> what is client mode?

Is your wifi adapter being used in managed mode to make a connection to an AP/wifi router?

My internet access comes in via cat6 cable so I was planning to take my wifi adapter directly into AP mode.

morrownr, 2024-01-17:
>  I really appreciate your support. Thank you.

You are welcome.

You are helping me also. You basically pushed me into updating my Guide for compiling a new kernel for the RasPiOS so now I need to find time to post it on the Main Menu. I'm also updating my AP guide so this is all good... but it goes slow as it takes a lot of work and I'm already busy on other issues. 

We'll get it done.

@morrownr

bjlockie, 2024-01-17:
What is unable to connect, a phone?
Try it without encryption (make the AP open).

morrownr, 2024-01-17:
Post your hostapd.conf file.

Have you ever setup hostapd before?
How are your networking skills?

FYI: The in-kernel driver and the out-of-kernel driver are not feature equal. Your ht capab and vht capab lines are likely wrong.

Did you follow my Bridged Wireless Access Point guide? I know it is a lot to take in if your experience level is low.

morrownr, 2024-01-17:
> ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]

> vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]

Would you like for me to give you some homework? I can give you some files and enough info for you to work on the new versions of the above?

The issue at hand is that the capabilities depend on the driver. I did a quick look and the in-kernel driver has somewhat different capabilities than the out-of-kernel driver so those line need to be a little different. They are close or hostapd would throw a fit. 

I am very busy today fixing an issue in my out-of-kernel scripts but will try to get back on this project later today or tomorrow as time permits.

hostapd is a very nice app but it takes time to learn and get it right.

Can I get your to post the results of:

$ lsusb -t

Let's see if the adapter is in USB3 mode. That is an issue that I'm working with a couple of others as I have time. The driver is not always going into USB3 mode when using a USB3 port.

@morrownr

morrownr, 2024-01-17:
> Here are the results of lsusb -t:

That does not look like lsusb -t. See below:

My dev box:

```
$ lsusb -t
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
    |__ Port 2: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 5000M
    |__ Port 4: Dev 4, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
    |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 5: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 5: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 6: Dev 4, If 0, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 6: Dev 4, If 1, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 6: Dev 4, If 2, Class=Video, Driver=uvcvideo, 480M
        |__ Port 6: Dev 4, If 3, Class=Video, Driver=uvcvideo, 480M
        |__ Port 7: Dev 5, If 1, Class=Wireless, Driver=btusb, 480M
        |__ Port 7: Dev 5, If 2, Class=Wireless, Driver=, 480M
        |__ Port 7: Dev 5, If 0, Class=Wireless, Driver=btusb, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 3: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M

```

My Pi4B:

```
 $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 480M

```

> f it works, add something,

Smart.

Something before I end this message: Remember that the in-kernel for the rtl8812bu is largely unproven in AP mode. I'm sure that somebody has tested it but they haven't showed up around here. Therefore, we really don't know what to expect. I hope it works but I do not know that it will. I hope to start setting up the AP here tomorrow but it will go slow as I have to rework my guide.

@morrownr

morrownr, 2024-01-19:
@Ekiben542 

Update: I ran onto an issue at another site that seems to indicate the in-kernel 8822BU driver is not working well in AP mode. I still want to test it and hope to have time soon.

What you may want to do is install the 88x2bu driver here at this site. If using `install-driver.sh` ro install it, it will automatically blacklist the in-kernel driver and reverse the blacklist if you use `remove-driver.sh` to remove it. The link is:

https://github.com/morrownr/88x2bu-20210702

The info in the example hostapd.conf files in my guide is for this driver. On a RasPi4B, you may find that you need to keep it in USB2 mode. This will give you the chance to test to see if your setup is working well.

@morrownr

morrownr, 2024-01-29:
@Ekiben542 

I haven't given up on this project but there is a problem. I cannot tell if it is a hardward issue with my pi4b or is something in RasPi. I'm not currently setup to test AP mode on other systems so this could take some time.

@morrownr