morrownr/USB-WiFi #345: SALE: Rokland currently has sales on several Linux compatible USB WiFi adapters...

morrownr, 2023-12-07:
Link: https://store.rokland.com/pages/weekenddeals-sale?mc_cid=f77d2ccb24

All adapters are 10-15% off depending on how much you buy.

For those looking for adapters for security analysis/pen testing, my recommendation is these two adapters:

ALFA AWUS036ACM (AC1200)
ALFA AWUS036ACHM (this one does not appear to be on sale) (AC600) (this one has very long range)

For those looking for managed (client) mode or AP mode, my recommendation is these two adapters:

ALFA AWUS036AXM (AXE3000)
ALFA AWUS036AXML (AXE3000)

Note: The 2 adapters above are very capable of WiFi 6 and I have experience running my AXML in AP mode W-Fi 6. WiFi-6E can be a challenge depending on the country you live in so I am not recommending them based on AP mode WiFi -6E (6 GHz operation).

If you find that you are interested in one of the above adapters and have questions, please ask.

If you have one of these adapters and have something to add, please do so.

Regards,

@morrownr

MR-pentestGuy, 2023-12-19:
Hi @morrownr, 

I need a wifi adapter where I can do all things like AP mode,packet injection,..etc on total WIFI pentesting. 
Which wifi adapter would you recommend since ALFA AWUS036ACHM is WIFI 4 and ALFA AWUS036ACM which has problems in VM and ALFA AWUS036AXML is good at AP but is not that good at packet injection. So what is your preference if you have to choose for all things. 

Regards,
mrpentestguy

morrownr, 2023-12-19:
Hi @MR-pentestGuy 

> So what is your preference if you have to choose for all kinds of wifi pentesting.

I am far more experienced with AP (master) mode than I am with monitor mode so someone like @ZerBea can do a better job of answering this question if he is available. He'll show up if he is available.

My answer: I have all 3 adapters. With my limited pen testing experience, I would go with the ACHM. I have to laugh sometimes when using the ACHM as its range allows one to reach out and touch things other adapter don't know is there. While the ACHM is only an AC600 adapter, in certain situations it can smoke AC1200 adapters because its range is such that it can stay connected and working while the AC1200 adapter has lost the connection. Both the ACM and ACHM fully support active monitor mode. The AXML driver, mt7921u, says it supports active monitor mode but there is currently a bug we are working to get fixed. If you don't turn active monitor mode on with the AXML, all monitor mode activities work fine as far as I know. One advantage the AXML has is that it supports 6 GHz and I have seen users reports saying that pen testing works fine. As far as the ACM goes, yes, I have seen reports of there being problems in VM's but I would not know as I use `dual boot` instead of VM's here. If you take VM's out of the picture, the ACM seems to work well for pen testing and it does support active monitor mode.

> I need a wifi adapter where I can do all things like AP mode...

My experience with all 3 adapters is that they work very well for AP mode. If you want WiFi-6 capability, you have to go with the AXML or AMX. I have an AP guide up here on the Main Menu.I even have a sample WiFi-6 hostapd.conf available. I run my AXML with WiFi 6 on the 5 GHz band. It works well. I am not there with WiFi 6e (6 GHz) yet. I've used the ACM and ACHM on a RasPi4B in AP mode for years. Rock solid.

Hope this helps,

@morrownr

morrownr, 2023-12-19:
I forgot to mention that all 3 of the adapters you mentioned and the AXM work well with OpenWRT. OpenWRT contains drivers for each. I mention this because OpenWRT is certainly an option for AP mode, especially if you want a routed setup.

@morrownr

MR-pentestGuy, 2023-12-20:
Hi @morrownr ,

Thanks for the detail explanation. 
really appreciate it & waiting for @ZerBea answer

regards,
mrpentestguy