morrownr/USB-WiFi #352: Wifi6 config - unable to bring up AP

slartibarfast, 2023-12-15:
Admission - I am a noob when it comes to Linux WiFi configuration.   I was pointed here by the linux support guys at brostrend, makers of the 1800ax wifi adapter I'm trying to configure.

I am trying to use your Wifi6 config to enable Wifi6 on this Realtek 8852u based wifi usb dongle.  I have the interface loaded and showing as DOWN under 'ip a'.

```
5: wlxc83a35b4484d: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether c8:3a:35:b4:48:4d brd ff:ff:ff:ff:ff:ff
```

Initial testing with v2.1 of hostapd and the posted WiFi 6 config (adjusted for my bridge and interface name) resulted in the system complaining about the frequency being unsupported.   Changing the op_class to 133 (or 134) has cleared that error, however it then complained about management frame protection.  

```
root@pve01:~# hostapd -d /etc/hostapd/hostapd6.conf
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd6.conf
ctrl_interface_group=0
Management frame protection is required in 6 GHz
1 errors found in configuration file '/etc/hostapd/hostapd6.conf'
Failed to set up interface with /etc/hostapd/hostapd6.conf
hostapd_init: free iface 0x55bf536ef9e0
Failed to initialize interface
```

Changing `ieee80211w` from 1 (in your config) to 2 got me by this, but now I'm greeted with

```
root@pve01:~# hostapd -d /etc/hostapd/hostapd6.conf
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd6.conf
ctrl_interface_group=0
Invalid AKM suite for 6 GHz
1 errors found in configuration file '/etc/hostapd/hostapd6.conf'
Failed to set up interface with /etc/hostapd/hostapd6.conf
hostapd_init: free iface 0x5560aa97c9e0
Failed to initialize interface
```

After some digging, I found the only option that would get me past that error was `wpa_key_mgmt=SAE`  -- however the AP would still not come up - and now I have no idea what to try next.  This is the debug log 

```
root@pve01:~# hostapd -d /etc/hostapd/hostapd6.conf
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd6.conf
ctrl_interface_group=0
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported vendor command: vendor_id=0x1a11 subcmd=4106
nl80211: Supported vendor command: vendor_id=0x1a11 subcmd=4107
nl80211: key_mgmt=0xf0f enc=0xfef auth=0x7 flags=0x8000008a4858c0 rrm_flags=0x0 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=9
nl80211: interface wlxc83a35b4484d in phy phy1
nl80211: Set mode ifindex 5 iftype 3 (AP)
nl80211: Setup AP(wlxc83a35b4484d) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x563546d98a50 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x563546d98a50 match=04 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x563546d98a50 match=0501 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x563546d98a50 match=0503 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x563546d98a50 match=0504 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x563546d98a50 match=06 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x563546d98a50 match=08 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x563546d98a50 match=09 multicast=0
[... 523 lines cut ...]
        Device supports SAE with AUTHENTICATE command
        Device supports scan flush.
        Device supports randomizing MAC-addr in scans.
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported TX frame types:
                 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * IBSS: 0xd0
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
        Supported extended features:

```

Suggestions on next step would be appreciated

bjlockie, 2023-12-15:
On 2023-12-14 22:11, slartibarfast wrote:
> Suggestions on next step would be appreciated
Return it? :-)

Realtek has really bad Linux support.

morrownr, 2023-12-15:
> Admission - I am a noob when it comes to Linux WiFi configuration.

There is no time like the present to learn.

Now, I am a little confused so let's get some terminology worked out:

WiFi 6 and WiFi 6e are not the same thing. WiFi 6 is supported on all 3 bands but only WiFi 6e indicates support for the 6 GHz band.

I think your adapter is based on the rtl8832bu chipset. I'll need you to confirm that. Are you using the rtl8852bu driver located here at this site?

If the rtl8832bu chipset is correct, then your adapter supports WiFi 6 but not WiFi 6e. The reason I mention this is that I am seeing you use hostapd.conf lines that are only for WiFi-6e.

If you are using my example WiFi-6 hostapd.conf, it was developed with an adapter with a mt7921au chipset. We will have to go over the file and change some settings for your adapters. There will also be some module parameters that will require change in the driver.

Last question for now: What hardware and distro are you using?

@morrownr 

P.S. @bjlockie is speaking words of wisdom with his last statement but let's give it a try and see how it works out.

slartibarfast, 2023-12-15:
Yes, I spent a 30+ year career in IT, and never stopped learning, no reason to think I would stop now.

I do not discount the sage advice of @bjlockie - I have learned that there seems to be little support for this chipset.... however, I was drawn to this product because the Amazon listing said it supported Debian and derived OSes, and my intention is to get it to run under Proxmox (v8.1)  It seems to be running a 6.5 kernel.  

My hardware is a Beelink EQ12 Pro. The N305 and its 8 Alder Lake "E" cores makes this a nice little machine. It has an Intel AX101(?) (a wifi5  chip I believe) that reports as **Network controller: Intel Corporation CNVi: Wi-Fi** - which I can't get past the same point - everything seems to start, and a hardware issue is found, and then hostapd shuts everything down.

(The AX101 being Wifi5 is why I purchased the external usb dongle - I have a lot of IOT devices in the house that I eventually want to hang off this, plus I wanted something with external antenna for better coverage in my house.   I have not disabled the AX101 as I cant find a bios option to do so as of yet.  A screw driver is not out of the question if needed.)

The actual chip that is running the usb dongle seems to be a 8852u based on the driver code downloaded and compiled.  This is a  Wifi6 USB dongle from Brostrend - BrosTrend USB WiFi 6 Adapter AX1800.   No 6e... just a 6.  So any changes to the config I may have done were simply trying to get around errors which were shown... not suggesting they were correct :)

The driver I am using was downloaded and compiled from the BrosTrend website.  Their script from `linux.brostrend.com/install` isn't foolproof, but manual kernel header installs isn't such a bad thing.  (side note - I do disklike that its not set up to recompile after kernel updates... I did have VMWare Workstation for Linux for a while, and it did this automatically)

I have been able to get the Brostrend device to run as a Wifi AP device using the (unmaintained) `https://github.com/oblique/create_ap` project.  Sadly, its just connecting as a 54Mbps device, so its not something I want to stick with... especially with WiFi6 emblazoned on the front of the dongle.

morrownr, 2023-12-15:
> Yes, I spent a 30+ year career in IT, and never stopped learning, no reason to think I would stop now.

Good stuff. I was never in IT. I learned programming with FORTRAN on an IBM Main Frame back in the Dark Ages and have kept myself current on one language or another over the years. I'm different in that my os path on small computers was CP/M, PC-DOS, DR-DOS, OS/2 and Linux. Don't ask questions about Windows because I do not know. I've been using Linux since around '94.

>  I have learned that there seems to be little support for this chipset.... 

I think that can be said of all Realtek chipsets. Realtek does not release these out-of-kernel drivers to the public nor do they take bug reports from the public. The driver require a black belt in C coding to maintain and we do have to maintain them as releases from Realtek are rare, if you can locate them. Realtek's strategy for Linux support is a puzzle to me. On the other hand, Mediatek's support is by the book. Mediatek's latest usb wifi driver was just added in kernel 6.7. The chipset is mt7925. It is WiFi 7. No adapters or cards are on the market yet but should be in 2024. This driver and the other modern Mediatek drivers are maintain in the Linux kernel and written in accordance with Linux Wireless Standards. Primary maintenance and development is handled by Mediatek employees but we get to play as well. Mediatek drivers lead to a Plug and Play experience that even my wife can handle because she has to do exactly nothing for things to work. There is no recompiling when a new kernel rolls in because the driver is already part of the compiled kernel so freshly updated driver find their way into your system whether you know it or not.

> My hardware is a Beelink EQ12 Pro. The N305 and its 8 Alder Lake "E" cores makes this a nice little machine. 

Sounds like a nice machine. That is about all I need to know about the machine for now.

> my intention is to get it to run under Proxmox (v8.1) It seems to be running a 6.5 kernel.

I am unfamiliar with this distro. Can you provide a link? Is Network Manager in use?

Is network connectivity provided by CAT5e or later ethernet cable?

> I have not disabled the AX101 as I cant find a bios option to do so as of yet. A screw driver is not out of the question if needed.)

Let's cross that bridge if we need to do so.

> The actual chip that is running the usb dongle seems to be a 8852u based on the driver code downloaded and compiled. This is a Wifi6 USB dongle from Brostrend - BrosTrend USB WiFi 6 Adapter AX1800.

Is the following the adapter you have?

https://www.brostrend.com/products/wifi-6-usb-adapter-ax1

If so, the chipset is rtl8832bu which uses the rtl8852bu driver.

The Brostrend version of the driver is packaged so as to make it easier for non-programmers to get it working in managed mode with few problems. For this project, I am going to suggest that you remove the Brostrend driver and use the version of the driver I have and maintain here:

https://github.com/morrownr/rtl8852bu

> Sadly, its just connecting as a 54Mbps device, so its not something I want to stick with... especially with WiFi6 emblazoned on the front of the dongle.

Understand. I'll bet we can get that up to 1200 mbps.

Before we just right in, I need to give you some homework. This site has a lot of information that can help with your project and if you take the time to do some reading... and let the reading soak in, it will help:

https://github.com/morrownr/USB-WiFi

That is the Main Menu for this site. I would recommend looking at menu items 1, 2, 4, 5 and 9.

That is enough for now. When you are ready, we can look at how to configure the driver you will upload from here as a replacement for the one you have installed. I will need to update some docs for this this driver and the hostapd.conf as we go.

@morrownr

slartibarfast, 2023-12-15:
> > my intention is to get it to run under Proxmox (v8.1) It seems to be running a 6.5 kernel.
> 
> I am unfamiliar with this distro. Can you provide a link? Is Network Manager in use?

Proxmox is a hypervisor, I'm running the community edition (ie, no support from Proxmox guys) and transitioning my home lab away from VMWare ESXi (my licenses expire early Feb)  Proxmox makes quite the capable VM manager, including features that VMWare makes you pay extra for.

Network Manager is installed, however there is no gui on this machine - its all managed from a web interface, or from command line.
 
> Is network connectivity provided by CAT5e or later ethernet cable?

Yes, I have a wired connection into my network.

> > The actual chip that is running the usb dongle seems to be a 8852u based on the driver code downloaded and compiled. This is a Wifi6 USB dongle from Brostrend - BrosTrend USB WiFi 6 Adapter AX1800.
> 
> Is the following the adapter you have?
> 
> https://www.brostrend.com/products/wifi-6-usb-adapter-ax1
>

Actually, its https://www.brostrend.com/products/usb-wifi-6-adapter-ax4

> If so, the chipset is rtl8832bu which uses the rtl8852bu driver.

After installng the Brostrend driver, /usr/src contains

drwxr-xr-x  7 root root 4096 Dec 14 20:42 rtl8852bu-1.19.3
```

root@pve01:/usr/src/rtl8852bu-1.19.3# ll
total 104
-rw-r--r-- 1 root root  6844 Sep  9 05:09 8852bu.conf
-rw-r--r-- 1 root root  3775 Jul 11 03:20 common.mk
drwxr-xr-x 6 root root 12288 Dec 14 20:42 core
-rw-r--r-- 1 root root   221 Sep  9 05:09 dkms.conf
-rwxr-xr-x 1 root root  5112 Sep  9 05:09 driverctl
drwxr-xr-x 4 root root 12288 Dec 14 20:42 include
-rw-r--r-- 1 root root   104 Jul 11 03:20 Kconfig
-rw-r--r-- 1 root root 22517 Sep  9 05:09 Makefile
drwxr-xr-x 3 root root  4096 Dec 14 20:42 os_dep
drwxr-xr-x 6 root root 20480 Dec 14 20:42 phl
drwxr-xr-x 2 root root  4096 Dec 14 20:42 platform
```
> The Brostrend version of the driver is packaged so as to make it easier for non-programmers to get it working in managed mode with few problems. For this project, I am going to suggest that you [...]

I think what I have is the same version of what you have, but for ease of configuration and debugging, I will clean install Proxmox to ensure all remnants of Brostrend are gone (not a big deal nothing running on this yet... and not the first time :) )

> That is enough for now. When you are ready, we can look at how to configure the driver you will upload from here as a replacement for the one you have installed. I will need to update some docs for this this driver and the hostapd.conf as we go.

Coffee is brewing... reading mode engaged

slartibarfast, 2023-12-15:
No rush to respond... I'm retired, I have lots of time to fiddle :)

Your docs suggest Realtek seems not to be supported as highly as the Brostrend group suggest... hopefully there is a solution :)

Current status

1. Proxmox reinstalled.  I've left it a the 6.5.11-4-pve kernel for now.   That is the release version of PVE 8.1, and should avoid any weirdities that the no subscription repos might introduce.
2. driver installed from your git repo (and AX4 device wlxc83a35b4484d showing)
3. hostapd configured with your Wifi6 config, and adjusted for my system.  These are the only changes made:

root@pve01:~# diff /etc/hostapd/hostapd.conf /etc/hostapd/hostapd.raw
16c16
< ssid=BeelinkExternal
---
> ssid=myPI-WiFi6
18c18
< wpa_passphrase=JustAnotherPW
---
> wpa_passphrase=myPW1234
53c53
< bridge=vmbr1
---
> bridge=br0
55c55
< interface=wlxc83a35b4484d
---
> interface=wlan0

4. tested, and just as unhappy as previous.  Results as per below:

```
root@pve01:~# hostapd -d /etc/hostapd/hostapd.conf
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd.conf
ctrl_interface_group=0
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported vendor command: vendor_id=0x1a11 subcmd=4106
nl80211: Supported vendor command: vendor_id=0x1a11 subcmd=4107
nl80211: key_mgmt=0xf0f enc=0xfef auth=0x7 flags=0x8000008a4858c0 rrm_flags=0x0 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=9
nl80211: interface wlxc83a35b4484d in phy phy0
nl80211: Set mode ifindex 4 iftype 3 (AP)
nl80211: Setup AP(wlxc83a35b4484d) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x555c8d8fca50 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=04 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=0501 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=0503 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=0504 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=06 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=08 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=09 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=0a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=11 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=12 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x555c8d8fca50 match=7f multicast=0
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x555c8d8fca50 match= multicast=0
nl80211: Enable Probe Request reporting nl_preq=0x555c8d8f96f0
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x555c8d8f96f0 match= multicast=0
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 7 (ifidx_reason 4)
[... 63 lines cut ...]
wlxc83a35b4484d: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlxc83a35b4484d wasn't started
hostapd_interface_deinit_free: driver=0x555c8d7ad760 drv_priv=0x555c8d8f78b0 -> hapd_deinit
nl80211: deinit ifname=wlxc83a35b4484d disabled_11b_rates=0
nl80211: Disable Probe Request reporting nl_preq=0x8888ddd405071e79
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=4)
netlink: Operstate: ifindex=4 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 4 iftype 2 (STATION)
nl80211: Teardown AP(wlxc83a35b4484d) - device_ap_sme=1 use_monitor=0
nl80211: Unsubscribe mgmt frames handle 0x8888ddd4050742d9 (AP teardown (dev SME))
hostapd_interface_free(0x555c8d8f19e0)
hostapd_interface_free: free hapd 0x555c8d8f3190
hostapd_cleanup_iface(0x555c8d8f19e0)
hostapd_cleanup_iface_partial(0x555c8d8f19e0)
hostapd_cleanup_iface: free iface=0x555c8d8f19e0
root@pve01:~# 

```
and the interface remains in a DOWN state (obviously)

```
4: wlxc83a35b4484d: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether c8:3a:35:b4:48:4d brd ff:ff:ff:ff:ff:ff
```

bjlockie, 2023-12-15:
> Your docs suggest Realtek seems not to be supported as highly as the 
> Brostrend group suggest... hopefully there is a solution :)

There might be a solution now but what happens when the out of kernel 
driver can't be made to work with new kernels

morrownr, 2023-12-15:
> 4: wlxc83a35b4484d: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether c8:3a:35:b4:48:4d brd ff:ff:ff:ff:ff:ff

In addition to "ip a" you might want to try "iw dev" when checking the interface.

> No rush to respond... I'm retired, I have lots of time to fiddle :)

Same here. I have been having some health problems lately and am not fully up to speed yet.

> hostapd -d /etc/hostapd/hostapd.conf

Not ready for that. I'd prefer a copy of your hostapd.conf... sort of, there is so much that I do not have eyes on. Let me think.

Okay. Let me see about changing my AP over to the same adapter as you have. Yes, I have one. I use a RasPi4B and it is dual band using 2 x usb wifi adapters.

You are using hostapd but what else are your doing to setup the networking? systemd-networkd? have you read my AP guide? It is menu item 9 on the Main Menu. I'm not saying you have to do your setup like my guide but if you know my recommended way, it could help us communicate.

I'll update my example WiFi-6 hostapd.conf when I move over to the new adapter.

Something I need to point out: My example WiFi-6 hostapd.conf has not been used with an adapter with the rtl8832bu chipset so I'll be working this as I go. It will take some time to determine the exact settings to optimize things.

Also, I still do not have a good handle on the Proxmox and what is in it. Is it based on Ubuntu or Debian?

slartibarfast, 2023-12-16:
> > 4: wlxc83a35b4484d: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
> > link/ether c8:3a:35:b4:48:4d brd ff:ff:ff:ff:ff:ff
> 
> In addition to "ip a" you might want to try "iw dev" when checking the interface.

```
phy#0
        Interface wlxc83a35b4484d
                ifindex 4
                wdev 0x1
                addr c8:3a:35:b4:48:4d
                type managed
                channel 36 (5180 MHz), width: 20 MHz (no HT), center1: 5180 MHz
                txpower 12.00 dBm

```

> > hostapd -d /etc/hostapd/hostapd.conf
> 
> Not ready for that. I'd prefer a copy of your hostapd.conf... sort of, there is so much that I do not have eyes on. Let me think.

Lines that are different from my hostapd.conf vs your posted.  only 4 lines different, 

---

+ ssid=BeelinkExternal

+ wpa_passphrase=JustAnotherPW

+ bridge=vmbr1

+ interface=wlxc83a35b4484d

---

I wanted to share the baseline before making any additional changes that might confuse the process as you mentioned that you needed to make changes.

> Okay. Let me see about changing my AP over to the same adapter as you have. Yes, I have one. I use a RasPi4B and it is dual band using 2 x usb wifi adapters.

Well, that should make things easier.... I hope.

> You are using hostapd but what else are your doing to setup the networking? systemd-networkd? have you read my AP guide? It is menu item 9 on the Main Menu. I'm not saying you have to do your setup like my guide but if you know my recommended way, it could help us communicate.

I dont believe so - systemctl shows it disabled

```
○ systemd-networkd.service - Network Configuration
     Loaded: loaded (/lib/systemd/system/systemd-networkd.service; disabled; preset: enabled)
     Active: inactive (dead)
TriggeredBy: ○ systemd-networkd.socket
       Docs: man:systemd-networkd.service(8)
             man:org.freedesktop.network1(5)

```
Nothing looks relevant in /etc/init.d/ either.

Everything I see online concerning network config just refers to adjusting /etc/network/interfaces including wired, wifi and bridges. Proxmox would **seem** to be restarting Networking through systemd (but not 100% sure on that).  I generally reboot after making changes, though changes through the WebGui require me to "Apply Configuration"

> I'll update my example WiFi-6 hostapd.conf when I move over to the new adapter.
>
> Something I need to point out: My example WiFi-6 hostapd.conf has not been used with an adapter with the rtl8832bu [...]

Thats why I wanted to share the results of the base config, in case you saw something right off the bat...

I did see recommended changes for hostapd in section 9, but I've always been an advocate of understanding the current state before complicating it with changes (especially when working blindly)

> Also, I still do not have a good handle on the Proxmox and what is in it. Is it based on Ubuntu or Debian?

/etc/debian_version says 12.2

I've never used Ubuntu (only Centos(RIP), Fedora and Debian), so I dont know if Ubuntu, as a Debian derivative, also carries the debian_version info.

morrownr, 2023-12-16:
> Lines that are different from my hostapd.conf vs your posted. only 4 lines different,

There should be more than 4 lines different. Here are 2 more lines that should be different:

```
# mt7921au  (HT capabilities 0x9ff)
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
```

```
# mt7921au (VHT capabilities 339071b2)
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
```

Those lines are specific to the mt7921au chipset. The same lines for the rtl8852bu chipset will be different. We need to calculate what they should be. This is getting close to rocket science and there are two ways to proceed. I can do it and give you the answers or I can go through the process of determining the answers with you. Your choice.

> Everything I see online concerning network config just refers to adjusting /etc/network/interfaces including wired, wifi and bridges.

Well, there are hundreds, if not thousands, of AP guides out there. There are hundreds of ways to set up the networking for an AP/router setup. In my AP guide I am showing users how to set up a BRIDGED setup. Bridged will work for most projects that users will want to do. If they really want a routed setup, I recommend they use OpenWRT. It sounds like a bridged setup will work for you. Therefore setting up your system networking for a bridged setup is what we need to do. Let me be clear about this: All we will end up doing is becoming frustrated if we are not on the same sheet of music. I strongly recommend, at least until you are more familiar with setting up an AP, that you use my AP guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

My guide only uses the following:

systemd-networkd (will need to be activated)
systemd-resolved (will need to be installed)
hostapd (will need to be installed)

Over the last several years I have tested many setups. My guide is the result of a LOT of testing. It leads to an extremely stable setup that is based on modern, well-mantained, components. It is also a setup that can provide the best performance that is available with your hardware. While the guide looks intimidating, as you become familiar with it, you should find that a lot of the guide is there for informational reasons so that you learn while using the guide. The guide currently has a lot of information but only goes to WiFi 5. The only changes needed for WiFi 6 are contained in the example WiFi 6 hostapd.conf which is still an external document. You need to ignore the internal hostapd.conf files and use the external WiFi 6 example.

What I am suggesting is that you start reading my guide, link above, and ask questions as you go. You can ignore the Raspberry Pi specific things. When you get to something you don't understand, stop and ask. It might be best if you do a clean install again unless you think you can restore networking to its original state. Without us having the networking setup properly, the hostapd setup does not matter... but I will be working on that while you get this networking stuff under control.

Fair enough?

@morrownr

slartibarfast, 2023-12-17:
I'm thinking this might be helpful in determining the appropriate values for configuration.  Unfortunately, the HT and VHT capabilities are not identified, and I have yet to find them referenced anywhere (there's no online datasheet that Google has indexed for the 8852bu - perhaps there's a clone chip?)

```

root@pve01:/usr/src/rtl8852bu-1.19.3# cat 8852bu.conf
# /etc/modprobe.d/8852bu.conf
#
# Purpose: Allow easy access to specific driver options.
#
# Edit the following line to change, add or delete options:
options 8852bu rtw_switch_usb_mode=0
#
# Note: To activate USB3 mode, change `rtw_switch_usb_mode=0` above to
# `rtw_switch_usb_mode=1`
#
# Note: The above `options` line is a good default for managed mode. Below is
# an example for AP mode. Modify as required after reading the documentation:
#options 8852bu rtw_switch_usb_mode=1 rtw_drv_log_level=0 rtw_vht_enable=2 rtw_dfs_region_domain=1
#
# After editing is complete, save this file (if using nano: Ctrl + x, y, Enter)
# and reboot to activate the changes.
#
# Documentation:
#
# -----
#
# USB options: ( rtw_switch_usb_mode )
#
# 0 = No switch (default)
# 1 = Switch from usb 2.0 to usb 3.0
# 2 = Switch from usb 3.0 to usb 2.0
#
# -----
#
# Log options ( rtw_drv_log_level )
#
# 0 = NONE (default)
# 1 = ALWAYS
# 2 = ERROR
# 3 = WARNING
# 4 = INFO
# 5 = DEBUG
# 6 = MAX
#
# -----
#
# VHT options ( rtw_vht_enable )
#
#  0 = Disable
#  1 = Enable (default)
#  2 = Force auto enable (use only for 5 GHz AP mode)
#
# Notes:
# - A non-default setting can degrade performance greatly in managed mode.
# - Option 2 allows 80 MHz channel width for 5GHz AP mode, such as when
#   you are using hostapd.
#
# -----
#
# DFS Options ( rtw_dfs_region_domain )
#
# 0 = NONE (default)
# 1 = FCC
# 2 = MKK
# 3 = ETSI
#
# Notes:
# - Activates DFS channels in AP mode.
# - DFS FCC 80 MHz channels for hostapd: 52(58), 100(106), 116(122) and 132(138)
# - For more information: https://en.wikipedia.org/wiki/List_of_WLAN_channels
[... 221 lines cut ...]
/sys/module/8852bu/parameters/rtw_stbc_cap:307
/sys/module/8852bu/parameters/rtw_switch_usb_mode:0
/sys/module/8852bu/parameters/rtw_sw_proto_bf_cap_phy0:130
/sys/module/8852bu/parameters/rtw_sw_proto_bf_cap_phy1:130
/sys/module/8852bu/parameters/rtw_tx_ampdu_amsdu:2
/sys/module/8852bu/parameters/rtw_tx_ampdu_num:0
/sys/module/8852bu/parameters/rtw_TxBBSwing_2G:255
/sys/module/8852bu/parameters/rtw_TxBBSwing_5G:255
/sys/module/8852bu/parameters/rtw_tx_bw_mode:817
/sys/module/8852bu/parameters/rtw_tx_nss:0
/sys/module/8852bu/parameters/rtw_tx_pwr_by_rate:1
/sys/module/8852bu/parameters/rtw_tx_pwr_lmt_enable:0
/sys/module/8852bu/parameters/rtw_usb_rxagg_mode:2
/sys/module/8852bu/parameters/rtw_user_regd_always_apply:0
/sys/module/8852bu/parameters/rtw_vcs_type:1
/sys/module/8852bu/parameters/rtw_vht_24g_enable:0
/sys/module/8852bu/parameters/rtw_vht_enable:1
/sys/module/8852bu/parameters/rtw_vht_rx_mcs_map:43690
/sys/module/8852bu/parameters/rtw_vo_edca:0
/sys/module/8852bu/parameters/rtw_vrtl_carrier_sense:1
/sys/module/8852bu/parameters/rtw_wifi_spec:0
/sys/module/8852bu/parameters/rtw_wireless_mode:63
/sys/module/8852bu/parameters/rtw_wmm_enable:1
/sys/module/8852bu/parameters/rtw_wowlan_sta_mix_mode:0
```

morrownr, 2023-12-17:
> 
```
# hw ht capab: ?
# ht_capab= to be determined
#
# hw vht capab: ?
# vht_capab= to be determined
```

You are correct. This file can be helpful... if the author would finish filling in the info.

Actually, if I can get you to do a `git pull` and take another look at the file, you will see updated information. The next question on your mind should be `how in the heck did he come up with that?` It takes 3 documents and `iw list`. Rocket science plays a roll as well. You have to put your EE hat on and go for it. I can show when you want.

I am having some problems with the driver in my RasPi4B dual band AP. I'm not sure if it the latest RasPiOS which is late Alpha quality or if it is the driver. I flipped setting back to use a mt7921au based adapter and it works perfectly:

```
$ iw dev
phy#1
	Interface wlxxxx (this is an Alfa AXML that uses a mt7921au chipset)
		ifindex 5
		wdev 0x100000001
		addr e0:e1:a9:35:c5:1f
		ssid myPI-WiFi6
		type AP
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
phy#0
	Interface wlxxxx (this is a Alfa ACM that uses a mt7612u chipset)
		ifindex 4
		wdev 0x1
		addr 00:13:ef:6f:0a:98
		ssid myPI-WiFi4
		type AP
		channel 11 (2462 MHz), width: 40 MHz, center1: 2452 MHz
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0

```

It just works and there is no need to install a driver for either adapter. Let me head @bjlockie off before he jumps in and says `I told you so.` Yes, I know. I'm trying to help so hold on and see where we get.

I have an empty partition on a x86_64 system so I think I will install a fresh download of Debian 12.4 and see what happens with something stable. I'll try to do it sometime this week.

@morrownr

morrownr, 2023-12-23:
@slartibarfast 

I'm checking in with you. As I said above, my initial attempts to get the rtl8852bu driver going in WiFi 6 AP mode were not successful. My additional attempts this week were also unsuccessful. Background: The process of making the rtl8852bu driver public was a challenge. I know this is not a very good driver. Most of my work concentrated on managed mode and a lot of basic stuff that simple was not working. I did a quick AP mode test with WiFi 4 and 2.4 GHz band while working on the driver due to me being tired of all the work I had done. It appears that I need to change the README to note that AP mode is not working well.

Is it possible that AP mode can be fixed? I don't know. I need to press on with other things I am working on for now. I'd say the best chance to see an AP mode that is working well would be with new release of the driver and when that will happen is unknown.

Therefore, to save yourself a lot of frustration, you may want to look at returning the adapter you have while looking for an adapter with a chipset known to work well. For WiFi 6 AP mode, the mt7921au chipset based adapters are known to work well. I have two adapters with the mt7921au chipset. Both work very well with the WiFi 6 hostapd.conf I have up and they work well with the AP guide I have up. I rank them as ultra stable. There are many ways to setup an AP and some ways are less stable than others.

The mt7921au based adapters that I have are the Alfa AXML and the Comfast CF-951AX. I have seen issues with the CF-951AX working with extension cables and powered hubs which is likely a power problem caused by the Comfast engineering in this adapter. The Comfast CF-953AX seems to be a better adapter based on the user reports I have seen. The Alfa AXM would also seem to be good adapter for this use case and the Netgear A8000 should be as well based on user reports. All of these adapters and more are shown in the Plug and Play list:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

The current state of the WiFi 6 Linux drivers from Mediatek and Realtek is such that Mediatek's driver are FAR superior.

slartibarfast, 2023-12-23:
Hey, thanks for the nudge

I was able to get the latest hostapd changes to bring the network device to an "UP" state, but it would not enter into AP mode unless I used that old, unsupported "createap" project that stuck me at 802.11a.  As the device is a Wifi6 adapter, that just wouldnt cut it...

So I feel I must agree with you and @bjlockie...  This realtek chip is more a PITA than I was lead to believe.   Does it load into Linux? Yes.  Does it do what I need it to do?  No.  Life is all about learning, right?  Even when things fail, its still a learning that can be put to use.

I have just ordered the ALFA AWUS036AXML.   Should get it by new years... and I can pick up on my journey when it arrives.

morrownr, 2023-12-24:
> Life is all about learning, right? Even when things fail, its still a learning that can be put to use.

I have learned this lesson many times over.

> I have just ordered the ALFA AWUS036AXML

I have enjoyed my AXML. It has some features that other adapters do not. The antennas are removal which will allow you to change out for directional antennas if you find a need. The USB cable that comes with it is high quality and it is USB-C on both ends. It has a USB-C to USB-A converter if you need it. Remember, there is no driver to compile unless you download mainline and compile your own kernel. Remember that in-kernel drivers consist of 2 or more files...the driver files and the firmware files. The firmware files are part of your distro, not part of the kernel. At times either you or your distro maintainer may have to update the firmware files .... like when an update comes out. Most people don't bother and just let their distro handle it.

This chipset should auto-detect whether the adapter is in a USB3 or USB2 port and adjust accordingly. The in-kernel drivers are like driving a car with an automatic transmission. There is no shifting to be done.

We keep a pinned issue here where we keep a log of bugs and feature requests:

https://github.com/morrownr/USB-WiFi/issues/107

The bug about the LED does not apply to you as Alfa used a clever solution for their adapters. We attempt to work with the Mediatek devs to get a resolution to issues. I can't think of any bugs on STA or AP mode currently that should cause any problems for you. If you have any problems, let us know. 

@morrownr

DarthShredder, 2024-01-06:
> I am trying to use your Wifi6 config to enable Wifi6 on this Realtek 8852u based wifi usb dongle

The BrosTrend AX1800 (ax4l) ... It took me a while to figure it all out. Maybe this will help someone.

The problem is that the 5Ghz frequencies are set to "no IR". This is because the WLAN stick is not set to a country. "iw reg set" does not work here either. 
This must be done in the module configuration /etc/modprobe.d/8852bu.conf . 
It looks like this for me:
"options 8852bu rtw_switch_usb_mode=1 rtw_he_enable=1 rtw_vht_enable=1 rtw_country_code=DE rtw_dfs_region_domain=3"

If you live in the USA, you might want "rtw_country_code=US rtw_dfs_region_domain=1"

Fun fact: after setting the country_code in the module, the blue LED on the stick also lights up

If you set this and reconnect the stick (you can also use "rmmod 8852bu" and "modprobe 8852bu"), you will still see no change under "iw reg get" (until hostapd is running), but the 5Ghz frequencies without "no IR" will now appear in "iw list" and with hostapd "hw_mode=a" will now work. Please note that the automatic channel check did not work for me and the interface went down again shortly after starting hostapd without finding a channel. Therefore you have to specify the channel in hostapd like in the 2.4Ghz setup before (example: "channel=44"). "iw yourdevice scan | grep freq" shows well what the neighbors are using. Using "iw list" you can then determine the channel number for an frequency that did not appear during the scan ...

In conclusion, I can say that Realtek WLAN sticks are not optimal under Linux. I am grateful that there are developers who create drivers that also cover the AP feature. 
But if I had to recommend a WLAN card for Linux, it would not be a Realtek. 
I also don't know why I always end up with Realtek WLAN cards/sticks  ¯\_(ツ)_/¯

slartibarfast, 2024-01-06:
Fantastic!

slartibarfast, 2024-01-06:
@DarthShredder,

I'm not using that realtek usb dongle anymore either, but this solved the same problem that I ran into with an ALFA device.

I added a `/etc/modprobe.d/local.conf` with your options and the hostapd config that I had been fighting with for most of the week came up as a Wifi6 (6e?) AP with no other changes after a reboot.

A couple things I'm not understanding though (but isn't stopping me from using it) is that my hostapd config says to use channel 36, but according to the spectrum analyzer on my phone, the actual channel is 42 at a distance of 3m (and the phone is less than 1m away)  The distance error is probably an issue with a noisy environment, so that doesn't really concern me.

Thank you for the key to unlocking the 5Ghz spectrum

alikaylan, 2024-01-13:
> @DarthShredder,
>
> I'm not using that realtek usb dongle anymore either, but this solved the same problem that I ran into with an ALFA device.
>
> I added a `/etc/modprobe.d/local.conf` with your options [...]

What are the options you used for ALFA? Thanks.

slartibarfast, 2024-01-14:
@alikaylan , I used more or less the same options line @DarthShredder suggested, except I changed the country code to US even though it wasnt my final choice. in my Wifi6 config I override the country code (which was not possible prior to adding the module options - it would always fail to set the country and the interface would never come up)

`options 8852bu rtw_switch_usb_mode=1 rtw_he_enable=1 rtw_vht_enable=1 rtw_country_code=US rtw_dfs_region_domain=3`

This is my current Wifi6 config:

```
auth_algs=3
beacon_int=100
bridge=vmbr0
channel=36
chanlist=36,40,44,48,52,56,60,64,100,104,108,112,116,120,124,128,132,136
country_code=CA
ctrl_interface_group=0
ctrl_interface=/var/run/hostapd
disassoc_low_ack=1
driver=nl80211
dtim_period=2
eapol_key_index_workaround=0
eap_server=0
fragm_threshold=-1
he_oper_centr_freq_seg0_idx=42
he_oper_chwidth=1
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
hw_mode=a
ieee80211ac=1
ieee80211ax=1
ieee80211d=1
#ieee80211h=1
ieee80211n=1
ieee80211w=1
#ieee80211w=2
ignore_broadcast_ssid=0
interface=wlan0
logger_stdout=-1
logger_stdout_level=2
logger_syslog=-1
logger_syslog_level=2
macaddr_acl=0
max_num_sta=255
multi_ap=0
multicast_to_unicast=0
okc=1
own_ip_addr=127.0.0.1
preamble=1
rts_threshold=-1
sae_require_mfp=1
skip_inactivity_poll=0
ssid=TestWifi
tx_queue_data2_burst=2.0
uapsd_advertisement_enabled=1
utf8_ssid=1
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
vht_oper_centr_freq_seg0_idx=42
vht_oper_chwidth=1
wmm_ac_be_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmax=10
wmm_ac_be_cwmin=4
wmm_ac_be_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_bk_aifs=7
wmm_ac_bk_cwmax=10
wmm_ac_bk_cwmin=4
wmm_ac_bk_txop_limit=0
wmm_ac_vi_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmax=4
wmm_ac_vi_cwmin=3
wmm_ac_vi_txop_limit=94
wmm_ac_vo_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_cwmin=2
wmm_ac_vo_txop_limit=47
wmm_enabled=1
wpa=2
wpa_key_mgmt=SAE WPA-PSK
wpa_pairwise=CCMP CCMP-256
wpa_passphrase=******

##########
# needed to add to /etc/modprobe.d/local.conf
#
# options rtw_switch_usb_mode=1 rtw_he_enable=1 rtw_vht_enable=1 rtw_country_code=US rtw_dfs_region_domain=1
#########
```

I stripped the comments out of @morrownr Wifi6 config and sorted the lines, but its essentially the same.  Just easier for me to find what I was looking for.

This config brings up a Wifi6 network, bridged to my system's main WAN interface through an existing bridge.  WPA2 security is in place, and I regularly see 400 Mbps out of the connection (with only my phone connected - I'm sure it will degrade once I have additional devices on it

alikaylan, 2024-01-14:
Thanks, much appreciated.

morrownr, 2024-01-15:
@DarthShredder 

> options 8852bu rtw_switch_usb_mode=1 rtw_he_enable=1 rtw_vht_enable=1 rtw_country_code=DE rtw_dfs_region_domain=3

For you to obtain 80 MHz channel width, change the following:

rtw_he_enable=2 rtw_vht_enable=2

It look like I need to update the docs in 8852bu.conf to make this setup more clear.

@morrownr