morrownr/USB-WiFi #353: Recommendation for AP on 5 GHz DFS Channels

m4rkusxxl, 2023-12-16:
In Performance_Comparison.md you are recommending for master mode (AP) only the "Alfa AWUS036ACH" with DFS support.
Which is realtek 8812au, which is not recommented in USB_WiFi_Chipsets.md and described as deadend in USB_WiFi_Adapter_out-of-kernel_drivers_for_Linux.md.

In The_Short_List you recommend for 5 GHz AP mt7610u (ALFA AWUS036ACHM) and mt7612u (ALFA AWUS036ACM & TEROW ROW02CD). But they seem to lack DFS support?!

I already use the ALFA AWUS036ACHM for AP on 2.4 GHz, but I wasnt able to get it running on 5 GHz DFS channels. The other 5 GHz channels are limited heavily in power and thus have only small range. (At least here in my country.)
Hostapd always bails out with an error:
```
nl80211: Start radar detection (CAC) 5500 MHz (ht_enabled=1, vht_enabled=1, he_enabled=0, bandwidth=80 MHz, cf1=5530 MHz, cf2=0 MHz)
  * freq=5500
  * he_enabled=0
  * vht_enabled=1
  * ht_enabled=1
  * bandwidth=80
  * channel_width=3
  * center_freq1=5530
  * center_freq2=0
nl80211: Failed to start radar detection: -16 (Device or resource busy)
DFS start_dfs_cac() failed, -1
Interface initialization failed
wlan0: interface state DFS->DISABLED
```

So my question:
**What USB wifi stick for an AP on 5 GHz DFS channels can be recommended?**
An in-kernel driver is prefered for easy long term support.

morrownr, 2023-12-16:
Hi @m4rkusxxl 

> In Performance_Comparison.md you are recommending for master mode (AP) only the "Alfa AWUS036ACH" with DFS support. Which is realtek 8812au, which is not recommented in USB_WiFi_Chipsets.md and described as deadend in USB_WiFi_Adapter_out-of-kernel_drivers_for_Linux.md.

All of the above is true and I stand by it.

> In The_Short_List you recommend for 5 GHz AP mt7610u (ALFA AWUS036ACHM) and mt7612u (ALFA AWUS036ACM & TEROW ROW02CD). But they seem to lack DFS support?!

This is also true. I recently sent a message to the Mediatek devs hoping that they can convince management that DFS support is important. Hopefully this works. Additional certifications are required for AP DFS support in many countries. If your country happens to not require this certification, you can add code to turn on the capability for the mt7612u driver:

mt76x2_init.c

@@ -702,6 +702,10 @@ static const struct ieee80211_iface_combination if_comb[] = {
		.max_interfaces = 8,
		.num_different_channels = 1,
		.beacon_int_infra_match = true,
+		.radar_detect_widths = BIT(NL80211_CHAN_WIDTH_20_NOHT) |
+				       BIT(NL80211_CHAN_WIDTH_20) |
+				       BIT(NL80211_CHAN_WIDTH_40) |
+				       BIT(NL80211_CHAN_WIDTH_80),
	}
};

In no way am I suggesting you do something that is illegal in your country. The above might work with the mt7610u driver but has not been tested here.

> The other 5 GHz channels are limited heavily in power and thus have only small range. (At least here in my country.)

I have an ACHM. The ACHM has really good range and it does not come from power as much as it does from the really high quality antenna and amp.

> What reliable USB wifi stick for an AP on 5 GHz DFS channels can be recommended for continious operation?
An in-kernel driver is prefered for easy long term support.

I wish I had a good answer for you. This is a problematic issue for Linux usb wifi adapter users. I have been doing as much as I know how to do by talking to sellers of Mediatek based usb wifi adapters and I have sent messages to the Mediatek devs hoping they can convince Mediatek to get the certifications so the capability can be added.

Depending on your situation, an Alfa ACH with the rtl8812au chipset might be the best answer... as long as you understand that Realtek has terminated their support for the out-of-kernel driver and it is likely that no in-kernel driver will ever happen. My rtl8812au based adapter is the Alfa ACH. Wow, it is a really good adapter and has range to spare. I have a slightly newer version of the driver code than is currently publicly available here. My intention is get it ready to be released as I have time and some volunteers to help. The rtl8812au out-of-kernel driver is probably the single best Realtek driver I have experienced. I'll try to keep it going as long as I can.

Now you know what I know.

Cheers,

@morrownr

m4rkusxxl, 2023-12-17:
Thank you very much for the comprehensive answer.

Here in Germany we have only two 80 MHz channels without DFS: 42 (with 0.2 W and pretty filled-up already) and 155 (with 0.025 W, which is useless).
I used 42, but the level was much lower than the 2.4 GHz wifi, so I thought aiming for DFS capable wifi would be beneficial.
The other channels 58 (0.2 W), 106 (1 W), 122 (1 W), 138 (0.025W) require all DFS, but are pretty free here at my home location.
That being said, I am pretty sure a certification would be needed here. But I do not need to get my existing mt7610u running at 5 GHz, I will have to keep it for 2.4 GHz. Not everything is ready for 5 GHz, yet.

So the result is the Alfa AWUS036ACH with [rtl8812au](https://github.com/morrownr/8812au-20210629). Not perfect driver-wise, but (currently) the only choice with the given requirements.

closed 2023-12-17 by m4rkusxxl

morrownr, 2023-12-18:
If you decide to get the ACH, contact me here if you are interested in helping me bring the new code to life. I will first create a private repo and I will add those interested in testing. It can take a while from start to going public but you might enjoy it.

@morrownr