morrownr/USB-WiFi #354: mt7612u awus036ACM [Bad usb?] "hci_xcd" "kernel error = vendor request" ?

dontholdmyhand, 2023-12-16:
Ive keep getting this error from inserting my usb adapter into my VM.
I cant get any wifi and the device wont even show in "ip link" or "iwconfig", but does appear in lsusb.
I have tried plugging it into different usb ports on my pc and tried plugging it without its extension cord.

Im currently using virtualbox running 6.5.0-kali3 (i have also tried parrot and i couldnt get it to work)

I have also tried  vmware and was met with similar results

the device DOES work properly on my windows host, except for after being used by vm and it tries to connect back to host i get this error message sometimes:
[image]
from windows device manager:
[image]

also sometimes my computer wont turn on if the adapter is plugged into the pc before boot

On VMs if i use usb 2 controller it wont appear in my "lsusb" but with usb 3 it does but it still wont bring wlan up

The error i keep getting on kali:
[image]
journal:
[image]

parrot: 
[image]

Does this mean the adapter itself is broken? or is there an issue on my part that i cant find?
I also have a awus036ACHM that works perfectly fine, minus having the set reg every time i connect it, and that fact that depreciated iwconfig on linux breaks the driver when switching to monitor mode but i got to fix that with the help of Zerbea @hcxdumptools

morrownr, 2023-12-18:
Hi @dontholdmyhand 

> Ive keep getting this error from inserting my usb adapter into my VM.

I may not be much help as I don't use usb wifi adapters with a VM. Hopefully someone with more knowledge will stop by this week. I have probably seen more reported issues regarding VM's over the last few years than any other single issue. The problem is almost never with the driver or adapter. It is almost always a problem with how the VM is configured.

>  have tried plugging it into different usb ports on my pc and tried plugging it without its extension cord.

There are things you have to configure in the VM for an adapter to work.

> I have updated the drivers on widowns through the link provided on alfa networks site on Windows[Host] and Kali[VM]

How did you update the driver for the Alfa ACM in Kali... given that it is an in-kernel driver?

> Does this mean the adapter itself is broken? 

Unlikely. I have both the Alfa ACM and ACHM here. They are well made and the Linux drivers are solid.

> or is there an issue on my part that i cant find?

My thoughts are that it is almost certainly a problem with the configuration of the VM. I have no problems with either adapter but I run on bare metal... with many distros.

>  ive yet to try this on a distro other than debian based.

Works fine here on other distros. Try them on bare metal.

> I also have a awus036ACHM that works perfectly fine, minus having the set reg every time i connect it, and that fact that depreciated iwconfig on linux breaks the driver when switching to monitor mode but i got to fix that with the help of Zerbea @hcxdumptools

Then don't use the depreciated iwconfig. Use `iw` and `ip`. I know @zerbea. He knows a lot more about monitor mode than I do. I more of a specialist in AP mode. Here is a link on the Rokland website:

https://store.rokland.com/pages/alfa-awus036acm-kali-virtual-box-instructions

Let me know if it helps and I will put the link here on this site.

Cheers,

@morrownr

dontholdmyhand, 2023-12-19:
> > I have updated the drivers on widowns through the link provided on alfa networks site on Windows[Host] and Kali[VM]
> 
> How did you update the driver for the Alfa ACM in Kali... given that it is an in-kernel driver?
> 
>> I didnt update them per say, i removed them from the mediatek dir, and downloaded mt7662u.bin and rom-patch.bin and replaced them just in case the ones on my kali vm were out of date

> > Does this mean the adapter itself is broken?
> 
> Unlikely. I have both the Alfa ACM and ACHM here. They are well made and the Linux drivers are solid.
> 
>> I meant could it be possible that the piece that i bought could be malfunctioning

> > or is there an issue on my part that i cant find?
> 

> My thoughts are that it is almost certainly a problem with the configuration of the VM. I have no problems with either adapter but I run on bare metal... with many distros.
> 
>> I currently cant bare metal any other os on my host cause its my only machine sadly, soon though hopefully thats the plan

> > I also have a awus036ACHM that works perfectly fine, minus having the set reg every time i connect it, and that fact that depreciated iwconfig on linux breaks the driver when switching to monitor mode but i got to fix that with the help of Zerbea @hcxdumptools
>
> Then don't use the depreciated iwconfig. Use `iw` and `ip`. I know @ZerBea. He knows a lot more about monitor mode than I do. I more of a specialist in AP mode. Here is a link on the Rokland website:
>
>> oh for sure i dont anymore at all, im still new to linux ive been a windows non technical person my entire life but im pivoting into it now, @ZerBea shun alot of lights on so many things for me, and sent me down many amazing rabbit hole(thank you) 

> https://store.rokland.com/pages/alfa-awus036acm-kali-virtual-box-instructions
> 
> Let me know if it helps and I will put the link here on this site.
> 
>>definitely will do, thank you so much, and ive also found 
https://forums.kali.org/showthread.php?128646-SOLVED-Alfa-AWUS036ACM-amp-Kali-2022-4-on-VB
>>from https://github.com/morrownr/7612u , but i hope its this cause the same issue was with vmware too so, hopefully its just a virtual machine issue.

thank you so much for your time and help

dontholdmyhand, 2023-12-19:
just coming to update that, https://store.rokland.com/pages/alfa-awus036acm-kali-virtual-box-instructions, did not work for me, i still got the same errors, when i was tryna get it to work before i posted this i tried a bunch of things like this but i only ever managed to get it register with lsusb but never show up as wlan0 now its not showing up as lsusb at all.

ill try with bare metal soon hopefully and update back

morrownr, 2023-12-19:
Have you investigated using multi-boot instead of a VM? I use multi-boot on all of my systems. I don't use Windows but rather I do it to have multiple distros available on each system for testing or for a backup. Tht would allow you to keep Windows and have one or more partitions of Linux. A menu on boot allows you to decide what os to boot.

morrownr, 2023-12-19:
Correction: In the above message I used the term multi-boot. I should have used `dual boot`. If you do a search on `linux dual boot` , you will find a lot of information.

I use dual boot on every system I have. Dual does not mean only 2 OS's installed as I have 4 on two test systems. Dual boot allows you to keep any existing OS in tack but allows to boot to other OS's based on a menu that comes up at boot time. Dual boot allows all installed OS's to work on bare metal so VM's and their problems are eliminiated.

dontholdmyhand, 2023-12-19:
> Correction: In the above message I used the term multi-boot. I should have used `dual boot`. If you do a search on `linux dual boot` , you will find a lot of information.
>
> I use dual boot on every [...]

i originally wanted to try a usb boot on my host instead of a vm, but my computer is very old and sensitive, im afraid i might break something to the point of no return.

i will be borrowing a laptop in a couple of days that i can test all sorts of boots for baremetalling and ill report back

morrownr, 2023-12-19:
Looking forward to your report.

I've been using Linux since '94 and setting up dual boot is just second nature for me but if you have no experience doing it, it would be best to practice first on a system where is does not matter if you mess things up.

dontholdmyhand, 2023-12-19:
> if you have no experience doing it, it would be best to practice first on a system where is does not matter if you mess things up.

exactly how i feel about it considering i cant replace the stuff i break lol

>I've been using Linux since '94 and setting up dual boot is just second nature for me

thats amazing actually i was barely 3 years old in 94 lol, but its never too early to start learning, i wish i knew i had a passion for this at an earlier age, ive basically been pcs and networks daily since i was a kid but never thought to delve deeper into them, instead i have thousands of hours on random video games lol

dontholdmyhand, 2023-12-22:
Just a quick little update, i managed to get the adapter to work by plugging it into a USB 3.1 instead of the 3.0 i was previously plugging it in, either my mobo is really weird with it or something is up, all i did was change my computer case and the front panel usb 3.1 decided it wanted to work this time.

for more context my mobo has 4 usb 3 ports in the back that i was trying to use too but kept getting error'd 
[image]

still going to testing on bare metal too soon

morrownr, 2023-12-22:
> either my mobo is really weird with it or something is up, all i did was change my computer case and the front panel usb 3.1 decided it wanted to work this time.

I have one desktop system where I experienced some USB3 issues so I got a card. The link below shows the card:

https://www.amazon.com/dp/B0BJPMVJ92

Ieft the USB2 ports working with the MB but but now USB3 support comes from this card. Very stable and compatible. USB3 is not mankind's greatest invention.

You would need to research whether that card would work with your MB.

@morrownr

dontholdmyhand, 2023-12-23:
> I have one desktop system where I experienced some USB3 issues so I got a card. The link below shows the card:
>
> https://www.amazon.com/dp/B0BJPMVJ92
>
> Ieft the USB2 ports working with the MB but but [...]

a card is pretty sweet lol but thankfully i didnt need to change anything other than the case and use the frontal 3.1 socket.

i just finished trying the ACM on a usb version of kali on a laptop i borrowed and it works perfectly, i can only assume itll work just fine as well on a partitioned normal build, thank you so much for the help, i was worried i bought a dud but it was just my weird ass computer/virtual machines. 

should i mark as closed or idk (im still new to github)

morrownr, 2023-12-23:
This is good news. I think you are headed in the right direction.

Please leave this issue open as it contains a lot of good information. I checked the numbers a couple of days ago. This site up to around 22,500 hits per week. I lot of people come by to read the info and news in these issues so I like to leave any issue that has good information open.

Good luck and stop by anytime.

@morrownr

dontholdmyhand, 2023-12-23:
> This is good news. I think you are headed in the right direction.
>
> Please leave this issue open as it contains a lot of good information. I checked the numbers a couple of days ago. This site up to [...]

I really appreciate it, and i hope my post will help someone out. 

last quick update, I tried it on bare metal and it works perfectly fine too.

have a good one

dontholdmyhand, 2023-12-24:
sorry this is like a very irrelevant question but idk where else to ask.

with the dual boot linux (im using kali) the entire hdd is visible and accessible on the desktop and under devices, is there anyway to disable access to that through or hide it through my kali

my main point of using kali is to delve deeper into messing with things and learn stuff but you know when youre learning youre bound to download some malicious stuff sometimes.
im thinking if someone managed to access my linux then by default they have access to my windows partition too

i understand that linux can see ntfs so its there but is there a way to unthere it?

am i just better off using a live usb kali instead?