morrownr/USB-WiFi #355: Station Mode not working for MT7921u / Comfast CF-953AX

cryptotabluvox, 2023-12-17:
This adapter works fine in kernel 6.3, 6.4, 6.5, 6.6.3. Recently when I upgraded my system (arch linux) from 6.6.3 to 6.6.5, the issue start happening.

The issue is when replugged the adapter whether on hub or direct USB 3 port, it will recognized in lsusb for a while, then it went missing. I have tried numerous ways to disable autosuspend using udev rules, re-install firmware as guided here, even reinstalling the system from scratch. In archiso boot medium, I was able to use my adapter fine because the 2023.12.01 archiso is running on kernel 6.6.3. I can't figure out what other way to fix this issue.

Below are the output of lsusb, dmesg and journalctl.

lsusb
Bus 002 Device 009: ID 0e8d:7961 MediaTek Inc. Wireless_Device

sudo dmesg | grep usb
[ 3290.952210] scsi host2: usb-storage 2-1.4:1.0
[ 3291.599721] usb 2-1.1: new SuperSpeed USB device number 27 using xhci_hcd
[ 3291.618435] usb 2-1.1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 3291.618447] usb 2-1.1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 3291.618451] usb 2-1.1: Product: Wireless_Device
[ 3291.618454] usb 2-1.1: Manufacturer: MediaTek Inc.
[ 3291.618457] usb 2-1.1: SerialNumber: 000000000
[ 3311.479337] usb 2-1.1: device not accepting address 27, error -62
[ 3322.362670] usb 2-1.1: device not accepting address 27, error -62
[ 3333.239283] usb 2-1.1: device not accepting address 27, error -62
[ 3344.122686] usb 2-1.1: device not accepting address 27, error -62
[ 3344.123571] usb 2-1.1: USB disconnect, device number 27
[ 3354.999292] usb 2-1.1: device not accepting address 28, error -62
[ 3365.879430] usb 2-1.1: device not accepting address 29, error -62
[ 3365.879836] usb 2-1-port1: attempt power cycle
[ 3376.975963] usb 2-1.1: device not accepting address 30, error -62
[ 3387.852709] usb 2-1.1: device not accepting address 31, error -62
[ 3387.853013] usb 2-1-port1: unable to enumerate USB device

[   34.466661] usb 2-1.1: device not accepting address 5, error -62
[   45.346659] usb 2-1.1: device not accepting address 6, error -62
[   45.347168] usb 2-1-port1: unable to enumerate USB device

[ 2905.519546] usb 2-1.1: new SuperSpeed USB device number 9 using xhci_hcd
[ 2905.537862] usb 2-1.1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 2905.537874] usb 2-1.1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 2905.537878] usb 2-1.1: Product: Wireless_Device
[ 2905.537881] usb 2-1.1: Manufacturer: MediaTek Inc.
[ 2905.537883] usb 2-1.1: SerialNumber: 000000000
[ 2926.199465] usb 2-1.1: device not accepting address 9, error -62
[ 2937.079478] usb 2-1.1: device not accepting address 9, error -62
[ 2947.959479] usb 2-1.1: device not accepting address 9, error -62

[ 3440.342776] Workqueue: usb_hub_wq hub_event
[ 3440.343067]  usb_reset_and_verify_device+0x1a3/0x470
[ 3440.343080]  usb_reset_device+0x133/0x280
[ 3440.343121]  usb_probe_interface+0xe3/0x2f0
[ 3440.343202]  usb_register_driver+0x8c/0x150
[ 3563.219415] Workqueue: usb_hub_wq hub_event
[ 3563.219617]  usb_reset_and_verify_device+0x1a3/0x470
[ 3563.219625]  usb_reset_device+0x133/0x280
[ 3563.219653]  usb_probe_interface+0xe3/0x2f0
[ 3563.219714]  usb_register_driver+0x8c/0x150

journalctl | grep mt7921u
Dec 17 00:54:03 archlinux kernel: usbcore: registered new interface driver mt7921u
Dec 17 00:54:03 archlinux kernel: mt7921u 2-1.1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
Dec 17 00:54:03 archlinux kernel: mt7921u 2-1.1:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
Dec 17 01:01:39 archlinux kernel: usbcore: registered new interface driver mt7921u
Dec 17 01:01:39 archlinux kernel: mt7921u 2-1.1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
Dec 17 01:01:39 archlinux kernel: mt7921u 2-1.1:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
Dec 17 01:10:18 archlinux kernel: usbcore: registered new interface driver mt7921u
Dec 17 01:10:18 archlinux kernel: mt7921u 2-1.1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
Dec 17 01:10:18 archlinux kernel: mt7921u 2-1.1:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
Dec 17 01:10:20 archlinux NetworkManager[519]: <info>  [1702746620.4810] rfkill6: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.3/ieee80211/phy1/rfkill6) (driver mt7921u)
Dec 17 03:04:44 archlinux kernel: usbcore: registered new interface driver mt7921u
Dec 17 03:04:44 archlinux kernel: mt7921u 2-1.1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
Dec 17 03:04:44 archlinux kernel: mt7921u 2-1.1:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
Dec 17 03:04:45 archlinux NetworkManager[483]: <info>  [1702753485.8723] rfkill5: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.3/ieee80211/phy1/rfkill5) (driver mt7921u)
Dec 17 03:06:39 archlinux kernel: mt7921u 2-1.1:1.3: timed out waiting for pending tx
Dec 17 03:07:31 archlinux kernel: mt7921u: probe of 2-1.1:1.3 failed with error -5
Dec 17 03:10:26 archlinux kernel: mt7921u: probe of 2-1.1:1.3 failed with error -5

Dec 17 04:26:52 archlinux sudo[3470]: hyprmann : TTY=pts/1 ; PWD=/home/hyprmann ; USER=root ; COMMAND=/usr/bin/nano /etc/udev/rules.d/90-mt7921u.rules
Dec 17 04:28:55 archlinux kernel: mt7921u: probe of 2-1.1:1.2 failed with error -22
Dec 17 04:28:55 archlinux kernel: mt7921u: probe of 2-1.1:1.3 failed with error -5
Dec 17 04:35:06 archlinux kernel: mt7921u: probe of 2-1:1.2 failed with error -22
Dec 17 04:35:06 archlinux kernel: mt7921u: probe of 2-1:1.3 failed with error -5
Dec 17 04:39:36 archlinux kernel: mt7921u: probe of 2-1.1:1.2 failed with error -22
Dec 17 04:39:36 archlinux kernel: mt7921u: probe of 2-1.1:1.3 failed with error -5

Dec 17 04:59:52 archlinux kernel: Modules linked in: ntfs3 ufs hfsplus hfs cdrom minix msdos jfs nls_ucs2_utils xfs ext4 mbcache jbd2 algif_aead crypto_null cbc des_generic libdes ecb algif_skcipher md4 algif_hash af_alg mt7921u mt792x_usb mt7921_common mt792x_lib mt76_connac_lib mt76_usb mt76 snd_seq_dummy snd_hrtimer snd_seq snd_seq_device ses enclosure scsi_transport_sas rtsx_usb_sdmmc rtsx_usb_ms mmc_core memstick rtsx_usb btusb uvcvideo btrtl btintel videobuf2_vmalloc btbcm uvc btmtk videobuf2_memops videobuf2_v4l2 bluetooth videodev videobuf2_common razerkbd(OE) ecdh_generic mc crc16 uas usb_storage cmac ccm snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic nvidia_drm(POE) nvidia_modeset(POE) pkcs8_key_parser nvidia_uvm(POE) nvidia(POE) snd_sof_pci_intel_icl snd_sof_intel_hda_common soundwire_intel snd_sof_intel_hda_mlink intel_tcc_cooling soundwire_cadence snd_sof_intel_hda snd_sof_pci x86_pkg_temp_thermal snd_sof_xtensa_dsp intel_powerclamp snd_sof coretemp snd_sof_utils snd_soc_hdac_hda kvm_intel snd_hda_ext_core
Dec 17 12:36:03 archlinux kernel:  mt7921u_probe+0xab/0x230 [mt7921u 5dd5f6ca5b00e9b24915aaa73636bb750e4c84ef]
Dec 17 12:36:03 archlinux kernel:  ? __pfx_mt7921u_driver_init+0x10/0x10 [mt7921u 5dd5f6ca5b00e9b24915aaa73636bb750e4c84ef]
Dec 17 12:38:06 archlinux kernel:  mt7921u_probe+0xab/0x230 [mt7921u 5dd5f6ca5b00e9b24915aaa73636bb750e4c84ef]
Dec 17 12:38:06 archlinux kernel:  ? __pfx_mt7921u_driver_init+0x10/0x10 [mt7921u 5dd5f6ca5b00e9b24915aaa73636bb750e4c84ef]

morrownr, 2023-12-26:
Status of this issue?

I agree with @Ekiben542 

This issue can be closed if the issue is fixed.

cryptotabluvox, 2023-12-26:
The issue is still happening on 6.6.8. Is there any workaround? Thanks.

morrownr, 2023-12-26:
> Is there any workaround?

We kinda need to know what the problem is first.

I have a test system that is using kernel 6.7 rc5 or 6. I'll take a look.

> Bus 002 Device 009: ID 0e8d:7961 MediaTek Inc. Wireless_Device

That is the default Mediatek vip/pid which is good but it doesn't tell me what specific adapter that you have.

Can you also provide computer hardware information?

> archlinux NetworkManager[519]: [1702746620.4810] rfkill6: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.3/ieee80211/phy1/rfkill6) (driver mt7921u)

That is interesting.

cryptotabluvox, 2023-12-27:
The adapter that I have is Comfast CF-953AX bought on AliExpress that's provided here.

As for the hardware information, it's an ASUS A409JP laptop. The CPU is i5-1035G1, 8GB Kingston DDR4 memory, 2TB Kingston KC3000 NVME. It has onboard wifi card, Intel Wireless-AC 9461 which registered as phy0/wlan0.

Please let me know if you need more information.

morrownr, 2023-12-27:
Nice laptop. The CD-953AX seems like a decent adapter as I continue to watch message traffic and see a lot of people happy with it.

Try the following and post the results:

$ rfkill list all

cryptotabluvox, 2023-12-27:
Thanks. Here's the results:

rfkill list all                                                                                                         ✔ 
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: asus-wlan: Wireless LAN
	Soft blocked: no
	Hard blocked: no
2: asus-bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
3: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

morrownr, 2023-12-27:
Well, that does not give us anything we can use.

I just did some testing on a test system with kernel 6.7 rc5 and the adapter works fine. I am stumped.

Have you looked around the arch forums? Maybe others are seeing the same thing.

You could boot the laptop with a usb flash driver with something like Ubuntu 23.10 on it but you said it was working until a very recent kernel upgrade in arch. 

My only suggestion right now to get you going is to backup to the last kernel that was working and wait until this gets sorted out.

cryptotabluvox, 2023-12-29:
Thanks, I will look around arch forums. I just tested on Pop OS kernel 6.6.6. Works fine. Not sure why it won't work in Arch. On Pop OS, it works flawlessly plug and play.

cryptotabluvox, 2023-12-29:
Apparently, this is issue is caused by linux-firmware package as stated in arch forum. I will try downgrade it and see how it goes.

https://bbs.archlinux.org/viewtopic.php?id=291391

cryptotabluvox, 2023-12-29:
For now, I finally managed to resolve this by reverting linux firmware to this version: linux-firmware-20230918.3672ccab-1-any.pkg.tar.zst.

Beforehand, I have removed these 3 files in /lib/firmware/mediatek.

WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_RAM_CODE_MT7961_1.bin
BT_RAM_CODE_MT7961_1_2_hdr.bin

Then, I downloaded linux-firmware-20230918.3672ccab-1-any.pkg.tar.zst and linux-firmware-20230918.3672ccab-1-any.pkg.tar.zst.sig and place both in /var/cache/pacman/pkg/

https://archive.archlinux.org/packages/l/linux-firmware/

Then, reboot. If still undetected, unplug the adapter and replugged it.
Voila, now I can use it with nmtui, nmcli and iwctl.

Thanks to @morrownr for suggesting to look in arch forum and the person who found the fix in arch forum: https://bbs.archlinux.org/viewtopic.php?id=291391

morrownr, 2023-12-30:
@cryptotabluvox 

FYI: On the Main Menu is the option:

8. Firmware - How to install firmware for Mediatek-based (and RTW88) USB WiFi adapters

I keep that `8. Firmware` current and post `News:` in Issues when firmware is added or updated. I do this for all WiFi 5 and later in-kernel drivers. I need to update the firmware list and post a message within the next week as we have new firmware now available for the mt7925 chipset for USB/PCIe. The driver is already in mainline as it went into kernel 6.7. The mt7925 is a WiFi 7 chipset. I expect adapters to be available at some point in 2024.