morrownr/USB-WiFi #356: Intel Corporation Wireless 3165 (rev 81)

mcondarelli, 2023-12-30:
Hi,
I have a similar problem, but on a different SBC: a Beelink mini PC based on Intel(R) Celeron(R) CPU N3450.
It has two integrated (PCI) network adapters (and I added two more USB ones, but that's another story):
```
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
02:00.0 Network controller: Intel Corporation Wireless 3165 (rev 81)
```
Hints about how to modify /etc/hostapd-WiFi*.conf would be welcome.
I need to use Intel drivers (iwlwifi), but other stuff is rather obscure to me.

Many Thanks in Advance

morrownr, 2023-12-30:
Hi @mcondarelli 

> I have a similar problem...

Similar to what? It is not clear to me what you are referring to.

> Hints about how to modify /etc/hostapd-WiFi*.conf would be welcome.

This is for `Intel Corporation Wireless 3165 (rev 81)` ?

I have a lot of experience with hostap but anything I suggest totally depends on what you are trying to do and that is not clear from your message. Help us out.

Regards,

@morrownr

mcondarelli, 2023-12-30:
Sorry, I was a bit too terse.
Let's try again ;)

My problem is to use `hostapd` (or `iwd` if that's better for any reason) to implement a fairly basic Access Point under Linux.

I have a small mini-PC (`lxd`) with several Ethernet adapters and a WiFi adapter (`Intel Corporation Wireless 3165 (rev 81)`).

This Mini-PC is based on ` Intel(R) Celeron(R) CPU N3450 @ 1.10GHz` and runs a fairly up-to-date Debian Bookworm release.
It is to be my Firewall.
Two Ethernet adapters are connected to my two modem/routers.
A third Ethernet adapter (USB3 `ASIX Electronics Corp. AX88179 Gigabit Ethernet`) connects with my home LAN.
Al this is already working reasonably well (actual config is a bit more complex, I can go into gory details, if useful).

What I would like to do (and I need some help with) is to implement a basic Access Point, possibly just bridging everything from "LAN" (after successful connection, of course).

I tried following tutorial, but first thing that happened is I managed to lock myself out from `lxd` because `/etc/systemd/network/30-config-bridge-br0.network' resulted in an incompatible IP address.

I can also see errors in `dmesg` I am unsure how to interpret, e.g.:
```
[24866.368496] br0: port 1(enxa0cec887415e) entered blocking state
[24866.368511] br0: port 1(enxa0cec887415e) entered disabled state
[25125.600948] r8169 0000:01:00.0 enp1s0: Link is Down
[25125.771009] r8169 0000:01:00.0: firmware: failed to load rtl_nic/rtl8168g-2.fw (-2)
[25125.772707] r8169 0000:01:00.0: firmware: failed to load rtl_nic/rtl8168g-2.fw (-2)
[25125.774360] r8169 0000:01:00.0: Direct firmware load for rtl_nic/rtl8168g-2.fw failed with error -2
[25125.774456] r8169 0000:01:00.0: Unable to load firmware rtl_nic/rtl8168g-2.fw (-2)
[25125.801947] Generic FE-GE Realtek PHY r8169-0-100:00: attached PHY driver (mii_bus:phy_addr=r8169-0-100:00, irq=MAC)
[25125.926132] r8169 0000:01:00.0 enp1s0: Link is Down
[25128.839923] r8169 0000:01:00.0 enp1s0: Link is Up - 1Gbps/Full - flow control off
[25128.839976] IPv6: ADDRCONF(NETDEV_CHANGE): enp1s0: link becomes ready
```

Before doing other tests I reached out for help.
I hope needs are clear, otherwise please ask, I will try to clarify.

Further (hopefully) relevant data:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: enp1s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    >>>> This is connected to my main modem/router <<<<
    link/ether 84:39:4c:00:47:34 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.99/24 brd 192.168.1.255 scope global enp1s0
       valid_lft forever preferred_lft forever
    inet6 fe80::8639:4cff:fe00:4734/64 scope link 
       valid_lft forever preferred_lft forever
3: enxa0cec8b35133: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    >>>> This is connected to my secondary modem/router (currently offline) <<<<
    link/ether a0:ce:c8:b3:51:33 brd ff:ff:ff:ff:ff:ff
4: enxa0cec887415e: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    >>>> This is connected to my internal LAN and should be bridged to WiFi <<<<
    link/ether a0:ce:c8:87:41:5e brd ff:ff:ff:ff:ff:ff
    inet 192.168.7.99/24 brd 192.168.7.255 scope global enxa0cec887415e
       valid_lft forever preferred_lft forever
5: wlp2s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    >>>> this is Intel Wireless adapter to be used <<<<
    link/ether 84:c5:a6:4c:ff:31 brd ff:ff:ff:ff:ff:ff
6: ORANGE: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    >>>> virtualized DMZ <<<<
    link/ether 00:16:3e:9d:92:d0 brd ff:ff:ff:ff:ff:ff
7: lxdbr0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    >>>> virtualized DMZ <<<<
    link/ether 00:16:3e:71:a4:e7 brd ff:ff:ff:ff:ff:ff
    inet 10.190.174.1/24 scope global lxdbr0
       valid_lft forever preferred_lft forever
8: mac6aef9d2d@enxa0cec887415e: <BROADCAST,MULTICAST,ALLMULTI,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 500
    >>>> virtualized interface for Firewall (backup modem) <<<<
    link/ether 00:16:3e:22:92:dc brd ff:ff:ff:ff:ff:ff
9: mac35bb3135@enp1s0: <BROADCAST,MULTICAST,ALLMULTI,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 500
    >>>> virtualized interface for Firewall (main modem) <<<<
    link/ether 00:16:3e:f4:94:6f brd ff:ff:ff:ff:ff:ff
10: tapc181c100: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master ORANGE state UP group default qlen 1000
    link/ether ca:8b:cc:83:0e:6c brd ff:ff:ff:ff:ff:ff
11: mac942ac72d@enxa0cec8b35133: <NO-CARRIER,BROADCAST,MULTICAST,ALLMULTI,UP,M-DOWN> mtu 1500 qdisc fq_codel state LOWERLAYERDOWN group default qlen 500
    link/ether 00:16:3e:0a:04:d4 brd ff:ff:ff:ff:ff:ff
13: vethe552b6fc@if12: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master ORANGE state UP group default qlen 1000
    link/ether 0e:4b:31:8d:7f:98 brd ff:ff:ff:ff:ff:ff link-netnsid 0
15: vetha4a8be97@if14: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master ORANGE state UP group default qlen 1000
    link/ether ca:93:e8:da:5f:e1 brd ff:ff:ff:ff:ff:ff link-netnsid 1
16: br0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether c2:46:84:d6:37:b5 brd ff:ff:ff:ff:ff:ff
    >>>> bridge I tried to setup following tutorial <<<<
```

morrownr, 2023-12-31:
> I tried following tutorial...

Which tutorial specifically are you using?

> Are you trying to setup 2GHz or 5GHz AP?

We may be able to help. However, this site is very specific to USB WiFi adapters and Intel does not do USB WiFi. I'm not trying to chase you away but rather just warning you that there is not much interaction here regarding Intel Wifi chipsets.

@morrownr

mcondarelli, 2023-12-31:
> > I tried following tutorial...
> 
> Which tutorial specifically are you using?

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

> 
> > Are you trying to setup 2GHz or 5GHz AP?

Possibly a dual-band, if that's not possible 5GHz would be preferred.

> 
> We may be able to help. However, this site is very specific to USB WiFi adapters and Intel does not do USB WiFi. I'm not trying to chase you away but rather just warning you that there is not much interaction here regarding Intel Wifi chipsets.

Understood.
Any help would be very welcome.
... even in the form of a redirect to more specific site ;)

Mauro

bjlockie, 2023-12-31:
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi

I don't know what iwlmvm is or if it applies. 

>AP mode on 2.4GHz (on devices driven by iwlmvm, note no 5GHz AP support due to LAR)

morrownr, 2023-12-31:
Hi again @mcondarelli 

> Possibly a dual-band, if that's not possible 5GHz would be preferred.

Okay. Now we are making progress. While I do not use or test with Intel wifi chipsets since they do not do usb, I do see a lot of comments in various places. @bjlockie speaks words of wisdom depending on what a person is doing. This site has a lot of users that make heavy use of master (AP) and monitor mode. AP mode users generally avoid Intel if their project involves 5 GHz AP mode because intel decided to do something that makes things hard to work in 5 GHz AP mode. If your end goal is to do a dual band setup, I would suggest the intel card for 2GHz and get a good USB WiFi adapter for 5GHz/6GHz. 6GHz is rather hit and miss depending on the laws of the country you live in at this point and I've not even got there yet as far as configuring hostapd in 6GHz yet. The example WiFi 6 hostapd.conf is for WiFi 6 (AX) on 5GHz. It seems to be very stable at this point. 

You are using the AP Guide called `Bridged Wireless Access Point` which is the guide I developed and maintain. I designed it to be an fast as possible as far as throughput is concerned and 24/7/365 reliable. There are a LOT of ways to setup an AP with Linux. My guide is different than most guides you will see out there in that I use systemd for the networking. I use hostapd to run the AP. 

I need to clarify your intent: ? The internet will come into your computer via ethernet port? The intel card will be used for 2GHz AP and 5GHz AP will be via usb wifi adapter? I think you mentioned that you have a couple of usb wifi adapters? If so, can you provide the wifi chipsets that are in the usb wifi adapters?

I'll need you to run the following for your intel chipset and post the results:

$ iw list

That will help us develop the info you need for hostapd.conf.

The networking setup in my guide should work with Debian 12 on x86_64 but I have not tested the latest version on the guide on that setup.

Cheers,

@morrownr

mcondarelli, 2024-01-01:
Thanks @morrownr 
Happy new Year!

I do attach full [iw_list.txt](https://github.com/morrownr/USB-WiFi/files/13804383/iw_list.txt)

Notice I have a rather complex situation more-or-less depicted as:

[image]

My metal is the blue box, which is a Beelink mini-PC called `lxd` and running [Incus](https://linuxcontainers.org/incus/). This machine has the (currently unused) Intel Wifi interface we are speaking about.

`lxd` also has three Ethernet adapters:
- `Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)` currently connected to `Fast Fiber`.
- `ASIX Electronics Corp. AX88179 Gigabit Ethernet` currently connected to `Slow DSL`
- `ASIX Electronics Corp. AX88179 Gigabit Ethernet` currently connected to `GREEN/LAN`; this is the interface I would like to bridge to WiFi.
- one virtual interface, managed by `Incus` is used to handle the internal and inaccessible `ORANGE/DMZ`.

Gory details of setup are available [here](https://blog.condarelli.it); this is actually served by `webserver` in the image above.

`lxd` is frontend/firewall/DMZ of my home net and I would like to enhance it to serve also as bridged A.P.

The `ip address` listing I posted a few messages ago was taken on `lxd` *after* I tried following your guide and thus you can see the `br0` interface.

if relevant:
- `lxd` is using standard Debian `systemd` `networking.service` and `systemd-networkd.service` to handle networking.
- I installed `hostapd`, `iw` and `firmware-iwlwifi`.
- One possible further stumbling block might be `Firewall` uses a Incus-managed interface (`mac35bb3135@enp1s0`) to connect to `GREEN/LAN` and not the plain `enp1s0`, but that is beyond the point; I would be more than happy if I manage to connect to A.P. and reach hosts on `GREEN/LAN`, I can deal with further redirects later ;)
- As said I would like to have full dual-band, if possible and 5G as second-best, but I understand there might be problems with Intel so I would be happy even with plain old 2.5G.

Many Thanks in Advance
Mauro