morrownr/USB-WiFi #357: I've failed to get a Fenvi AX1800 working

MSSProvenance, 2024-01-02:
Thank in advance to anyone who can help. I guess I'm an "intermediate" Linux user: neither a complete newbie nor highly competent (I generally prefer a GUI to a CLI).

Partly based on the recommendation at https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md I bought a Fenvi AX1800 from AliExpress.

I think I have correctly followed the instructions in Section 2 at https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md and my /lib/firmware/mediatek directory now contains both these files:
WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_RAM_CODE_MT7961_1.bin 
After a reboot, the Fenvi does not seem to be able to connect to wifi, so I have to revert to my TP-Link wifi adaptor.

I'm using Ubuntu 22.04.3 LTS. If someone is willing to help me diagnose the problem, what other info do you need? (If possible please give me the commands I need to use).

morrownr, 2024-01-02:
Hi @MSSProvenance 

What kernel are you using?

$ uname -r

MSSProvenance, 2024-01-02:
> What kernel are you using?

**5.15.0-91-generic**

Thanks!

morrownr, 2024-01-02:
> 5.15.0-91-generic

Won't work. You need at least kernel 5.18 for managed mode and 5.19 for AP mode. The driver is simply not in 5.15 because that kernel was released long before the driver was added to mainline.

Recommend you go to the Plug and Play list and start reading. There is a lot of information before I start listing adapters. It will explain the drivers and firmware.

Adding the firmware is fine but if the kernel has no driver, you need to upgrade your kernel. I'd have to go check but 5.19 should be available to you as a HWE kernel. Install it.

@morrownr

MSSProvenance, 2024-01-02:
Ah! Apologies. I had read about the issue with the kernal, but I obviously didn't read it closely enough, because I thought that my release of Ubuntu uses a recent-enough kernal.  
I've now done a bit of searching and see the issue:  it seems that while a **new** install of my version of Ubuntu would indeed have the 5.19 kernal, an **upgraded** version (like mine) does not. 
I will try the HWE kernel, and will come back here to let you know how it goes ...

Thanks again for your help. Much appreciated :-)

MSSProvenance, 2024-01-02:
OK, 
`uname -r`
now produces:
`6.2.0-39-generic`
A step in the right direction! 

The AX1800 is recognised (as it was before), but still does not connect. In case this helps,
`$ nmcli`
produces the following, in which the first line is in red:
```
wlx90de80fc04bf: disconnected        
"MediaTek Wi-Fi"
        wifi (mt7921u), 90:DE:80:FC:04:BF, hw, mtu 1500

enp3s0: unavailable
        "Realtek RTL8111/8168/8411"
        ethernet (r8169), 90:2B:34:BD:B7:8A, hw, mtu 1500

lo: unmanaged
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536

DNS configuration:
        servers: 194.168.4.100 194.168.8.100
        interface: wlx98ded01a4aaf

```

morrownr, 2024-01-02:
> wlx90de80fc04bf

I see a WiFi interface.I usually use `$ iw dev` instead of NM but an interface is an interface.

Have you `Forget connection` the previous connections?

Tell me what you are trying to connect to?

Describe what you are seeing?

bjlockie, 2024-01-03:
Type dmesg in the CLI after you insert the AX1800 and after it tries to connect.

MSSProvenance, 2024-01-03:
To try to answer your questions:

`iw dev `
produces this output:
```
phy#2
	Interface wlx98ded01a4aaf
		ifindex 6
		wdev 0x200000001
		addr 98:de:d0:1a:4a:af
		ssid VM2738188
		type managed
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

I have not attempted to "Forget connection" (should I?).

I am not sure what you mean by

> Tell me what you are trying to connect to?

Do you want my wifi router drtails?

> Describe what you are seeing?

I see nothing! (When I connect to wifi with my TP-link dongle, the only thing I "see" is a small wifi icon in the corner of my monitor.)

If I reboot the PC with the Fenvi plugged in, the PC starts as normal, but the wifi did not connect. If I remove the Fenvi from the USB port, and then plugged it back in. Still nothing. If I run `dmesg` this is the output:

```
[    0.000000] microcode: microcode updated early to revision 0x21, date = 2019-02-13
[    0.000000] Linux version 6.2.0-39-generic (buildd@lcy02-amd64-045) (x86_64-linux-gnu-gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2 (Ubuntu 6.2.0-39.40~22.04.1-generic 6.2.16)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.0-39-generic root=UUID=0b543551-e296-4c65-a815-5a8ff278b32f ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ec00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000040003fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040004000-0x0000000040004fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040005000-0x00000000d5be9fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d5bea000-0x00000000d642efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d642f000-0x00000000d66aefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000d66af000-0x00000000d66b3fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000d66b4000-0x00000000d66f6fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000d66f7000-0x00000000d6de4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d6de5000-0x00000000d6fddfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d6fde000-0x00000000d6ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d7800000-0x00000000df9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041f5fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Gigabyte Technology Co., Ltd. To be filled by O.E.M./H61MA-D2V, BIOS F4 03/23/2012
[... 1090 lines cut ...]
[   40.895840] audit: type=1107 audit(1704270711.929:154): pid=1412 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="CheckAuthorization" mask="send" name=":1.8" pid=3120 label="snap.snap-store.ubuntu-software" peer_pid=1431 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   43.251292] audit: type=1400 audit(1704270714.285:155): apparmor="DENIED" operation="open" class="file" profile="snap.snap-store.ubuntu-software" name="/etc/appstream.conf" pid=3120 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[  311.985126] audit: type=1107 audit(1704270983.021:156): pid=1412 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/timedate1" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.107" pid=2764 label="snap.firefox.firefox" peer_pid=4271 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
```

If I unplug and re-insert the Fenvi, and run `dmesg` again, the final lines are:

```
[  504.035398] usb 2-1.4: USB disconnect, device number 3
[  511.168220] usb 2-1.4: new high-speed USB device number 6 using ehci-pci
[  511.282603] usb 2-1.4: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  511.282612] usb 2-1.4: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[  511.282615] usb 2-1.4: Product: Wireless_Device
[  511.282617] usb 2-1.4: Manufacturer: MediaTek Inc.
[  511.282620] usb 2-1.4: SerialNumber: 000000000
[  511.384252] usb 2-1.4: reset high-speed USB device number 6 using ehci-pci
[  511.532150] mt7921u 2-1.4:1.0: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a

[  511.793242] mt7921u 2-1.4:1.0: WM Firmware Version: ____010000, Build Time: 20231109190959
[  513.461235] mt7921u 2-1.4:1.0 wlx90de80fc04bf: renamed from wlan0
```

Does that help?

MSSProvenance, 2024-01-03:
Excellent. Thanks again.

morrownr, 2026-03-12:
Hi @Alex-Denysiuk 

> They have locked the TX power limit to a maximum of 3 dBm directly in the device's EEPROM/firmware.

We have established an issue to work this problem:

https://github.com/morrownr/USB-WiFi/issues/700#

Hope to see you there.

> If you are experiencing random disconnects...

There are a couple of documents on the site Main Menu that could be helpful:

https://github.com/morrownr/USB-WiFi

Scroll down to the following documents:

Frequently Asked Questions

USB Ports can be a source of problems with USB WiFi adapters

One last note: inexpensive adapters may have low cost antennas and amps which can effect range and performance.