morrownr/USB-WiFi #359: News: Firmware for the mt7925 (WiFi 7) Chipset Posted

morrownr, 2024-01-04:
Hello All,

The driver for the new mt7925 chipset was merged into kernel 6.7 and now the first release of the firmware for the mt7925 chipset is available. I have updated the firmware guide that is on the Main Menu:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

Before you get in a hurry to grab the firmware and upgrade your kernel to 6.7, there is no hardware available to buy. None. Nothing. My best guess is that we will starting seeing cards and USB adapters based on the mt7925 at some point in 2024. Mediatek keeps getting support for their products in the kernel earlier and earlier. They are getting close to what Intel and AMD do...and they are the Gold Standard.

A walk down memory lane tells us that USB WiFi support in Linux 10-15 years ago was not what it is today. The trend has been up over the last 6 years and continues to get better.

Have a Happy New Year.

@morrownr

fakemanhk, 2024-01-06:
> Mediatek keeps getting support for their products in the kernel earlier and earlier. They are getting close to what Intel and AMD do...and they are the Gold Standard.

That's why I immediately pay for the first Banana Pi BPI-R4, and waiting for their WiFi 7 module to come out for a build.

Ten0, 2024-01-18:
Well... there's one in there:
https://www.asus.com/motherboards-components/motherboards/workstation/pro-ws-trx50-sage-wifi/

Doesn't work because my kernel is too old :')

fs30000, 2024-07-06:
> Well... there's one in there: https://www.asus.com/motherboards-components/motherboards/workstation/pro-ws-trx50-sage-wifi/
> 
> Doesn't work because our kernel is too old ATM, and then there's also this: https://lore.kernel.org/all/dbb87db4cef3620c44fe678a81a062bef11c557f.camel@mediatek.com/T/ :') EDIT: indeed there's this but pre-applying [the following kernel patch](https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=eaab5c300eab) on kernel 6.7 makes everything work

How to apply only the mt7925 patches? I mean, any easy way to apply all the respective patches? I just want to update the driver and recompile the kernel.

Ten0, 2024-07-06:
Kernel 6.9 has Wifi working I think, patch fixes specifically bluetooth. I'm not sure if it's still necessary to have the bluetooth patch at this point (I'm not sure how far the patch has gone yet), but as far as I'm concerned I'm on NixOS so I select kernel and apply the patch simply by adding:
```nix
  # Enforce very recent kernel to get the wifi working:
  # (ATM this is mainline but not even stable)
  boot.kernelPackages = pkgs.linuxKernel.packages.linux_6_9;
  # However that isn't enough to make bluetooth work because of this:
  # https://lore.kernel.org/all/dbb87db4cef3620c44fe678a81a062bef11c557f.camel@mediatek.com/
  # Rest of discussion: https://lore.kernel.org/all/170542082933.25552.3927828232982053152.git-patchwork-notify@kernel.org/
  # Patch here: https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=eaab5c300eab
  boot.kernelPatches = [{
    name = "Bluetooth: btusb: Add new VID/PID 13d3/3602 for MT7925";
    patch = ./bluetooth-mt7925.patch;
  }];
```
to my [`configuration.nix`](https://nixos-and-flakes.thiscute.world/nixos-with-flakes/get-started-with-nixos#configuring-the-system-using-etc-nixos-configuration-nix).
(where `./bluetooth-mt7925.patch` is https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/patch/?id=eaab5c300eab901be0040aa03bf22b20cc6ce0a5)

fs30000, 2024-07-06:
So, that's similar to ansible. Does it recompile the kernel with that conf? That sounds nice.

Ten0, 2024-07-06:
> So, that's similar to ansible. Does it recompile the kernel with that conf? That sounds nice.

Yes it does. In NixOS fundamentally you declare how to build your entire system in the nix language (it has reasonable default values for properties you don't specify), and for each package either you've got it common enough that it cache hits the official nixpkgs binary cache, or it compiles stuff. That makes overriding specific packages easy while also not spending an absurd amount of time compiling everything that you don't change. NixOS is great 🙂
It has a steep learning curve though.

ilikenwf, 2025-02-22:
Anyone else notice the latest update of the mt7925e firmware seems to break bss fast transitions?

morrownr, 2025-02-22:
@ilikenwf 

You might get more eyes on this problem if you also post a new issue. Adding some details might help. Is the mt7925e based adapter in AP mode or Managed mode? What firmware versions have you tested and what were the results? How are you setting up fast transitions?

fs30000, 2025-02-24:
> Anyone else notice the latest update of the mt7925e firmware seems to break bss fast transitions?

Here, on a banana pi r4, the mt7925e is crashing the system each 2-3 days. I have emailed the developers, at least, i used the email addresses listed on the kernel mailing list. So far no replies whatsoever.

morrownr, 2025-02-24:
@fs30000 

Are you using kernel 6.12.13?

fs30000, 2025-02-24:
> [@fs30000](https://github.com/fs30000)
> 
> Are you using kernel 6.12.13?

I have tried these kernels:
6.12.0
6.12.7
6.13-rc1

morrownr, 2025-02-24:
Okay, the problem is likely not linking to a recent patch that has now been fixed.

What are the details? Logs, distro?

fs30000, 2025-02-24:
Debian 12 on a Banana pi R4.

Stack  trace:

```root@banana1 /root $ cat  /var/lib/systemd/pstore/console-ramoops-0
0: port 5(wlP1p1s0) entered blocking state
[   11.482023] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   11.487586] lanbr0: port 5(wlP1p1s0) entered forwarding state
[   11.501158] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   11.512361] wireguard: wg0: Interface created
[   11.523127] wireguard: wg0: Peer 1 created
[   11.527595] wireguard: wg0: Peer 2 created
[   11.532013] wireguard: wg0: Peer 3 created
[   11.536419] wireguard: wg0: Peer 4 created
[   18.507157] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[   18.514487] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[   18.520783] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[   18.529227] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[51158.178929] mt7925e 0001:01:00.0 wlP1p1s0: left allmulticast mode
[51158.185072] mt7925e 0001:01:00.0 wlP1p1s0: left promiscuous mode
[51158.191136] lanbr0: port 5(wlP1p1s0) entered disabled state
[51160.618247] lanbr0: port 5(wlP1p1s0) entered blocking state
[51160.623828] lanbr0: port 5(wlP1p1s0) entered disabled state
[51160.629442] mt7925e 0001:01:00.0 wlP1p1s0: entered allmulticast mode
[51160.635923] mt7925e 0001:01:00.0 wlP1p1s0: entered promiscuous mode
[51161.009202] lanbr0: port 5(wlP1p1s0) entered blocking state
[51161.014799] lanbr0: port 5(wlP1p1s0) entered forwarding state
[137557.334991] mt7925e 0001:01:00.0 wlP1p1s0: left allmulticast mode
[137557.341332] mt7925e 0001:01:00.0 wlP1p1s0: left promiscuous mode
[137557.347481] lanbr0: port 5(wlP1p1s0) entered disabled state
[137559.743309] lanbr0: port 5(wlP1p1s0) entered blocking state
[137559.748990] lanbr0: port 5(wlP1p1s0) entered disabled state
[137559.754669] mt7925e 0001:01:00.0 wlP1p1s0: entered allmulticast mode
[137559.761242] mt7925e 0001:01:00.0 wlP1p1s0: entered promiscuous mode
[137560.133730] lanbr0: port 5(wlP1p1s0) entered blocking state
[137560.139412] lanbr0: port 5(wlP1p1s0) entered forwarding state
[223956.767220] mt7925e 0001:01:00.0 wlP1p1s0: left allmulticast mode
[223956.773454] mt7925e 0001:01:00.0 wlP1p1s0: left promiscuous mode
[223956.779628] lanbr0: port 5(wlP1p1s0) entered disabled state
[223959.179429] lanbr0: port 5(wlP1p1s0) entered blocking state
[223959.185098] lanbr0: port 5(wlP1p1s0) entered disabled state
[223959.190796] mt7925e 0001:01:00.0 wlP1p1s0: entered allmulticast mode
[223959.197356] mt7925e 0001:01:00.0 wlP1p1s0: entered promiscuous mode
[223959.591977] lanbr0: port 5(wlP1p1s0) entered blocking state
[223959.597654] lanbr0: port 5(wlP1p1s0) entered forwarding state
[310356.047398] mt7925e 0001:01:00.0 wlP1p1s0: left allmulticast mode
[310356.053676] mt7925e 0001:01:00.0 wlP1p1s0: left promiscuous mode
[310356.059878] lanbr0: port 5(wlP1p1s0) entered disabled state
[310358.417000] lanbr0: port 5(wlP1p1s0) entered blocking state
[310358.422686] lanbr0: port 5(wlP1p1s0) entered disabled state
[310358.428367] mt7925e 0001:01:00.0 wlP1p1s0: entered allmulticast mode
[310358.434942] mt7925e 0001:01:00.0 wlP1p1s0: entered promiscuous mode
[310358.797712] lanbr0: port 5(wlP1p1s0) entered blocking state
[310358.803394] lanbr0: port 5(wlP1p1s0) entered forwarding state
[396754.787714] mt7925e 0001:01:00.0 wlP1p1s0: left allmulticast mode
[396754.793925] mt7925e 0001:01:00.0 wlP1p1s0: left promiscuous mode
[396754.800107] lanbr0: port 5(wlP1p1s0) entered disabled state
[396757.171913] lanbr0: port 5(wlP1p1s0) entered blocking state
[396757.177595] lanbr0: port 5(wlP1p1s0) entered disabled state
[396757.183275] mt7925e 0001:01:00.0 wlP1p1s0: entered allmulticast mode
[396757.189846] mt7925e 0001:01:00.0 wlP1p1s0: entered promiscuous mode
[396757.562218] lanbr0: port 5(wlP1p1s0) entered blocking state
[396757.567900] lanbr0: port 5(wlP1p1s0) entered forwarding state
[411772.953185] Unable to handle kernel paging request at virtual address ffffffffffffffa0
[411772.961210] Mem abort info:
[411772.964091]   ESR = 0x0000000096000006
[411772.967917]   EC = 0x25: DABT (current EL), IL = 32 bits
[411772.973304]   SET = 0, FnV = 0
[411772.976444]   EA = 0, S1PTW = 0
[411772.979661]   FSC = 0x06: level 2 translation fault
[... 34 lines cut ...]
[411773.230770]  genl_rcv_msg+0x1e4/0x26c
[411773.234512]  netlink_rcv_skb+0x5c/0x128
[411773.238426]  genl_rcv+0x38/0x50
[411773.241646]  netlink_unicast+0x2e4/0x33c
[411773.245647]  netlink_sendmsg+0x17c/0x3b4
[411773.249648]  ____sys_sendmsg+0x18c/0x2f8
[411773.253650]  ___sys_sendmsg+0x80/0xdc
[411773.257390]  __sys_sendmsg+0x80/0xec
[411773.261042]  __arm64_sys_sendmsg+0x24/0x30
[411773.265216]  invoke_syscall+0x48/0x110
[411773.269046]  el0_svc_common.constprop.0+0x40/0xe0
[411773.273828]  do_el0_svc+0x1c/0x28
[411773.277221]  el0_svc+0x30/0xd0
[411773.280357]  el0t_64_sync_handler+0x10c/0x138
[411773.284791]  el0t_64_sync+0x19c/0x1a0
[411773.288534] Code: d280003a f9420381 12001c7b aa1c03e0 (785a0021)
[411773.294702] ---[ end trace 0000000000000000 ]---
[411773.301162] pstore: backend (ramoops) writing error (-28)
[411773.306638] Kernel panic - not syncing: Oops: Fatal exception
[411773.312460] SMP: stopping secondary CPUs
[411773.316463] Kernel Offset: disabled
[411773.320027] CPU features: 0x000,00001020,00800000,8200420b
[411773.325589] Memory Limit: none
[411773.330499] Rebooting in 10 seconds..
```

morrownr, 2025-02-24:
> 6.14.0-rc1-bpi-r4

There have been a couple emergency patches that went in over the last two weeks. One of which I reported to linux-wireless. One of the Mediatek devs reached out within a few hours and had a fix a fix for us to test within a day. The new patch went in within 2 days. rc1 could very well have that problem or another one. rc1's are the bleeding edge of test versions.

I have a Mini PC with a mt7925 based M.2 card. I just tested with 6.14.0 rc4, 6.12.16. The distro is Ubuntu 24.10. I am not seeing this. Keep in mind that the default kernel on Debian 12 is 6.1. Debian only officially goes to newer kernels once every 2 years. The mt7925 driver only entered the kernel with 6.7. For us to convince the kernel devs to look at this, we need to figure it out and show results from something that should be solid. 

Recommend you test 6.14 rc4 and 6.12.16. Let me know what you find out.

ilikenwf, 2025-02-24:
I am thinking my issue may be something with some newer openwrt builds - my mtk7925 is in a laptop, my aps are openwrt and may be facing some bug since upgrading.

Laptop is running 6.13.4

fs30000, 2025-02-24:
> There have been a couple emergency patches that went in over the last two weeks. One of which I reported to linux-wireless. One of the Mediatek devs reached out within a few hours and had a fix a [...]

Ah forgot to mention the  6.14-rc1 version. 

Ok, i'll update the kernel soon. Hopefully it will fix the crashes. I have a qualcomm QCNCM865, but also has problems, like not being able to change the country code. So hostpad fails to start. I tried the patches from Openwrt, but compilation fails. BE1400 also sucks right now. So mediatek is the one i am hopefull about.

The issue here is that this is a arm64 system, much less priority  than  x86/x64 systems. That's why fixes take a longer time, if they actually come.

ilikenwf, 2025-03-06:
Well, for whatever reason the kernel module shipped in 6.8.9 works far better for me than 6.13.5 for some reason. I'm using it as a client device in a laptop; under 6.13, it randomly disconnects and sometimes will just sit there and not connect as networkmanager spins.

morrownr, 2025-03-06:
@ilikenwf 

There have been some opps that have needed work lately. You are probably better off sticking with a version that works for now while some of the fixes work their way through the system.

Basically the flow of patches adding more features to the WiFi 7 drivers and supporting stacks is really heavy from all of the main WiFi driver suppliers. I'm talking Intel, Qualcomm, Mediatek and Realtek. You just cannot add as much work as is going in right now without something causing problems. These modern WiFi 6 and 7 drivers are big and very complex. We are not in the age of WiFi 4 anymore.

You might also make sure that you are using the November firmware for the mt7925 if you haven't installed it.

ilikenwf, 2025-03-06:
Thanks for the tip. I'll keep my eye out and may just switch back to an intel card for the time being...it actually works better, I just don't trust CNVI despite having ME disabled and using coreboot.

morrownr, 2025-03-06:
What I do with my main dev box is I have it using 6.12.9. Things went downhill with 6.12.13. I'm running a mt7922 PCIe card and various USB adapters that vary depending on what I need. I have a M.2 card mt7925 card in my Mini PC. It runs 6.11.x. I would have to go look and I cannot recall any problems. Both of those cards smoke along at over 1.2 Gbps. 1 G ethernet cable sure seems slow these days <he he>.

My comment was actually aimed at all of the main wireless makers. Some of the bugs have to do with the wireless stack and not specific drivers but I have been watching WiFi 7 problem reports from all makers.

ilikenwf, 2025-03-06:
Ah very nice! I may have to pop it in a non-daily machine at some point then. 

I did see there are some patches for this card in the queue on the mediatek patch site...

morrownr, 2025-03-06:
One of the bugs that I reported:

https://patchwork.kernel.org/project/linux-wireless/patch/20250218033343.1999648-1-mingyen.hsieh@mediatek.com/

I did not discover it but 2 users here reported it and one did a bisect so I reported it. The Mediatek dev was on it in hours and had a patch to test within a day. Have you checked your log?

FWIW: That patch should have gone directly in but is still sitting there. Not good.

morrownr, 2025-03-07:
@ilikenwf 

I see a message in linux-wireless this morning that an new set of firmware files for the mt7925 is inbound so it should be in the normal place in a few days. No idea what it fixed.

fs30000, 2025-03-07:
Mine just crashed 20 minutes ago. Going to fill a bug. 6.14.0-rc1 here.

morrownr, 2025-03-07:
@fs30000 

Many, if not most, of the devs do not pay attention to bugzilla. I'm not saying that I agree with this but that is how it is.

What does get their attention is a well done message to linux-wireless, especially if the issue is bisected and the problematic patch is identified.

Here is the result of a bug report that I recently sent to linux-wireless:

https://patchwork.kernel.org/project/linux-wireless/patch/20250218033343.1999648-1-mingyen.hsieh@mediatek.com/

The story: Recently one user reported a problem here on USB-WiFi. I told him I would keep an eye out for confirmation. The next day another user reported the same thing in a different system. I asked if either knew how to bisect the bug. One said yes and bisected to find the bad patch. I reported the problem. Within 12 hours one of the Mediatek devs answered and a few hours later send a patch for use to test. Both users tested and the patch worked. The dev sent the patch to linux-wireless. The entire time from me submitting the parch until the patch was tested and submitted was maybe 2 days.

How about you start a new bug here with a good title and if I can get a confirmation or can duplicate it myself, we can press on and do what is needed to get it in front of a Mediatek dev.

Do you know how to bisect?

fs30000, 2025-03-10:
@morrownr If by bisecting you mean, compiling  the kernel with debug symbols and running the debugger and stuff,  then no. I never did that. Isn't that stack trace enough?

This happens an a banana pi r4, arm64, do you got a arm64 machine?

Anyway, i just swapped the mt7925 with an older mt7922, and it works much better now. So far, no crashes and iperf3 actually reports double the speed. So, not sure if it's worth it to keep trying to fix this...

morrownr, 2025-03-10:
> This happens an a banana pi r4, arm64, do you got a arm64 machine?

I have a 64 bit RasPi4B. I have no capability to run a mt7925 M.2 card. My mt7925 M.2 card is in a AMD64 Mini PC.

> Anyway, i just swapped the mt7925 with an older mt7922, and it works much better now. 

The new firmware for the 7925 hit the repos this morning. Recommend you try it. As usual, no idea what is fixed but something is fixed or we would not have new firmware.

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

nbattista89, 2025-03-22:
Hi Team,

The Bananapi R4 with OpenWrt 24.10 (stock) the wifi mt7922 or mt7925 are stable? The range covered is good?

morrownr, 2025-03-23:
Hi @ilikenwf 

I have a full size PCIe card based on the mt7922 chip and a M.2 card based on the mt7925 chip.

mt7922: (wifi router is WiFi 6 capable. WiFi 6 is in use, channel width 160. 5 Ghz channel 100)

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.128 port 52904 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   130 MBytes  1.09 Gbits/sec    0   1.25 MBytes       
[  5]   1.00-2.00   sec   135 MBytes  1.13 Gbits/sec    0   1.25 MBytes       
[  5]   2.00-3.00   sec   131 MBytes  1.10 Gbits/sec    0   1.25 MBytes       
[  5]   3.00-4.00   sec   130 MBytes  1.09 Gbits/sec    0   1.25 MBytes       
[  5]   4.00-5.00   sec   130 MBytes  1.09 Gbits/sec    0   1.25 MBytes       
[  5]   5.00-6.00   sec   129 MBytes  1.08 Gbits/sec    0   1.25 MBytes       
[  5]   6.00-7.00   sec   131 MBytes  1.10 Gbits/sec    0   1.25 MBytes       
[  5]   7.00-8.00   sec   131 MBytes  1.10 Gbits/sec    0   1.25 MBytes       
[  5]   8.00-9.00   sec   129 MBytes  1.08 Gbits/sec    0   1.25 MBytes       
[  5]   9.00-10.00  sec   134 MBytes  1.12 Gbits/sec    0   1.25 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.28 GBytes  1.10 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.28 GBytes  1.10 Gbits/sec                  receiver

```

The mt7925 card does about 1.2 Gbps on average so a little faster but it is in a different location. I have noticed nothing to report on range. I have just used them and have not done a range test so far. The only thing to report is a recent report of a bug that slowed down throughput with the mt7925 but the fix patch has been posted. Both adapters are really solid in my experience.

LC43, 2025-03-24:
hi all, i just wanted to show my appreciation for this thread. 
i too got confused with the kernel updates that broke my wifi. reading this thread and this repo resources helped me figure out the solution.

i have the mt7925 :
```
driver: mt7925e
version: 6.12.12-061212-generic
firmware-version: ____000000-20241104133053
```
and this kernel + firmware seems to be the most stable i could try.

i also found this:
https://lore.kernel.org/regressions/CACr2tqv6ZsEt1heUKvYeSOT9joYqCFsepn6j8sqVO7ErhKvGeQ@mail.gmail.com/T/#m5d8592bc3770cf55d7b3eec0c31b596c98e0c0a2

which describes the problem with the firmware update.

morrownr, 2025-03-24:
Just to add more to the conversation:

https://github.com/morrownr/USB-WiFi/issues/431#issuecomment-2746121697

I had downloaded and installed the Ubuntu 25.04 test version with kernel 6.14 rcx a couple of weeks ago. I saw the performance regressions as pointed out in the above issue. It appears that there is a patch already.

@LC43 

There is an even more recent set of firmware files for the mt7925 than the one you are using. It is from March 2025. The firmware installation guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

morrownr, 2025-03-24:
> thanks for the heads up, but according to that email, using the new firmware will affect the speeds. ill have to wait :)

I must have missed it, can you point me to the email that addressed the new firmware affecting the speeds?

LC43, 2025-03-25:
i saw this on the link i shared.  Jean-François INGELAERE emailed some lists in the kernel. so i dont know their email, just the name. Maybe it was the wrong list? because there's no follow up.

> With 6.12.12 + firmware-202412x speed is correct (1Gbps/800Mbps)
> With 6.12.12 + firmware-202503x speed is NOT correct but between acceptable and unacceptable (350Mbps/25Mbps)
> With 6.12.13 + firmware-202412x speed is NOT correct but between acceptable and unacceptable (280Mbps/80Mbps)

morrownr, 2025-03-25:
@LC43 

Yen says:

```
Currently, these patches are in the maintainer’s tree. It should soon
be included in 6.14 and then ported Back to 6.12.x and 6.13.x.
```

I did not see a request to take the patch directly to Linus to be included in 6.14. If the patch is in the maintainers tree, then it will likely go into 6.15 and then be backported to 6.14 and 6.12. It is good to know the patch is going in but it may not show up as fast we would like.

fs30000, 2025-03-25:
> Hi Team,
> 
> The Bananapi R4 with OpenWrt 24.10 (stock) the wifi mt7922 or mt7925 are stable? The range covered is good?

 With the mt7925 it is crashing the whole system each 2-3 days or even more frequently if i connect my phone more times. This is a secondary AP here. With the mt7922 it is completly stable and speeds are a lot better.

 But not 130MB/s like @morrownr  shows here. He's using it as a client with 160Hz width and i use it as AP with hostapd with just 80Hz. If i try with 160Hz, always crashes with some radar detection scan error. I don't know how to fix it.

LC43, 2025-06-27:
hi, i just noticed 6.15.2 kernel, which seems to include 
$ ethtool -i wlp195s0 
driver: mt7925e
version: 6.15.2-061502-generic
firmware-version: ____000000-20250305133013

and looking at the [repo](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/mediatek/mt7925/BT_RAM_CODE_MT7925_1_1_hdr.bin), seems there's a two more fixes coming soon:

> 2025-05-28	[mediatek MT7925: update bluetooth firmware to 20250526153203](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/mediatek/mt7925/BT_RAM_CODE_MT7925_1_1_hdr.bin?id=c799f5d3045f2495ceeefcc2b9055ef42843c0e2)	Chris Lu	1	-0/+0
> 2025-04-30	[mediatek MT7925: update bluetooth firmware to 20250425073330](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/mediatek/mt7925/BT_RAM_CODE_MT7925_1_1_hdr.bin?id=8357180b9ba022f9e8fda502dcf952e247da4c07)	Chris Lu	1	-0/+0
> 2025-03-10	[mediatek MT7925: update bluetooth firmware to 20250305133215](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/mediatek/mt7925/BT_RAM_CODE_MT7925_1_1_hdr.bin?id=ae25a3cb0fd890ad83ab058cdb13e6056908be29)	Chris Lu	1	-0/+0

i'll be using this 20250305133013 to see if the speeds have improved.

pcraciunoiu, 2025-06-28:
I just tried 6.15.4 which is available on mainline, and bluetooth still doesn't work for me. My WiFi speeds are ~80% of the ISP capacity I'm supposed to have at 800mbps, so I think that's working fine.

Does bluetooth work for anyone? If so, what did you do? I'm on Ubuntu 25.04 FWIW.

morrownr, 2025-06-28:
@pcraciunoiu 

> I just tried 6.15.4 which is available on mainline, and bluetooth still doesn't work for me. 

I assume you are talking about mt7925 based M.2 card. Right?

> Does bluetooth work for anyone? If so, what did you do? I'm on Ubuntu 25.04 FWIW.

I just rechecked on my Mini PC that has a mt7925 M.2 card and Ubuntu 25.04. Bluetooth and WiFi seem to be working fine. I tested with the default kernel 6.14 and I do have a 6.15 kernel installed so I tested with it also. Bluetooth can do various things...like tethering, file transfer and more. What is not working for you?

@LC43 

> i'll be using this 20250305133013 to see if the speeds have improved.

That is the firmware that shipped with Ubuntu 25.04 and it has not been updated by Ubuntu. There is a newer version available if you want to install it. When I first installed Ubuntu 25.04, the slow wifi problem was still there but within a week or two, when the first kernel upgrade came in, it was fixed. I see iperf3 speeds of around 1.2 gbps with my WiFi 6 router on 5 GHz using 160 MHz channel width.

LC43, 2025-06-29:
@pcraciunoiu 

i got bluetooth working recently with 6.12, if i remember correctly, when it included [this](https://patchwork.kernel.org/project/bluetooth/patch/53ccc7377341b64ff3fbdde3df28cbd14f245340.camel@nuclearsunshine.com/).

```bash
> rfkill list
0: asus-wlan: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: asus-bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
3: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```
and

```bash
> btmgmt info
Index list with 1 item
hci0:	Primary controller
	addr 28:D0:43:29:97:13 version 13 manufacturer 70 class 0x6c010c
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech cis-central cis-peripheral 
	current settings: powered ssp br/edr le secure-conn wide-band-speech cis-central cis-peripheral iso-broadcaster sync-receiver 
	name zenbaiten
	short name 
hci0:	Configuration options
	supported options: public-address 
	missing options:
```

going to install 6.15.4, thanks for the tip to upgrade @morrownr

morrownr, 2025-06-29:
@pcraciunoiu 

> I don't understand why Bluetooth is not working for me but it does for you.

We may have cards with the same chipset but our cards could have different VID/PIDs. Run and post the results of the following command:

$ lsusb

Where I run the command, I get:

bla bla bla: ID 13d3: 3604 IMC Network Wireless_Device

pcraciunoiu, 2025-06-29:
@morrownr here it is:

```
$ lsusb
...
Bus 003 Device 003: ID 2c7c:7009 Quectel Wireless Solutions Co., Ltd. Wireless_Device
```

Or another way I came across

```
$ lspci -knn | grep Net -A2; lsusb
c2:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7925]
	Subsystem: Quectel Wireless Solutions Co., Ltd. Device [1eac:8005]
	Kernel driver in use: mt7925e
```

pcraciunoiu, 2025-06-29:
@morrownr also, if it helps, I followed [this guide](https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md) and copied those 3 files to `/lib/firmware/mediatek/mt7925` but that didn't fix my bluetooth either.

morrownr, 2025-06-29:
@pcraciunoiu 

> Bus 003 Device 003: ID 2c7c:7009 Quectel Wireless Solutions Co., Ltd. Wireless_Device

There is what we are looking for. Let's do some educational stuff first:

Bluetooth has been around for several years now. It started as a USB2 technology and it still is a USB2 technology. Yes, I know a M.2 or PCIe card is not running on the USB bus but that Bluetooth support, whether in a card or an adapter is USB2 technology. That is why I had you run `lsusb` because that is how we look for USB IDs.

The device ID has to be in the appropriate file in the kernel so that your hardware is identified to the correct driver. We need to now go to the file in the kernel where the device ID should be located:

/drivers/bluetooth/btusb.c

I keep a clone of the following repo here:

https://github.com/torvalds/linux

Some guy named Torvalds maintains it. Anyway, a quick `git pull` and it is up to date on my system. Open `btusb.c` in test editor and we see the following section for the mt7925:

```
	/* Additional MediaTek MT7925 Bluetooth devices */
	{ USB_DEVICE(0x0489, 0xe111), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x0489, 0xe113), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x0489, 0xe118), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x0489, 0xe11e), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x0489, 0xe124), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x0489, 0xe139), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x0489, 0xe14f), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x0489, 0xe150), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x0489, 0xe151), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x13d3, 0x3602), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x13d3, 0x3603), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x13d3, 0x3604), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x13d3, 0x3608), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x13d3, 0x3613), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x13d3, 0x3628), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
	{ USB_DEVICE(0x13d3, 0x3630), .driver_info = BTUSB_MEDIATEK |
						     BTUSB_WIDEBAND_SPEECH },
```

My mt7925 based M.2 card has a device ID of 13d3:3604. Look in the above list and you see it listed which means the Linux kernel knows about my bluetooth capability. Now look for your device ID of 2c7c:7009. We don't see it. There is the problem.

What is the corrective action? First we need to add your device ID to kernel code on your system and compile it to see if what we think actually works. Do you know how to download the source for your kernel? And do you know how to compile and install it?

Once we have figured out the above and we get a successful test, it is time to submit a patch to the appropriate places. My work in the kernel is almost exclusively in linux-wireless which is for wifi but I can likely walk you through what it will take to get the patch sent in.

> also, if it helps, I followed this guide and copied those 3 files to /lib/firmware/mediatek/mt7925 but that didn't fix my bluetooth either.

That cannot help because the device ID is not in the kernel but I do know the guy that maintains that firmware guide. He does a good job... ha ha.

morrownr, 2025-06-29:
@LC43 

> updated to 6.15.4 , same firmware, also looks good ( similar speeds to other devides ). updated firmware to latest https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek/mt7925, but wifi & bluetooth stopped working.

Oh my. That is something to investigate. My test was with the default firmware that came with Ubuntu 25.04. Let me install the latest firmware to see what happens here.

bjlockie, 2025-06-29:
Isn't there a way to add USB id's without recompiling the kernel?

pcraciunoiu, 2025-06-29:
@morrownr thanks for the detailed reply.

I do not know how to download or compile the kernel. But I can follow a guide if it's not crazy advanced.

I also tried patching btusb.c per this thread, but had some errors.

https://github.com/dekomote/btusb-mt7925-extended-pids-dkms/issues/1#issuecomment-3016748044

Happy to do what makes the most sense as I am no expert at installing Bluetooth drivers, especially as updates are released to avoid re-breaking it.

pcraciunoiu, 2025-06-29:
@morrownr also from that GH issue I linked is this, maybe someone is already working on a patch?
https://lkml.org/lkml/2025/6/6/699

morrownr, 2025-06-30:
@pcraciunoiu 

> also from that GH issue I linked is this, maybe someone is already working on a patch? https://lkml.org/lkml/2025/6/6/699

That indicates that the patch is complete and has gone in. When it will show up in the kernel is not clear. I would guess it will go into kernel 6.17 based on the timing. Will it be backported to earlier kernels? That was not clear. However, at least you know help is on the way and it saves me from teaching you how to be a kernel developer. Ha ha.

@bjlockie 

> Isn't there a way to add USB id's without recompiling the kernel?

There is. It is just a temp fix until the read fix is in the kernel but that could work well in this case since it appears the needed patch has already been submitted.

An example is in The Plug and Play List in the AXE3000 section:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md#axe3000---usb30---24-ghz-5-ghz-and-6-ghz-wifi-6e

The two adapters that show the example are the Netgear A8000 and TP-Link  TXE50UH. I suspect the driver that would be the target is `btmtk`.

morrownr, 2025-06-30:
@pcraciunoiu 

The method that @bjlockie is talking about, using udev to get the bluetooth working, should work but my experience is with wifi, not bluetooth so I think we will have to mess around to figure out what works. I would suggest method 2, which is temporary, until we figure out what works.

Here is the Method 2 example from the Netgear A8000 USB WiFi adapter:
```
Method 2: From a terminal, enter and execute the following commands:

su
modprobe mt7921u
echo 0846 9060 > /sys/bus/usb/drivers/mt7921u/new_id
```

We will need to figure out which driver to use modprobe on and then we will need to figure out the location of new_id to send the ID.

pcraciunoiu, 2025-06-30:
@morrownr so if I understand, I would be running

```sh
modprobe mt7925e
echo 2c7c 7009 > /sys/bus/usb/drivers/mt7925e/new_id
```

Is that right?

The folder `mt7925` doesn't exist in that `drivers` parent folder yet.

How would I figure out the next steps from here?

morrownr, 2025-06-30:
@pcraciunoiu 

> Is that right?

No, this is for bluetooth, not wifi. That was for a wifi adapter.

Try the following:

su
modprobe btusb
echo 2c7c 7009 > /sys/bus/usb/drivers/btusb/new_id

pcraciunoiu, 2025-06-30:
@morrownr alright, I did that, but after the echo it's just a file... I'm not sure where to go from here. Sorry if this is such a basic question, I skimmed through the link you posted but that's where method 2 seems to end.

If I try to turn on the bluetooth it doesn't work still. I would imagine I have to start or restart some service?

morrownr, 2025-06-30:
@LC43 

> i made a mistake while copying the files. all good now.

That is good. One less problem is always good.

pcraciunoiu, 2025-06-30:
@morrownr I tried a few more things, here is some output in case it helps

```
$ hcitool dev
Devices:
# lists nothing
$ rfkill list all
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```

And this code
```
$ hciconfig hci0 up
Can't init device hci0: Device or resource busy (16)
```

The dmesg looks to match

```
[   10.844837] systemd[1]: systemd-hibernate-clear.service - Clear Stale Hibernate Storage Info was skipped because of an unmet condition check (ConditionPathExists=/sys/firmware/efi/efivars/HibernateLocation-8cf2644b-4b0b-428f-9387-6d876050dc67).
[   11.052027] Bluetooth: Core ver 2.22
[   11.064346] NET: Registered PF_BLUETOOTH protocol family
[   11.064354] Bluetooth: HCI device and connection manager initialized
[   11.064360] Bluetooth: HCI socket layer initialized
[   11.064363] Bluetooth: L2CAP socket layer initialized
[   11.064372] Bluetooth: SCO socket layer initialized
[   11.129716] Bluetooth: hci0: Opcode 0x0c03 failed: -16
[   11.594582] mt7925e 0000:c2:00.0: WM Firmware Version: ____000000, Build Time: 20250526153043
[   12.104044] amdgpu 0000:c5:00.0: amdgpu: [drm] Optional firmware "amdgpu/isp_4_1_0.bin" was not found
[   12.148179] [drm] Loading DMUB firmware via PSP: version=0x09001700
[   12.148330] amdgpu 0000:c5:00.0: amdgpu: Found VCN firmware Version ENC: 1.23 DEC: 9 VEP: 0 Revision: 9
```

The output from bluetooth service:

```
$ systemctl status bluetooth.service

● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
     Active: active (running) since Mon 2025-06-30 09:24:18 MDT; 13min ago
 Invocation: 8e9f2c99a9f84315a46ceb4c31c56938
       Docs: man:bluetoothd(8)
   Main PID: 6977 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 109876)
     Memory: 768K (peak: 1.8M)
        CPU: 19ms
     CGroup: /system.slice/bluetooth.service
             └─6977 /usr/libexec/bluetooth/bluetoothd

Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: src/plugin.c:init_plugin() System does not support bap plugin
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: src/plugin.c:init_plugin() System does not support bass plugin
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: src/plugin.c:init_plugin() System does not support mcp plugin
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: src/plugin.c:init_plugin() System does not support vcp plugin
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: src/plugin.c:init_plugin() System does not support micp plugin
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: src/plugin.c:init_plugin() System does not support ccp plugin
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: src/plugin.c:init_plugin() System does not support csip plugin
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: src/plugin.c:init_plugin() System does not support asha plugin
Jun 30 09:24:18 paul-mini-x1 bluetoothd[6977]: Bluetooth management interface 1.23 initialized
```

`bluetoothctl` does not see any controllers. Probably makes sense as there is no driver.

I don't have any other ideas. Is there something else I can try?

Does it make more sense to build the kernel myself now? Is that difficult on Ubuntu 25?

morrownr, 2025-06-30:
@pcraciunoiu 

> bluetoothctl does not see any controllers. Probably makes sense as there is no driver.

Exactly. Well, actually the driver is there but without the device ID, the driver is not being told to service your bluetooth device.

> I don't have any other ideas. Is there something else I can try?

You could do some searching to see if anyone else has used the method I showed you to add the device ID. I wish I had the exact answer for you but I have little experience with bluetooth except for the things that tend to overlap with wifi. If I had a better understanding of the flow of the bluetooth code, I could probably figure out the answer but I do not at this point.

There is the option to compile the kernel after adding the device ID to the appropriate file. That is not a permanent solution for a system that will be upgraded regularly but we could give it a try to check that the patch that went in will solve your problem.

> Does it make more sense to build the kernel myself now? Is that difficult on Ubuntu 25?

You can give it a try. I do not regularly build kernels for Ubuntu. I use Ubuntu mainline to test newer kernels on Ubuntu but that won't be any immediate help as I think that patch would be set to go into kernel 6.17 and work has not started on kernel 6.17. Here is a guide for compiling a kernel for Ubuntu:

https://canonical-kernel-docs.readthedocs-hosted.com/latest/how-to/develop-customise/build-kernel/

We would need to add a couple of lines of code to a specific file when you get to the following part of the guide:

Modify kernel configuration

You might want to read the guide before starting to see if you have any questions.

FYI: I do regularly build kernels as part of troubleshooting and testing but I do this mostly with Raspberry Pi OS. The compile process for kernels can vary depending on the distro. I'm sure I can clarify anything you need help with.

morrownr, 2025-06-30:
@pcraciunoiu 

From Google AI (which is not always correct):

```
The command echo 2c7c 7009 > /sys/bus/usb/drivers/btusb/new_id is a valid way to inform the btusb driver (a kernel module responsible for handling most Bluetooth USB adapters) to recognize a specific USB device ID (2c7c:7009) as a Bluetooth adapter, assuming that ID corresponds to a compatible device. 

Explanation:

/sys/bus/usb/drivers/btusb/new_id: This is a special file within the Linux sysfs filesystem. Sysfs exposes information about devices and kernel modules in a hierarchical file structure. Writing a Vendor ID (VID) and Product ID (PID) pair (like "2c7c 7009") to the new_id file of a driver essentially tells the driver to attempt to bind to a device with that specific ID.
btusb Driver: This is a generic kernel driver that supports a wide range of Bluetooth USB dongles.
echo "2c7c 7009": This command outputs the VID (2c7c) and PID (7009) to the standard output.

>: This redirects the standard output of the echo command to the specified file, in this case, /sys/bus/usb/drivers/btusb/new_id. 

Important Considerations:

Compatibility: This command will only work if the USB device with the VID:PID you specify (2c7c:7009 in your example) is actually a Bluetooth adapter and is supported by the btusb driver or a compatible kernel module.
Temporary Change: Writing to the new_id file through /sys/bus/usb/drivers/btusb/new_id is generally a temporary measure. The device will likely not be recognized after a reboot unless you implement a more permanent solution, like adding the device ID to the kernel driver or using udev rules.

Permissions: You will likely need root privileges to write to /sys/bus/usb/drivers/btusb/new_id. You can achieve this by using sudo before the command. 

If your goal is to add a USB device ID to the kernel driver permanently, you should consider filing a bug report to have the device ID included in the kernel source code. 

To make the change persistent, you can use udev rules to bind the device to the driver upon plugging it in. However, the command you provided is a valid method to test whether the driver can recognize the specific device ID at a given moment. 
```