morrownr/USB-WiFi #362: Unable to bring up the Netgear AX8000 card

kinhsman, 2024-01-11:
Really appreciate it if someone could help me with this issue.

I followed the steps on this guide and get the following error:
```
root@DietPi:~# systemctl status hostapd
○ hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: inactive (dead) since Thu 2024-01-11 02:16:06 GMT; 585ms ago
   Duration: 5.074s
       Docs: man:hostapd(8)
    Process: 1246 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS $DAEMON_CONF (code=exited, status=0/SUCCESS)
   Main PID: 1247 (code=exited, status=0/SUCCESS)
        CPU: 12ms

Jan 11 02:16:01 DietPi systemd[1]: Starting hostapd.service - Access point and authentication server for Wi-Fi and Ethernet...
Jan 11 02:16:01 DietPi systemd[1]: Started hostapd.service - Access point and authentication server for Wi-Fi and Ethernet.
Jan 11 02:16:06 DietPi hostapd[1247]: wlan0: IEEE 802.11 Configured channel (36) or frequency (5180) (secondary_channel=1) not found from the channel list of the current mode (2) IEEE 802.11a
Jan 11 02:16:06 DietPi hostapd[1247]: wlan0: IEEE 802.11 Hardware does not support configured channel
Jan 11 02:16:06 DietPi systemd[1]: hostapd.service: Deactivated successfully.
root@DietPi:~#
```
I am suspecting something is wrong with the channel configuration in my hostapd.conf file.

Below is the file content:

root@DietPi:~# cat /etc/hostapd/hostapd-WiFi6.conf
country_code=US
interface=wlan0
ssid=!MobileAX-5G
wpa_passphrase=sang1234
hw_mode=a
ieee80211d=1
ieee80211h=1
ieee80211n=1
ieee80211ac=1
ieee80211ax=1
wmm_enabled=1
preamble=1
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMEE][BF-ANTENNA-4][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
vht_oper_chwidth=1
he_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42
he_oper_centr_freq_seg0_idx=42
he_su_beamformer=1
he_su_beamformee=1
he_mu_beamformer=1
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256 SAE
rsn_pairwise=CCMP
root@DietPi:~#

I am running the DietPi VM and after upgrading the kernel to 6.5.0, it was able to detect the card:
```
root@DietPi:~# iw list
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 482 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
[... 642 lines cut ...]
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
root@DietPi:~#
```

kinhsman, 2024-01-11:
hostapd.log file:
```
Configuration file: /etc/hostapd/hostapd-WiFi6.conf
nl80211: TDLS supported
nl80211: TDLS external setup
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: key_mgmt=0x1ff0f enc=0xfef auth=0x7 flags=0x24003d10d35bfbe0 rrm_flags=0x30 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=4
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 4 iftype 3 (AP)
nl80211: Failed to set interface 4 to mode 3: -16 (Device or resource busy)
nl80211: Try mode change after setting interface down
nl80211: Set mode ifindex 4 iftype 3 (AP)
nl80211: Mode change succeeded while interface is down
nl80211: Setup AP(wlan0) - device_ap_sme=0 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x55c8a5b08a20
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x55c8a5b08a20 match= multicast=0
nl80211: Register frame type=0x0 (WLAN_FC_STYPE_ASSOC_REQ) nl_handle=0x55c8a5b08a20 match= multicast=0
nl80211: Register frame type=0x20 (WLAN_FC_STYPE_REASSOC_REQ) nl_handle=0x55c8a5b08a20 match= multicast=0
nl80211: Register frame type=0xa0 (WLAN_FC_STYPE_DISASSOC) nl_handle=0x55c8a5b08a20 match= multicast=0
nl80211: Register frame type=0xc0 (WLAN_FC_STYPE_DEAUTH) nl_handle=0x55c8a5b08a20 match= multicast=0
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x55c8a5b08a20 match= multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=04 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=0501 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=0503 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=0504 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=06 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=08 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=09 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=0a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=11 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=12 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c8a5b08a20 match=7f multicast=0
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 4 (ifidx_reason -1)
nl80211: if_indices[16]: 4(-1)
nl80211: Do not open EAPOL RX socket - using control port for RX
phy: phy0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code 00, new country code US
Continue interface setup after channel list update
ctrl_iface not configured!
RTM_NEWLINK: ifi_index=4 ifname=wlan0 operstate=2 linkmode=0 ifi_family=0 ifi_flags=0x1002 ()
nl80211: Ignore interface down event since interface wlan0 is up
RTM_NEWLINK: ifi_index=4 ifname=wlan0 operstate=2 linkmode=0 ifi_family=0 ifi_flags=0x1003 ([UP])
Channel list update timeout - try to continue anyway
nl80211: Regulatory information - country=00
nl80211: 2402-2472 @ 40 MHz 20 mBm
nl80211: 2457-2482 @ 20 MHz 20 mBm (no IR)
nl80211: 2474-2494 @ 20 MHz 20 mBm (no OFDM) (no IR)
nl80211: 5170-5250 @ 80 MHz 20 mBm (no IR)
nl80211: 5250-5330 @ 80 MHz 20 mBm (DFS) (no IR)
nl80211: 5490-5730 @ 160 MHz 20 mBm (DFS) (no IR)
nl80211: 5735-5835 @ 80 MHz 20 mBm (no IR)
nl80211: 57240-63720 @ 2160 MHz 0 mBm
[... 20 lines cut ...]
nl80211: send_mlme -> send_frame_cmd
nl80211: Frame command failed: ret=-22 (Invalid argument) (freq=0 wait=0)
wlan0: AP-DISABLED
hostapd_cleanup(hapd=0x55c8a5b01fd0 (wlan0))
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=0x55c8a446d760 drv_priv=0x55c8a5b03850 -> hapd_deinit
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=4)
netlink: Operstate: ifindex=4 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 4 iftype 2 (STATION)
nl80211: Failed to set interface 4 to mode 2: -16 (Device or resource busy)
nl80211: Try mode change after setting interface down
nl80211: Set mode ifindex 4 iftype 2 (STATION)
nl80211: Mode change succeeded while interface is down
nl80211: Teardown AP(wlan0) - device_ap_sme=0 use_monitor=0
nl80211: Unsubscribe mgmt frames handle 0x8888dd402d3802a9 (AP teardown)
hostapd_interface_free(0x55c8a5b00620)
hostapd_interface_free: free hapd 0x55c8a5b01fd0
hostapd_cleanup_iface(0x55c8a5b00620)
hostapd_cleanup_iface_partial(0x55c8a5b00620)
hostapd_cleanup_iface: free iface=0x55c8a5b00620
random: getrandom() support available
```

morrownr, 2024-01-11:
Hi @kinhsman 

> I followed the steps on this guide...

Which guide? The one here in USB-WiFi?

> my hostapd.conf 

I have an example for a WiFi 6 (5 GHz) hostapd.conf: Iit is on the Main Menu)

 https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.con

```
  * 5180 MHz [36] (20.0 dBm) (no IR)
  * 5200 MHz [40] (20.0 dBm) (no IR)
  * 5220 MHz [44] (20.0 dBm) (no IR)
  * 5240 MHz [48] (20.0 dBm) (no IR)
```

no IR means cannot initiate radiation which is a real problem for AP mode. Did you set the country code in DietPi. In RasPiOS, you can do it with raspi-config. I'm not sure how to do it with DietPi.

I also saw where you mentioned VM. If you are running a VM, you need to give the details.

Regards,

@morrownr

kinhsman, 2024-01-11:
Hi @morrownr 
Thank you for your response

> Which guide? The one here in USB-WiFi?
Yes, the one you provided. I also tried the hostapd-WiFi6.conf file you provided as well but it returned the same error.
Additionally, I had this DietPi VM working on another 2.4 GHz WiFi card (TP-LINK TL-WN722N v1) without issues.

>Did you set the country code in DietPi. In RasPiOS, you can do it with raspi-config. I'm not sure how to do it with DietPi.
Is this what you are refering to?
[image]

I also saw where you mentioned VM. If you are running a VM, you need to give the details.
The DietPi VM is running x86_x64 image kernel 6.5.0
The Netgear AX8000 card is passthrough-ed and recognized by the system:
```
root@DietPi:~# iw dev
phy#0
	Interface wlan0
		ifindex 4
		wdev 0x1
		addr 94:18:65:5b:04:79
		type managed
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
root@DietPi:~# 
root@DietPi:~# lsusb
Bus 003 Device 002: ID 0846:9060 NetGear, Inc. Wireless_Device
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
root@DietPi:~# 

root@DietPi:~# usb-devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=12  MxCh= 2
D:  Ver= 1.10 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0001 Rev=06.05
S:  Manufacturer=Linux 6.5.0-0.deb12.1-amd64 uhci_hcd
S:  Product=UHCI Host Controller
S:  SerialNumber=0000:00:01.2
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   2 Ivl=255ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=0627 ProdID=0001 Rev=00.00
S:  Manufacturer=QEMU
S:  Product=QEMU USB Tablet
S:  SerialNumber=28754-0000:00:01.2-1
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=81(I) Atr=03(Int.) MxPS=   8 Ivl=10ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=06.05
S:  Manufacturer=Linux 6.5.0-0.deb12.1-amd64 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:01:1b.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 4
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev=06.05
S:  Manufacturer=Linux 6.5.0-0.deb12.1-amd64 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:01:1b.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=03 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  1
P:  Vendor=0846 ProdID=9060 Rev=01.00
S:  Manufacturer=MediaTek Inc.
S:  Product=Wireless_Device
S:  SerialNumber=000000000
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=160mA
I:  If#= 0 Alt= 0 #EPs= 9 Cls=ff(vend.) Sub=ff Prot=ff Driver=mt7921u
E:  Ad=04(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=05(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=06(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=07(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=08(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=09(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=84(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=85(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=86(I) Atr=03(Int.) MxPS=   2 Ivl=125us
root@DietPi:~# 
```

morrownr, 2024-01-11:
> Additionally, I had this DietPi VM working on another 2.4 GHz WiFi card (TP-LINK TL-WN722N v1) without issues.

Country code is not that big of a deal for the 2.4 GHz band it is pretty standardized around the world. 5 GHz is another story so basically 5 GHz is shut down unless you have the correct country code info in place. I can tell by the info you posted that the lack of a country code setting is your core problem.

> Is this what you are refering to?

No. Look for the following:

L4 WLAN Country Set legal wireless channels for your country

@morrownr

kinhsman, 2024-01-11:
Hi @morrownr 
> No. Look for the following:
L4 WLAN Country Set legal wireless channels for your country

Is that not part of the hostapd.conf parameters?
I set the country code to US and selected channel 36 (which is allowed in the US)
If there is another place to set the country code other than the hostapd.conf file, can you please educate me?

Thanks!

Below is the content of the hostapd file:

```
root@DietPi:~# cat /etc/hostapd/hostapd-WiFi6.conf
ssid=!myPI-WiFi6
wpa_passphrase=12341234
hw_mode=a
channel=36
chanlist=36
country_code=US
ieee80211d=1
bridge=br0
interface=wlan0
driver=nl80211
logger_syslog=1
logger_syslog_level=2
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
beacon_int=100
dtim_period=2
multicast_to_unicast=0
tx_queue_data2_burst=2.0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
okc=1
wpa=2
wpa_pairwise=CCMP CCMP-256
wpa_key_mgmt=SAE WPA-PSK
ieee80211w=1
sae_require_mfp=1
ieee80211n=1
wmm_enabled=1
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
ieee80211ax=1
he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=42
```

closed 2024-01-11 by kinhsman