morrownr/USB-WiFi #365: AWUS036ACM : Extremely slow bitrate as a 2.4ghz AP

djheroez, 2024-01-16:
Hi,

I'm using the AWUS036ACM as AP on a Pi 4B. It's connected directly to the USB-2 port (I've also tried the USB-3 port with no luck) and no other device is connected to the Pi.

The bitrate is extremely slow but it's also unstable, sometimes it gives me ~2mb/s or ~30mb/s but never faster than that.

```bash
Connecting to host 10.0.1.1, port 5201
[  5] local 10.0.1.166 port 48598 connected to 10.0.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   256 KBytes  2.10 Mbits/sec    0   36.8 KBytes       
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    1   35.4 KBytes       
[  5]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec    0   28.3 KBytes       
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    2   19.8 KBytes       
[  5]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec    0   22.6 KBytes       
[  5]   5.00-6.00   sec   512 KBytes  4.19 Mbits/sec    0   33.9 KBytes       
[  5]   6.00-7.00   sec   256 KBytes  2.10 Mbits/sec    0   39.6 KBytes       
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    3   29.7 KBytes       
[  5]   8.00-9.00   sec   256 KBytes  2.10 Mbits/sec    1   25.5 KBytes       
[  5]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec    0   31.1 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.62 MBytes  1.36 Mbits/sec    7             sender
[  5]   0.00-10.01  sec  1.54 MBytes  1.29 Mbits/sec                  receiver

iperf Done.
```

My hostapd config is:

```bash
interface=wlan1
driver=nl80211
# bridge=br0
hw_mode=g
channel=6
country_code=US
ieee80211n=1
beacon_int=100
ignore_broadcast_ssid=0
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
rsn_pairwise=CCMP
ssid=WIPI
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-3839]
macaddr_acl=0
logger_syslog=1
logger_syslog_level=2
logger_stdout=1
logger_stdout_level=2
max_num_sta=50
dtim_period=2
rts_threshold=2347
fragm_threshold=2346
wmm_enabled=1
```

`# dmesg | grep mt` does not show anything weird I think:

```bash
[    4.956706] systemd[1]: Set up automount proc-sys-fs-binfmt_misc.automount - Arbitrary Executable File Formats File System Automount Point.
[    8.980059] mt76x2u 2-2:1.0: ASIC revision: 76120044
[    9.041606] mt76x2u 2-2:1.0: firmware: direct-loading firmware mt7662_rom_patch.bin
[    9.041649] mt76x2u 2-2:1.0: ROM patch build: 20141115060606a
[    9.189447] mt76x2u 2-2:1.0: firmware: direct-loading firmware mt7662.bin
[    9.189489] mt76x2u 2-2:1.0: Firmware Version: 0.0.00
[    9.189510] mt76x2u 2-2:1.0: Build: 1
[    9.189525] mt76x2u 2-2:1.0: Build Time: 201507311614____
[    9.941140] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   10.035046] usbcore: registered new interface driver mt76x2u
```

morrownr, 2024-01-19:
Replace your line with:

ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]

I think are trying to give the driver capabilities it does not have.

Have you looked at the hostapd log?

There could be networking setup problems? What guide did you use?

@morrownr

djheroez, 2024-01-20:
Hi @morrownr,

First of all, thanks a lot for your reply. What you suggested improved a ton the bitrate transfer:

```bash
Connecting to host 10.0.1.1, port 5201
[  5] local 10.0.1.166 port 50432 connected to 10.0.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  8.25 MBytes  69.2 Mbits/sec    0    372 KBytes       
[  5]   1.00-2.00   sec  5.88 MBytes  49.3 Mbits/sec    0    444 KBytes       
[  5]   2.00-3.00   sec  3.75 MBytes  31.5 Mbits/sec    1    310 KBytes       
[  5]   3.00-4.00   sec  2.75 MBytes  23.1 Mbits/sec    0    331 KBytes       
[  5]   4.00-5.00   sec  2.88 MBytes  24.1 Mbits/sec    0    378 KBytes       
[  5]   5.00-6.00   sec  3.88 MBytes  32.5 Mbits/sec    0    411 KBytes       
[  5]   6.00-7.00   sec  4.75 MBytes  39.8 Mbits/sec    0    438 KBytes       
[  5]   7.00-8.00   sec  5.62 MBytes  47.2 Mbits/sec    1    313 KBytes       
[  5]   8.00-9.00   sec  8.25 MBytes  69.2 Mbits/sec    0    344 KBytes       
[  5]   9.00-10.00  sec  6.62 MBytes  55.5 Mbits/sec    0    363 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  52.6 MBytes  44.1 Mbits/sec    2             sender
[  5]   0.00-10.03  sec  50.7 MBytes  42.4 Mbits/sec                  receiver

iperf Done.
```
But I'm noticing 2 things:
1. When I do an internet speed test, it starts fast at ~70mbps or even ~90mpbs but then it drops to ~30mbps or ~2mpbs
2. The AP seems to get congested somehow, it starts dropping packages when for example I'm watching a YouTube video.

This is the output of the hostapd service:

```bash
systemd[1]: Starting hostapd@wlan1.service - Access point and authentication server for Wi-Fi >
hostapd[1005]: random: getrandom() support available
hostapd[1005]: Configuration file: /etc/hostapd/wlan1.conf
hostapd[1005]: nl80211: TDLS supported
hostapd[1005]: nl80211: TDLS external setup
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:1
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:5
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:2
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:4
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:10
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:8
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:9
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:6
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:13
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:11
hostapd[1005]: nl80211: Supported cipher 00-0f-ac:12
hostapd[1005]: nl80211: Using driver-based off-channel TX
hostapd[1005]: nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 00>
hostapd[1005]: nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8>
hostapd[1005]: nl80211: Use separate P2P group interface (driver advertised support)
hostapd[1005]: nl80211: key_mgmt=0x1ff0f enc=0xfef auth=0x7 flags=0x4000531db5bfae0 rrm_flags=>
hostapd[1005]: nl80211: interface wlan1 in phy phy0
hostapd[1005]: nl80211: Set mode ifindex 4 iftype 3 (AP)
hostapd[1005]: nl80211: Failed to set interface 4 to mode 3: -16 (Device or resource busy)
hostapd[1005]: nl80211: Try mode change after setting interface down
hostapd[1005]: nl80211: Set mode ifindex 4 iftype 3 (AP)
hostapd[1005]: nl80211: Mode change succeeded while interface is down
hostapd[1005]: nl80211: Setup AP(wlan1) - device_ap_sme=0 use_monitor=0
hostapd[1005]: nl80211: Subscribe to mgmt frames with AP handle 0x55bc43a7f0
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x55bc43a7f0 m>
hostapd[1005]: nl80211: Register frame type=0x0 (WLAN_FC_STYPE_ASSOC_REQ) nl_handle=0x55bc43a7>
hostapd[1005]: nl80211: Register frame type=0x20 (WLAN_FC_STYPE_REASSOC_REQ) nl_handle=0x55bc4>
hostapd[1005]: nl80211: Register frame type=0xa0 (WLAN_FC_STYPE_DISASSOC) nl_handle=0x55bc43a7>
hostapd[1005]: nl80211: Register frame type=0xc0 (WLAN_FC_STYPE_DEAUTH) nl_handle=0x55bc43a7f0>
hostapd[1005]: nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x55bc43a>
hostapd[1005]: nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55bc43a7f0>
hostapd[1005]: nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55bc43a7f0>
hostapd[1005]: nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55bc43a7f0>
[... 17 lines cut ...]
hostapd[1005]: nl80211: 902-904 @ 2 MHz 30 mBm
hostapd[1005]: nl80211: 904-920 @ 16 MHz 30 mBm
hostapd[1005]: nl80211: 920-928 @ 8 MHz 30 mBm
hostapd[1005]: nl80211: 2400-2472 @ 40 MHz 30 mBm
hostapd[1005]: nl80211: 5150-5250 @ 80 MHz 23 mBm
hostapd[1005]: nl80211: 5250-5350 @ 80 MHz 24 mBm (DFS)
hostapd[1005]: nl80211: 5470-5730 @ 160 MHz 24 mBm (DFS)
hostapd[1005]: nl80211: 5730-5850 @ 80 MHz 30 mBm
hostapd[1005]: nl80211: 5850-5895 @ 40 MHz 27 mBm (no outdoor) (no IR)
hostapd[1005]: nl80211: 5925-7125 @ 320 MHz 12 mBm (no outdoor) (no IR)
hostapd[1005]: nl80211: 57240-71000 @ 2160 MHz 40 mBm
hostapd[1005]: nl80211: Added 802.11b mode based on 802.11g information
hostapd[1005]: nl80211: Mode IEEE 802.11g: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2>
hostapd[1005]: nl80211: Mode IEEE 802.11a: 5180 5200 5220 5240 5260[RADAR] 5280[RADAR] 5300[RA>
hostapd[1005]: nl80211: Mode IEEE 802.11b: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2>
hostapd[1005]: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
hostapd[1005]: Scan for neighboring BSSes prior to enabling 40 MHz channel
hostapd[1005]: 40 MHz affected channel range: [2407,2487] MHz
hostapd[1005]: wlan1: nl80211: scan request
hostapd[1005]: nl80211: Passive scan requested
hostapd[1005]: Scan requested (ret=0) - scan timeout 10 seconds
hostapd[1005]: Interface initialization will be completed in a callback
hostapd[1005]: ctrl_iface not configured!
systemd[1]: Started hostapd@wlan1.service - Access point and authentication server for Wi-Fi a>
```

djheroez, 2024-01-22:
I played a little more with the `ht_capab` and found that this setup gives me a more stable ~90-100mbps internet speed:

`ht-capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][RX-STBC1][MAX-AMSDU-3839]`

closed 2024-01-22 by djheroez