morrownr/USB-WiFi #368: How to instruct the OS to load the driver when the USB ID is not recognized?

fakemanhk, 2024-01-18:
I have an used IO-DATA WHG-AC433US dongle on hand, USB ID is 04BB:0955, looking up in DeviWiki the same ID gave me slightly different name [WN-AC433UK](https://deviwiki.com/wiki/I-O_DATA_WN-AC433UK), which should be MT7610U, and under Windows 10 I saw it loaded a Ralink driver (which I think it's before the merge to Mediatek?!) which makes me believe that MT7610U should be the correct one.

Under linux even with kernel 6.1 it can't recognize properly so I want to try to force the OS to use MT7610U and test if my guess is correct, is there any instruction?

morrownr, 2024-01-19:
Hi @fakemanhk 

You can try the following to see if it works:

```
$ su -
# modprobe mt7610u
# echo 04BB 9055 > /sys/bus/usb/drivers/mt7610u/new_id
```

If that works then you do have a mt7610u based adapter and can go with a more permenent solution which can be out in Main Menu item 2 Plug and Play list. Look for the Netgear A8000 adapter for an example. You will have to change a few things. If you have problems, come back and let us know.

@fakemanhk

fakemanhk, 2024-01-20:
Cool thanks!! It works!! Below are the log from OpenWrt 23.05.2 after doing what you told me. But on my Linux Mint the folder has locked and I am unable to edit, so I can't test it with my laptop. But in both OpenWrt 23.05.2 or latest Linux Mint, both are using module name _**mt76x0u**_ instead.

From OpenWrt I have created an AP which also works correctly and transferring data without issue.

So now is the question: Any way we can report and add this back to official Linux upstream kernel?

```
Sun Jan 21 02:17:27 2024 kern.info kernel: [ 4480.513520] usb 1-1.4: new high-speed USB device number 7 using dwc_otg
Sun Jan 21 02:17:27 2024 kern.info kernel: [ 4480.669326] usb 1-1.4: New USB device found, idVendor=04bb, idProduct=0955, bcdDevice= 1.00
Sun Jan 21 02:17:27 2024 kern.info kernel: [ 4480.682249] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sun Jan 21 02:17:27 2024 kern.info kernel: [ 4480.692022] usb 1-1.4: Product: WHG-AC433US
Sun Jan 21 02:17:27 2024 kern.info kernel: [ 4480.698536] usb 1-1.4: Manufacturer: I-O DATA
Sun Jan 21 02:17:27 2024 kern.info kernel: [ 4480.705158] usb 1-1.4: SerialNumber: XXXXXXXXXX
Sun Jan 21 02:17:27 2024 kern.info kernel: [ 4480.803681] usb 1-1.4: reset high-speed USB device number 7 using dwc_otg
Sun Jan 21 02:17:28 2024 kern.info kernel: [ 4480.955490] mt76x0u 1-1.4:1.0: ASIC revision: 76100002 MAC revision: 76502000
Sun Jan 21 02:17:29 2024 kern.info kernel: [ 4482.007238] mt76x0u 1-1.4:1.0: EEPROM ver:02 fae:01
```

morrownr, 2024-01-22:
> But on my Linux Mint the folder has locked and I am unable to edit, so I can't test it with my laptop

Try:

$ sudo su -

> module name mt76x0u

Well, I should have looked. Glad you figured it out.

> So now is the question: Any way we can report and add this back to official Linux upstream kernel?

Yes.  I don't have a guide but can point you in the right direction if you are interested.

@morrownr

fakemanhk, 2024-01-22:
Even with root I cannot edit that file in Linux Mint, weird.

morrownr, 2024-01-22:
Do you have a password set for root?

fakemanhk, 2024-01-22:
> Do you have a password set for root?

For sure I have root right because I did many things else, but I have no idea why those files in /sys is locked and not even root can modify them under Linux Mint.

fakemanhk, 2024-01-26:
> Yes. I don't have a guide but can point you in the right direction if you are interested.

I would like to try if it's not too complicated.

morrownr, 2024-01-26:
@fakemanhk 

I don't think the process is too complicated but you need to learn the process one bite at a time so as to understand it. I will start putting together a "one stop shopping guide" for folks here while working with you.

I recommend that you first go to the following site with the purpose of joining linux-wireless:

https://wireless.wiki.kernel.org/en/developers/mailinglists

It would be good if your email app has the ability to filter incoming mail on "linux-wireless" so that all linux-wireless email can go in a folder of its own. The purpose of doing this is so that you can read/scan traffic and get an idea of how things work. This also will give you an idea of what is going on. Since this is the USB WiFi site, I scan it frequently to see what is going on with drivers and firmware related to USB Wifi. You sometimes see me posting NEWS in Issues when I see something that might be of interest to folks that stop by.

Let me know how this goes and when we are ready, we can take another bite.

@morrownr