morrownr/USB-WiFi #369: "could not add STA to kernel driver" mt7921u , ALFA AWUS036AXML

Simon566, 2024-01-19:
Hi,

im on kernel 6.6.12 and running hostapd on the mt7921u device

Is this a hardware limit ? i didnt count the exact number of connected clients , its around 12 only

Jan 19 07:24:59 odroid-h3 hostapd[213153]: wlx00c0cab38118: STA 94:e2:3c:f2:29:f7 IEEE 802.11: Could not add STA to kernel driver
Jan 19 07:24:59 odroid-h3 hostapd[213153]: wlx00c0cab38118: STA 94:e2:3c:f2:29:f7 IEEE 802.11: Could not add STA to kernel driver
Jan 19 07:24:59 odroid-h3 hostapd[213153]: wlx00c0cab38118: STA 94:e2:3c:f2:29:f7 IEEE 802.11: Could not add STA to kernel driver
Jan 19 07:24:59 odroid-h3 hostapd[213153]: wlx00c0cab38118: STA 94:e2:3c:f2:29:f7 IEEE 802.11: Could not add STA to kernel driver
Jan 19 07:25:01 odroid-h3 hostapd[213153]: wlx00c0cab38118: STA 94:e2:3c:f2:29:f7 IEEE 802.11: Could not add STA to kernel driver
Jan 19 07:25:02 odroid-h3 hostapd[213153]: wlx00c0cab38118: STA 94:e2:3c:f2:29:f7 IEEE 802.11: Could not add STA to kernel driver

regards,
Simon

Simon566, 2024-01-19:
i can --PARTLY-- answer that myself.

It looks like the mt7921u driver has a very low limit for associated stations, something like 16 or so -> mt792x.h
#define MT792x_WTBL_SIZE	20
#define MT792x_WTBL_RESERVED	(MT792x_WTBL_SIZE - 1)
#define MT792x_WTBL_STA		(MT792x_WTBL_RESERVED - MT792x_MAX_INTERFACES)

the "worktable" aka WTBL seems to be pretty small , no idea why.

Can we ramp up that limit ?

morrownr, 2024-01-19:
Hi @Simon566 

It helps me understand if I know what hardward and software is involved. You mentioned hostapd but nothing else. Please provide some details.

Can you post your hostapd.conf file?
What AP guide did you use?

I use my ALFA AXML in AP mode but this report is a new one on me.

@morrownr

Simon566, 2024-01-20:
Hi,

im running:
- Odroid H3 (X86-64), Intel 5105 cpu
- debian bookworm
- hostapd (latest git mainline)

i didnt use any specific guideline but the sample hostapd,conf from the SW maintainer website and adapted it, see here:

##### hostapd configuration file ##############################################
# Empty lines and lines starting with # are ignored

# AP netdevice name (without 'ap' postfix, i.e., wlan0 uses wlan0ap for
# management frames); ath0 for madwifi
interface=wlx00c0cab38118

# In case of madwifi, atheros, and nl80211 driver interfaces, an additional
# configuration parameter, bridge, may be used to notify hostapd if the
# interface is included in a bridge. This parameter is not used with Host AP
# driver. If the bridge parameter is not set, the drivers will automatically
# figure out the bridge interface (assuming sysfs is enabled and mounted to
# /sys) and this parameter may not be needed.
#
# For nl80211, this parameter can be used to request the AP interface to be
# added to the bridge automatically (brctl may refuse to do this before hostapd
# has been started to change the interface mode). If needed, the bridge
# interface is also created.
bridge=br-guest
#multicast_to_unicast=1

# Driver interface type (hostap/wired/madwifi/test/none/nl80211/bsd);
# default: hostap). nl80211 is used with all Linux mac80211 drivers.
# Use driver=none if building hostapd as a standalone RADIUS server that does
# not control any wireless/wired driver.
# driver=hostap
driver=nl80211

# hostapd event logger configuration
#
# Two output method: syslog and stdout (only usable if not forking to
# background).
#
# Module bitfield (ORed bitfield of modules that will be logged; -1 = all
# modules):
# bit 0 (1) = IEEE 802.11
# bit 1 (2) = IEEE 802.1X
# bit 2 (4) = RADIUS
# bit 3 (8) = WPA
# bit 4 (16) = driver interface
# bit 5 (32) = IAPP
# bit 6 (64) = MLME
#
# Levels (minimum value for logged events):
#  0 = verbose debugging
#  1 = debugging
#  2 = informational messages
#  3 = notification
#  4 = warning
#
logger_syslog=1
logger_syslog_level=3
logger_stdout=1
logger_stdout_level=3

# Dump file for state information (on SIGUSR1)
#dump_file=/tmp/hostapd.dump

# Interface for separate control program. If this is specified, hostapd
# will create this directory and a UNIX domain socket for listening to requests
# from external programs (CLI/GUI, etc.) for status information and
[... 1638 lines cut ...]
# masks may be used if the driver supports them (e.g., swap the locally
# administered bit)
#
# BSSIDs are assigned in order to each BSS, unless an explicit BSSID is
# specified using the 'bssid' parameter.
# If an explicit BSSID is specified, it must be chosen such that it:
# - results in a valid MASK that covers it and the dev_addr
# - is not the same as the MAC address of the radio
# - is not the same as any other explicitly specified BSSID
#
# Please note that hostapd uses some of the values configured for the first BSS
# as the defaults for the following BSSes. However, it is recommended that all
# BSSes include explicit configuration of all relevant configuration items.
#
#bss=wlan0_0
#ssid=test2
# most of the above items can be used here (apart from radio interface specific
# items, like channel)

#bss=wlan0_1
#bssid=00:13:10:95:fe:0b
# ...

regards,
Simon

Simon566, 2024-01-23:
The MT792x_WTBL_SIZE  as seen above has something to do with the RAM which is available inside the USB device or not ?

The ALFA device is able to load the BT firmware assoiated with the chipset, but hasnt got any bluetooth in real.
-> there should be plenty of RAM available to ramp up that limit in case bluetooth isnt needed !!

is this a limitation by the firmware or can we just happily increase the #define, recompile and give it a go ?

Simon566, 2024-03-04:
Hi @morrownr ,

is there any chance that a mediatek dev might look at this ? fixing/increasing the max conntected client amount (to maybe 64?) shouldnt be a big issue if there is enough RAM ( which i really suspect for this device)

see here a related OpenWRT ticket:
https://forum.openwrt.org/t/reference-for-number-of-supported-stations-in-different-wifi-hardware/156535

regards,
Simon

morrownr, 2024-03-04:
> is there any chance that a mediatek dev might look at this ?

Ask them:

https://wireless.wiki.kernel.org/en/users/drivers/mediatek

Simon566, 2024-08-28:
i repeatedly contacted Felix Fietkau , lets see . i havent got any reply

kbmetz, 2025-01-12:
This is still an issue, is there any way to work around this? Or is is possible to try and recompile the module?

I have a AX3000 with the mt7921u module loaded running hostapd. I've tried this on Raspbian, Debian and Ubuntu 24.04. When I start hostapd everything runs fine. Anywhere between 16-20 clients I start getting the 'cloud not add STA to kernel driver'. This makes the USB stick unusable as an AP in my opinion.