morrownr/USB-WiFi #370: Range comparison between AWUS036AXML and AWUS036ACHM.

VicentG3, 2024-01-19:
Hi,

I have been trying to find out if the new AWUS036AXML is comparable in range against the older Alfa AWUS036ACHM. Is the new adapter a better option overall or is the older one still a better choice?

Can someone who has both compare them or help me decide.

Thanks.

morrownr, 2024-01-19:
Hi @VicentG3 

I have both. The one you depends on entirely on the features give priority to. You mentioned range. If range is your only considerization then the ACHM is the winner as it is what Alfa calls a Max Power adapter. Alfa periodically makes an adapter to have better range that basically anything else out there and the ACHM is one of those. The AXML is a step down in that it is a High Power adapter. Both cost more than most competing adapters as they use better materials  and have features many adapters do not. Both have antennas that are removable to you can replace them with long range one direction antennas if you want. Alfa makes those. Just get the ones for you adapters.

Both also come with extension cables that are good quality. The cable for the AXML is USB-C on both ends but has a little adapter to convert one end to USB-A in case you need it. Both adapters are capable of running at full speed indefinitely as the their heat output is very low. Both adapters are rated at the top of the list for Kali Linux compatibility with the ACHM being rately slight higher as it supports things like Active Monitor mode without issue. The AXML supports monitor mode very well but there is currently a bug in Active Monitor mode.

Both are plug and play on nearly all mainstream desktop distros but it you need support for really old distros, the ACHM is the one to buy as they driver has been in the kernel since 2018.

Tell us what else you need to do with the adapter?

@morrownr

VicentG3, 2024-01-20:
Hi @morrownr thanks for the feedback

Mainly mi objective is to learn about WiFi pentesting (mainly Kali so compatibility is not an issue) so I was looking for a adapter that has and above average range but I do not need something with extreme range. I don't know if the AXML has better range than the average WiFi adapter considering that is only high power as you said. Also it would be a positive point if the adapter is fast for daily use in things like downloading games or browsing the web. 

Is the ACHM much better in range or the difference is not really that big?

Thanks for all the help.

morrownr, 2024-01-20:
> mainly Kali 

Will you be using Kali on bare metal or in a VM?

> Also it would be a positive point if the adapter is fast for daily use in things like downloading games or browsing the web.

What do you consider fast? What is the fastest speed your internet connect can support?

> Is the ACHM much better in range or the difference is not really that big?

I would consider the ACHM to have much better range that basically any dual-band adapter on the market today. Here is a link to an old range test that I need to update:

https://github.com/morrownr/USB-WiFi/blob/main/home/Performance_Comparison.md

You will note in the test that there is a lot of difference in the `signal level` of the adapters and that directly applies to range.The Alfa ACH also is a Max Power adapter with outstanding range but it uses a rtl8812au chipset that has poor Linux support for a lot of things including monitor mode.

Zerbea has a list on his hcxsumptool site that spells out what he things are the best monitor mode adapters:

https://github.com/ZerBea/hcxdumptool/discussions/361

Note that he lists the Alfa ACM and ACHM as the best for monitor mode. He does not currently list the AXML as the driver for the mt7921au chipset has a bug with active monitor mode. Monitor mode works very well but active monitor mode is not working so he is keeping it off the list. The ACM is a really good AC1200 adapter and it works well but has problems running in VMs.

My opinion based on the information you have shared so far is that a good adapter for you at this point would be the ACHM. It does monitor mode and active monitor capability as well as anything you might want to do with Kali as well or better than any adapter available. It is plug and play on Linux and Windows. It works in VMs. It is what I call a 24/7/365 adapter in that it can run full speed indefinitely without problems. It is an AC600 class adapter and while it is far from the speediest adapter available. it can sustain 200+ Mbps on 5 GHz which is really not bad at all.

At some point in the future, if you find that you want a faster adapter, you can get on with the mt79au chipset like the AXML or we should be seeing new WiFi 7 adapters based on the new mt7925 chipset sometime this year. The driver for the mt7925, USB and PCIe, went into Linux kernel 6.7 and the firmware is in place on the Linux website for distros to grab.

Hope this helps.

@morrownr

VicentG3, 2024-01-21:
> Will you be using Kali on bare metal or in a VM?

I only use Kali on VM.

Thanks for all the feedback, I think i will stick with ACHM because all the support, and for learning I prefer not to worry about software incompatibility problems. In addition mi internet connection is not much faster than 200 Mbps  so being limited by that is not a big deal.
The AWUS036AXML is nice but the lack of some functionalities is quite deal breaking and the extra speed is a nice addition but is not that necessary.

Thank you very much for your time and for sharing your extensive knowledge it has been very helpful.

arbv, 2024-04-12:
@morrownr Can you tell me please if the SMA connectors are only for antennas that work in 5GHz and 6GHz bands or they will for 2.4 Ghz range too?

I have found [an image](https://preview.redd.it/awus036axml-range-v0-uw0esr3fl48c1.jpg?width=1163&format=pjpg&auto=webp&s=147d792ca16cf7bd9480fec6c7cd4ed8be174a88) that suggests that for 2.4 Ghz PCB antennas are used.

morrownr, 2024-04-12:
> Can you tell me please if the SMA connectors are only for antennas that work in 5GHz and 6GHz bands or they will for 2.4 Ghz range too?

Are you saying that you want to replace the existing antennas with ones that are turned more toward the 2.4 GHz band?

If that is the case, I think that would work but I am not an EE. I could test but my AXML is busy to doing other testing right now. It should be free in a few days if you want me to test. I do have extra antennas that are specific to the 2.4 GHz band.

morrownr, 2024-04-12:
My ACHM is available for testing now. Would that work? You were not specific about what hardware you have.

arbv, 2024-04-15:
> Are you saying that you want to replace the existing antennas with ones that are turned more toward the 2.4 GHz band?
>
> If that is the case, I think that would work but I am not an EE. I could test [...]

No-no. I was wondering if the SMA antennas are responsible for all Wi-Fi bands or only a subset of them (like being for 5 or 6 GHz only), as the device has internal 2.4 GHz PCB antennas additionally to the ones connected via SMA. Now, as I have received the device, I can tell that the internal 2.4 GHz PCB antennas appear to be used for Bluetooth only (that is depicted right on the packaging).

morrownr, 2024-04-17:
> I was wondering if the SMA antennas are responsible for all Wi-Fi bands or only a subset of them...

This is a good question. The answer is that if you have a tri-band adapter, you would actually need 5 antennas to be fully optimized for all three bands.

xnvmx, 2024-08-24:
Hi @morrownr 

I was also wondering about comparing the above cards in terms of range and I'm also curious how the Alfa AWUS1900 card compares to them. Have you had this card and tested it in this regard?

Has anything changed and is there anything better in terms of range than Alfa AWUS036ACHM?

> I would consider the ACHM to have much better range that basically any dual-band adapter on the market today. Here is a link to an old range test that I need to update:
> 
> https://github.com/morrownr/USB-WiFi/blob/main/home/Performance_Comparison.md

When can we expect the update?

morrownr, 2024-08-25:
Hi @xnvmx 

> I was also wondering about comparing the above cards in terms of range and I'm also curious how the Alfa AWUS1900 card compares to them. Have you had this card and tested it in this regard?

I do not own an Alfa 1900. Based info from Alfa and many discussions with those who do own it, I would estimate that it has better than average range but not what you would get with the Alfa achm. Alfa breaks their extended range adapter into these categories: Standard Power, High Power and Max Power. The 1900 is a high power rated adapter. We know that power is not what gives the range, it is the quality of the AMP and antennas but let's press on. The Alfa ACM is also a high power rated adapter. I do have one of those and it does have good range but nowhere near the ACHM. I also have an Alfa ACH which is also, like the ACHM, a Max Power adapter, and it has nearly as good of range as the ACHM. The problem with the 1900 and ACH has been poor driver support from Realtek.

A new standards compliant driver for the ACH is under review currently and will hopefully make it into kernel 6.12. Does this driver compete well with the Mediatek drivers? No, not yet. If you need things Active monitor mode, you need to stick with an Alfa ACHM or ACM. If you do pen testing or the like, go with the ACHM.

Regarding the 1900, it has suffered from a particulatly bad Realtek driver. There are plans to work on a new standards compliant driver but don't count on it until you see it in the kernel. I recommend Linux users stay away from the 1900 or any rtl8814au based adapter due to poor driver support from Realtek.

> Has anything changed and is there anything better in terms of range than Alfa AWUS036ACHM?

Nothing to this point. One adapter that I have has surprisingly good range: EDUP EP-AX1672. It is shown in the Plug and Play List as are the Alfa ACM and Alfa ACHM.

> When can we expect the update?

Good question. It actually takes a lot of planning to produce something that can yield meaningful results. I have moved since the last test so I would need to determine a good location for the adapters to be tested and then there is the issue of which adapters to test and what band and what channel. Your suggestions are welcome.