morrownr/USB-WiFi #372: Need help with mt7921au based adapter in a proxmox VM

kinhsman, 2024-01-21:
**Problem:**
I am encountering performance issues with mt7921au-based adapters when used in a Proxmox VM with USB passthrough. During iperf speed tests to a wired client on the same VLAN, I consistently achieve only around ~150 Mbps. I have moved the VM to my gaming Windows PC and run it via VMware workstation with a USB3.0 passthough, and the results are about the same

**Hardware:**

- Tested USB adapters: Fenvi FU-AX1800 and Comfast CF-953AX
- Proxmox hardware: Intel N100 CPU (4x core; 3.40 GHz)
- VM configuration: 1xCPU (host), 2GB RAM, USB 3.0 passthrough. I have also experimented with 4xCPUs with no improvement.
- Tested Operating Systems: DietPi OS (Debian based), OpenWRT

**Comparison with Bare Metal Setup:**
When I ran the USB dongle on the same OS but in a bare metal setup (tested with a NanoPi R6S), I achieved optimal results of around ~650 Mbps.

**Question:**
Has anyone successfully configured any of these adapters to achieve optimal results? While I'm aiming for approximately ~300 Mbps, the current speed of 150 Mbps is far from ideal. Any suggestions would be greatly appreciated.

kinhsman, 2024-01-21:
Resolved:
I have came across the following proxmox threads suggested to PCIe passthrough the entire USB controller instead of USB passthrough to get native speed
https://www.reddit.com/r/Proxmox/comments/xks9dq/usb_passthrough_really_slow/
https://forum.proxmox.com/threads/how-do-i-add-an-entire-usb-controller-to-a-vm.113445/

I am now able to get 300Mbps after this change.

```
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.01  sec  45.8 MBytes  38.3 Mbits/sec                  sender
[  4]   0.00-10.01  sec  45.8 MBytes  38.3 Mbits/sec                  receiver
[  6]   0.00-10.01  sec  41.2 MBytes  34.6 Mbits/sec                  sender
[  6]   0.00-10.01  sec  41.2 MBytes  34.6 Mbits/sec                  receiver
[  8]   0.00-10.01  sec  43.4 MBytes  36.3 Mbits/sec                  sender
[  8]   0.00-10.01  sec  43.4 MBytes  36.3 Mbits/sec                  receiver
[ 10]   0.00-10.01  sec  45.8 MBytes  38.3 Mbits/sec                  sender
[ 10]   0.00-10.01  sec  45.8 MBytes  38.3 Mbits/sec                  receiver
[ 12]   0.00-10.01  sec  45.6 MBytes  38.2 Mbits/sec                  sender
[ 12]   0.00-10.01  sec  45.6 MBytes  38.2 Mbits/sec                  receiver
[ 14]   0.00-10.01  sec  45.4 MBytes  38.0 Mbits/sec                  sender
[ 14]   0.00-10.01  sec  45.4 MBytes  38.0 Mbits/sec                  receiver
[ 16]   0.00-10.01  sec  45.4 MBytes  38.0 Mbits/sec                  sender
[ 16]   0.00-10.01  sec  45.4 MBytes  38.0 Mbits/sec                  receiver
[ 18]   0.00-10.01  sec  45.5 MBytes  38.1 Mbits/sec                  sender
[ 18]   0.00-10.01  sec  45.5 MBytes  38.1 Mbits/sec                  receiver
[SUM]   0.00-10.01  sec   358 MBytes   300 Mbits/sec                  sender
[SUM]   0.00-10.01  sec   358 MBytes   300 Mbits/sec                  receiver

iperf Done.
```

closed 2024-01-21 by kinhsman

DatHardie, 2024-11-03:
> Resolved: I have came across the following proxmox threads suggested to PCIe passthrough the entire USB controller instead of USB passthrough to get native speed https://www.reddit.com/r/Proxmox/comments/xks9dq/usb_passthrough_really_slow/ https://forum.proxmox.com/threads/how-do-i-add-an-entire-usb-controller-to-a-vm.113445/
> 
> I am now able to get 600Mbps after this change.
> 
> ```
> [ ID] Interval           Transfer     Bandwidth
> [  4]   0.00-10.00  sec   101 MBytes  84.8 Mbits/sec                  sender
> [  4]   0.00-10.00  sec   101 MBytes  84.8 Mbits/sec                  receiver
> [  6]   0.00-10.00  sec  98.8 MBytes  82.8 Mbits/sec                  sender
> [  6]   0.00-10.00  sec  98.8 MBytes  82.8 Mbits/sec                  receiver
> [  8]   0.00-10.00  sec  97.4 MBytes  81.6 Mbits/sec                  sender
> [  8]   0.00-10.00  sec  97.4 MBytes  81.6 Mbits/sec                  receiver
> [ 10]   0.00-10.00  sec  96.4 MBytes  80.8 Mbits/sec                  sender
> [ 10]   0.00-10.00  sec  96.4 MBytes  80.8 Mbits/sec                  receiver
> [ 12]   0.00-10.00  sec  94.5 MBytes  79.2 Mbits/sec                  sender
> [ 12]   0.00-10.00  sec  94.5 MBytes  79.2 Mbits/sec                  receiver
> [ 14]   0.00-10.00  sec  92.6 MBytes  77.7 Mbits/sec                  sender
> [ 14]   0.00-10.00  sec  92.6 MBytes  77.7 Mbits/sec                  receiver
> [ 16]   0.00-10.00  sec  89.9 MBytes  75.4 Mbits/sec                  sender
> [ 16]   0.00-10.00  sec  89.9 MBytes  75.4 Mbits/sec                  receiver
> [ 18]   0.00-10.00  sec  86.6 MBytes  72.6 Mbits/sec                  sender
> [ 18]   0.00-10.00  sec  86.6 MBytes  72.6 Mbits/sec                  receiver
> [SUM]   0.00-10.00  sec   757 MBytes   635 Mbits/sec                  sender
> [SUM]   0.00-10.00  sec   757 MBytes   635 Mbits/sec                  receiver
> 
> iperf Done.
> ```

Did you also used the CF-953AX on a N100 ?
Having the same problem here.

kmod-mt7921u is installed in openwrt.

After setting up passthrough over pci keyboard and mouse LED start flickering for a few seconds,
So something is happening for sure, but the there is no new interface appearing in the interface list in openwrt.
Even after VM and Proxmox restart.

Maybe you have some comments that could help.