morrownr/USB-WiFi #373: Looking for insights for Improving OpenWRT Setup

joopdo, 2024-01-25:
I've been running my current setup for 1.5 years now (since Jun 2022), utilizing an Alfa AWUS036ACHM and a COMFAST CF-WU782AC.

Here's the breakdown of my OpenWRT setup:

AWUS036ACHM: Running in 2.4 GHz, catering to LAN network and Guest network.
CF-WU782AC: Also operating in 2.4 GHz, specifically for the IOT network.

While the AWUS036ACHM has proven excellent for range in both LAN and Guest networks, I'm looking to extend the range for my IOT network.

Considerations:

I'd like to somehow integrate IOT into my AWUS036ACHM, but it appears limited to a maximum of 2 networks due to interface restrictions.

```` 
valid interface combinations:
* #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2,
   total <= 2, #channels <= 1, STA/AP BI must match
````
I'm considering adding a 5 GHz or 6 GHz band AP to the network for faster mobile and laptop up/downloads. Any recommendations?

How would you approach the situation?
I was anticipating the AWUS036ACH, but it seems to support a maximum of 2 networks as well ([source](https://github.com/morrownr/USB-WiFi/issues/34#issuecomment-1005423153)). Any suggestions on combining 'Guest' and 'IOT' networks efficiently?

morrownr, 2024-01-25:
Hi @joopdo 

> How would you approach the situation?

A military saying: A good plan never survives contact with the enemy.

But having a plan is important. I like to write down a plan but then I test like crazy to see what is actually working like I want.

The ACHM is flexible as it has good range on both bands. The 782AC is also flexible as it is dual band but the range is not like the ACHM... but the range is not bad at all, I think I would first ensure the IoT is taken care of. If IoT is handled well by both adapters, then you have the flexibility of what to use for 5 GHz. I usually use 5 GHz for guests. 

My own setup uses 2.4 for IoT and 2 FHD streams to TVs. Basically, anything that will annoy the wife or rest of the family I have on one 2.4 GHz channel. 5 GHz is for things that require more bandwidth and guest APs. So, it appears that the plan I would start with 2.4 GHz on the 782AC and 5 GHz on the ACHM. There are a lot of variables so what works best may be far from what I am throwing out.

I have 2 adapters based on the mt7921au chipset. They are a pleasure to use. Both have WiFi 6 and WiFi 6e capability. WiFi 6 works well on the 5 GHz band for situations that require a lot of throughput. I also have made an example hostapd.conf in the AP section of the Main Menu. It works well. I have not messed with WiFi 6e much yet as it gets complicated, many countries do not have 6 GHz laws in place and 6 GHz drops power and range compared to 5 GHz. I will get around to working with 6 GHz one of these days but not now.

My main devs box has a PCIe card based on the mt7922 chipset which is capable of 160 MHz channel width. My main wifi router is a WiFi 6 capable and 160 MHz channel width capable. I also usually have 2 to 4 usb wifi adapters connected. I do a lot of testing here. When I need speed, I connect the PCIe card to WiFi 6:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 45276 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   125 MBytes  1.05 Gbits/sec    0    993 KBytes       
[  5]   1.00-2.00   sec   130 MBytes  1.09 Gbits/sec    0    993 KBytes       
[  5]   2.00-3.00   sec   129 MBytes  1.08 Gbits/sec    0    993 KBytes       
[  5]   3.00-4.00   sec   130 MBytes  1.09 Gbits/sec    0    993 KBytes       
[  5]   4.00-5.00   sec   130 MBytes  1.09 Gbits/sec    0    993 KBytes       
[  5]   5.00-6.00   sec   126 MBytes  1.06 Gbits/sec    0    993 KBytes       
[  5]   6.00-7.00   sec   128 MBytes  1.07 Gbits/sec    0    993 KBytes       
[  5]   7.00-8.00   sec   130 MBytes  1.09 Gbits/sec    0    993 KBytes       
[  5]   8.00-9.00   sec   129 MBytes  1.08 Gbits/sec    0    993 KBytes       
[  5]   9.00-10.00  sec   126 MBytes  1.06 Gbits/sec    0    993 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.25 GBytes  1.08 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.25 GBytes  1.07 Gbits/sec                  receiver

iperf Done.

```

Yes, that is over 1 Gbps you are seeing. Smoking... wireless that is faster than 1 Gbps ethernet cable.

@morrownr

joopdo, 2024-01-26:
Thank you for the detailed information @morrownr ! I appreciate you taking the time to share your knowledge and personal experiences with us here. 

The Wifi 6 throughput test results (Cudy WR3000 I saw in another issue?) are impressive - something to look forward too!

I will keep your guidance on MT7921/7922 chipsets, WiFi 6/6E capabilities, and 160MHz channel widths in mind. For now I just want to thank you again for generously providing all this valuable perspective to aid my own wireless planning. Your helpful spirit is commendable.

morrownr, 2024-01-26:
@joopdo 

Thanks for the kind words. I try to keep in mind some famous words that seem applicable:

"Ask not what your operating system can do for you, ask what you can do for your operating system."

Well, maybe that wasn't an exact quote but you get the idea. Have a great day.

@morrownr

joopdo, 2024-04-22:
@morrownr Hope all is fine with you. I recently acquired a AWUS036AXML adapter (next to AWUS036ACHM) and I've been trying to set up multiple networks (specifically, SSID_5G and SSID_IOT) on my OpenWRT router using this adapter. However, I've encountered some difficulties that I believe might be related to the adapter's capabilities, I'd like to share my findings to understand if they are correct.

Upon reviewing the adapter's specifications I noticed that the adapter supports the following interface combinations:

valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 2

Based on this information, I assumed that I could configure **_two_** separate wireless networks using the two available "managed" interfaces. However, despite my attempts, I haven't been successful in achieving this configuration.

I also just came across your overview page, and only now I read the comment that the adapter is a "single-state adapter," which made me wonder if this means that there is indeed no way to setup two SSID's.

Could you please provide some guidance or clarification on how I can properly configure the AWUS036AXML adapter to support multiple networks on OpenWRT? I would greatly appreciate any insights or suggestions you might have regarding this matter.

morrownr, 2024-04-23:
Hi @joopdo 

>  Is it possible?

It is not clear what you are trying to do.

> Based on this information, I assumed that I could configure two separate wireless networks 

All usb wifi adapters that I am aware of only contain a single radio so two separate networks is not going to work. The capability lines that you posted show only one interface of type AP at a time. You could also have another interface of type managed if you want.

> I read the comment that the adapter is a "single-state adapter," 

Single state means that it does not have an onboard Windows driver which can cause problems. Linux users want single-state adapters.

> Could you please provide some guidance or clarification on how I can properly configure the AWUS036AXML adapter to support multiple networks on OpenWRT?

If you want your OpenWRT setup to have two APs, you need two adapters or maybe use an onboard wifi capability. I usually use an adapter like the AXML for 5 GHz and something that has a rt5370 for 2.4 Ghz band or it could be a lot of different adapters depending on exactly what you want. 

I don't know if I have helped or made a mess. 

@morrownr

joopdo, 2024-04-23:
@morrownr I just wanted to drop a quick note to say a big thank you for taking the time to provide such detailed guidance on configuring the AWUS036AXML adapter with OpenWRT. Your insights are incredibly helpful, and I really appreciate your patience in explaining the nuances of setting up multiple networks with this adapter!

I guess the right word that I was looking for is 'interface'. I want two interfaces on a single AP, but that seems not possible with this device. So it has to be a single interface on one AP.

joopdo, 2024-04-23:
Thank you so much for your detailed explanation and clarification. I really appreciate you taking the time to delve into the intricacies of the AWUS036AXML adapter and its capabilities.

Apologies if my original question caused any confusion by mixing up terminologies. Indeed, what I meant to inquire about was the possibility of having two Access Points (APs) on the AXML adapter. With your guidance, I now understand that this isn't feasible due to the limitation on the number of AP interfaces.

> Below shows my 2 wireless interfaces on my RasPi4B Bridged AP

Your example with the Raspberry Pi setup really helped illustrate the concept. Having two interfaces in AP mode on different channels using separate USB WiFi adapters makes perfect sense. It's a clear demonstration of how the capabilities of different chipsets dictate the setup possibilities. I'll do this with the COMFAST CF-WU782AC I mentioned earlier, which is also capable of this.

Your explanation regarding the #{ AP, P2P-GO } <= 1 limitation on the AXML adapter clarified why having two AP interfaces simultaneously isn't possible. I kind of knew it but I wanted to make sure I wasn't overlooking something. Fun fact, yesterday I even tried setting up two radio's in OpenWRT as follows, and it came up! What I tried was defining two radio's with the same path, unfortunately, only one wifi-iface came online:

```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel 'auto'
	option hwmode '11g'
	option path 'platform/usb@fe800000/fe800000.usb/xhci-hcd.0.auto/usb6/6-1/6-1.4/6-1.4.2/6-1.4.2:1.3'
	option htmode 'HE20'
	option cell_density '0'

config wifi-iface 'guest_radio0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'name_GUEST'
	option encryption 'psk2'
	option key ''
	option network 'GUESTNETWORK'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel 'auto'
	option hwmode '11g'
	option path 'platform/usb@fe800000/fe800000.usb/xhci-hcd.0.auto/usb6/6-1/6-1.4/6-1.4.2/6-1.4.2:1.3'
	option htmode 'HE20'
	option cell_density '0'

config wifi-iface 'guest_radio1'
	option device 'radio1'
	option mode 'ap'
	option ssid 'name_5G'
	option encryption 'psk2'
	option key ''
	option network 'LAN'
````

But it was funny that it one radio did come up and it didn't just froze. As doing this with the 'AWUS036ACHM' if you setup a double setup like that, it would freeze. Anyway, at that point I realized that I tried about everything and two AP's are not possible in any way, hence my reachout to you earlier. I know now what is the technical limitations of the AXML. It's fascinating to see how these technical constraints shape the setup options.

Overall, your insights have been incredibly helpful in understanding the nuances of wireless networking configurations, especially within the context of OpenWRT. Your expertise is invaluable, and I'm grateful for your willingness to share it.

Thank you once again for your patience and guidance. It's making a lot more sense now, and I'm looking forward to applying this knowledge to my setup effectively :+1:

morrownr, 2024-04-23:
> Overall, your insights have been incredibly helpful in understanding the nuances of wireless networking configurations, especially within the context of OpenWRT. Your expertise is invaluable, and I'm grateful for your willingness to share it.

You are welcome but you need to know that you do not become an expert in a few days. There is a lot to learn but once you get to a certain level, you can really do setups that work really well. The basic setup that I use with my RasPi4B is outlined in the Main Menu item 8 AP Mode... . On the line under item 8 is an example hostapd.conf for the mt7921au chip which what your AXML uses, The AXML can do WiFi 6. I would recommend WiFi 6 and not WiFi 6e (6 GHz) until you are really good at this. I use and enjoy OpenWRT as well and I think it is the best solution for a routed AP setup as opposed to a bridged setup. Remember that in OpenWRT you have to install the wifi drivers for the adapters you want to use. The AXML uses the mt7921u driver and the ACHM uses the mt7610u driver. I recommend the AXML for a 5 GHz, wifi 6 AP and the ACHM for a 2.4 Ghz WiFi 4 AP. Recommend that the ACHM use WPA2 AES with 20 MHz channel width and the AXML use mixed WPA2/WPA3 and 80 MHz channel width,

Let me know if you have any more questions.

@morrownr