morrownr/USB-WiFi #374: Future WiFi Speeds

morrownr, 2024-01-25:
For many years now we have been measuring WiFi speed in Mbits/sec (Mbps). Now we need to start thinking about Gbps. The below output from iperf3 is me testing with my mt7922 based PCIe card, The nt7922 is capable of WiFi 6 with a channel width of 160 MHz. I am using channel 116 DFS as it is clear for the channel width. There are no other APs on DFS channel in my are so this is clean air. My WiFi router is dual band and is capable of WiFi 6, channel width 160. It is based on the mt7982 SoC in case you are interested. 

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 45276 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   125 MBytes  1.05 Gbits/sec    0    993 KBytes       
[  5]   1.00-2.00   sec   130 MBytes  1.09 Gbits/sec    0    993 KBytes       
[  5]   2.00-3.00   sec   129 MBytes  1.08 Gbits/sec    0    993 KBytes       
[  5]   3.00-4.00   sec   130 MBytes  1.09 Gbits/sec    0    993 KBytes       
[  5]   4.00-5.00   sec   130 MBytes  1.09 Gbits/sec    0    993 KBytes       
[  5]   5.00-6.00   sec   126 MBytes  1.06 Gbits/sec    0    993 KBytes       
[  5]   6.00-7.00   sec   128 MBytes  1.07 Gbits/sec    0    993 KBytes       
[  5]   7.00-8.00   sec   130 MBytes  1.09 Gbits/sec    0    993 KBytes       
[  5]   8.00-9.00   sec   129 MBytes  1.08 Gbits/sec    0    993 KBytes       
[  5]   9.00-10.00  sec   126 MBytes  1.06 Gbits/sec    0    993 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.25 GBytes  1.08 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.25 GBytes  1.07 Gbits/sec                  receiver

iperf Done.

```

If I test a direct ethernet connection, I will get about 950 Mbps so I am seeing wireless that is faster than cable unless I get a new WiFi router with higher speed ethernet ports.

Why am I posting this? The driver for the new mt7925 chipset went into kernel 6.7 and the firmware has now been posted. The mt7925 is a WiFi 7 chipset for USB and PCIe and it supports 160 MHz channel width. We do not currently have any USB adapters with this chipset on the market but I suspect they will come this year.

@morrownr

bjlockie, 2024-01-25:
What router do you have?
I'm using a raspberry pi running openwrt.

morrownr, 2024-01-25:
How is it going @bjlockie ?

> What router do you have?

Cudy WR3000:

https://www.amazon.com/gp/product/B0BRK3CYY3

I needed something to test WiFi. This is a low cost router but it does what I needed it to do. For many years I have been stopping by OpenWRT when it is time to get a new router. This was interesting. It comes with an old version of OpenWRT as the backend and Cudy supplies an intermediate image to flash to convert to full OpenWRT... which I have done.

It uses the mt7981 SoC. Interestingly enough, OpenWRT has worked up a wifi router idea that they are working with BananaPi to build and distribute and it uses the same SoC:

https://forum.openwrt.org/t/openwrt-one-celebrating-20-years-of-openwrt/183684

The Cudy version does not have a usb port but OpenWRT One will. In fact, OpenWRT One will have several additional features but then it will likely cost more.

> I'm using a raspberry pi running openwrt.

My Pi4B runs my own AP guide as shown here with RasPiOS and is plugged into the Cudy via ethernet cable. I use this setup to test many things.

@morrownr