morrownr/USB-WiFi #378: Wifi adapter not showing up (ALFA AWUS036AXML)

mickyb5169, 2024-02-10:
I am getting no Wi-Fi showing up, Bluetooth appears for about 3 seconds then disappears

Verified that not a hardware problem, different USB ports do not change outcome and interestingly works perfectly in Windows with driver installed

Currently I'm running kernel version 6.6.14-200.fc39.x86_64 on fedora workstation, extra information about my computer can be found at https://linux-hardware.org/?probe=cde1b607e4

I feel problem could be understood from here.
I tried this command
Dmesg –follow

I plugged USB wifi in and got:
[  361.323604] usb 3-2: new high-speed USB device number 2 using xhci_hcd
[  361.513245] usb 3-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  361.513252] usb 3-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  361.513256] usb 3-2: Product: Wireless_Device
[  361.513259] usb 3-2: Manufacturer: MediaTek Inc.
[  361.513262] usb 3-2: SerialNumber: 000000000
[  361.534768] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[  364.650568] Bluetooth: hci0: Device setup in 3047631 usecs
[  364.650577] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  366.675730] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[  368.723783] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[  370.771825] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[  370.948979] usb 3-2: reset high-speed USB device number 2 using xhci_hcd
[  376.259992] usb 3-2: device descriptor read/64, error -110
[  392.132366] usb 3-2: device descriptor read/64, error -110
[  392.412344] usb 3-2: reset high-speed USB device number 2 using xhci_hcd
[  397.764487] usb 3-2: device descriptor read/64, error -110
[  413.636848] usb 3-2: device descriptor read/64, error -110
[  413.916837] usb 3-2: reset high-speed USB device number 2 using xhci_hcd
[  418.717748] usb 3-2: Device not responding to setup address.
[  423.725614] usb 3-2: Device not responding to setup address.
[  423.933050] usb 3-2: device not accepting address 2, error -71
[  424.109046] usb 3-2: reset high-speed USB device number 2 using xhci_hcd
[  428.909013] usb 3-2: Device not responding to setup address.
[  433.917870] usb 3-2: Device not responding to setup address.
[  434.125269] usb 3-2: device not accepting address 2, error -71
[  434.125525] mt7921u: probe of 3-2:1.3 failed with error -5
[  434.125794] usb 3-2: USB disconnect, device number 2
[  434.302274] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[  439.749430] usb 3-2: device descriptor read/64, error -110
[  455.621793] usb 3-2: device descriptor read/64, error -110
[  455.901767] usb 3-2: new high-speed USB device number 4 using xhci_hcd
[  461.253922] usb 3-2: device descriptor read/64, error -110
[  477.126286] usb 3-2: device descriptor read/64, error -110
[  477.230462] usb usb3-port2: attempt power cycle
[  477.674268] usb 3-2: new high-speed USB device number 5 using xhci_hcd
[  482.474348] usb 3-2: Device not responding to setup address.
[  487.479210] usb 3-2: Device not responding to setup address.
[  487.686493] usb 3-2: device not accepting address 5, error -71
[  487.862502] usb 3-2: new high-speed USB device number 6 using xhci_hcd
[  492.662717] usb 3-2: Device not responding to setup address.
[  497.670578] usb 3-2: Device not responding to setup address.
[  497.878729] usb 3-2: device not accepting address 6, error -71
[  497.878921] usb usb3-port2: unable to enumerate USB device

Currently running kennel version 
I have checked driver use with this
Lsusb -t
And got:
|__ Port 004: Dev 000, If 0, Class=Wireless, Driver=[none], M
            |__ Port 004: Dev 000, If 1, Class=Wireless, Driver=btusb, M
            |__ Port 004: Dev 000, If 2, Class=Wireless, Driver=[none], M
            |__ Port 004: Dev 000, If 3, Class=Vendor Specific Class, Driver=mt7921u, M

I have downloaded and copied the patch, Wi-Fi ram code and Bluetooth ram code from option 3 from here https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

What could be causing this Wi-Fi not to show in Linux?
Any help greatly appreciated, hope not to pull any more hair out!

morrownr, 2024-02-10:
Hi @mickyb5169 

> I have downloaded and copied the patch, Wi-Fi ram code and Bluetooth ram code from option 3 from here

Let's make sure about the details regarding the firmware. You went to the right place. Section 3 is for your adapter. You mentioned the ram code file but there are two firmware files for USB and I suspect if you have one of one version and the other of another version, we might see what you are seeing. Confirm that you installed the following two files:

WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_RAM_CODE_MT7961_1.bin

Both of those files are required and need to be the same version so reinstall according to the instructions if you are not sure.

Additionally, you can delete `BT_RAM_CODE_MT7961_1_2_hdr.bin` because usb adapters can't support bluetooth if USB3 is to be used so makers turn bluetooth off. I see it is still trying to load so you can delete it as follows:

sudo rm /lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin

Let me know if this helps.

@morrownr

mickyb5169, 2024-02-11:
i deleted BT_RAM_CODE_MT7961_1_2_hdr.bin, I downloaded and copied WIFI_MT7961_patch_mcu_1_2_hdr.bin &
WIFI_RAM_CODE_MT7961_1.bin
to /lib/firmware/mediatek/

i rebooted and still no deal
dmesg --follow
still produce roughly the same problem
[  457.343295] usb 3-2: new high-speed USB device number 7 using xhci_hcd
[  462.789462] usb 3-2: device descriptor read/64, error -110
[  478.661843] usb 3-2: device descriptor read/64, error -110
[  478.941800] usb 3-2: new high-speed USB device number 8 using xhci_hcd
[  484.294970] usb 3-2: device descriptor read/64, error -110
[  500.167327] usb 3-2: device descriptor read/64, error -110
[  500.270524] usb usb3-port2: attempt power cycle
[  500.714299] usb 3-2: new high-speed USB device number 9 using xhci_hcd
[  505.514419] usb 3-2: Device not responding to setup address.
[  510.519278] usb 3-2: Device not responding to setup address.
[  510.726529] usb 3-2: device not accepting address 9, error -71
[  510.903546] usb 3-2: new high-speed USB device number 10 using xhci_hcd
[  515.703849] usb 3-2: Device not responding to setup address.
[  520.710715] usb 3-2: Device not responding to setup address.
[  520.918781] usb 3-2: device not accepting address 10, error -71
[  520.919003] usb usb3-port2: unable to enumerate USB device
[  550.376663] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  550.376688] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  550.376692] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  550.376697] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  587.200567] pcieport 0000:00:03.2: AER: Corrected error received: 0000:01:00.0
[  587.200582] pcieport 0000:01:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[  587.200587] pcieport 0000:01:00.0:   device [1002:1478] error status/mask=00003000/00002000
[  587.200592] pcieport 0000:01:00.0:    [12] Timeout               
[  608.150704] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  608.150730] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  608.150734] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  608.150739] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  620.448207] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  620.448243] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  620.448247] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  620.448252] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  621.956535] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  621.956552] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  621.956554] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  621.956556] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  633.127030] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  633.127061] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  633.127068] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  633.127075] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  670.423997] pcieport 0000:00:03.2: AER: Corrected error received: 0000:00:03.2
[  670.424012] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  670.424030] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  670.424031] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  689.545120] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  689.545156] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  689.545160] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  689.545165] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  690.261463] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  690.261479] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  690.261480] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  690.261483] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  726.960591] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  726.960616] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  726.960620] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  726.960625] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  733.769165] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  733.769182] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[  733.769183] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[  733.769186] pcieport 0000:00:03.2:    [ 6] BadTLP                
[  789.899138] pcieport 0000:00:03.2: AER: Multiple Corrected error received: 0000:00:03.2
[  789.899177] pcieport 0000:00:03.2: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Receiver ID)
[... cut ...]
[ 1109.635708] pcieport 0000:00:03.2:   device [8086:6f0a] error status/mask=00000040/00002000
[ 1109.635715] pcieport 0000:00:03.2:    [ 6] BadTLP

lsusb -t
shows:
 Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 002: Dev 011, If 0, Class=[unknown], Driver=[none], 480M
    |__ Port 002: Dev 011, If 1, Class=Wireless, Driver=[none], 480M
    |__ Port 002: Dev 011, If 2, Class=[unknown], Driver=[none], 480M
    |__ Port 002: Dev 011, If 3, Class=Vendor Specific Class, Driver=mt7921u, 480M

any more ideas?

morrownr, 2024-02-11:
Well, if it were me, I would go into my bios and note all of the settings that have to do with usb and PCIe... and then do a lot of research.

You are showing multiple usb error codes. This makes me think of what could cause mutiple error codes and the first thing that pops into mind is a power problem. Could it be that you have so many usb devices plugged in that you are exceeding the amount of power that the usb subsystem can provide? To test, remove all of the usb devices you can while still being able to start and control the computer and see if the adapter works. If it does, then the problem is likely power related.

However, you also have numerous errors that have to do with PCIe and that should not have anything to do with usb. These errors may be fixed with a bios setting but you would have to research and test.

@morrownr

mickyb5169, 2024-02-12:
i have disabled all energy saving features in bios already and rechecked usb settings, nothing obvious. 

I only have 2 things plugged in, a bluetooth dongle for wireless keyboard/mouse and the alfa awus036axml so i have my doubts that it is power related especially when the wifi adapter functions in windows on the same computer in the same port.

pci express errors have not resurfaced

mickyb5169, 2024-02-14:
it is 100% an issue with Fedora Linux 39 workstation edition, i downloaded and ran a live distribution of ubuntu 23.10.1 and the wifi worked fine straight from the start. I'm debating about wiping fedora from the drive and installing ubuntu as long as i can get wayland windowing system working it should be good. Would you like me to gather anymore details about fedora before i move on?

morrownr, 2024-02-14:
Now we know. I know fedora has been working with adapters with the same chipset that your axml uses but they must have messed something up. It really sounds like an issue somewhere in the wifi stack and not in the driver itself. Distros are very complicate things and require people that know what they are doing and even then, things happen. You might want to file a bug with fedora so that they know there is a problem. They will need to know the driver and the driver is mt7921u.

> I'm debating about wiping fedora from the drive and installing ubuntu as long as i can get wayland windowing system working it should be good.

I use Debian on my dev box and Ubuntu on my laptop. I lean toward Debian or distros based on Debian because they are so stable. Ubuntu uses Debian below the desktop but stays a little closer to the bleeding edge than Debian.

@morrownr

jmnicolas90, 2024-02-15:
I can confirm: I tried it on a laptop and a desktop both running Fedora 39 with current patches and the AWUS036AXML isn't detected.

morrownr, 2024-02-15:
@jmnicolas90 

Something is wrong in Fedora 39. Maybe some detective work with give you something to report to Fedora:

The driver (module) for the mt7921au chipset is called mt7921u.ko. You can check on the driver by going to the following location:

```
cd /usr/lib/modules/$(uname -r)/kernel/drivers/net/wireless/mediatek/mt76/mt7921
```

Show the files:

```
ls -l
```

The file you are looking for is `mt7921u.ko`

If the file is not there... that is a problem and should be reported.

You can also confirm that the 2 wifi firmware files are in place but going to:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

Section 3 is the one you want for the AXML. 

If the firmware files are in place also, then the problem is not with the driver and things really turn into more detective work. You might want to post in the appropriate Fedora forum to see how many others are seeing the same thing. Right now I am only seeing this with Fedora. If it was a general problem with the driver for the mt7921au chipset then I'm sure we would be seeing a lot of posts here from many users.

jmnicolas90, 2024-02-16:
Hi, here is the content from the directory:

`xxx@xxx:/usr/lib/modules/6.7.4-200.fc39.x86_64/kernel/drivers/net/wireless/mediatek/mt76/mt7921$ ll
total 124
-rw-r--r--. 1 root root 52652  5 févr. 01:00 mt7921-common.ko.xz
-rw-r--r--. 1 root root 22936  5 févr. 01:00 mt7921e.ko.xz
-rw-r--r--. 1 root root 24644  5 févr. 01:00 mt7921s.ko.xz
-rw-r--r--. 1 root root 18624  5 févr. 01:00 mt7921u.ko.xz
`
Did they forget to unzip them? :smiley: 
I'm currently looking at filing a bug report but it's quite complicated (I have no idea which component to choose) and I don't enough Linux knowledge to explain it well. I'll reference this thread if I manage to submit the bug.

morrownr, 2024-02-16:
> Did they forget to unzip them?

The Linux kernel supports compressed drivers so that is okay.

> -rw-r--r--. 1 root root 18624 5 févr. 01:00 mt7921u.ko.xz

There is the driver.

> I'm currently looking at filing a bug report but it's quite complicated 

Maybe it is a good idea to go to an appropriate fedora forum, describe the problem and ask for guidance on filing a bug. It can be hard to pick a component when it could be several components that could cause this.

You'll get there. When you find out what the problem/fix it, let us know.

@morrownr

hasamsote, 2024-02-21:
Hi, I have a similar issue running Manjaro on the 6.5 kernel. The adapter is detected if I `modprobe -r btusb`.

morrownr, 2024-02-21:
I'll do some testing. In the mean time, post anything that has to do with adapter when you run:

$ sudo dmesg

hasamsote, 2024-02-27:
> I'll do some testing. In the mean time, post anything that has to do with adapter when you run:
> 
> $ sudo dmesg

Ok, I'm following up on this. I deblacklisted bluetooth and btusb, and also updated my kernel to 6.7 and firmware to the newest version available (linux-firmware 20240115.9b6d0b08-2). The wi-fi adapter actually initialized correctly on boot, but I unplugged it and plugged it in again, and that resulted in a freeze (this happens at 112). I later plugged it back into the previous port at 377 and got a different error this time, however the result seems to have been the same.

```
[    2.562486] usb 2-3: new SuperSpeed USB device number 2 using xhci_hcd
[    2.582612] usb 2-3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[    2.582616] usb 2-3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[    2.582618] usb 2-3: Product: Wireless_Device
[    2.582620] usb 2-3: Manufacturer: MediaTek Inc.
[    2.582621] usb 2-3: SerialNumber: 000000000
[    2.741946] ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20230628/utaddress-204)
[    2.741953] ACPI: OSL: Resource conflict; ACPI support missing from driver?
[    2.741953] ACPI: OSL: Resource conflict: System may be unstable or behave erratically
[    7.331574] Bluetooth: Core ver 2.22
[    7.331600] NET: Registered PF_BLUETOOTH protocol family
[    7.331602] Bluetooth: HCI device and connection manager initialized
[    7.331607] Bluetooth: HCI socket layer initialized
[    7.331609] Bluetooth: L2CAP socket layer initialized
[    7.331614] Bluetooth: SCO socket layer initialized
[    7.367997] usbcore: registered new interface driver btusb
[    7.370615] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[   11.379266] Bluetooth: hci0: Device setup in 3917239 usecs
[   11.379271] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   13.473841] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[   15.404254] fbcon: Taking over console
[   15.412538] Console: switching to colour frame buffer device 240x67
[   15.606681] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[   17.740088] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[   17.867845] usb 2-3: reset SuperSpeed USB device number 2 using xhci_hcd
[   17.898005] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[   18.023344] Bluetooth: hci0: Device setup in 126330 usecs
[   18.023349] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   18.104374] usbcore: registered new interface driver mt7921u
[   18.108355] mt7921u 2-3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
[   18.108847] Bluetooth: hci0: AOSP extensions version v1.00
[   18.108853] Bluetooth: hci0: AOSP quality report is supported
[   18.118250] mt7921u 2-3:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
[   19.740993] mt7921u 2-3:1.3 wlp2s0f0u3i3: renamed from wlan0
[   22.813179] wlp2s0f0u3i3: authenticate with <MAC1> (local address=<MAC2>)
[   22.952700] wlp2s0f0u3i3: send auth to <MAC1> (try 1/3)
[   22.956150] wlp2s0f0u3i3: authenticated
[   22.956678] wlp2s0f0u3i3: associate with <MAC1> (try 1/3)
[   22.971319] wlp2s0f0u3i3: RX AssocResp from <MAC1> (capab=0x1411 status=0 aid=16)
[   23.013497] wlp2s0f0u3i3: associated
[   23.276332] wlp2s0f0u3i3: Limiting TX power to 40 (40 - 0) dBm as advertised by <MAC1>
[   40.011750] usb 3-4: reset high-speed USB device number 2 using xhci_hcd
[  112.865079] usb 2-3: USB disconnect, device number 2
[  112.891795] wlp2s0f0u3i3: deauthenticating from <MAC1> by local choice (Reason: 3=DEAUTH_LEAVING)
[  114.144174] mt7921u 2-3:1.3: timed out waiting for pending tx
[  124.493775] usb 4-3: new SuperSpeed USB device number 3 using xhci_hcd
[  124.511654] usb 4-3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  124.511663] usb 4-3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  124.511667] usb 4-3: Product: Wireless_Device
[  124.511671] usb 4-3: Manufacturer: MediaTek Inc.
[  124.511674] usb 4-3: SerialNumber: 000000000
[  124.518711] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[  127.122149] Bluetooth: hci0: Device setup in 2541602 usecs
[  127.122153] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  129.326812] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[  131.461642] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[  133.596305] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[  138.932594] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[  144.268744] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[  144.475494] usb 4-3: device not accepting address 3, error -62
[  149.818289] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[  155.154316] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[  155.361068] usb 4-3: device not accepting address 3, error -62
[  160.704066] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[  377.735038] usb 2-3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  377.735043] usb 2-3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[... cut ...]
[  399.996025] usb 2-3: Device not responding to setup address.
[  404.165029] usb 2-3: Device not responding to setup address.
[  404.372222] usb 2-3: device not accepting address 3, error -71
[  407.850656] usb 2-3: Device not responding to setup address.
[  408.057812] usb 2-3: Device not responding to setup address.
[  408.264153] usb 2-3: device not accepting address 3, error -71
[  409.018890] mt7921u: probe of 2-3:1.3 failed with error -5
[  409.019139] usb 2-3: USB disconnect, device number 3
```

hasamsote, 2024-02-27:
And here's a successful initialization after I do `sudo modprobe -r btusb`. Though interestingly the initialization process actually causes btusb module to be reinserted. It's very consistent though, if I remove `btusb` before plugging the device in, it worked three times out of three (though it takes a while, longer than usual for a wi-fi adapter), and 0 times out of three if I don't.

```
[ 1612.369987] usb 2-3: new SuperSpeed USB device number 4 using xhci_hcd
[ 1612.391915] usb 2-3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 1612.391923] usb 2-3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 1612.391927] usb 2-3: Product: Wireless_Device
[ 1612.391930] usb 2-3: Manufacturer: MediaTek Inc.
[ 1612.391933] usb 2-3: SerialNumber: 000000000
[ 1612.587112] usb 2-3: reset SuperSpeed USB device number 4 using xhci_hcd
[ 1612.673953] mt7921u 2-3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a

[ 1612.931145] mt7921u 2-3:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
[ 1613.252097] usbcore: registered new interface driver btusb
[ 1613.253821] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[ 1614.549086] mt7921u 2-3:1.3 wlp2s0f0u3i3: renamed from wlan0
[ 1617.345143] Bluetooth: hci0: Device setup in 3996385 usecs
[ 1617.345152] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 1618.560301] Bluetooth: hci0: Mediatek coredump end
[ 1619.393793] Bluetooth: hci0: Mediatek coredump end
[ 1619.530610] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[ 1620.385759] wlp2s0f0u3i3: authenticated
[ 1621.664300] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[ 1623.797945] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[ 1624.562750] usb 2-3: reset SuperSpeed USB device number 4 using xhci_hcd
[ 1624.590588] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[ 1624.725434] mt7921u 2-3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a
[ 1624.735362] mt7921u 2-3:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
[ 1627.498608] Bluetooth: hci1: Device setup in 2841510 usecs
[ 1627.498616] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 1629.559260] Bluetooth: hci1: Opcode 0x0c03 failed: -110
[ 1631.693181] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[ 1633.826481] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[ 1634.207582] usb 2-3: reset SuperSpeed USB device number 4 using xhci_hcd
[ 1634.239026] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[ 1634.297432] mt7921u 2-3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20231109190918a

[ 1634.307869] mt7921u 2-3:1.3: WM Firmware Version: ____010000, Build Time: 20231109190959
[ 1634.352185] Bluetooth: hci1: Device setup in 112662 usecs
[ 1634.352189] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 1634.434191] Bluetooth: hci1: AOSP extensions version v1.00
[ 1634.434195] Bluetooth: hci1: AOSP quality report is supported
```

morrownr, 2024-02-28:
@hasamsote 

I need you to simplify things for me. What is not working with your AXML?

If it is the bluetooth, ALFA turned bluetooth support off in the chipset of this adapter. You cannot use wifi in usb3 mode if bluetooth is active so all sellers of mt7921au chipset based adapters that I am aware have turned the capability off so wifi can run with usb3.

I am aware that the capability is detected and the driver is loaded in some cases... however bluetooth will not actually work. This is a bug somewhere that needs to be fixed.

The problems you are seeing with plugging and unplugging are not familiar. I can do some tests later this week on my AXML to see if I can duplicate if you want.

@morrownr

hasamsote, 2024-02-28:
What doesn't work is the wi-fi component. The adapter isn't detected by NetworkManager and thus can't be used to connect to any network either. However deactivating or otherwise messing with the bluetooth modules seems to fix the wi-fi issue.

hasamsote, 2024-02-28:
Ok, now it's randomly disconnecting when I try to transfer files. I'm definitely returning it.

morrownr, 2024-02-28:
You might burn a copy of say Ubuntu 23.10 that works well from a flash drive and boot/run it to see the results. I've had good results from my AXML but I understand that not everything works well together. What you are describing may not be this adapter/driver causing the problem but it is easy to check that out.

Cheers,

@morrownr

sanderhollaar, 2024-03-17:
> Hi, I have a similar issue running Manjaro on the 6.5 kernel. The adapter is detected if I `modprobe -r btusb`.

Heh, I have two ALFA AWUS036AXML adapters and have searched for weeks as to why these don't work:

    "Bluetooth: hci1: Opcode 0x0c03 failed: -110"
    "xhci_hcd 0000:3b:00.0: Timeout while waiting for setup device command"
    "usb 4-2: device not accepting address 2, error -62"

On any of my NixOS computers even with latest firmware 20240312 as of 5 days ago.

On my Thinkpad P70, if I plug in the adapter and execute `sudo modprobe -r btusb` all `btusb` entries listed in `lsmod | grep btusb` disappear, the laptop bluetooth device listed with `hcitool dev` disappears and an extra `wlan` appears.

Now, if I execute `sudo modprobe -r btusb` before I plug in the adapter, at adapter plug in both the laptop bluetooth device and the AWUS036AXML bluetooth device appear, as well as the AWUS036AXML wlan adapter.

The wlan adapter works, confirmed by bringing the internal wlan down.

The bluetooth adapter also seems to work. I can scan and pair my headset. It fails with `br-connection-profile-unavailable` but that seems unrelated.

Same results on an IdeaPad Duet 3i 10.

sanderhollaar, 2024-03-17:
> Ok, now it's randomly disconnecting when I try to transfer files.
[..]
> It seems to be reliably triggered by connecting somewhere with `sftp` and transferring a file, at least I had it happen twice in a row.

I've transferred a 3.9GB kernel repository with `scp` initiated on a laptop to the one that now uses the AWUS036AXML. And after that an `sftp` initiated on the AWUS036AXML laptop to a local server fetching another kernel repository. No hick ups or disconnects.

Laptop is running kernel `6.8.1` and firmware `20240312` fwiw.

morrownr, 2024-03-17:
Hi @sanderhollaar 

It is good to meet you.

> Heh, I have two ALFA AWUS036AXML adapters and have searched for weeks as to why these don't work

I noticed some problems with my mt7922 based PCIe card and my AXML about 3 weeks ago. If I recall correctly, the problems started after I upgraded the firmware files. At the time I was busy on other projects and just switched to other devices as I have 30+ usb wifi adapters here... it kinda goes with running this site. Most folks don't upgrade firmware files as new files are available so I did not think it was an immediate concern.

> No hick ups or disconnects. Laptop is running kernel 6.8.1 and firmware 20240312 fwiw.

Does this mean that things are working well now?

I'm going to take a look at the current state of affairs and figure out what I can figure out. Let there be no doubt that Mediatek has struggled with the firmware for the mt7921. I can see some learning going on. I think we have had around 10 new drops of firmware files over the last 1.5 years which is unusual but then wifi 6(e) plus maintaining backward compatibility is incredibly complex.

sanderhollaar, 2024-03-17:
> It is good to meet you.

Likewise, thank you for your repository.

> Most folks don't upgrade firmware files as new files are available so I did not think it was an immediate concern.

NixOS is quite fast with `linux-firmware` updates: https://github.com/NixOS/nixpkgs/commits/master/pkgs/os-specific/linux/firmware/linux-firmware

> > No hick ups or disconnects. Laptop is running kernel 6.8.1 and firmware 20240312 fwiw.
> 
> Does this mean that things are working well now?

During my `scp` and `sftp` transfer of two kernel repositories, yes.

I've tested `scp` and `sftp` as @hasamsote reported - https://github.com/morrownr/USB-WiFi/issues/378#issuecomment-1970051361 - reproducible disconnects with `sftp`. These adapters are not in use as they did not work at all until today when I learned about the `modprobe -r btusb` workaround in https://github.com/morrownr/USB-WiFi/issues/378#issuecomment-1957717612

> I'm going to take a look at the current state of affairs and figure out what I can figure out. Let there be no doubt that Mediatek has struggled with the firmware for the mt7921.

You mentioned - https://github.com/morrownr/USB-WiFi/issues/378#issuecomment-1936902937 -

> Additionally, you can delete `BT_RAM_CODE_MT7961_1_2_hdr.bin` because usb adapters can't support bluetooth if USB3 is to be used so makers turn bluetooth off.

Will bluetooth not work at all?

Bluetooth is mentioned at https://www.alfa.com.tw/products/awus036axml

> AWUS036AXML is the WiFi 6/6E (802.11ax) 2x2 Tri-band 2.4 GHz / 5 GHz / 6 GHz and Bluetooth 5.2 high-performance USB adapter.

morrownr, 2024-03-17:
> AWUS036AXML is the WiFi 6/6E (802.11ax) 2x2 Tri-band 2.4 GHz / 5 GHz / 6 GHz and Bluetooth 5.2 high-performance USB adapter.

[USB3_Radio_Frequency_Interference.pdf](https://github.com/morrownr/USB-WiFi/files/14629022/USB3_Radio_Frequency_Interference.pdf)

Read the intel white paper. Unless some engineer has figured out a way around this fundamental flaw in USB3, this should not work. No adapter makers have been releasing BT in adapters that can run in USB3. This dates back to 2012 or so.

I's say that Alfa's engineers are not talking to their webmaster.

bjlockie, 2024-03-18:
> Will bluetooth not work at all?
> 
> Bluetooth is mentioned at https://www.alfa.com.tw/products/awus036axml

Maybe in USB2 mode?
If it does, Mediatek is not outright lying. :-)

sanderhollaar, 2024-03-18:
> Read the intel white paper.

Unless I'm missing something, the paper talks about performance degradation and ways to improve performance. It does not say 'cannot work'.

> No adapter makers have been releasing BT in adapters that can run in USB3.

I find it hard to believe engineers design and produce a device like ALFA AWUS036AXML and advertise it with usb3/btw/wifi if they cannot make it function. It would be much easier to leave out the BT part?

morrownr, 2024-03-18:
> Unless I'm missing something, the paper talks about performance degradation and ways to improve performance. It does not say 'cannot work'.

I think intel was part of the group that designed USB3 so they are wording themselves carefully. I think a more honest way to have written this is "We really screwed up and there are many designs that are simply not going work well." Well, one of those designs that won't work well is USB WiFi adapters. Here is an example:

https://www.realtek.com/en/products/communications-network-ics/item/rtl8822bu

Take note that the rtl8822bu chipset is limited to USB2 as it has BT turned on. The part of the chipset that does wifi is very capable of USB3 but the chipset is limited to USB2 due to BT. The far more popular rtl8812bu supports USB3 as BT is turned off. It sells much better because an AC1200 chipset is really speed limited by USB2. An AXE3000 is really limited by USB2. 

> I find it hard to believe engineers design and produce a device like ALFA AWUS036AXML and advertise it with usb3/btw/wifi if they cannot make it function. It would be much easier to leave out the BT part?

Yes. I think the BT support is turned off and the ad is wrong. I do not know why the mistake was made. I am not aware of any usb wifi adapters using the mt7921 chipset that have BT turned on. The PCIe adapters with the mt7921 chipset can have BT on as the wifi is not going through a usb bus.

wangluyi1982, 2024-03-26:
also confirmed with the 6.8 kernel, the workaround is the modprobe -r btusb first, then plug in the usb wifi adapter.

morrownr, 2024-03-26:
@wangluyi1982 

> also confirmed with the 6.8 kernel, the workaround is the modprobe -r btusb first, then plug in the usb wifi adapter.

Thank you and the others for the confirmation and workaround. I have been busy and just now tested this on Debian 12 with kernel 6.6. I am going to make another issue to point out this workaround and...

I'm going to add this problem to the dedicated mt7921au issue.

https://github.com/morrownr/USB-WiFi/issues/107

It is not yet clear to me why a bluetooth driver would have anything to do with this adapter and it may be that the cause of the problem is a patch to the btusb driver, not the mt7921u driver or firmware. Regardless of what caused this, I'll try to get the issue in front of the Mediatek devs.

@deren

FYI: The issue is that our usb wifi adapters with mt7921au chipsets appeared to stop working starting in early Feb. 24 on some distros. At first it appeared to be a fedora specific problem but now it is obvious that is not the case. My test from today is on Debian 12 with kernel 6.6. 

It is not clear how a bluetooth driver could be causing the problem as the workaround is: (report from post immediately above this one)

> also confirmed with the 6.8 kernel, the workaround is the modprobe -r btusb first, then plug in the usb wifi adapter.

I have an additional adapter with the mt7921au chipset so I will test it to rule out the problem being device specific and report here. This problem seems to be growing as distros are upgraded so appears likely that a patch somewhere is causing this. I'll try to narrow down the cause as I am able.

@morrownr

nvllsvm, 2024-04-12:
Another workaround is configuring the `btusb` module with the option `reset=N`. The module option is described as `Send HCI reset command on initialization` by modinfo, so it seems likely that the bug is in one of the wifi modules. My guess is that it gets interrupted at an initialization step where the module ends up hanging on I/O with the adapter. 

**/etc/modprobe.d/btusb.conf**
```
options btusb reset=N
```

Wifi reliably connects and seems to remain stable. I'm not sure what ramifications this has to other bluetooth devices on the system.

## dmesg with `options btusb reset=Y` (default)
```
[  590.820599] page_pool_release_retry() stalled pool shutdown: id 12, 7 inflight 60 sec
[  592.790691] usb 2-2: new SuperSpeed USB device number 9 using xhci_hcd
[  592.812538] usb 2-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  592.812553] usb 2-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  592.812559] usb 2-2: Product: Wireless_Device
[  592.812564] usb 2-2: Manufacturer: MediaTek Inc.
[  592.812568] usb 2-2: SerialNumber: 000000000
[  592.825364] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[  595.273676] page_pool_release_retry() stalled pool shutdown: id 8, 53 inflight 182 sec
[  595.918711] Bluetooth: hci1: Device setup in 3025309 usecs
[  595.918727] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  598.100566] Bluetooth: hci1: Opcode 0x0c03 failed: -110
[  600.233899] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[  602.367269] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[  607.700635] xhci_hcd 0000:08:00.3: Timeout while waiting for setup device command
[  613.033986] xhci_hcd 0000:08:00.3: Timeout while waiting for setup device command
[  613.240555] usb 2-2: device not accepting address 9, error -62
[  618.580433] xhci_hcd 0000:08:00.3: Timeout while waiting for setup device command
[  623.913954] xhci_hcd 0000:08:00.3: Timeout while waiting for setup device command
[  624.120592] usb 2-2: device not accepting address 9, error -62
[  629.460690] xhci_hcd 0000:08:00.3: Timeout while waiting for setup device command
[  634.793989] xhci_hcd 0000:08:00.3: Timeout while waiting for setup device command
[  635.000578] usb 2-2: device not accepting address 9, error -62
```

## dmesg with `options btusb reset=N`
```
[  674.817354] usb 2-2: new SuperSpeed USB device number 10 using xhci_hcd
[  674.836111] usb 2-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  674.836126] usb 2-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  674.836132] usb 2-2: Product: Wireless_Device
[  674.836137] usb 2-2: Manufacturer: MediaTek Inc.
[  674.836141] usb 2-2: SerialNumber: 000000000
[  674.848766] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[  675.515915] Bluetooth: hci1: urb 0000000024d9281c failed to resubmit (2)
[  678.857558] Bluetooth: hci1: Device setup in 3919595 usecs
[  678.857574] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  680.873894] Bluetooth: hci1: Opcode 0x1003 failed: -110
[  680.873929] Bluetooth: hci1: command 0x1003 tx timeout
[  680.873949] Bluetooth: hci1: failed to reset (-19)
[  680.873960] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[  683.007221] Bluetooth: hci1: command 0x1003 tx timeout
[  683.007221] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[  683.007235] Bluetooth: hci1: failed to reset (-19)
[  683.132610] usb 2-2: reset SuperSpeed USB device number 10 using xhci_hcd
[  683.159811] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[  683.182817] usbcore: registered new interface driver mt7921u
[  683.184688] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a

[  683.283123] Bluetooth: hci1: Device setup in 124964 usecs
[  683.283139] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  683.344318] Bluetooth: hci1: AOSP extensions version v1.00
[  683.344333] Bluetooth: hci1: AOSP quality report is supported
[  683.344855] Bluetooth: MGMT ver 1.22
[  683.441053] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20240219111038
[  686.215540] wlan10: authenticate with xx:xx:xx:xx:xx:xx (local address=xx:xx:xx:xx:xx:xx)
[  686.732919] wlan10: send auth to xx:xx:xx:xx:xx:xx (try 1/3)
[  686.814433] wlan10: authenticate with xx:xx:xx:xx:xx:xx (local address=xx:xx:xx:xx:xx:xx)
[  686.815747] wlan10: send auth to xx:xx:xx:xx:xx:xx (try 1/3)
[  686.846817] wlan10: authenticated
[  686.850262] wlan10: associate with xx:xx:xx:xx:xx:xx (try 1/3)
[  686.856556] wlan10: RX AssocResp from xx:xx:xx:xx:xx:xx (capab=0x1511 status=0 aid=1)
[  686.872477] wlan10: associated
[  687.036636] wlan10: Limiting TX power to 27 (30 - 3) dBm as advertised by xx:xx:xx:xx:xx:xx
```

gbcox, 2024-04-12:
> Another workaround is configuring the `btusb` module with the option `reset=N`. The module option is described as `Send HCI reset command on initialization` by modinfo.

Just did that and rebooted.  Works for me.  I'll keep monitoring and LYK if I notice any problems.  I don't use bluetooth on my laptop, so this isn't an issue for me.

Thanks much for taking the time to post the workaround.  Much easier than having to enter a command at every boot.

gbcox, 2024-04-12:
At least in Fedora, this appears to be fixed in the latest kernel.  I just tested and it is working for me.

* Thu Apr 11 2024 Justin M. Forbes <jforbes@fedoraproject.org> [6.8.5-301]
- nouveau: fix devinit paths to only handle display on GSP. (Dave Airlie)
- Add bluetooth bug to Bugsfixed for 6.8.6 (Justin M. Forbes)
- Bluetooth: l2cap: Don't double set the HCI_CONN_MGMT_CONNECTED bit (Archie Pusaka)

morrownr, 2024-04-12:
@gbcox 

Thanks for updating us. This is good to hear. It is interesting that a bug in something that is not wifi can cause such serious problems with wifi but it happens... rare, but it happens.

All of Linux appreciates the testing and reporting that you users of bleeding edge distros like fedora do.

gbcox, 2024-04-13:
I've attached the dmesg output below.  I found when unplugging the wifi that it doesn't come back.  This only
works if the card is plugged in when you boot the machine.  If you remove the wifi after the boot you have to enter
the command `modprobe -r btusb ` to get it to reconnect.  The fix using the btusb.conf file no longer works with the latest kernel update.

[  334.924549] mt7921u 1-1.1:1.3: timed out waiting for pending tx
[ 2250.767247] usb 1-1.1: new high-speed USB device number 6 using ehci-pci
[ 2250.861504] usb 1-1.1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 2250.861524] usb 1-1.1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 2250.861530] usb 1-1.1: Product: Wireless_Device
[ 2250.861534] usb 1-1.1: Manufacturer: MediaTek Inc.
[ 2250.861538] usb 1-1.1: SerialNumber: 000000000
[ 2250.921257] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[ 2250.961256] Bluetooth: hci1: urb 000000006fe9bf34 failed to resubmit (2)
[ 2255.982314] Bluetooth: hci1: Device setup in 4996671 usecs
[ 2255.982340] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 2258.003279] Bluetooth: hci1: Opcode 0x0c03 failed: -110
[ 2260.051285] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[ 2262.099295] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[ 2262.176410] usb 1-1.1: reset high-speed USB device number 6 using ehci-pci
[ 2267.428311] usb 1-1.1: device descriptor read/64, error -110
[ 2283.301340] usb 1-1.1: device descriptor read/64, error -110
[ 2283.480344] usb 1-1.1: reset high-speed USB device number 6 using ehci-pci
[ 2288.932389] usb 1-1.1: device descriptor read/64, error -110
[ 2292.772415] usbcore: deregistering interface driver btusb
[ 2294.612701] mt7921u: probe of 1-1.1:1.3 failed with error -5
[ 2294.613670] usb 1-1.1: USB disconnect, device number 6
[ 2294.663983] usbcore: registered new interface driver btusb
[ 2294.785350] Bluetooth: hci0: BCM: chip id 86
[ 2294.788172] Bluetooth: hci0: BCM: features 0x2f
[ 2294.806212] Bluetooth: hci0: fedoralinux #1
[ 2294.806237] Bluetooth: hci0: BCM4335C0 (003.001.009) build 0000
[ 2294.809379] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[ 2294.809398] Bluetooth: hci0: BCM: 'brcm/BCM4335C0-0489-e079.hcd'
[ 2294.809402] Bluetooth: hci0: BCM: 'brcm/BCM-0489-e079.hcd'
[ 2294.937698] Bluetooth: MGMT ver 1.22
[ 2306.536409] usb 1-1.1: new high-speed USB device number 7 using ehci-pci
[ 2306.630788] usb 1-1.1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 2306.630809] usb 1-1.1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 2306.630815] usb 1-1.1: Product: Wireless_Device
[ 2306.630820] usb 1-1.1: Manufacturer: MediaTek Inc.
[ 2306.630824] usb 1-1.1: SerialNumber: 000000000
[ 2306.687750] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[ 2306.709307] Bluetooth: hci1: urb 00000000888a185a failed to resubmit (2)
[ 2311.752032] Bluetooth: hci1: Device setup in 4998394 usecs
[ 2311.752057] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 2313.811484] Bluetooth: hci1: Opcode 0x0c03 failed: -110
[ 2315.859544] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[ 2317.907472] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[ 2318.564908] mt7921u: probe of 1-1.1:1.3 failed with error -5
[ 2318.565688] usb 1-1.1: USB disconnect, device number 7
[ 2320.995974] usbcore: deregistering interface driver btusb
[ 2334.040525] usb 1-1.1: new high-speed USB device number 8 using ehci-pci
[ 2334.134087] usb 1-1.1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 2334.134107] usb 1-1.1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 2334.134113] usb 1-1.1: Product: Wireless_Device
[ 2334.134118] usb 1-1.1: Manufacturer: MediaTek Inc.
[ 2334.134122] usb 1-1.1: SerialNumber: 000000000
[ 2334.284769] usb 1-1.1: reset high-speed USB device number 8 using ehci-pci
[ 2334.418969] mt7921u 1-1.1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a
[ 2334.580078] usbcore: registered new interface driver btusb
[ 2334.632292] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[ 2334.688899] Bluetooth: hci0: BCM: chip id 86
[ 2334.691886] Bluetooth: hci0: BCM: features 0x2f
[ 2334.709877] Bluetooth: hci0: fedoralinux
[ 2334.709884] Bluetooth: hci0: BCM4335C0 (003.001.009) build 0000
[ 2334.713204] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[ 2334.713208] Bluetooth: hci0: BCM: 'brcm/BCM4335C0-0489-e079.hcd'
[ 2334.713210] Bluetooth: hci0: BCM: 'brcm/BCM-0489-e079.hcd'
[ 2334.778893] mt7921u 1-1.1:1.3: WM Firmware Version: ____010000, Build Time: 20240219111038
[ 2334.840424] Bluetooth: MGMT ver 1.22
[ 2336.449412] mt7921u 1-1.1:1.3 wlp0s18u1u1i3: renamed from wlan0
[ 2338.905422] Bluetooth: hci1: Device setup in 4223412 usecs
[ 2338.905440] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 2338.988962] Bluetooth: hci1: AOSP extensions version v1.00
[ 2338.988978] Bluetooth: hci1: AOSP quality report is supported
[ 2338.989647] Bluetooth: MGMT ver 1.22
[ 2343.322719] wlp0s18u1u1i3: authenticate with 7e:45:58:f7:a9:d1 (local address=96:7a:3e:6d:07:c5)
[ 2343.856806] wlp0s18u1u1i3: send auth to 7e:45:58:f7:a9:d1 (try 1/3)
[ 2343.925508] wlp0s18u1u1i3: authenticate with 7e:45:58:f7:a9:d1 (local address=96:7a:3e:6d:07:c5)
[ 2343.926315] wlp0s18u1u1i3: send auth to 7e:45:58:f7:a9:d1 (try 1/3)
[ 2343.949615] wlp0s18u1u1i3: authenticated
[ 2343.950498] wlp0s18u1u1i3: associate with 7e:45:58:f7:a9:d1 (try 1/3)
[ 2343.999703] wlp0s18u1u1i3: RX AssocResp from 7e:45:58:f7:a9:d1 (capab=0x1111 status=0 aid=6)
[ 2344.014106] wlp0s18u1u1i3: associated
[ 2344.271499] wlp0s18u1u1i3: Limiting TX power to 30 (30 - 0) dBm as advertised by 7e:45:58:f7:a9:d1

gbcox, 2024-04-13:
> @gbcox
>
> Thanks for updating us. This is good to hear. It is interesting that a bug in something that is not wifi can cause such serious problems with wifi but it happens... rare, but it happens. [...]

Unfortunately, doesn't look like it is completely fixed.  If you unplug the wifi, then plug it back in, it doesn't come back without the `modprobe -r btusb`.

morrownr, 2024-04-13:
@gbcox 

Thanks for the update. I read your post on RedHats bugzilla. It might be worthwhile to post at the Linux bugzilla as well. Something that could improve the report is that it helps get the correct developers attention if you put the driver name in the title of the report. Developers may not have a clue if a Alfa AXXX adapter has their chipset or another brand and they are busy so they will not take the time to did into each bug to see if it applies to them. Reports have to jump out and get the attention of the correct developer. The driver is mt7921u so I recommend adding `driver = mt7921u` or something close to that to the title.

https://bugzilla.kernel.org/

gbcox, 2024-04-13:
> @gbcox
>
> Thanks for the update. I read your post on RedHats bugzilla. It might be worthwhile to post at the Linux bugzilla as well. Something that could improve the report is that it helps get the [...]

Thanks for the tip.  I'll do that.

morrownr, 2024-04-13:
@gbcox 

Well, after thinking about it, maybe you should also add `driver=btusb` in addition to `driver=mt7921u` since we have some confusion going on.

gbcox, 2024-04-13:
> @gbcox
> 
> Well, after thinking about it, maybe you should also add `driver=btusb` in addition to `driver=mt7921u` since we have some confusion going on.

ok, I'll do that also.  I'm also trying to install older kernels to see if I can determine when it broke.

morrownr, 2024-04-13:
> I'm also trying to install older kernels to see if I can determine when it broke.

Good idea. It would be great to figure out which patch broke things but that could be a challenge as some patches get backported to older kernels so the problem could be in older supported kernels going way back now. I suspect we will see more people having this problem when Ubuntu 24.04 is released in a couple of weeks with kernel 6.8.

arbv, 2024-04-13:
@nvllsvm Thank you for your fix. Works for me. I am not sure if the Bluetooth part of the device is functional, but for my case it does not matter.

petershh, 2024-05-08:
Hello,

I'm using COMFAST CF-953AX adapter and I have the same issue. I can work it around by `modprobe -r btusb` or setting `options btusb reset=N`. Thank you for the workaround!

Moreover, this bug causes delays in boot process when udev does coldplug. I've observed that in multiple distros with different init systems.

Also, the adapter manages to make UEFI on my motherboard upset because POST gets slower too. And Windows sometimes fails to detect the adapter, though replugging it makes it work.

I can provide my kernel logs if needed.

nvllsvm, 2024-05-15:
After updating my Arch install to Linux 6.9 (linux-zen `6.9.zen1-1`), wifi works without any workarounds. Bluetooth does not.

I'm guessing this is due to the `BT_RAM_CODE_MT7961_1a_2_hdr.bin` firmware not existing in the `20240510.b9d2bf23-1` firmware package.

```
[  309.750767] usb 2-1.4: new SuperSpeed USB device number 7 using xhci_hcd
[  309.764236] usb 2-1.4: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  309.764242] usb 2-1.4: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  309.764244] usb 2-1.4: Product: Wireless_Device
[  309.764246] usb 2-1.4: Manufacturer: MediaTek Inc.
[  309.764247] usb 2-1.4: SerialNumber: 000000000
[  309.835756] bluetooth hci1: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1a_2_hdr.bin failed with error -2
[  309.835762] Bluetooth: hci1: Failed to load firmware file (-2)
[  309.835765] Bluetooth: hci1: Failed to set up firmware (-2)
[  309.835767] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  309.914641] usb 2-1.4: reset SuperSpeed USB device number 7 using xhci_hcd
[  309.931549] bluetooth hci1: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1a_2_hdr.bin failed with error -2
[  309.931555] Bluetooth: hci1: Failed to load firmware file (-2)
[  309.931557] Bluetooth: hci1: Failed to set up firmware (-2)
[  309.931559] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  309.956150] mt7921u 2-1.4:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a

[  310.215513] mt7921u 2-1.4:1.3: WM Firmware Version: ____010000, Build Time: 20240219111038
```

morrownr, 2024-05-24:
> After updating my Arch install to Linux 6.9 (linux-zen 6.9.zen1-1), wifi works without any workarounds. Bluetooth does not.

Ummm... the Alfa AXML does not have bluetooth support. It is a wifi only adapter. WiFi adapters that have bluetooth support are limited to USB2 due to interference problems.

nvllsvm, 2024-05-24:
> > After updating my Arch install to Linux 6.9 (linux-zen 6.9.zen1-1), wifi works without any workarounds. Bluetooth does not.
> 
> Ummm... the Alfa AXML does not have bluetooth support. It is a wifi only adapter. WiFi adapters that have bluetooth support are limited to USB2 due to interference problems.

The official product page lists the AWUS036AXML as supporting both WiFi and Bluetooth 5.2
Lhttps://www.alfa.com.tw/products/awus036axml?variant=39754360684616

morrownr, 2024-05-24:
> The official product page lists the AWUS036AXML as supporting both WiFi and Bluetooth 5.2

The official product page is wrong. Go ahead and contact Alfa with a question on this issue.

I have never seen a USB3 WiFi adapter with bluetooth active. Around 2012 it was discovered that USB3 can and will interfere with bluetooth and the rest is history. Somebody forgot to tell the person at Alfa that does the web pages.

nvllsvm, 2024-05-24:
> The official product page is wrong. Go ahead and contact Alfa with a question on this issue.
>
> I have never seen a USB3 WiFi adapter with bluetooth active. Around 2012 it was discovered that USB3 can [...]

I confirmed the device has Bluetooth after passing it through to a Windows 11 VM and installing the drivers from https://store.rokland.com/products/alfa-awus036axml-802-11ax-wifi-6-1800-mbps-tri-band-wifi-usb-adapter-w-bluetooth.

I paired my phone and was able to simultaneously maintain that connection while connected to my phone and either a 2.4ghz or 5ghz wifi network. 
[image]
[image]
[image]

morrownr, 2024-05-24:
> I confirmed the device has Bluetooth after passing it through to a Windows 11 VM...

Did you check to see if you were in USB3 mode?
Were you able to do anything with bluetooth?

If so, an engineer has come up with something.

How long ago did you buy your adapter?

nvllsvm, 2024-05-25:
> > I confirmed the device has Bluetooth after passing it through to a Windows 11 VM...
> 
> Did you check to see if you were in USB3 mode? Were you able to do anything with bluetooth?

Yup - it works flawlessly in USB 3.0 mode. I simultaneously played audio over Bluetooth while running iperf3. No hiccups or stuttering in the audio.

2.4ghz wifi
[image]

5ghz wifi
[image]

 
> How long ago did you buy your adapter?

In April 2024.

morrownr, 2024-05-25:
@nvllsvm 

> Yup - it works flawlessly in USB 3.0 mode. I simultaneously played audio over Bluetooth while running iperf3. No hiccups or stuttering in the audio.

I withdraw my previous statements. It appears we have a really good engineer out there at Alfa. As far as I know, this would be the first USB3 WiFi adapter that has successfully integrated bluetooth when wifi is running at USB3. I have a mt7922 PCIe card and it is really good but bluetooth and wifi on the same device is no big deal for pcie devices.

I tried to read back up this thread that is way too long at this point... I got lost. Does bluetooth work on your adapter with Linux?

abdoun20, 2024-05-26:
i think its a BIOS settings issue , i have the same problem with the ALFA network card when i plug it to my pc the LED stays turned on, it doesnt work on my laptop[dell precision M4800] , it keeps connecting and disconnecting on windows and shows your same error  : [
[ 461.253922] usb 3-2: device descriptor read/64, error -110
[ 477.126286] usb 3-2: device descriptor read/64, error -110
] during the boot process of ubuntu. 
 but when i plug it on my brother's pc [dell latitude E6430] with older BIOS version it works just fine and the LED keeps blinking as normal .

morrownr, 2024-05-26:
> when i plug it to my pc the LED stays turned on...

As far as I can tell, that is currently normal behavior on Linux for usb wifi adapters. Basically the LED is serving as a power on/off indicator. This is the case with my Alfa axml and my Edup adapter that has the same chip. 

> i think its a BIOS settings issue 

It could be or it could be because of the latest iteration of usb. My main dev box is 11 years old but it is an i7 chip so i have plenty of power. I do have a mdern laptop so I'll do a test to see what I can find out.

What kernels are you using and have you upgraded the firmware to the latest?

bouk, 2024-10-03:
As an extra data point, we also started seeing this after upgrading `linux-firmware`. We were already on a recent kernel version so I don't think it's an issue in there.

We went from `linux-firmware` version 20231030 to 20240909

wellitsmez, 2024-11-10:
hello @morrownr please can you confirm with me the fccid for alfa AXML mine is 2AB8788321 i think it's fake one

morrownr, 2024-11-10:
Hi @wellitsmez 

I just checked mine and it has the same fccid as you posted above. Mine came from a well known US Alfa distributor.

What makes you think it is fake?

wellitsmez, 2024-11-10:
@morrownr the performance is really bad , when in monitor mode it doesn't show the ssid just like length of ssid <10> or <0> the driver is getting crashed on VIF mode after 20 min

wellitsmez, 2024-11-10:
@morrownr also if the lsusb showes Bus 003 Device 003: ID  MediaTek Inc. Wireless_Device it might not be fake , how can i verify that i have a legit version please

Stekke, 2024-11-15:
> Now, if I execute `sudo modprobe -r btusb` before I plug in the adapter, at adapter plug in both the laptop bluetooth device and the AWUS036AXML bluetooth device appear, as well as the AWUS036AXML wlan adapter.

Thanks so much for mentioning this. I'm on NixOS. With I have the adapter inserted during boot, and login to KDE, everything hangs.
I suspect it's KDE trying to do something with bluetooth.
However when I login while the adapter is unplugged, run the modprobe -r btusb to remove my old bluetooth dongle, and the plug the adapter back in, then WiFi works (and bluetooth through a different older usb2 device as well).

It looks like there is some fairly recent discussion about the bluetooth functionality of this chipset on the linux kernel mailing list: https://lore.kernel.org/all/?q=MT7921AUN

morrownr, 2024-11-15:
@wellitsmez 

> it's weird that it has tx-power = 3dBm also why the power mangment mode is on doesn't make any sense to me

As best I can tell this falls into the category of unsupported feature. Nearly all of Mediatek's drivers show a fixed 3 dBm but the adapters are actually running at the correct txpwr given the country code and band. It does seem strange that Mediatek's drivers are overall the most feature rich and stable but this little issue is there ... users report it all the time. I have tried unsuccessfully to make sure the Mediatek devs are aware of the concern but have so far been unsuccessful.

FjakaLife, 2025-01-10:
Has anyone made any progress on this issue?

morrownr, 2025-01-10:
@FjakaLife 

I think a strong push to solve this could happen this year.

[7 more replies]