morrownr/USB-WiFi #380: Fenvi AX-1800 with different chipset

flipreverse, 2024-02-15:
Hi!

I've just bought a [Fenvi AX-1800](https://de.aliexpress.com/item/1005005061605028.html) stick on AliExpress. However, it does not present itself with a mt7921u chipset.
It announces itself as `0bda:b832 Realtek Semiconductor Corp. 802.11ac WLAN Adapter`. I have to use the 8852bu driver.
FYI, the Linux kernel now supports the new USB id `0e8d:7961` for mt7921u by default.

Cheers,
Alex

morrownr, 2024-02-15:
@flipreverse 

Alex,

Thanks for the message. I need to clear this up with additional information in the Plug and Play List as soon as possible and I would appreciate your help.

The link you posted above goes to a page not found so if you could repost the link to the adapter that you bought, I would appreciate it.

I need to see if the model number on the adapter you bought is different that the adapter link in the Plug and Play list.

The link I use in the Plug and Play List is as follows and it has worked well:

https://www.aliexpress.us/item/3256805749323751.html?gatewayAdapt=glo2usa4itemAdapt

Description: Chip: MT7921

Model Number: WIFI6 FU-AX1800

> I have to use the 8852bu driver.

Are using the 8852bu driver that I have up here? I am going to recommend that you return that adapter as my work to the get the 8852bu out-of-kernel driver going was a real battle that took far too much of my time. This driver and the other Realtek WiFi 5 out-of-kernel drivers are not good drivers. Don't bet on having much success with AP mode or monitor mode. You might even think about posting a review for the product you bought that warns Linux users to avoid it.

To repeat: Anything you can do to help me clarify this issue will be greatly appreciated. As you can see, buying usb wifi adapters for Linux is like walking in a mine field. I think the most important part of this site is the Plug and Play List that has specific links and details so I am very aggressive in making sure the information there is really good,

@morrownr

flipreverse, 2024-02-17:
> The link you posted above goes to a page not found so if you could repost the link to the adapter that you bought, I would appreciate it.

That's interesting, because that link works for me. I'll give it another try:
https://de.aliexpress.com/item/1005005061605028.html?gatewayAdapt=glo2usa4itemAdapt
> 
> Description: Chip: MT7921

Mine shows: `Chipset：Realtek RTL8832`
> 
> Model Number: WIFI6 FU-AX1800
> 
> > I have to use the 8852bu driver.
> 
> Are using the 8852bu driver that I have up here? I am going to recommend that you return that adapter as my work to the get the 8852bu out-of-kernel driver going was a real battle that took far too much of my time. This driver and the other Realtek WiFi 6 out-of-kernel drivers are not good drivers. Don't bet on having much success with AP mode or monitor mode. You might even think about posting a review for the product you bought that warns Linux users to avoid it.

Yeah, I'm using that one. And yes, the AP support is not the best. `FT-PSK` doesn't work, and after some time my devices have issues connecting. Moreover, there are some error messages in the kernel log during init. Finally, upon unloading the module the kernel complains about runtime pm counter underflow.

> To repeat: Anything you can do to help me clarify this issue will be greatly appreciated. As you can see, buying usb wifi adapters for Linux is like walking in a mine field. I think the most [...]

Yeah, why not? Shall I write some sentences about that adapter?
> 
> @morrownr

morrownr, 2024-02-18:
> That's interesting, because that link works for me. I'll give it another try:

Still did not work. It goes to the site and throws some adapters up but not anything released to our discussion. This is likely caused by me being in a different country than you.

> Yeah, why not? Shall I write some sentences about that adapter?

Find the model number for the rtl8852bu based adapter and let me know what it is. I have already made some changed in the Plug and Play list warning users to be careful to check ads to make sure they say Chipset: mt7921. 

Thanks.

@morrownr