morrownr/USB-WiFi #382: Question about the simultaneous client/AP capability of the AWUS036ACM

jfh92, 2024-02-17:
Hi,
I have a Rpi 4B running Buster and a AWUS036ACM USB Wifi dongle connected to it on a USB3 port. I am looking at configuring it as an access point and a client using a virtual interface.
When I type:

`iw list | grep "valid interface combination" -A 8,`
I get:
` #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2, total <= 2, #channels <= 1, STA/AP BI must match`
Interpreting this result after reading about it, I thought I could have the managed and AP features at the same time although not sure. I am looking for advice on this topic! 
Thinking it could be a valid configuration, I tried to implement this configuration following the information at the following address: [here](https://raspberrypi.stackexchange.com/questions/89803/access-point-as-wifi-router-repeater-optional-with-bridge?noredirect=1&lq=1) With or without systemd-networkd, I can't get it to work. I can either have the AP or the client but not both.

I could check that if I remove the USB dongle wifi, the combination AP with station is working using the Rpi 4B wifi hardware. On the other hand, if I disable the Rpi4 wifi setting `dtoverlay=disable-wifi` in `/boot/config.txt`, I have the issue back where the usb wifi dongle does not work with both AP and client active. The command `journalctl -xe` indicates that the AP interface could not be set up with the explanation: `Name not unique on network`. I investigated more with the default raspbian network management. Depending on the boot, sometimes the access point is up and other times the client is up but never both at the same time. In one instance, after a boot, the access point was active and the client was not. I managed to allocate a different MAC address to the client. I did not get the `Name not unique` issue. Instead, the client interface started to scan for Wifi networks and it looked good. It found a network but then it failed with the following messages:
```
wlxxxx: nl80211: MLME command failed (auth): count=0, ret=-22 (Invalid argeument)
wlxxxx: SME: Authentication request to the driver failed
```
Out of curisoity I activated the deterministic network interface naming which did not change the behavior of the networking interfaces: either access point or client up but not both at the same time.

I am looking for some guidance.
Thank you!

bjlockie, 2024-02-17:
> either access point or client up but not both at the same time.

I don't think any chipset lets you have both at the same time.
Use the builtin broadcom as the client and the AWUS036ACM as the AP or wire the client.
Wired is probably better than having two wireless that might interfere with each other.

jfh92, 2024-02-17:
Hi,
It is my undesrtanding that the wifi chipset of the Rpi 4b can be configured to operate both as an access point and as a client. There are a few tutorials one of which I indicated in my first post. I am looking at trying to implement the same feature with a wifi usb dongle that would not have the same limitations as the Rpi 4b wifi.
Thank you!

morrownr, 2024-02-18:
@jfh92 

>  #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2, total <= 2, #channels <= 1, STA/AP BI must match

This line does indeed tell you what the support for VIF (Virtual Interface) is for a specific adapter.

Here is an example from my mt7922 based PCIe card:

valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 2

In this case, you can have 2 interfaces in managed mode and they can be on different channels. You could have one interface in managed mode and one in P2P-client mode on different channels. You could also have one interface in managed mode and one in AP mode and both be on different channels. Most usb adapters that support VIF only support one channel so one interface in managed mode and the other in AP mode would both use the same channel. That works but will cut your overall speed but that may not matter depending on what you are doing.

The follow is a link to a FAQ on the Alfa web site:

https://docs.alfa.com.tw/Support/Kali_Linux/FAQ/

I think they need to update the information to include the AXML and AXM adapters also as they support VIF.

> Interpreting this result after reading about it, I thought I could have the managed and AP features at the same time although not sure.

You are correct. The ACM can indeed support VIF and run managed and AP mode interfaces at the same time. However, and this is true to all wireless, setting up the networking is where the difficulties come in.

The link you provided to is to advice that is very old. I would disregard it.

There are a lot of ways to do various types of networking and I am not fully convinced that I understand exactly what you are trying to do. Can you take your time and write out in details how you want the flow of packets to work in your setup? Also, spell out exactly what you are trying to accomplish so that we might offer alternatives if you are trying to do something that is very hard to do.

@morrownr

jfh92, 2024-02-18:
Hi,
Thank you for the information.
The first goal is to learn about the capabilities of the Rpi. The project to experiment with the Rpi is to have a sort of internet of things which could manage some sensors and also provide wireless router capabilities. It worked with the outdated instructions I gave a link to but the wireless networking was very slow using the Rpi wifi hardware. I was trying to replace the Rpi wifi hardware with the one from a supposedly good USB wifi dongle. I selected, based on the data I found on the internet and on your github in particular, the AWUS036ACM as it has USB3, good wifi and good support on the Rpi. 
I tried different configurations to set the wireless router in addition to the internal Rpi wifi hw acting as both AP and client. One being two ACM connected to the Rpi 4b since I was not able up to now to set up one ACM in AP/client mode. One dongle as an AP and the other as a client. It works with systemd-networkd using  outdated information for Buster but the throughput is still disappointing considering the hardware used. It is also experimental because the setup is bulky and expensive.
I read in different articles, including one of yours, that a wireless router on Rpi with Raspian OS is probably not a good setup. I thought I would investigate this a bit further with OpenWrt. That is yet to come.
In the end, which might be the right approach, as suggested by many, I should try to get a dedicated wireless router and manage the sensors with the Rpi.
Best regards
JF

morrownr, 2024-02-18:
> The first goal is to learn about the capabilities of the Rpi.

It has some decent capabilities and can be fun to play with. It also has weaknesses, many of which are a result of using Broadcom components. Broadcom has never supported Linux very well at all so using their components when it was known that Linux would the operating system is questionable.

The onboard wifi is an example of poor Broadcom Linux support. The first thing I do when I burn a new sd is turn off the onboard wifi. Another problem, and it is a big problem unless you understand it and work to mediate it, is the USB subsystem. The problems here are the result of RasPi engineering decisions. By standard, a USB subsystem with 2 USB3 ports and 2 USB2 ports, should be capable of providing 2800 mA of power. However, the RasPi4B can only provide up to 1200 mA of power so it is common for users to see stability problems. Many assume a powered hub will help but RasPi's are notorious for not working well, or at all, with powered hubs. The USB3 hub chipset is shaky. Well, buggy may be a better way to put it. Adapters such as the ACM need to have Scatter-Gather support turned off for stable operation. Also, adapters with mt7921au chipset should have Scatter-Gather turned off. It is easy to turn off. At least we can get stable opertion with the Mediatek based adapters. I have never been able to get rtl8812bu based adapters to run stable in USB3. In the driver repo here, I simply recommend to stick with USB2. Another thing you need to know is not to use 2  USB adapters with the same chipset in the same computer, whether it be a RasPi or anything else. My main dev box here often has up to 4 usb wifi adapters plugged in at any given time. Many adapters will work well in the same machine... as long as all adapters use a different chipset. My RasPi4B is usually used as a wifi AP with an Afla ACM and Alfa ACHM. The ACM uses a mt7612u chipset and the ACHM uses a mt7610u chipset and they work very well together to provide a dual band AP. They also use less power than many other adapters and that is good given what we know about the lack of power that is available. I also usually have a webcam installed but it uses little power and those are the only 3 things I have plugged into the Pi.  

> ...the throughput is still disappointing considering the hardware used.

What throughput did you see and what were you expecting?

With my ACM connected to my RasPi4B in AP mode, I normally see throughput of 400 Mbps or more. as measured by iperf3.

> I read in different articles, including one of yours, that a wireless router on Rpi with Raspian OS is probably not a good setup.

My opinion is that using OpenWRT for a routed setup is usually going to work better. 

> I thought I would investigate this a bit further with OpenWrt. That is yet to come.

OpenWRT supports the ACM. You will need to install the driver that OpenWRT provides.

> In the end, which might be the right approach, as suggested by many, I should try to get a dedicated wireless router and manage the sensors with the Rpi.

This could be the best solution. I can't say because I still don't really understand exactly what you are trying to do.

@morrownr

jfh92, 2024-02-18:
Hi,
Thank you for detailing the answer! Very interesting. One aspect I was not aware of is to have different usb wifi chipsets rather than identical ones. I will try that.
In terms of trying to describe what I want to do, I could summarize it as a travel router offering weather station capabilities. In terms of router I have been focusing on a full wireless router. 
I am reluctant to go to OpenWRT because the weather station aspect of the project uses drivers that were developed for Raspian OS and I am not sure they would operate on OpenWRT.
The test setup I have is to set up the Rpi as a wireless router. The Rpi is connected to a wifi hotspot as a client. I am connected to the Rpi with a phone via wifi and there the Rpi is an access point. I can access the internet with my phone through the Rpi. With the two ACMs, I would have at best 25Mbps using speedtest. Quite poor performance that I can't really explain. I will disable the scather gather and see if it improves the throughput. However, the additional performance issue I have is that this throughput varies over time quite significantly and can go down to 7Mbps for instance. I can't explain why the throughput varies so much. My Rpi in these tests is not busy at all. I appreciate that the speedtest approach can vary depending on the internet quality.
Thank you again!
JF