morrownr/USB-WiFi #383: Slow TX and absymal RX speeds with mt7610u device

danielshub, 2024-02-20:
I have a Panda PAU0a AC600 USB WiFi adapter (mt7610u chipset) installed in a Intel NUC13 i5 that is running Arch Linux. I configured hostapd based on your guide.

```
 # cat /etc/hostapd/hostapd.conf
ssid=NUC13AP
bridge=home0
interface=wlan0
wpa_passphrase=HelloWorld

country_code=US
hw_mode=a
vht_oper_chwidth=1

channel=36
vht_oper_centr_freq_seg0_idx=42

driver=nl80211

ieee80211d=1
ieee80211h=1
ieee80211ac=1
ieee80211n=1
wmm_enabled=1

auth_algs=1
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

# Panda Pau0a
ht_capab=[HT40-][HT40+][GF][SHORT-GI-20][SHORT-GI-40][RX-STBC1][DSSS_CCK-40]
vht_capab=[SHORT-GI-80][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

```
I can start the AP on the NUC 13 without issue

```
# iw wlan0 info
Interface wlan0
	ifindex 44
	wdev 0x400000001
	addr 9c:ef:d5:f9:40:2e
	ssid NUC13AP
	type AP
	wiphy 4
	channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
	txpower 16.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	133	0	0	0	0	14984		158

```

I can also connect from a laptop

```
$ iw wlp0s20f3 info
Interface wlp0s20f3
	ifindex 2
	wdev 0x1
	addr f0:de:f1:80:47:c0
	ssid NUC13AP
	type managed
	wiphy 0
	channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
	txpower 22.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0

```

The problem is my TX and RX speeds, especially the RX speed, are much slower than I would expect. I measured the speeds by running a iperf3 server on the NUC13 and a iperf3 client on the laptop and placing the laptop and NUC13 about 1 foot apart with no obstructions between them

```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  51.6 MBytes  43.3 Mbits/sec    5             sender
[  5][TX-C]   0.00-10.01  sec  50.5 MBytes  42.3 Mbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec  3.25 MBytes  2.73 Mbits/sec   32             sender
[  7][RX-C]   0.00-10.01  sec  2.38 MBytes  1.99 Mbits/sec                  receiver
```

If I connect the laptop to the NUC13 via a wired 1000 Mbps connection, I get 900+ Mbps transfer rates. If I connect the laptop to a Unifi 802.11ac AP (channel 40, 80 MHz bandwidth) I can get 200+ Mbps.

Any ideas about what I may have screwed up?

morrownr, 2024-02-20:
Hi @danielshub 

> Any ideas about what I may have screwed up?

I'd say you have screwed up exactly nothing. If you had asked me what I would expect from an AP setup with an AC600 adapter on the 5 GHZ band using an 80 MHz channel width, I would have said 210-215 Mbps if you have almost no congestion on the channel you are using. I would have said about 950 Mbps for the 1 GHz ethernet cable.

> If I connect the laptop to a Unifi 802.11ac AP (channel 40, 80 MHz bandwidth) I can get 200+ Mbps.

That is almost perfect. But you do need to be careful about having wireless devices too close as it can cause things that will degrade the speed. AC600 can have links speeds up to 433 Mbps and since it is a half duplex technology, cut that in half and you can expect 216.5 Mbps per second in ideal conditions. Double that speed for an AC1200 adapter.

Currently I have an AC600 usb wifi adapter and an AXE5400 PCIe wifi card in the box I am using:

AC600

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.103 port 57460 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  24.1 MBytes   202 Mbits/sec    0    556 KBytes       
[  5]   1.00-2.00   sec  22.3 MBytes   187 Mbits/sec    0    721 KBytes       
[  5]   2.00-3.00   sec  21.2 MBytes   178 Mbits/sec    0    805 KBytes       
[  5]   3.00-4.00   sec  22.5 MBytes   189 Mbits/sec    0    843 KBytes       
[  5]   4.00-5.00   sec  22.5 MBytes   189 Mbits/sec    0    884 KBytes       
[  5]   5.00-6.00   sec  21.2 MBytes   178 Mbits/sec    0    884 KBytes       
[  5]   6.00-7.00   sec  21.2 MBytes   178 Mbits/sec    0    884 KBytes       
[  5]   7.00-8.00   sec  22.5 MBytes   189 Mbits/sec    0    884 KBytes       
[  5]   8.00-9.00   sec  23.8 MBytes   199 Mbits/sec    0    929 KBytes       
[  5]   9.00-10.00  sec  21.2 MBytes   178 Mbits/sec    0    929 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   223 MBytes   187 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   219 MBytes   184 Mbits/sec                  receiver

iperf Done.

```

AXE5400 (Wifi 6, 160 MHz channel width on a clean DFS channel, mt7922 chipset)

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 49124 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  78.6 MBytes   660 Mbits/sec    0    988 KBytes       
[  5]   1.00-2.00   sec   104 MBytes   870 Mbits/sec    0    988 KBytes       
[  5]   2.00-3.00   sec   106 MBytes   891 Mbits/sec    0    988 KBytes       
[  5]   3.00-4.00   sec   102 MBytes   860 Mbits/sec    0    988 KBytes       
[  5]   4.00-5.00   sec   104 MBytes   870 Mbits/sec    0    988 KBytes       
[  5]   5.00-6.00   sec   105 MBytes   881 Mbits/sec    0    988 KBytes       
[  5]   6.00-7.00   sec   104 MBytes   870 Mbits/sec    0    988 KBytes       
[  5]   7.00-8.00   sec   104 MBytes   870 Mbits/sec    0    988 KBytes       
[  5]   8.00-9.00   sec   105 MBytes   881 Mbits/sec    0    988 KBytes       
[  5]   9.00-10.00  sec   104 MBytes   870 Mbits/sec    0    988 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1016 MBytes   852 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  1014 MBytes   850 Mbits/sec                  receiver

iperf Done.

```

200 Mbps is not slow. I'll bet that the average internet service in the world is likely slower than 200 Mbps. Mediatek merged their new mt7925 wifi driver recently in kernel 6.7. It is WiFi 7, 160 MHz channel width capable. The driver supports usb adapters and pcie cards. We should start seeing new adapters sometime this year. These adapters should smoke but you will need an AP/router that supports WiFi 6/7 and 160 MHz channel width if you want on the fast train. It really depends on what you need for what you are doing.

Hope this helps,

@morrownr

danielshub, 2024-02-20:
@morrownr you missed the part where my mt7610u AP is giving me TX speeds of only 40 Mbps and RX speeds of 2 Mbps. The other numbers were my attempts to make sure I didn't miss something obvious. Thanks for the reassurance that part of what I am doing and seeing makes sense.

morrownr, 2024-02-20:
@danielshub 

> you missed the part where my mt7610u AP is giving me TX speeds of only 40 Mbps and RX speeds of 2 Mbps.

You are correct. I managed to skip this and misread what you were saying. Sorry about that.

```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  51.6 MBytes  43.3 Mbits/sec    5             sender
[  5][TX-C]   0.00-10.01  sec  50.5 MBytes  42.3 Mbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec  3.25 MBytes  2.73 Mbits/sec   32             sender
[  7][RX-C]   0.00-10.01  sec  2.38 MBytes  1.99 Mbits/sec                  receiver
```

One foot is too close. What do you see when you back off to 5-10 feet?

How is the congestion on channel 36?

morrownr, 2024-02-21:
How about you try the following to see what happens:

channel=149
vht_oper_centr_freq_seg0_idx=155

> Any idea why the TX and RX speeds are so different when using iperf3 --bidir

Not yet. Let me study your hostapd.conf file.

danielshub, 2024-02-21:
When I set
```
channel=149
vht_oper_centr_freq_seg0_idx=155
``
I get the expected channel and even better performance, but still asymmetrical TX/RX
```
Interface wlp0s20f3
	ifindex 2
	wdev 0x1
	addr f0:de:f1:80:47:c0
	ssid NUC13AP
	type managed
	wiphy 0
	channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
	txpower 22.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0

```
```

- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec   235 MBytes   197 Mbits/sec    0             sender
[  5][TX-C]   0.00-10.02  sec   232 MBytes   195 Mbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec  53.4 MBytes  44.8 Mbits/sec   27             sender
[  7][RX-C]   0.00-10.02  sec  50.1 MBytes  41.9 Mbits/sec                  receiver
```
Thanks for the help and advice.

morrownr, 2024-02-21:
Okay, making progress. It appears channel 149 has less congestion in your location. Now we know.

What is wrong with rx? It is not clear to me yet. Have you tried using a different usb port?