morrownr/USB-WiFi #384: ALFA AWUS036AXML mt7921au WiFi6e access point

andrew-kovalenk0, 2024-02-22:
Bassicly I need to run access point on Raspberry Pi Zero 2 W:
Operating System: Raspbian GNU/Linux 11 (bullseye)
Kernel: Linux 6.1.21-v7+
Architecture: arm

Following the guides written in this repo I built newer hostapd 2.11 version. Then completelly repeat this instructions [AP Mode - How to build a bridged wireless access point](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md), but without enternet bridge, because I don't need internet connection, only local wireless 6 ghz network. I used this config file: [hostapd-WiFi6e.conf](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6e.conf).

And this is the result:

<details>
<summary>systemctl status hostapd</summary>

```
Andrei@raspberrypi:~ $ systemctl status hostapd
● hostapd.service - Access point and authentication server for Wi-Fi and Ethern>
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor prese>
     Active: activating (auto-restart) (Result: exit-code) since Wed 2024-02-21>
       Docs: man:hostapd(8)
    Process: 1043 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON>
        CPU: 24ms

Feb 21 23:54:48 raspberrypi systemd[1]: Stopped Access point and authentication>
Feb 21 23:54:48 raspberrypi systemd[1]: Starting Access point and authenticatio>
Feb 21 23:54:48 raspberrypi hostapd[1046]: Line 125: unknown configuration item>
Feb 21 23:54:48 raspberrypi hostapd[1046]: SAE: Enabling SAE H2E on 6 GHz
Feb 21 23:54:48 raspberrypi hostapd[1046]: 1 errors found in configuration file>
Feb 21 23:54:48 raspberrypi hostapd[1046]: Failed to set up interface with /etc>
Feb 21 23:54:48 raspberrypi hostapd[1046]: Failed to initialize interface
Feb 21 23:54:48 raspberrypi systemd[1]: hostapd.service: Control process exited>
Feb 21 23:54:48 raspberrypi systemd[1]: hostapd.service: Failed with result 'ex>
Feb 21 23:54:48 raspberrypi systemd[1]: Failed to start Access point and authen>
lines 1-17/17 (END)...skipping...
● hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2024-02-21 23:54:46 CET; 1s ago
       Docs: man:hostapd(8)
    Process: 1043 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS $DAEMON_CONF (code=exited, status=1/FAILURE)
        CPU: 24ms

Feb 21 23:54:48 raspberrypi systemd[1]: Stopped Access point and authentication server for Wi-Fi and Ethernet.
Feb 21 23:54:48 raspberrypi systemd[1]: Starting Access point and authentication server for Wi-Fi and Ethernet...
Feb 21 23:54:48 raspberrypi hostapd[1046]: Line 125: unknown configuration item 'ht_coex'
Feb 21 23:54:48 raspberrypi hostapd[1046]: SAE: Enabling SAE H2E on 6 GHz
Feb 21 23:54:48 raspberrypi hostapd[1046]: 1 errors found in configuration file '/etc/hostapd/hostapd-WiFi6.conf'
Feb 21 23:54:48 raspberrypi hostapd[1046]: Failed to set up interface with /etc/hostapd/hostapd-WiFi6.conf
Feb 21 23:54:48 raspberrypi hostapd[1046]: Failed to initialize interface
Feb 21 23:54:48 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Feb 21 23:54:48 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Feb 21 23:54:48 raspberrypi systemd[1]: Failed to start Access point and authentication server for Wi-Fi and Ethernet.
```

</details>

Some additional info about the system:
<details>
<summary>lsusb -t</summary>

```
Andrei@raspberrypi:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 5, If 1, Class=Wireless, Driver=btusb, 480M
        |__ Port 4: Dev 5, If 2, Class=Wireless, Driver=, 480M
        |__ Port 4: Dev 5, If 0, Class=Wireless, Driver=btusb, 480M
        |__ Port 4: Dev 5, If 3, Class=Vendor Specific Class, Driver=mt7921u, 480M
```

[... 617 lines cut ...]
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ BEACON_RATE_HT ]: HT beacon rate setting
		* [ BEACON_RATE_VHT ]: VHT beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

</details>

So I'll be glad to discuss what changes could I try to test.

Best regards,
Andrei Kavalenka

morrownr, 2024-02-22:
Hi @andrew-kovalenk0 

This is going to be fun.

You just beat me to the idea. I was going to suggest that you start with a WiFi 6, 5 GHz band setup first so as to learn a few things while working with a known good setup. The WiFi6e example hostapd.conf file that is down in the /home/AP_mode folder is not linked to the Main Menu yet as it is not ready for prime time. I guess I need to add a note about it being an alpha quality example at this point.

You mentioned using the Bridged Wireless Access Point guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

That should be a good guide for what you are trying to do. Now, from the guide:

`Further updates and testing will need to happen before this guide can be used with versions of RasPiOS prior to the 2023-10-10 version. `

I see that you are using an old version of RasPiOS. I am serious about the above statement. I know how many changes had to be made with the release of RasPiOS that is based on Debian 12. I really do recommend that you take a hard look at starting this project with a fresh burn of RasPiOS 12-05-2023 64 bit... is your Pi 64 or 32 bit? I am not familiar with this Pi so if you could pass along a link that gives the details, I would appreciate it.

With that said, I  might be able to walk you through some changes so that the guide works with the old RasPiOS but it could take a lot of work as there are some serious changes in networking.

Let me know what you think of the above and what questions you have.

@morrownr

andrew-kovalenk0, 2024-02-22:
> Let me know what you think

I'll install newer RasPiOS in that case. For some reaseon raspi imager offers only Debain 11 x32 for [Zero 2 W](https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/), but it's capable to run Debain 12 x64. After setting up a new system I'll try run WiFi6 access point and will come back with results.

andrew-kovalenk0, 2024-02-26:
I installed Raspy OS based on Debain 12, update hostapd to 2.11, repeat all steps from [AP Mode - How to build a bridged wireless access point](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md), but without bridge steps, because I don't need them. Result is the same - I can see hostspot on my phone, but can't connect to it, it fails on the step of receiving ip adres. Here is a log:
<details>
<summary>systemctl status hostapd</summary>

```
andrei@andrei:~ $ systemctl status hostapd
● hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: active (running) since Mon 2024-02-26 18:12:37 CET; 3min 11s ago
       Docs: man:hostapd(8)
    Process: 757 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid $DAEMON_OPTS $DAEMON_CONF (code=exited, status=0/SUCCESS)
   Main PID: 763 (hostapd)
      Tasks: 1 (limit: 169)
        CPU: 216ms
     CGroup: /system.slice/hostapd.service
             └─763 /usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd-WiFi6.conf

Feb 26 18:12:36 andrei systemd[1]: Starting hostapd.service - Access point and authentication server for Wi-Fi and Ethernet...
Feb 26 18:12:37 andrei hostapd[757]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Feb 26 18:12:37 andrei systemd[1]: Started hostapd.service - Access point and authentication server for Wi-Fi and Ethernet.
Feb 26 18:14:46 andrei hostapd[763]: wlan0: STA 9e:e5:12:78:09:a5 IEEE 802.11: associated (aid 1)
Feb 26 18:15:04 andrei hostapd[763]: wlan0: STA 9e:e5:12:78:09:a5 IEEE 802.11: disassociated
Feb 26 18:15:05 andrei hostapd[763]: wlan0: STA 9e:e5:12:78:09:a5 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

</details>

@morrownr any ideas/suggestions how can I fix it?

morrownr, 2024-02-27:
> I installed Raspy OS based on Debain 12, update hostapd to 2.11, ..

If you installed the 2023-12-05 64 bit RasPiOS (based on Debian 12) then you don't need to compile hostapd for what you are doing. I know, so much to keep up with.

If your setup is per my guide then a log file is created in your home directory... hostapd.log. Open it in a text editor and see if it is giving us any info.

@morrownr

andrew-kovalenk0, 2024-02-27:
Unfortunately I don't see a log file in my home directory. Double checked all the steps but everything seems to be correct. Here is hostapd service file, maybe there something is uncorrect:

<details>
<summary>systemctl status hostapd</summary>

```
[Unit]
Description=Access point and authentication server for Wi-Fi and Ethernet
Documentation=man:hostapd(8)
After=network.target
#ConditionFileNotEmpty=/etc/hostapd/hostapd.conf

[Service]
Type=forking
PIDFile=/run/hostapd.pid
Restart=on-failure
RestartSec=2
Environment=DAEMON_CONF=/etc/hostapd/hostapd-WiFi6.conf
Environment=DEAMON_OPTS="-d -K -f /home/andrei/hostapd.log"
#EnvironmentFile=-/etc/default/hostapd
ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS $DAEMON_CONF

[Install]
WantedBy=multi-user.target
```

</details>

andrew-kovalenk0, 2024-02-27:
Update: I tried wifi5 config, but result is the same. I'm thinking that the reason could be in bridge settings (I don't configure bridge from Ethernet port, because I don't need it. On Debain 11 I configured a bridge to Ethernet port (and on raspberry zero 2w there is no Ethernet port) with a same result) or in absence of Internet connection.

morrownr, 2024-02-29:
> I'm thinking that the reason could be in bridge settings 

Could be. I don't have a better suggestion.

andrew-kovalenk0, 2024-02-29:
Any ideas why there isn't log file in the home directory?

andrew-kovalenk0, 2024-02-29:
Yes, in this message you could find my service file https://github.com/morrownr/USB-WiFi/issues/384#issuecomment-1966324318

andrew-kovalenk0, 2024-02-29:
After some research I found that problem could be in the absence of dhcp server, so now I'm trying to install and configure it.

morrownr, 2024-02-29:
dhcp server ...

Did you install and enable the following?

sudo apt install systemd-resolved
sudo systemctl enable systemd-resolved

andrew-kovalenk0, 2024-02-29:
Yes, but after disabling networkmanager and without Ethernet connection there is no IP address at my wlan0 interface (checked through if config). And I think that is the reason.

morrownr, 2024-02-29:
> there is no IP address at my wlan0 interface

Did you set an IP address?

andrew-kovalenk0, 2024-02-29:
Yes, but it had no effect.

morrownr, 2024-02-29:
How did you set it?

I'm trying to see if we can avoid you installing dhcp as it is not clear how that will help and it will probably conflict with systemd-resolved.

andrew-kovalenk0, 2024-02-29:
`sudo ifconfig wlan0 192.168.0.1`, and then restart hostapd by `sudo systemctl restart hostapd`. And yes, I was trying to install dnsmasq and it also uses 53 port as systemd-resolved.

morrownr, 2024-02-29:
> ifconfig

I'd avoid using the above as it has been depreciated for several years.

Recommend you dig into systemd.networkd to see if you can set it there. This is a less than common setup for networking so I would just have to dig myself. If I had a good idea, I would tell you.

andrew-kovalenk0, 2024-03-24:
Hi, finally returned back to my project. So I updated my system to Debain 12, disabled NetworkManager, configured wlan0 interface in /etc/notwork/interfaces to has static IP address (didn't install and use systemd-netwrokd because it' more complicated without any benefits in my case), configured DHCP server with dnsmasq, so my phone obtain IP address while connecting to WAP. I downloaded your WIFI6 config where commented bridge=br0 line and everything works! But with WIFI6e config hostapd won't start with error: `Line 125: unknown configuration item 'ht_coex'`, so maybe you have any advices what can I try? @morrownr

jadtl, 2024-11-15:
Chiming in late but I have a very similar configuration where I'm also setting up a NAT instead of bridging wlan to ethernet, and for me it was enough to comment the `ht_coex` for the config to work.

kphanipavan, 2025-06-15:
Hello,

I recently bought an EDUP EP1672, which uses the same MT7921AU chipset. I am trying to use it as a hotspot using a raspberry pi 5, raspios 64bit with latest updates. Basically same as author's use case. The hotspot turns on, but the clients cannot communicate with each other. I made sure that `ap_isolate` is off in hostapd config. Can any of you confirm if clients connected to the hotspot from the usb card can communicate with eachother?

I am working on wifi 6 support for the lnxrouter project: https://github.com/garywill/linux-router/pull/82. Clients can discover fine on my realtek m.2 card, but cannot communicate when connected to the usb card.