morrownr/USB-WiFi #390: Deauth has stop working for 5GHz on MediaTek chipsets since kernel linux-image-6.1.0-12-amd64

LightMoon, 2024-02-29:
I have noticed, I am no longer able to deauth against the 5Ghz channels on kernels 6.3, 6.5 and 6.6; although it works fine on kernel 6.1.

This issue is only related to the 5Ghz band and can be reproduced by following the command.
aireplay-ng -0 10 -a <BSSID> wlan1mon

Can this be confirmed that the bug exists and is being tracked?

Wireless cards that I've tried and are problematic:

Mediatek MT7921AUN | Alfa AWUS036AXML
Mediatek MT7921AUN | Alfa AWUS036AXM
MediaTek MT7612U | Alfa AWUS036ACM

This issue is indifferent to CPU architecture and can be reproduced on arm ex, Raspi 5.

morrownr, 2024-02-29:
How are you going into monitor mode?

LightMoon, 2024-03-01:
I have tried [your script ](https://github.com/morrownr/Monitor_Mode) and also airmon-ng, both are not working.
with Monitor_Mode script which pauses the processes,
`airodump-ng --channel 132 --bssid A0:04:60:43:xx:xx wlan1mon`

```
aireplay-ng -0 10 -a A0:04:60:43:xx:xx wlan1mon             
19:25:05  Waiting for beacon frame (BSSID: A0:04:60:43:xx:xx) on channel 132
19:25:15  No such BSSID available.
```

with airmon-ng:

```
airmon-ng check kill
airmon-ng start wlan1 
```
`airodump-ng --channel 132 --bssid A0:04:60:43:xx:xx wlan1mon`

```└─# aireplay-ng -0 10 -a A0:04:60:43:2E:3B wlan1mon
19:26:08  Waiting for beacon frame (BSSID: A0:04:60:43:xx:xx) on channel 132
NB: this attack is more effective when targeting
a connected wireless client (-c <client's mac>).
19:26:08  Sending DeAuth (code 7) to broadcast -- BSSID: [A0:04:60:43:xx:xx]
19:26:09  Sending DeAuth (code 7) to broadcast -- BSSID: [A0:04:60:43:xx:xx]
19:26:09  Sending DeAuth (code 7) to broadcast -- BSSID: [A0:04:60:43:xx:xx]
```

morrownr, 2024-03-01:
> This results are for the card - Mediatek MT7921AUN | Alfa AWUS036AXML

I have one those. I'll do some testing and see what I come up with. So, 5 GHz band I'll use kernel 6.6.

LightMoon, 2024-03-03:
> > This results are for the card - Mediatek MT7921AUN | Alfa AWUS036AXML
> 
> I have one those. I'll do some testing and see what I come up with. So, 5 GHz band I'll use kernel 6.6.
By any chance have you managed to do some testing? I am waiting for your confirmation before opening a bug case on kernel.org. Hope you can make time to test and keep me posted.

morrownr, 2024-03-03:
Sorry for the delay. I'm not feeling well but will do it when able.

Heyholiday067, 2024-07-12:
Hi,this issue is still not fixed in the latest kernel 6.8

Heyholiday067, 2024-07-12:
Hi @LightMoon 
You said the adapter can work on 5ghz in kernel 6.1.0 , can it also work on kernel 6.0.0 ?
I am considered downgrading my kali now to solve this problem temporary.

LightMoon, 2024-07-12:
I haven't tested it on the 6.0.0 kernel, but I can confirm that it won't work on the DFS channels with any kernel version higher than 6.1.0. You might want to follow up on this ticket that's open on the kernel upstream. - https://bugzilla.kernel.org/show_bug.cgi?id=218638

Heyholiday067, 2024-07-12:
@LightMoon What you said "higher than 6.1.0" include kernel 6.1.0 ? Or i misunderstood your words!
By the way,you said dfs is not working and i found this repo but havent tested it yet !
https://github.com/singe/wifi-frequency-hacker
It can bypass most wifi restriction ! Do you think that is the new kernel making more restrictions on wifi?

morrownr, 2024-07-12:
@LightMoon 

Sorry for the delay but I have been having health problems. I checked my to-do list today and took a look at this issue again.

What does the following command show for your 5 GHz frquencies:

$ iw list

Look under `Frequencies:`.

I see your country is US. Is this problem happening on the regular channels or just on DFS channels?

I just don't see deauth being supported on DFS channels on a device that probably does not have radar detection capabilities.

Ankit1999Raikwar, 2024-07-13:
@LightMoon Hello,
                      Recently i purchased the AWUS036axml but its driver has some issues. while sending the death packets getting 
the error mentioned below. even I'm using the kali kernel version 6.0.1.

Sending DeAuth (code 7) to broadcast -- BSSID:

Heyholiday067, 2024-07-14:
@Ankit1999Raikwar Hi,i suggest you to open a new issue.This issue is addressed to mt7612u !
Open a new issue for a greater support.

Ankit1999Raikwar, 2024-07-15:
@LightMoon Thanks. i will create the new issue.

giannis371, 2024-08-01:
@LightMoon
I just bought an AWUS036ACM with MT7612U and have the same problem.
Did you find a solution?

morrownr, 2024-08-01:
Gents,

This particular problem came up a while back and I think it was reported by @ZerBea. His site is:

https://github.com/ZerBea/hcxdumptool

In fact, I think I saw him on linux-wireless saying that a submitted fix fixed the problem but this is not on my radar right now as I have too many other things going.

I think the backstory is that an Intel kernel dev submitted a patch for mac80211 (part of the Linux wireless stack) and it broke Mediatek's driver. Check in with ZerBea if you have time and want more info. A tremendous amount of work on the Linux wireless stack has been in progress for many months to add additional capabilities such as WiFi 7 and the tremendous complexities that it brings so shit is going to happen. The way the problem would have ended up in kernel 6.1 is that 6.1 is still a supported kernel and the patch was designated to go back into all supported kernels which has broken stuff all over the place. The patch will have to work its way back to kernel 6.1 as well.

ZerBea, 2024-08-02:
Two patched caused this problem.
This one broke monitor mode (on all on nearly all vendors): https://github.com/ZerBea/hcxdumptool/discussions/454
That was fixed, but the fix broke mt76 series drivers: https://github.com/ZerBea/hcxdumptool/discussions/465
This patch haven't found its way in to the Linux kernel yet.

ZerBea, 2024-08-02:
KALI and aireplay-ng have been mentioned above, so I suspect, you are professional penetration testers.
I apologize for not being a professional penetration tester. I only use simple tools for the latest test, like:
```
the patched kernel
Arch Linux
hcxlabtool
hcxpcapngtool
hashcat
```

Hardware is an ALFA AWUS036ACM
```
$ lsusb
ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
```
Here is the result (target is on 5GHz), running a patched Linux kernel (mentioned above):
```
$ uname -r
6.10.0-1-git-12246-g786c8248dbd3-dirty

$ sudo ./hcxlabtool -i wlp5s0f3u3  --rds=1
...
0 ERROR(s) during runtime
0 TX ERROR(s) during runtime
9584 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on sigterm

$ hcxpcapngtool -o test.22000 20240802094440-wlp5s0f3u3.pcapng
hcxpcapngtool 6.3.4-33-g62901cb reading from 20240802094440-wlp5s0f3u3.pcapng...
summary capture file
--------------------
file name................................: 20240802094440-wlp5s0f3u3.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 6.10.0-1-git-12246-g786c8248dbd3-dirty
application..............................: hcxdumptool 7.0
interface name...........................: wlp5s0f3u3
interface vendor.........................: 00c0ca
...
EAPOL pairs written to 22000 hash file...: 2 (RC checked)

$ hashcat -m 22001 --nonce-error-corrections=100 test.22000 test.list
hashcat (v6.2.6-848-gc1a10518f) starting
...
Recovered........: 2/2 (100.00%) Digests (total), 2/2 (100.00%) Digests (new)
```

ZerBea, 2024-08-02:
I forgot to mention that Management Frame Protection is enabled on my target:
```
RSN Capabilities: 0x00cc
    .... .... .... ...0 = RSN Pre-Auth capabilities: Transmitter does not support pre-authentication
    .... .... .... ..0. = RSN No Pairwise capabilities: Transmitter can support WEP default key 0 simultaneously with Pairwise key
    .... .... .... 11.. = RSN PTKSA Replay Counter capabilities: 16 replay counters per PTKSA/GTKSA/STAKeySA (0x3)
    .... .... ..00 .... = RSN GTKSA Replay Counter capabilities: 1 replay counter per PTKSA/GTKSA/STAKeySA (0x0)
    .... .... .1.. .... = Management Frame Protection Required: True
    .... .... 1... .... = Management Frame Protection Capable: True
    .... ...0 .... .... = Joint Multi-band RSNA: False
    .... ..0. .... .... = PeerKey Enabled: False
    ..0. .... .... .... = Extended Key ID for Individually Addressed Frames: Not supported
    .0.. .... .... .... = OCVC: False
```
The target is not vulnerable against a stupid DEAUTHENTICATION attack.

ZerBea, 2024-08-19:
Packet injection is working on:
```
$ uname -r
6.10.5-arch1-1
```

attack device (wlp22s0f0u9u3):
```
$ lsusb
Bus 001 Device 009: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
```

independent monitor device (wlp48s0f4u2u4):
```
$ lsusb
Bus 005 Device 003: ID 148f:5572 Ralink Technology, Corp. RT5572 Wireless Adapter
``` 

Test taget is on channel 100
`$ sudo hcxdumptool  -i wlp22s0f0u9u3 -c 100b --rds=1`

Deauthentication frame received on independent monitor interface:
```
Frame 419: 56 bytes on wire (448 bits), 56 bytes captured (448 bits) on interface wlp48s0f4u2u4, id 0
    Section number: 1
    Interface id: 0 (wlp48s0f4u2u4)
        Interface name: wlp48s0f4u2u4
    Encapsulation type: IEEE 802.11 plus radiotap radio header (23)
    Arrival Time: Aug 19, 2024 09:08:22.860605747 CEST
    UTC Arrival Time: Aug 19, 2024 07:08:22.860605747 UTC
    Epoch Arrival Time: 1724051302.860605747
    [Time shift for this packet: 0.000000000 seconds]
    [Time delta from previous captured frame: 0.000127698 seconds]
    [Time delta from previous displayed frame: 0.000127698 seconds]
    [Time since reference or first frame: 36.742583634 seconds]
    Frame Number: 419
    Frame Length: 56 bytes (448 bits)
    Capture Length: 56 bytes (448 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: radiotap:wlan_radio:wlan]
Radiotap Header v0, Length 30
    Header revision: 0
    Header pad: 0
    Header length: 30
    Present flags
        Present flags word: 0xa000402e
            .... .... .... .... .... .... .... ...0 = TSFT: Absent
            .... .... .... .... .... .... .... ..1. = Flags: Present
            .... .... .... .... .... .... .... .1.. = Rate: Present
            .... .... .... .... .... .... .... 1... = Channel: Present
            .... .... .... .... .... .... ...0 .... = FHSS: Absent
            .... .... .... .... .... .... ..1. .... = dBm Antenna Signal: Present
            .... .... .... .... .... .... .0.. .... = dBm Antenna Noise: Absent
            .... .... .... .... .... .... 0... .... = Lock Quality: Absent
            .... .... .... .... .... ...0 .... .... = TX Attenuation: Absent
            .... .... .... .... .... ..0. .... .... = dB TX Attenuation: Absent
            .... .... .... .... .... .0.. .... .... = dBm TX Power: Absent
            .... .... .... .... .... 0... .... .... = Antenna: Absent
            .... .... .... .... ...0 .... .... .... = dB Antenna Signal: Absent
            .... .... .... .... ..0. .... .... .... = dB Antenna Noise: Absent
            .... .... .... .... .1.. .... .... .... = RX flags: Present
            .... .... .... .... 0... .... .... .... = TX flags: Absent
            .... .... .... ..0. .... .... .... .... = data retries: Absent
            .... .... .... .0.. .... .... .... .... = Channel+: Absent
            .... .... .... 0... .... .... .... .... = MCS information: Absent
            .... .... ...0 .... .... .... .... .... = A-MPDU Status: Absent
            .... .... ..0. .... .... .... .... .... = VHT information: Absent
            .... .... .0.. .... .... .... .... .... = frame timestamp: Absent
            .... .... 0... .... .... .... .... .... = HE information: Absent
[... 121 lines cut ...]
        Flags: 0x00
            .... ..00 = DS status: Not leaving DS or network is operating in AD-HOC mode (To DS: 0 From DS: 0) (0x0)
            .... .0.. = More Fragments: This is the last fragment
            .... 0... = Retry: Frame is not being retransmitted
            ...0 .... = PWR MGT: STA will stay up
            ..0. .... = More Data: No data buffered
            .0.. .... = Protected flag: Data is not protected
            0... .... = +HTC/Order flag: Not strictly ordered
    .000 0000 0000 0000 = Duration: 0 microseconds
    Receiver address: b2:54:a8:12:b1:13
    Destination address: b2:54:a8:12:b1:13
    Transmitter address: b0:0a:d5:12:6c:42
    Source address: b0:0a:d5:12:6c:42
    BSS Id: b0:0a:d5:12:6c:42
    .... .... .... 0000 = Fragment number: 0
    0000 0000 1000 .... = Sequence number: 8
    [WLAN Flags: ........]
IEEE 802.11 Wireless Management
    Fixed parameters (2 bytes)
        Reason code: Class 3 frame received from nonassociated STA (0x0007)
```

Important information:
The same test failed on longterm Kernel linux-lts 6.6.46-1!
Packet injection on channels >= 100 is completely broken!

ZerBea, 2024-08-21:
@Heyholiday067 
I don't know, because conditions change on every new kernel version. There is a lot of work in progress.

Test device:
```
$ lsusb
ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
```

While 6.10.6 is working as expected:
```
$ uname -r
6.10.6-arch1-1

$ sudo hcxdumptool -i wlp5s0f4u2 --rcascan=active -c 100b
...
1017 Packet(s) captured by kernel
840 Packet(s) dropped by kernel
57 PROBERESPONSE(s) captured
```

6.6.47 failed:
```
$ uname -r
6.6.47-1-lts

$ sudo hcxdumptool -i wlp5s0f4u2 --rcascan=active -c 100b
...
^C
402 Packet(s) captured by kernel
210 Packet(s) dropped by kernel
Warning: no PROBERESPONSES received (frame injection may not work as expected)
Possible reasons:
 no AP in range
 frames are filtered out by BPF
 driver is broken
 driver does not support frame injection
```

The real funny part.
This device has never worked: TP-Link TL-WN8200ND(N) Ver:3.0
```
$ lsusb
Bus 003 Device 004: ID 2357:0126 TP-Link 802.11n NIC
```

Running kernel 6.6.47 packet injection is working, changing a channel not!
```
$ sudo hcxdumptool -i wlp5s0f4u2 --rcascan=active -c 100b
...
^C
106 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
48 PROBERESPONSE(s) captured
```

After awhile, the driver died:
`[13863.526769] usb 1-4: Failed to initialize RF`

ZerBea, 2024-08-22:
@KALI users
As long as you do not report this issue on KALI Linux Bug Tracker, you will not get a fix!

morrownr, 2025-01-08:
@Heyholiday067 

Thanks for the update. 2024 was a challenging year for wireless. So many API changes. I watch the bug fixes as they come into linux-wireless but it is not always possible to determine which one fixed a specific bug.

ZerBea, 2025-01-08:
@Heyholiday067 
Thanks for the head up.
If you run hcxdumptool/hcxlabtool there is no need to DEAUTHENTICATE a CLIENT on 5GHz.
From README.md:
`hcxdumptool is able to capture handshakes from 5/6GHz clients on 2.4GHz. (Only one single M2 from the client is required. Use hcxpcapngtool to convert to a format Hashcat or JtR understands.)`
This is an AP-LESS attack to retrieve an EAPOL M1M2ROGUE.
It is also possible to downgrade a WPA3 CLIENT to WPA2:
https://github.com/ZerBea/hcxdumptool/discussions/485#discussioncomment-11588215

As a consequence of the recent changes on Linux kernels, I've replaced my reference/testing WiFi hardware completely.
Now I'm running rtl8xxxu devices (bgn) as reference (AWUS035NHV, TP-LINK: TL-WN822N, TL-WN823N., TL-WN722N V2/V3)
Additional I'm testing rtw88 (USB) devices. But to retrieve a PMKID or an EAPOL M1M2ROGUE or an EAPOLM2M3, older bgn devices are much more suitable than AC or AX devices.
To increase range, hcxdumptool/hcxlabtool always choose the lowest rate and the smallest bandwidth. That exactly is the playground of the bgn devices.
A look (Wireshark) at the IE_TAG (Supported Rates and Extended Supported Rates) of a PROBEREQUEST confirm that a CLIENT can't resist to connect:
```
Tag: Supported Rates 1, 2, 5.5, 11, [Mbit/sec]
Tag: Extended Supported Rates 6, 9, 12, 18, 24, 36, 48, 54, [Mbit/sec]
```

morrownr, 2025-01-08:
@Heyholiday067 

> By the way,i want to know what is the max txpower value of mt7612u,i cant find it on the web(even with chatgpt)

Use the following command:

$ iw list

Then look at what it says for the channels in the 3 bands. The numbers will reflect your regulatory environment.

Right now the Mediatek drivers set txpower based on the max for the regulatory environment but I am hoping that we can get that changed so that we can adjust to lower levels. There are many reasons to be able to adjust to lower levels.

morrownr, 2025-01-12:
@Heyholiday067 

> However,i dont believe that those numbers are the max power of the hardware

I have seen people burn adapters by upping txpwr. That was back when it was easier to adjust power. I'm not an EE so about half of what I say on this topic could be wrong.

> which part in source code is about txpower?

I haven't worked on the Mediatek drivers yet. I hope to learn more about them at some point. When you are looking at the source, you may not find it as you might expect it. The regdb as used by the kernel has more to do with txpwr than the driver itself. I also think makers can set their own max pwr which may be less than regdb and what is used is the lesser of the two numbers. The makers will know what the tolerance for heat is.