morrownr/USB-WiFi #392: Alfa Network AWUS036AXML: WDS AP does not work

patrakov, 2024-03-02:
It gave another try today to OpenWrt snapshot (which uses 6.1.79 kernel with the WiFi stack backported from 6.6.15) on x86_64 hardware, using my two USB WiFi adapters:

* Alfa Network AWUS036ACM
* Alfa Network AWUS036AXML

I tried to set up WDS AP on each adapter. With AWUS036ACM, it just works. With AWUS036AXML, the WDS client connects, but the traffic passes only for that client, not for the bridged network behind it.

The `phy1-ap0.sta1` interface is not getting created.

The `iw phy` output is attached below.

[iw-phy.txt](https://github.com/morrownr/USB-WiFi/files/14469542/iw-phy.txt)

I think that this driver limitation should be documented.

morrownr, 2024-03-02:
Hi @patrakov 

> I think that this driver limitation should be documented.

I agree but I also think it should be fixed. Some options:

OpenWRT maintains a downstream repo for the mt76 series of drivers:

https://github.com/openwrt/mt76

Patches flow back and forth between this repo and the kernel. Posting your report at this site is worth a try. Some of the Mediatek devs are part of the OpenWRT project and they watch their mt76 repo. A suggestion:

- You are I know what chipset is in the adapters but the devs may not have all of that memorized and all they care about is the chipset so writing the reporting like the following might help:

Alfa Network AWUS036ACM (mt7612u)
Alfa Network AWUS036AXML (mt7921au)

I would add the chipset for the problem driver (mt7921au) in the title also.

@morrownr

zDEFz, 2024-03-08:
I cannot even use the entire Device, it immediately disconnects (Archlinux latest)
Trying to install rtl88xxau-aircrack-dkms-git makes my kernel no more load.

morrownr, 2024-03-08:
Hi @zDEFz 

Not much to go on in your message.

Your adapter is an Alfa AXML?

What kernel is in your Arch?

Why are you installing `rtl88xxau-aircrack-dkms-git`?

zDEFz, 2024-03-08:
> Hi @zDEFz
> 
> Not much to go on in your message.
> 
> Your adapter is an Alfa AXML?
> 
> What kernel is in your Arch?
> 
> Why are you installing `rtl88xxau-aircrack-dkms-git`?

I have read somewhere it helps.

Yes, AXML.

Finding: I need to blacklist btusb and bluetooth and my device works.

bjlockie, 2024-03-08:
You do not need the rtl (Realtek) stuff.

zDEFz, 2024-03-08:
> > I have read somewhere it helps.
> 
> No, the `rtl88xxau-aircrack-dkms-git` driver is for adapter with the rtl8812au chipset. You don't have that... at least not in the Alfa AXML adapter. Besides that, it you do need a rtl8812au driver, I have a more modern driver here.
> 
> > Blacklist btusb andbluetooth and the device works.
> 
> Exactly which files did you blacklist?
> 
> I want to get to the bottom of this.

ok I write you a guide on what I did

## Issue Resolution Guide

### Preliminary Steps:

1. Open terminal with administrative privileges:
   ```sh
   sudo nano /etc/modprobe.d/blacklist.conf
   ```

2. Add the following lines to blacklist Bluetooth and its related module:
   ```sh
   blacklist bluetooth
   blacklist btusb
   ```

3. Disable Bluetooth service:
   ```sh
   sudo systemctl disable bluetooth
   ```

4. Reboot

### Obtaining Firmware:

Please follow the instructions outlined in the third link provided [here](https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md).

### Reenabling Bluetooth Services:

1. Reopen the blacklist configuration file:
   ```sh
   sudo nano /etc/modprobe.d/blacklist.conf
   ```

2. Remove or comment out the lines that blacklist `bluetooth` and `btusb`.

3. Reenable Bluetooth service:
   ```sh
   sudo systemctl enable bluetooth
   sudo systemctl start bluetooth
   ```

That helped! Thank you for sharing the firmware guide!

zDEFz, 2024-03-08:
Nevermind...

Although the device now doesnt entirely disappear, pairing in bluetooth mode does not work.

zDEFz, 2024-03-08:
@morrownr So grabbing the firmware did help a lot. But I did not compile a new kernel as described in https://github.com/morrownr/USB-WiFi/blob/main/home/Compiling_the_Linux_Mainline_Kernel.md

I will try that, too.

morrownr, 2024-03-08:
Why is it that you want to compile a new kernel?

zDEFz, 2024-03-08:
> Why is it that you want to compile a new kernel?

Screw it, I send that thing back.
No time.

bjlockie, 2024-03-09:
FYI 

I don't think Bluetooth works in USB 3 mode.
I think you can force USB 2.

ntzb, 2024-04-11:
@patrakov what firmware were you running?
I'm planning to use WDS on the `Fenvi FU-AX1800` (with chip `mt7921au`) and it sad to hear it doesn't work

patrakov, 2024-04-11:
> @patrakov what firmware were you running?

The default one in OpenWrt 23.05.3. That is:

```
root@OpenWrt:~# ls -l /lib/firmware/mediatek/WIFI_*.bin
-rw-r--r--    1 root     root         92192 Mar 22 22:09 /lib/firmware/mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
-rw-r--r--    1 root     root        722420 Mar 22 22:09 /lib/firmware/mediatek/WIFI_RAM_CODE_MT7961_1.bin
root@OpenWrt:~# md5sum /lib/firmware/mediatek/WIFI_*.bin
2af4730a69f5372acd1c49547326919e  /lib/firmware/mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
68b488ee424fb50e073186e009786314  /lib/firmware/mediatek/WIFI_RAM_CODE_MT7961_1.bin
```

bjlockie, 2024-04-11:
I think the command `ethtool [wifi device]` shows the version.

morrownr, 2024-04-11:
Firmware info:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

ntzb, 2024-04-11:
in perfect timing, my `Fenvi FU-AX1800` adapter has just arrived.
for WDS, I'm trying to set the interface to 4addr:
`iw dev wlan1usb1ax1800 set 4addr on`
but getting in response:
`command failed: Operation not supported (-95)`

am I missing something?

morrownr, 2024-04-11:
@ntzb 

There are some pretty smart people stop by here but they will likely never see this issue at the end of an old post. Recommend you come up with an appropriate title and explanation and post a new issue.

bjlockie, 2024-04-11:
Are you root?
If not, maybe you need to be `sudo -s`

morrownr, 2024-04-11:
Saw this is a very old reply to a question:

```
# undo the earlier configuration
iw dev wlan0 set 4addr off

# add a second interface
iw dev wlan0 interface add wds.wlan0 type managed 4addr on
ip link set dev wds.wlan0 addr $ADDR
ip link set dev wds.wlan0 up

Here $ADDR must match the WDS device address configured in the router; other than that, it can be any valid MAC address. The original MAC of wlan0 then remains for "normal" usage.
```

morrownr, 2024-04-11:
This might be good info for you to know about the mt7921u driver:

```
valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 2

```

Also, I am not an expert at what you are doing but should you not take the interface down before you try to make a change? This question kind of follows on to the posting I made above. I do some limited monitor mode work you gotta take the interface down before you to do certain things. Also, if NetworkManager is trying to manage the interface you are trying to use, NM wins so it could be that you need to configure NM to not manage the interface. This would case the reply that you saw.

ntzb, 2024-04-12:
> Are you root? If not, maybe you need to be `sudo -s`

yeah, everything was run as root, in debian

> This might be good info for you to know about the mt7921u driver:
>
> ```
> valid interface combinations:
> 		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
> 		   total <= 2, #channels <= 2
> ``` [...]

what does this actually mean? (about the valid interface combinations).

to make this clearer:
I'm running debian 12, kernel 6.1.0-18, connecting to the  J3455 celeron mini PC, using the built in wifi chip.
on first boot with the usb device, I updated it's firmware according to the mentioned page. right now:
```
root@wds:~# ethtool -i wlan1usb1ax1800
driver: mt7921u
version: 6.1.0-18-amd64
firmware-version: ____010000-20240219111038
expansion-rom-version:
bus-info: 2-3:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```
iw phy shows:
```
Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
```

I renamed the device from wlx[mac-address-without-semicolons] to `wlan1usb1ax1800` via `/etc/systemd/network/10-wlan1usb1ax1800.link`, and then tested wireless connectivity by adding the following in `/etc/network/interfaces`:
```
# fenvi ax1800
allow-hotplug wlan1usb1ax1800
iface wlan1usb1ax1800 inet dhcp
        wpa-ssid SSID
        wpa-psk  PASSWORD
```
once I verified network connection using the new device, I commented out that part above, and tried enabling 4addr before creating the bridge, using `iw dev wlan1usb1ax1800 set 4addr on`, and got the error `command failed: Operation not supported (-95)`.
I also tried turning the device down: `ip link set dev wlan1usb1ax1800 down`, and then running the command - same result.
after that, my searches led me to try creating a startup service, running `set 4addr on` before network initialization (`Before=network.target systemd-networkd.service`), still no success.
adding a second interface also fails:
```
root@wds:~# iw dev wlan1usb1ax1800 interface add wds.wlan0 type managed 4addr on
command failed: Operation not supported (-95)
```

at this point, I'm questioning if:
1. the chip actually supports WDS and 4addr (according to [this](https://unix.stackexchange.com/a/680164/205424), since `AP` was listed, it should be supported?)
2. the driver has any issues
3. the firmware has any issues
4. debian 12 has any issues

according to the first post, it seems one of 1-3 may be true...
I don't really know what to do next

morrownr, 2024-04-12:
Let me see what I can explain.

Here is an example of adding a new interface, giving it a type and turning 4addr on. I expected you to use your information when I posted the example. Note that the interface in the command below is an already existing interface from a Alfa ACM adapter that uses an mt7612u chipset. I gave the new interface the name of myNewIface, made it a managed interface and turned 4 addr on. See how it look by using $ iw dev.

```
$ sudo iw dev wlx00c0caadcb84 interface add myNewIface type managed 4addr on
$ iw dev
phy#3
	Interface myNewIface
		ifindex 7
		wdev 0x300000002
		addr 00:c0:ca:ad:cb:84
		type managed
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
		4addr: on
	Interface wlx00c0caadcb84
		ifindex 6
		wdev 0x300000001
		addr 00:c0:ca:ad:cb:84
		type managed
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
phy#0
	Interface wlp3s0
		ifindex 4
		wdev 0x1
		addr e8:fb:1c:7e:4d:cb
		ssid myPI-WiFi6
		type managed
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0

```

```
valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 2
```

> what does this actually mean? (about the valid interface combinations).

You adapter based on the mt7921au chipset can support multiple interfaces on the same adapter. I just showed you an example above with my phy#3. The interface combinations supported by any wireless device on Linux can be found with this command:

$ iw list

That command will list a LOT of information so you have to look for the section I posted. This ability to add multiple interfaces can come in very handy depending on what you are doing.

> the chip actually supports WDS and 4addr ?

My adapters with the mt7921au chipset are busy right now so I tested 4addr on the older generation chipset and 4addr came to life as you can see in my posting above. I am no expert with WDS. I suspect WDS will work given the knowledge to set it up.

> the driver has any issues ?

The mt7921u driver is really solid but all drivers have issues of one kind or another. I run this driver is several modes, to include AP mode, and have found it to be really good.

> the firmware has any issues ?

Same answer as for the driver.

> debian 12 has any issues

It could have an issue but I don't think it is the problem in this case.

> I'm running debian 12, kernel 6.1.0-18

You can upgrade to kernel 6.6 very easily. How do you add .deb packages to your installation?

> I don't really know what to do next

Don't get frustrated. Learn what you are doing one step at a time. If you are following a guide, you can give me a link and I will offer my advice on whether it is a good guide or not.

ntzb, 2024-04-12:
I followed multiple stackexchange/superuser/debian q&a, for creating WDS, and it all boiled down to the need for enabling 4addr on the interface.

anyway, it's not working:
```
wds@wds:~$ sudo iw dev wlx90de80d6883f interface add myNewIface type managed 4addr on
command failed: Operation not supported (-95)
```
I tried upgrading to the latest kernel I could easily find
```
wds@wds:~$ uname -r
6.8.5-1-liquorix-amd64
```

maybe I should replicate your env, what OS/kernel are you running?
it would really be helpful if you could test your mt7921au devices for 4addr, but even then, could there be a difference between makers? e.g. alfa vs fenvi

morrownr, 2024-04-12:
>it would really be helpful if you could test your mt7921au devices for 4addr...

I have new adapter coming early next week. I might find the time to do a temp change in setup here to see what I can do before then. Pay very careful attention to the example I gave you above. Keep in mind that you have to take your existing interface down before you do this.

```
$ sudo iw dev wlx00c0caadcb84 interface add myNewIface type managed 4addr on
$ iw dev
phy#3
	Interface myNewIface
		ifindex 7
		wdev 0x300000002
		addr 00:c0:ca:ad:cb:84
		type managed
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
		4addr: on
```

Notice how it says "4addr: on". 

> but even then, could there be a difference between makers? e.g. alfa vs fenvi

Extremely unlikely. This is a driver thing, not a hardware thing.

ntzb, 2024-04-12:
the interface was not up before the `4addr` manipulation.

seems like I've ran my course on this adapter, unless you can think of something else to try...
I'll wait for your testing, and then we'll know more

morrownr, 2024-04-12:
> seems like I've ran my course on this adapter, unless you can think of something else to try...

I really can't think of anything else right now. 

> I'll wait for your testing, and then we'll know more

This is on my to-do list. My setup is different than most people in that running this site causes me to have to organize things. I'll get to this but right now a couple of other tests are in progress with the adapter that I need to use. I will have another adapter early next week so this won't take too long.

Could you do something in return? Forget about 4addr for a while and just use the fenvi to see how you like it? I'm always looking for reports to add to the Plug and Play List.

My place is interesting. I have probably around 30 usb wifi adapters. I do a lot of work and testing to help users out. I can't be an expert on everything but I can often get to the bottom of problems. Have a good weekend.

ntzb, 2024-04-14:
I tried two other OS: arch (latest) and clear linux (latest)
both of them presented the same error:
```
sudo iw dev wlan1 interface add myNewIface type managed 4addr on
command failed: Operation not supported (-95)
```

it seems that mt7921au does not support 4addr.
@morrownr is there any way to get in touch with the mediatek driver developers? I'd like to verify this.

possibly related issues:
https://github.com/morrownr/USB-WiFi/issues/407, https://github.com/morrownr/USB-WiFi/issues/396

regarding regular usage of the adapter - it's a bit tricky, because it was bought specifically to be used in a bridge setup with the J3455 celeron mini PC I mentioned.
my other devices are wired to an existing gl.inet router (I dislike that solution, so wanted to replace it).
I'll see if I can make changes in my setup to allow the regular usage.

bjlockie, 2024-04-14:
Does the mt7612 support 4addr?

patrakov, 2024-04-15:
At least the USB version, yes.

ntzb, 2024-04-15:
@morrownr until you have some news with the tests, I [setup proxy arp](https://raspberrypi.stackexchange.com/a/88955) with the device, and so far (~1d) it's working just fine.
between the 2402Mb/s (5GHz) router, and the 1201Mb/s USB WiFi device, I get ~730Mb/s from ~9m distance.
the setup, as mentioned, is proxy arp, with the host device being an x86 debian machine.
don't see any weird behaviours, or ping spikes (again, so far)

patrakov, 2024-04-15:
Regarding the benchmark, I have some questions:

* What are the reported signal levels on both ends?
* Does the USB WiFi device act as an access point or as a client?

ntzb, 2024-04-15:
it functions as a client.

router side (openwrt):
```
1080.6 Mbit/s, 80 MHz, HE-MCS 10, HE-NSS 2
1200.9 Mbit/s, 80 MHz, HE-MCS 11, HE-NSS 2
```

client side (not at the exact same moment):
```
freq: 5500
RX: 2244127469 bytes (5183119 packets)
TX: 419339166 bytes (375195 packets)
signal: -44 dBm
rx bitrate: 1200.9 MBit/s 80MHz HE-MCS 11 HE-NSS 2 HE-GI 0 HE-DCM 0
tx bitrate: 1080.6 MBit/s 80MHz HE-MCS 10 HE-NSS 2 HE-GI 0 HE-DCM 0

bss flags:      short-slot-time
dtim period:    2
beacon int:     100
```

patrakov, 2024-04-15:
Can you do an opposite type of a benchmark, with the USB WiFi acting as an AP?

ntzb, 2024-04-15:
the router does not have a USB slot unfortunately.
I'll see if I can think of a simple way to make this test case you suggest

morrownr, 2024-04-18:
@patrakov 

After getting some time to look at this issue, it appears that WDS is not supported in AP mode currently. I was able to get positive results for 4addr in managed mode. I am going to write this up in the pinned issue where we track bugs/feature request.

Are you aware of any usb wifi adapters that do support WDS in AP mode?

ntzb, 2024-04-18:
@morrownr what's `managed` mode? did you get 4addr working in a regular client mode?

patrakov, 2024-04-18:
> Are you aware of any usb wifi adapters that do support WDS in AP mode?

Yes - Alfa Network AWUS036ACM.

morrownr, 2024-04-18:
> @morrownr what's `managed` mode? did you get 4addr working in a regular client mode?

Managed mode, Client mode. STA.... all the same thing.

Yes, 4addr came right up in client mode.

morrownr, 2024-04-18:
@patrakov 

So, this is working with the mt7612u driver but not with the mt7921u driver?

Here is the link to the Mediatek Linux wifi page. Note thr emaill addresses at the bottom. When you email the, remember they communicate with chipsets and drivers, not product brand and model number so Alfa AXML may not mean anything to them. Be nice and make sure you tell them it works in the mt7612u driver but not the mt7921u driver. You might also add a link to this issue in case they need background info.

@morrownr

vincent-zouzou, 2025-10-28:
> https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/net/wireless/mediatek/mt76?id=565ddaaab9a15424a9037f929bce685aa34ffd0b
>    mt76: mt7921: disable 4addr capability    
>    4addr is not supported by the driver/firmware yet

> https://github.com/openwrt/mt76/blob/master/mt792x_core.c#L652

From above, I think mt792x would never support 4addr mode

@morrownr