morrownr/USB-WiFi #393: AWUS1900 driver on RPI CM4 (5.10.17-V8+)

bking46, 2024-03-03:
Hi All,

I wanted to ask for your support please with any guidance on getting this adapter to work on my system. When trying to install the driver I keep getting an error stating that the kernel headers are missing/incorrect for 5.10.17-V8+. I have installed the raspberry pi kernel headers.  Going down the route of installing the lastest raspberry pi OS would not be possible as I need to keep the current setup due to extensive work on other software I have already installed and working. Any suggestions to get this driver working  on this setup would be much appreciated.  I have a back-up of this setup so would be willing to try anything experimental. 

Many thanks!

bjlockie, 2024-03-04:
I think it is a Realtek RTL8814AU.

Read [this](https://github.com/morrownr/8814au)

bking46, 2024-03-04:
@bjlockie Thank you for your input. That is what I have followed for the install however I get the following errors (have tried both methods for install):

[image]
[image]
[image]

bjlockie, 2024-03-04:
`apt-cache search linux-headers
`

bking46, 2024-03-04:
Thanks, this is the output:
Linux raspberrypi 5.10.17-v8+ #5 SMP PREEMPT Thu Jun 29 13:36:01 CST 2023 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Mar  4 09:02:49 2024
pi@raspberrypi:~ $ apt-cache search linux-headers
linux-libc-dev-amd64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-arm64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-armel-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-armhf-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-i386-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-ppc64el-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-s390x-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mips-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mips64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mips64el-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mips64r6-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mips64r6el-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mipsel-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mipsn32-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mipsn32el-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mipsn32r6-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mipsn32r6el-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mipsr6-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mipsr6el-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-alpha-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-hppa-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-m68k-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-powerpc-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-ppc64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-riscv64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-sh4-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-sparc64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-x32-cross - Linux Kernel Headers for development (for cross-compiling)
linux-headers-5.10.0-26-arm64 - Header files for Linux 5.10.0-26-arm64
linux-headers-5.10.0-26-armmp - Header files for Linux 5.10.0-26-armmp
linux-headers-5.10.0-26-armmp-lpae - Header files for Linux 5.10.0-26-armmp-lpaelinux-headers-5.10.0-26-cloud-arm64 - Header files for Linux 5.10.0-26-cloud-arm64
linux-headers-5.10.0-26-common - Common header files for Linux 5.10.0-26
linux-headers-5.10.0-26-common-rt - Common header files for Linux 5.10.0-26-rt
linux-headers-5.10.0-26-rt-arm64 - Header files for Linux 5.10.0-26-rt-arm64
linux-headers-5.10.0-26-rt-armmp - Header files for Linux 5.10.0-26-rt-armmp
linux-headers-5.10.0-28-arm64 - Header files for Linux 5.10.0-28-arm64
linux-headers-5.10.0-28-armmp - Header files for Linux 5.10.0-28-armmp
linux-headers-5.10.0-28-armmp-lpae - Header files for Linux 5.10.0-28-armmp-lpaelinux-headers-5.10.0-28-cloud-arm64 - Header files for Linux 5.10.0-28-cloud-arm64
linux-headers-5.10.0-28-common - Common header files for Linux 5.10.0-28
linux-headers-5.10.0-28-common-rt - Common header files for Linux 5.10.0-28-rt
linux-headers-5.10.0-28-rt-arm64 - Header files for Linux 5.10.0-28-rt-arm64
linux-headers-5.10.0-28-rt-armmp - Header files for Linux 5.10.0-28-rt-armmp
linux-headers-armmp - Header files for Linux armmp configuration (meta-package)
linux-headers-armmp-lpae - Header files for Linux armmp-lpae configuration (meta-package)
linux-headers-rt-armmp - Header files for Linux rt-armmp configuration (meta-package)
linux-headers-arm64 - Header files for Linux arm64 configuration (meta-package)
linux-headers-cloud-arm64 - Header files for Linux cloud-arm64 configuration (meta-package)
linux-headers-rt-arm64 - Header files for Linux rt-arm64 configuration (meta-package)
linux-headers-5.10.0-25-arm64 - Header files for Linux 5.10.0-25-arm64
linux-headers-5.10.0-25-armmp - Header files for Linux 5.10.0-25-armmp
linux-headers-5.10.0-25-armmp-lpae - Header files for Linux 5.10.0-25-armmp-lpaelinux-headers-5.10.0-25-cloud-arm64 - Header files for Linux 5.10.0-25-cloud-arm64
linux-headers-5.10.0-25-common - Common header files for Linux 5.10.0-25
linux-headers-5.10.0-25-common-rt - Common header files for Linux 5.10.0-25-rt
linux-headers-5.10.0-25-rt-arm64 - Header files for Linux 5.10.0-25-rt-arm64
linux-headers-5.10.0-25-rt-armmp - Header files for Linux 5.10.0-25-rt-armmp
linux-headers-5.10.0-27-arm64 - Header files for Linux 5.10.0-27-arm64
linux-headers-5.10.0-27-armmp - Header files for Linux 5.10.0-27-armmp
[... cut ...] 0.0-27-armmp-lpaelinux-headers-5.10.0-27-cloud-arm64 - Header files for Linux 5.10.0-27-cloud-arm64
linux-headers-5.10.0-27-common - Common header files for Linux 5.10.0-27
linux-headers-5.10.0-27-common-rt - Common header files for Linux 5.10.0-27-rt
linux-headers-5.10.0-27-rt-arm64 - Header files for Linux 5.10.0-27-rt-arm64
linux-headers-5.10.0-27-rt-armmp - Header files for Linux 5.10.0-27-rt-armmp
raspberrypi-kernel-headers - Header files for the Raspberry Pi Linux kernel
pi@raspberrypi:~ $

bjlockie, 2024-03-04:
I am guessing but try:
`apt install linux-headers-arm64`

If that doesn't work, I have no more ideas.

bking46, 2024-03-04:
@bjlockie, Thanks for the suggestion. These were already installed. Output says "...is already the newest version (5.10.200-2)

morrownr, 2024-03-04:
The RasPi devs have been breaking Linux kernel headers for a long time. The version of the distro that you have installed is dated so it would take a lot of work to determine exactly which one of the many problems is the one you are experiencing. I understand that you want to stick with the installation you have but unless you have the skills to determine the problem and fix, it likely will not be fixed.

You do have options. Adapters with mt7612u chipsets should work fine on that kernel as no compiling is required. You can take a look at the Plug and Play List for more information.

@morrownr

bking46, 2024-03-05:
@morrownr, thank you for letting me know. This is quite unfortunate for me as I have both the AWUS036AXML, AUS1900 and none work with this setup. I was really keen on trying out everything to get at least one of these working with this setup. But from what I could tell from your response there is no easy way of doing this. I am willing to tryout/experiment on getting either of these to work if you can point me in the right direction 😉. Yesterday I managed to install the driver on this system using the following commands:
$ apt install debhelper dpkg-dev dkms libelf-dev bc 
$ dpkg-buildpackage -b --no-sign
$ cd ..
$ dpkg -i rtl8814au-dkms_5.8.5.1_all.deb
But it skipped kernel 5.10.17-V8+ and installed it for Linux 5.10.28.arm64 if I recall correctly. 
However the adapter didn't show up. My guess is that without the 5.10.17-V8+ kernel headers installed it will never work.

morrownr, 2024-03-05:
@bking46 

> But from what I could tell from your response there is no easy way of doing this. 

With the adapters you mentioned above, this is correct. In the case of the AXML, it is such a modern adapter that the driver did not come along until kernel 5.18 so your kernel 5.10 is too old. In the case of the 1900, this rtl8814au chipset has never been well supported by Realtek. The driver I have up is just the best I and some others can make of it but it is not a great driver... and it needs to be compiled, which is where you are running into problems. It is not that I would not like to help you further but I know how much work it is likely to take to figure out what the fix would be and my to-do list is already far to long.

The AXML is a really good adapter. I have one. What you might do is burn a new sd, if that is what you are doing, and try the AXML with your hardware. I do recommend using the 2023-12-05 or later RasPiOS 64 bit. If this works well, you need to consider whether it is really worth keeping that old installation or whether your time is best spent porting the setup to a newer version of RasPiOS. Like I said, adapters with the mt7612u or even the mt7610u chips should work with the 5.10 kernel as the drivers have been in the Linux kernel since 4.19.

Using USB WiFi adapters with Linux has improved greatly over the last 6 years and we are likely to see very big improvements over the next year but when something in your project includes something that is dated, it is a good idea to ask questions. Should we expect good support for the AXML or other modern chipset on Windows 7? Probably not. Should we expect good support for the AXML on Linux for the next 20 years? Probably so as that is realistic. There are people here that can make good recommendations based on the details of any specific project.

Good luck.

@morrownr

bking46, 2024-03-05:
@morrownr Thank you very much for taking the time to explain this to me, I really appreciate it.  I forgot to mention in my replies earlier that I would want to migrate to a newer version of RaspiOS however at the moment I am stuck on this since the device (Uconsole) doesn't yet function correctly on other version, the kernel is heavily patch and so far I have failed to patch it to make the built-in screen to work with newer rpi OS's. The device as setup now by me works perfectly with all other software I have on it, it's just these adapters drivers that are causing challenges for me. You did raise a very good point that instead of insisting on getting this outdated OS to work with AWUS1900 I should focus and invest more time in getting the newer OS to work on this device and hopefully the AXML would work straight out of the box. Again, thanks for all your support and ideas with this.

All the best!

closed 2024-03-05 by bking46