morrownr/USB-WiFi #394: can't update firmware for 7921

bjlockie, 2024-03-03:
I'm trying to update the firmware of my PCI card but it is still using the old one. :-(
I use lubuntu (Ubuntu variant).

```
/lib/firmware/mediatek$ ls -l *7961*
-rw-r--r-- 1 root root 532206 Mar  2 14:52 BT_RAM_CODE_MT7961_1_2_hdr.bin
-rw-r--r-- 1 root root 380339 Dec 15 07:49 BT_RAM_CODE_MT7961_1_2_hdr.bin.zst.old
-rw-r--r-- 1 root root  92192 Mar  2 14:52 WIFI_MT7961_patch_mcu_1_2_hdr.bin
-rw-r--r-- 1 root root  47846 Dec 15 07:49 WIFI_MT7961_patch_mcu_1_2_hdr.bin.zst.old
-rw-r--r-- 1 root root 797300 Mar  2 14:52 WIFI_RAM_CODE_MT7961_1.bin
-rw-r--r-- 1 root root 384849 Dec 15 07:49 WIFI_RAM_CODE_MT7961_1.bin.zst.old

$ ethtool -i wlp4s0
driver: mt7921e
version: 6.5.0-21-generic
firmware-version: ____010000-20231109190959
expansion-rom-version: 
bus-info: 0000:04:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

morrownr, 2024-03-03:
Where did you get the new files?

bjlockie, 2024-03-03:
```
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
 
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/mediatek/WIFI_RAM_CODE_MT7961_1.bin

wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin
```

Even if it got corrupted firmware, why is it reporting it is using the old ones?

morrownr, 2024-03-03:
New files came through linux-wireless last week but have not been posted to the site yet. You updated with the same version.

bjlockie, 2024-03-03:
Dec 2023 is the latest version at the kernel website?
Are the new files available on another website?

morrownr, 2024-03-04:
> Dec 2023 is the latest version at the kernel website?

Yes

> Are the new files available on another website?

May be but not one that I know of.

If you click on `log` , you can see when files are uploaded.

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/mediatek

morrownr, 2024-03-07:
@bjlockie 

The new firmware for mt7921 and mt7922 is now posted.

Akumar201, 2025-03-30:
@morrownr Hey, I am trying to find the driver can you please locate where it is as I can not see it under tree.

https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

this is my machine specs 
```

lspci -nnk | grep -iA3 net
09:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7927]
    Subsystem: Foxconn International, Inc. Device [105b:e124]
0a:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. Device [10ec:8126] (rev 01)
    Subsystem: ASUSTeK Computer Inc. Device [1043:88e0]
    Kernel driver in use: r8126
    Kernel modules: r8126
```

morrownr, 2025-03-30:
Hi @Akumar201 

The wifi chip in your system is a mt7927. The Linux driver for that chip has not been merged into mainline yet. The driver for the mt7925 has been in the kernel since 6.7. I have an M.2 card and a PCie card based on the mt7925. Very nice. Will we see a mt7927 driver? Probably but we do not know when. Mediatek merged a new driver for the mt7990 chip a couple of days ago. Many of the wireless chips that Mediatek makes are Linux only so we know Linux is heavily supported by Mediatek. I suspect they are a little behind as many new chips have been shipping over the last 2 years.

What you could do for now is get a mt7925 M.2 card, if your system uses M.2 for wireless, and replace the card. We are talking about maybe 20 USD. The replacement card could always be used to  upgrade something else later if the mt7927 driver shows.

You could also get a currently available USB WiFi adapter as they are flexible. Go to the Main Menu:

https://github.com/morrownr/USB-WiFi

Menu item 2, The Plug and Play List, has a lot of information about and links to many USB WiFi adapters that work very well with Linux.

If you have questions, let me know.