morrownr/USB-WiFi #395: COMFAST CX-951AX not work properly

Lingzilla, 2024-03-06:
Hi, I bought a COMFAST CX-951AX on Alipress and hopfully wanna just plug and play

I got this with `lsusb` command:

` Bus 002 Device 008: ID 0e8d:7961 MediaTek Inc. Wireless_Device`

and I got these errors in the `dmesg` output:

```
[  820.059522] usb 2-1: new SuperSpeed USB device number 14 using xhci_hcd
[  820.075140] usb 2-1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  820.075153] usb 2-1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  820.075156] usb 2-1: Product: Wireless_Device
[  820.075158] usb 2-1: Manufacturer: MediaTek Inc.
[  820.075160] usb 2-1: SerialNumber: 000000000
[  820.127657] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[  822.803088] Bluetooth: hci1: Device setup in 2656161 usecs
[  822.803103] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  824.852402] Bluetooth: hci1: Opcode 0x0c03 failed: -110
[  826.901342] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[  828.948313] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[  834.198267] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  839.828196] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  840.036150] usb 2-1: device not accepting address 14, error -62
[  845.461108] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  851.092022] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  851.299999] usb 2-1: device not accepting address 14, error -62
[  856.724081] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  862.356874] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  862.563854] usb 2-1: device not accepting address 14, error -62
[  867.987809] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  873.619733] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  873.827698] usb 2-1: device not accepting address 14, error -62
[  873.846765] mt7921u: probe of 2-1:1.3 failed with error -5
[  873.847029] usb 2-1: USB disconnect, device number 14
[  878.256397] perf: interrupt took too long (2502 > 2500), lowering kernel.perf_event_max_sample_rate to 79000
[  879.251651] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  884.883858] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  885.091553] usb 2-1: device not accepting address 15, error -62
[  890.515584] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  896.147469] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  896.355347] usb 2-1: device not accepting address 16, error -62
[  896.363436] usb usb2-port1: attempt power cycle
[  902.300138] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  907.923297] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  908.131121] usb 2-1: device not accepting address 17, error -62
[  913.556007] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  919.186932] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[  919.394905] usb 2-1: device not accepting address 18, error -62
[  919.403022] usb usb2-port1: unable to enumerate USB device
```

bjlockie, 2024-03-06:
What kernel and distro and hardware?
Delete the Bluetooth firmware and/or try a different USB port.

Lingzilla, 2024-03-06:
I uses Fedora39 on a laptop with N4020 CPU, and plugged this dongle into the one of USB ports on this machine.

I have already tried a different port, but still got the same issue.

And I tried to use it on my newer machine with ArchLinux installed and Intel 1240P processor, still this problem.

I have also tried this on Windows 11 with COMFAST offical Windows driver installed, but got error

`" This device is not functioning well with error number -10, this device can not be started. "`

morrownr, 2024-03-06:
@Lingzilla 

I also have a cx-951ax. When I first got it, I went through about what you are going through. Over a period of a few days it started working. I think the problem was due to a manufacturing issue. My best guess is that there was something left on one or more of the usb pins in the adapter and it was not making good contact. I think putting it in and out many times helped it make better contact and it started working. I still can't use it with extention cables or powered hubs but can use it with an L shaped adapter. 

It actually does a decent job for what I use it for these days but Comfast's quality control is not impressive... at least with this adapter. I have read a lot of good comments from others that seem to indicate they have had better success with the 953ax and 952ax as far as Comfast adapters go. I also have an Alfa AXML which is far better quality but then the price is much higher.

If the adapter does not work well within a few days, you should probably return it and look at something else. There are other low cost adapters with the same chipset in the Plug and Play List.

Cheers,

@morrownr

Lingzilla, 2024-03-08:
> What kernel and distro and hardware? Delete the Bluetooth firmware and/or try a different USB port.

how to archive this (I use archlinux)?

only remove the firmware used by this chip?

I finally got it work by using `modprobe -r btusb` to disable bluetooth usb module in kernel to get the wifi functioning.

Lingzilla, 2024-03-12:
THX. Everything is working fine now :)

closed 2024-03-12 by Lingzilla

morrownr, 2024-03-12:
Glad to hear it. Enjoy.