morrownr/USB-WiFi #396: Layer 2 Bridging in Managed/Client mode

2193219831, 2024-03-07:
Hi, 

First off, thank you so much for this awesome repository! Really helps having everything in one place. 
Not sure what keywords to use, but I'm looking to purchase a WiFi adapter that supports L2 bridging in managed / client mode. Is that the same thing as WDS and is 4addr support needed?

Ideally, the Wi-Fi adapter should be capable of this op: 
~ip l set wlan0 master br0

Any help narrowing this down would be highly appreciated!

morrownr, 2024-03-08:
Hi @2193219831 

Are you looking for WiFi5 capability?

Are you looking for WiFi6 capability?

Are you looking for 6 GHz band capability?

What speeds are you looking for? AC1200/AC600/AXE3000?

What will you be doing with the adapter?

2193219831, 2024-03-08:
Hi @morrownr  

Essentially, I'm trying to reproduce this setup from https://gist.github.com/zOrg1331/a2a7ffb3cfe3b3b821d45d6af00cb8f6 but with wireless interfaces. 

I'm using wireguard to connect to my home network from my girlfriend's and would like to have Layer 2 Transparency using gretap or vxlan on top of wireguard. Ethernet isn't really an option for me.

Anything >= WiFi 5+ would be ideal, but if that's not an option, I'd be willing to sacrifice on speed.

2193219831, 2024-03-09:
Nevermind, what I wanted to do is quite flaky and would require all APs to be compatible with WDS or 4addr mode.
If I insist on using wlan (wlan0) to connect and wlan (wlan1) to broadcast, it would be best to NAT the IP traffic and place all LAN devices on the internal NATted router. I'd get true Layer 2 transparency over wireguard with all devices on wlan1

morrownr, 2024-03-09:
Copy all. Sometimes you have to back up and start over. I've had to do that many times. 4addr and WDS are missing from Realtek out-of-kernel drivers. Your best bet is the Plug and Play List on the Main Menu.

2193219831, 2024-03-09:
I bought the NetGear AXE3000 and it doesn't seem to support either WDS or 4addr.
Also, I'm trying to setup AP mode on the raspberry Pi 5, but hostapd won't take off....

morrownr, 2024-03-09:
> Also, I'm trying to setup AP mode on the raspberry Pi 5, but hostapd won't take off....

What guide for AP mode are your using?

What adapter are you using?

What operating system are you using?