morrownr/USB-WiFi #398: News: Status of Project to improve in-kernel support for rtl8821cu/rtl8811cu chipsets

morrownr, 2024-03-08:
As many of you may be aware, usb support for the rtw88 in-kernel driver was added with kernel 6.2. I maintain an out-of-kernel driver here for the same chipsets and noticed reports from users that the in-kernel driver was not working well. I confirmed this with testing. A few months ago I opened an issue in my local repo having to do with a Project to fix the problems. Things should be working much better with the following patches heading into kernel 6.9:

Bitterblue Smith (5):
      wifi: rtw88: 8821cu: Fix firmware upload fail
      wifi: rtw88: 8821cu: Fix connection failure
      wifi: rtw88: 8821c: Fix beacon loss and disconnect
      wifi: rtw88: 8821c: Fix false alarm count

Nick Morrow (1): (co-developed with Larry Finger)
      wifi: rtw88: Add missing VID/PIDs for 8811CU and 8821CU

If you have an adapter based on the rtl8821cu/rtl8811cu chipsets, please test and report when you are able to install kernel 6.9. Work has so far centered on basic functionality so if you are able to test AP mode, monitor mode and P2P, please do so and report here.

Thanks go to those who are working on this project. If this project is successful, adapters with these chipsets can be added to the Plug and Play List. These chipsets allow for tiny adapters which many people have ask for.

@morrownr

fakemanhk, 2024-04-03:
Do you mean with kernel 6.9 we can expect plug and play without doing anything?

morrownr, 2024-04-03:
> Do you mean with kernel 6.9 we can expect plug and play without doing anything?

Yes. Managed mode appears to be working well with the exception that some adapters may experience slower than expected speeds. We are working that problem. AP mode is still problematic but work on that can happen hopefully once managed mode is fully good to go. On adapters that have bluetooth, it seems to be working well.

So, if you are using kernel 6.9, let us know how it goes.

This project also seems to have caused an effort to create an in-kernel driver for the rtl8821/11au chipset that has been popular with users over the years. This may happen soon.