morrownr/USB-WiFi #399: AXE3000 on OpenWRT

black-jmyntrn, 2024-03-08:
lsusb shows: Bus 001 Device 009: ID 0bda:c832 Realtek 802.11ax WLAN Adapter

What's interesting, running my own build Generic Device would show up in wireless. 

After copying over WIFI_RAM_CODE_MT7961_1.bin and WIFI_MT7961_patch_mcu_1_2_hdr.bin, I then removed the /etc/config/wireless followed by wifi config and a reboot. Now, it doesn't even show Generic Device 

Any tips to get this adapter working in openwrt?

morrownr, 2024-03-08:
> After copying over WIFI_RAM_CODE_MT7961_1.bin and WIFI_MT7961_patch_mcu_1_2_hdr.bin

Those firmware files are for cards or adapters using the mt7921au, mt7921k and mt7921 chipsets. Your chipset is a Realtek chipset. Without looking, it is probably a rtl8832bu?

> Any tips to get this adapter working in openwrt?

Not really. There isn't much support for adapters that use Realtek out-of-kernel drivers in OpenWRT. The reason is that it is hard to do.

The solution I use for OpenWRT is adapters with the following chipsets as OpenWRT already has support:

mt7610u
mt7612u
mt7921au

You can install the drivers with Luci in the Software section. I can show you how if you get an adapter. There is a list adapters in the Plug and Play List on the Main Menu.

@morrownr

black-jmyntrn, 2024-07-27:
its the below chipset.

Bus 001 Device 005: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)

morrownr, 2024-07-27:
Hi @black-jmyntrn 

>  ID 0bda:1a2b

That means the adapter is a multi-state adapter... it has a Windows driver onboard. Once the state is changed, you can see the ID of the wifi device. Do you know how to use usb_modeswitch to manually switch the adapter out of its CDROM mode?

black-jmyntrn, 2024-11-09:
yep. will give it a go

black-jmyntrn, 2025-04-18:
to followup I did use usb modeswitch and now the id is  0bda:c832