morrownr/USB-WiFi #400: New 802.11ac MT7612U adaptor - PIX-LINK LV-UAC04

fakemanhk, 2024-03-12:
I was walking in a mall with shops selling lots of stuff from China and then this PIX-LINK LV-UAC04 drew my attention (interestingly official website doesn't have it)

Below is the lsusb details from my Chromebook (I don't have other Linux PC with me due to traveling now):

```
Bus 002 Device 008: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x0e8d MediaTek Inc.
  idProduct          0x7612 MT7612U 802.11a/b/g/n/ac Wireless Adapter
  bcdDevice            1.00
  iManufacturer           1 MediaTek Inc.
  iProduct                2 802.11ac WLAN
  iSerial                 3 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x007a
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          2 802.11ac WLAN
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              800mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           8
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11ac WLAN
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x07  EP 7 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f41e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us 
    Deep BESL value    61440 us 
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         180 micro seconds
Device Status:     0x000c
  (Bus Powered)
  U1 Enabled
  U2 Enabled
```

Though I haven't tested it now but first impression is good, can be recognized by Chromebook is a positive signal, however this dongle is flat and a bit wide (looks similar to COMFAST CF-953AX) so you might find difficulties to use on Raspberry Pi if other USB devices connected.

Online selling channel:
[AliExpress](https://www.aliexpress.com/item/1005003407274228.html)
[Amazon India](https://www.amazon.in/Jutek-PIX-UAC04-Wireless-Adapter/dp/B07LGH4RLK)

morrownr, 2024-03-12:
@fakemanhk 

Thanks for the report. Can I get you to send me a report when you are home? I'll add this adapter to the Plug and Play List if it gets your approval.

>  this dongle is flat and a bit wide (looks similar to COMFAST CF-953AX) so you might find difficulties to use on Raspberry Pi if other USB devices connected.

I have a fix for that. This is what I do with my RasPi and laptops: (right angle usb adapter)

amazon.com/gp/product/B07S6B5X76/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

> can be recognized by Chromebook is a positive signal

Yes, Chromebooks are supporting several USB WiFi adapters these days. In fact, my wife uses ChromeOS-Flex on her desktop system these days as she likes a really simple interface:

https://chromeenterprise.google/intl/en_uk/os/chromeosflex/

Last week I tested several adapters on ChromeOS-Flex and it supports many adapters and chipsets in the Plug and Play List here. The following chipsets worked wonderfully: (this list is also probably good for recent Chromebooks)

mt7612u
mt7610u
rt5370
mt7601u
ar9271

I tried the mt7921au chipset but not yet. Android and ChromeOS always seem to use kernels that are a few years old so maybe someday. I also tried several Realtek chipset based adapters and did not find any that worked.

Thanks again for the report and I look forward to your review when you are able to send it. I need a recommendation as to whether this adapter should go in the Plug and Play List.

@morrownr

fakemanhk, 2024-03-14:
@morrownr Just borrowed a Windows laptop from others and I booted with Linux Mint 21.3 USB Live (still using kernel 5.15) to have a quick test on it.

As expected, I do nothing with the OS, plug the PIX-LINK dongle and immediately usable, and this is a single-state device (since I didn't do anything with USB modswitch, and by default the USB ID has a product ID 7612 when I plug in so I assume this, hope I am understanding this correctly)

Below are the dmesg when I connect the laptop to my OpenWrt router, router end shows clearly that the connection is `866.7 Mbit/s, 80 MHz, VHT-MCS 9, VHT-NSS 2, Short GI`
```
[  356.452657] usb 4-1.1: new SuperSpeed USB device number 4 using xhci_hcd
[  356.474036] usb 4-1.1: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[  356.474048] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  356.474052] usb 4-1.1: Product: 802.11ac WLAN
[  356.474055] usb 4-1.1: Manufacturer: MediaTek Inc.
[  356.474057] usb 4-1.1: SerialNumber: 000000000
[  356.600691] usb 4-1.1: reset SuperSpeed USB device number 4 using xhci_hcd
[  356.622136] mt76x2u 4-1.1:1.0: ASIC revision: 76120044
[  356.650333] mt76x2u 4-1.1:1.0: ROM patch build: 20141115060606a
[  356.801039] mt76x2u 4-1.1:1.0: Firmware Version: 0.0.00
[  356.801048] mt76x2u 4-1.1:1.0: Build: 1
[  356.801051] mt76x2u 4-1.1:1.0: Build Time: 201507311614____
[  357.551790] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[  357.552946] usbcore: registered new interface driver mt76x2u
[  357.563216] mt76x2u 4-1.1:1.0 wlx00e0bb330922: renamed from wlan0
[  411.622377] wlx00e0bb330922: authenticate with xx:xx:xx:xx:xx:xx
[  411.676019] wlx00e0bb330922: send auth to xx:xx:xx:xx:xx:xx (try 1/3)
[  412.305637] wlx00e0bb330922: authenticate with xx:xx:xx:xx:xx:xx
[  412.305644] wlx00e0bb330922: send auth to xx:xx:xx:xx:xx:xx (try 1/3)
[  412.717450] wlx00e0bb330922: authenticated
[  412.720295] wlx00e0bb330922: associate with xx:xx:xx:xx:xx:xx (try 1/3)
[  412.721917] wlx00e0bb330922: RX AssocResp from xx:xx:xx:xx:xx:xx (capab=0x111 status=0 aid=3)
[  412.723870] wlx00e0bb330922: associated
[  412.773556] wlx00e0bb330922: Limiting TX power to 24 (27 - 3) dBm as advertised by xx:xx:xx:xx:xx:xx
````

Running some speed test (uplink speed 100Mbps only, the transfer speed hit 96-98Mbps and I am sure it can do better with higher uplink speed when I test at home), with a test download (3.5GB Zorin ISO download) the speed sustained at max. uplink speed all the time (11.3MB/s), I can feel that the dongle a bit warm which is acceptable, there is a LED but it doesn't lit under Linux (tested with Windows I can confirm that it has light).

I tried to test AP mode by plugging it to my GL-INET MT1300, but it looks like there is problem on MTK USB side (I searched around and see there were a few reports regarding the MT7621A USB with external USB device incompatibility, I already used a powered USB hub so definitely nothing related to power), and I don't have enough time to test the AP mode on the borrowed laptop (I only had about half hour time to check).

Above are all the very first impression but I strongly believe that this is a "plug-and-play" one, will test a bit more, especially with AP mode when I go home (I will have GL-INET MT3000 coming very soon, not sure if that one works with this dongle or not), I do have the angled adapter already so I will also test with Raspberry Pi 4, NanoPi R4S together.

fakemanhk, 2024-03-14:
> Yes, Chromebooks are supporting several USB WiFi adapters these days. In fact, my wife uses ChromeOS-Flex on her desktop system these days as she likes a really simple interface:

I also have ChromeOS Flex on my Lenovo X61 at home, in general Flex doesn't use very new kernel so sometimes I am worrying about the hardware support, I only tested my Asus USB-AC51 (MT7610U), and a DLink RT2870 (2.4GHz 802.11n very old USB WiFi) which are known to work.

fakemanhk, 2024-03-22:
@morrownr 

Just get back home and tested this with my Zorin 17.1 (kernel 6.5) fresh install, with USB Mode Switching turned off it can still be recognized immediately, so it's a true single-state device.

morrownr, 2024-03-24:
@fakemanhk 

Thanks for the report. This sounds like a good adapter to add to the Plug and Play List.

For the Plug and Play List, I need some links to trusted sellers so that users have links to the product. Can you find some links?

And can you do a review as you have time as it is good to have a review also.

Thanks,

@morrownr

morrownr, 2024-03-24:
Here are some links that I have found. You can check and verify that this is the correct adapter:

https://www.aliexpress.us/item/3256803220959476.html

https://www.amazon.in/-/hi/dp/B07LGH4RLK

More links would be good.

morrownr, 2024-03-24:
I added an entry for this adapter in the Plug and Play List. Please take a look.

I can add your review when it is ready. Please add any info that you think is of benefit to users.

Thanks.

fakemanhk, 2024-04-03:
Did more tests these few days, with NanoPi R4S with OpenWrt 23.05.3 as AP.

Below are Waveform test and Fast.com test results, this is something out of my expectation, at about 1m distance (signal level roughly -30dBm) with my phone there is no issue to get ~500Mbps download while upload is even faster.

Previously I saw the comments about Netgear A6210 which doesn't have a good range, I took my phone to another room, and/or even 1 level below (wooden home), I was still able to get 300-350Mbps speed.

Using laptop to test ~10GB file transfer to server through this WiFi AP, everything working as expected, smooth and fast, the dongle only a bit warm.

However since this dongle is flat so it might block other devices, when I tried to put angled adaptor to my NanoPi R4S I got some USB bus reset issue, this doesn't happen when I use other Linux machine so probably not the dongle's problem. 

The verdict is, if you plan to use it to build some kind of travel router, it definitely can do the job nicely.

[image]

[image]

fakemanhk, 2024-04-03:
> Here are some links that I have found. You can check and verify that this is the correct adapter:
> 
> https://www.aliexpress.us/item/3256803220959476.html
> 
> https://www.amazon.in/-/hi/dp/B07LGH4RLK
> 
> More links would be good.

Those links should be the same as the one I posted at original post (at the bottom)

And this shop from Singapore seems legit as well:

https://www.electronicscrazy.sg/wireless-usb-adapter-1200m-2.4-5ghz-dual-band-wifi-portable-router-lv-uac04/

morrownr, 2024-04-03:
Hi @fakemanhk 

It sounds like you are getting along with this adapter so far. I added the links to the entry in the Plug and Play List. Thanks.

Do you want to write a short review or should I link to this thread?

fakemanhk, 2024-04-03:
> Hi @fakemanhk
> 
> It sounds like you are getting along with this adapter so far. I added the links to the entry in the Plug and Play List. Thanks.
> 
> Do you want to write a short review or should I link to this thread?

As I am not a professional writer, I think it might be good to link to this and see if someone else has problem with it.

Currently I have set a few of my devices using this AP for couple hours and not seeing any problem so far, so I do receommend it.

morrownr, 2024-04-03:
Okay, I used a link to this issue for a review. Let us know if you have further information whether it is good or bad. 

Thanks again.

yun14u, 2024-06-17:
I did a research and bought this USB wifi adapter.  It works well on my Lenovo laptop running Ubuntu 24.04 (Noble Numbat).

morrownr, 2024-06-17:
Thanks for the report. This adapter is listed in the Plug and Play List. The mt7612u chip that this adapter uses is well supported in Linux.