morrownr/USB-WiFi #402: COMFAST AX-952 v2: available, Realtek based, works with 8852bu.ko

patrakov, 2024-03-14:
I have ordered a COMFAST AX-953 card, but the seller shipped the wrong item: COMFAST AX-952 v2. I am going to return it in the next hour. Before I do so, here is some information.

lsusb:

```
Bus 003 Device 060: ID 3574:6121 Realtek 802.11ac WLAN Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x3574 Realtek
  idProduct          0x6121 802.11ac WLAN Adapter
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac WLAN Adapter
  iSerial                 3 00e04c000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x004a
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           8
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11ac WLAN Adapter
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
[... 140 lines cut ...]
iperf Done.
[aep@aep-haswell ~]$ iperf3 -c 192.168.10.1 -p 9999 -R
Connecting to host 192.168.10.1, port 9999
Reverse mode, remote host 192.168.10.1 is sending
[  5] local 192.168.12.244 port 40498 connected to 192.168.10.1 port 9999
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  26.2 MBytes   220 Mbits/sec                  
[  5]   1.00-2.00   sec  27.6 MBytes   232 Mbits/sec                  
[  5]   2.00-3.00   sec  28.4 MBytes   238 Mbits/sec                  
[  5]   3.00-4.00   sec  28.5 MBytes   239 Mbits/sec                  
[  5]   4.00-5.00   sec  28.2 MBytes   237 Mbits/sec                  
[  5]   5.00-6.00   sec  28.1 MBytes   236 Mbits/sec                  
[  5]   6.00-7.00   sec  28.5 MBytes   239 Mbits/sec                  
[  5]   7.00-8.00   sec  28.9 MBytes   242 Mbits/sec                  
[  5]   8.00-9.00   sec  28.9 MBytes   242 Mbits/sec                  
[  5]   9.00-10.00  sec  28.9 MBytes   242 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   286 MBytes   240 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   282 MBytes   237 Mbits/sec                  receiver

iperf Done.
```

This again confirms my negative recommendation against the bulkier but slower ALFA Network AWUS036AXML.

morrownr, 2024-03-14:
Thanks for the heads up. This is exactly why I have not added the CF-952 to the Plug and Play List. That model number has been confusing since the beginning.

fakemanhk, 2024-03-15:
Seller confuses this with CF953AX easily, actually this happened also when I ordered my CF953AX, CF952AX arrived instead.

fjh1997, 2026-02-22:
Hi @patrakov and @morrownr,

I just wanted to chime in and confirm that I’ve also received this exact model (COMFAST CF-952AX v2) with the same hardware ID 3574:6121.

Based on my testing, while the ID might seem obscure, it is indeed an RTL8852BU chipset. For anyone looking for a modern and stable driver, this device is now effectively "Plug and Play" (or at least very easy to set up) using the in-kenel driver on kernel 6.17+. 

I've checked the iw phy output for this RTL8852BU (3574:6121) device. While it clearly supports HE capabilities and monitor mode is listed under Supported interface modes, the driver (rtw89_8852bu) seems to lack a functional Injection path for USB devices in the current 6.17+ kernel mainline.like：
`ioctl(6, SIOCGIWFREQ, 0xffffd3e96678) = -1 ENOTTY (Inappropriate ioctl for device)`

The ENOTTY error from aireplay-ng confirms that the legacy WEXT path is gone, and the nl80211 interface isn't responding to frame injection requests. It appears this chip is a beast for high-speed AX networking, but for security auditing (injection), we might still be waiting for better driver maturity.

```
root@localhost:~# iw phy
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 8
	max scan IEs length: 394 bytes
	max # sched scan SSIDs: 8
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0 RX 0
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x19ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15, 32
		HE Iftypes: managed
			HE MAC Capabilities (0x0801100a8060):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				All Ack
				BSR
				Maximum A-MPDU Length Exponent: 2
				OPS
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x02701e1f0dc0270e91bd00):
[... 402 lines cut ...]
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xb0 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ POWERED_ADDR_CHANGE ]: can change MAC address while up
```