morrownr/USB-WiFi #403: Another two COMFAST MT7612U adaptors

fakemanhk, 2024-03-15:
Well, since I'm in greater China region now so it's easy to get more interesting stuff at lower price, I found the following 2 devices were never mentioned in your Wiki so I am going to write about it (I'm still not home yet)

[CF-WU782AC v2](https://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=214) (I don't know if v1 is the same)

[CF-WU785AC](https://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=215)

Both devices are having a lot of similar things so I am not going to split them into 2 posts.

Common things:

- MT7612U based (2.4GHz 400Mbps + 5GHz 866Mbps)
- Single-state
- USB 3.0 (Micro B type at device end)
- Desktop style (not little dongle anymore, in fact they are quite huge in size)
- 18dBm High transmit power (but **_iwconfig_** is showing 20dBm??)

Under Linux Mint 21.3, I don't have to do anything and plug-and-play immediately, and very good signal quality due to high gain antenna (CF-782AC v2 has 2 x 6dBi while CF-785AC has 4 x 6dBi), even on load you could hardly feel it warm due to large casing. Desktop style ensures that you won't have issue like the CF-953AX wide & flat style which will block other USB/ethernet devices when using with Raspberry Pi style SBCs.

fakemanhk, 2024-03-18:
My bad.....the CF-WU782AC vs in fact is not "single-state" (previously I plugged this into Linux Mint PC which I guess it automatically switched mode so that I didn't notice?), just now when I was trying to plug this into Chromebook I saw a storage device coming up, with ID 0e8d:2870.

For CF-WU785AC, for sure it's single-state (plugged into Chromebook and recognized)

10muhtesem, 2024-03-19:
@fakemanhk  Hello, are you satisfied with the CF-WU785AC adapter? I'm considering getting one myself but still can't decide. The type of adapter I need is one that costs under $20-25, supports monitor mode packet injection in Kali, and optionally has AP support. It should also be a LONG-RANGE adapter (the most important aspect).

morrownr, 2024-03-19:
@fakemanhk 

The best way to tell for sure that an adapter is single-state is for you to turn usb_modeswitch off. Go to:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md

then go to the last section called:

How to deactivate usb_modeswitch

follow the instructions. I think you will find the 785AC is multi-state. I have one and just tested it. It is multi-state. I used to have the 785AC and 782AC in the Plug and Play List but about a year ago, I decided that there are plenty of single-state adapters available and that I would no longer list multi-state adapters. The 785AC is a solid adapter.

@10muhtesem 

> The type of adapter I need is one that costs under $20-25, 

> It should also be a LONG-RANGE adapter (the most important aspect).

This adapter does not exist. The very high quality antennas and amps that are required for long range are not cheap. You can choose one of those characteristics but not both.

Suggestions:

ALFA AWUS036ACHM - $39.97 - single-state
https://store.rokland.com/collections/wi-fi-usb-adapters/products/alfa-awus036achm-802-11ac-dual-band-high-power-ac1200-mediatek-wifi-usb-adapter

Panda PAU0B - $27.99 - single-state
https://www.amazon.com/dp/B08NPX2X4Z/

Thoughts: I think both of these adapters would meet your needs. Both are in the mt7610u section of the Plug and Play List. The following interface modes are supported:

Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO

The driver is in-kernel and is solid. Active monitor mode is supported as is interface combinations:

valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2,
		   total <= 2, #channels <= 1, STA/AP BI must match

The drivers for the following chipsets are also full featured but the better quality adapters get farther away from your price range :

mt7612u - some users have problems with VM's and this driver. Not so with the mt7610u.
mt7921au - currently active monitor mode is broken. Regular monitor mode is fine.
mt7925 - the driver is in the kernel but no adapters are on the market yet. This is a WiFi 7 chipset.

Overall, the Alfa ACHM is the gold standard for pen testing and security analysis. The Panda is likely a good choice as well but it will have somewhat shorter range than the ACHM. I have the Alfa adapter. The range is, well, very impressive. I think the Panda  has above average range based on what I hear from people and reading reviews so it might be a good choice for you. 

@morrownr

10muhtesem, 2024-03-19:
@morrownr Thank you for your response but i will be ordering from AliExpress because I have a coupon, so unfortunately neither of the adapters you recommended is available at the prices you mentioned. There is only a replica product with an RT3070L chipset, claimed to be of the same quality as the Alfa AWUS036NH, available in this price range, but I'm not sure about it :/
Do you have any recommendations among these 4 adapters? 

COMFAST CF-WU785AC (MT7612UN) - 4* 6dBi antennas (not sure about it)
COMFAST CF-938AC (RTL8814AU) - It has AP support
EDUP EP-AC1686 (RTL8812BU) - 1* 5dBi antennas
FENVI AC1300 (RTL8812BU) - It has AP support (the cheapest one)

If it's accurate, adapter COMFAST CF-WU785AC with the 6dBi antenna seems more logical to me, but I still wanted to get your opinion. I apologize if I'm bothering you.

10muhtesem, 2024-03-19:
> Will you be running in a VM?

No, dual boot with Windows.

morrownr, 2024-03-19:
So, dual boot and buy from AliExpress.

Comfast CF-953AX
https://www.aliexpress.us/item/3256804585621404.html?spm=a2g0o.productlist.main.11.57f814426UG3h9&algo_pvid=629168ad-29e3-47c2-8bcf-22de809c1af5&algo_exp_id=629168ad-29e3-47c2-8bcf-22de809c1af5-5&pdp_npi=4%40dis%21USD%2122.38%2115.89%21%21%2122.38%2115.89%21%40210318e817108724035035273ec6b2%2112000037006351662%21sea%21US%210%21AB&curPageLogUid=HMHhnEp5jTwE&utparam-url=scene%3Asearch%7Cquery_from%3A

Fenvi FU-AX1800
https://www.aliexpress.us/item/3256805749323751.html?gatewayAdapt=glo2usa4itemAdapt

Both have the mt7921au chipset that is one of the most Linux compatible chipsets available. They won't have that long range like the Alfa ACHM but it could be a good start for you. Make sure you double check that both of the above have the mt7921au chipset.

> COMFAST CF-938AC (RTL8814AU) - It has AP support

Avoid any adapter that uses the rtl8814au chipset.

> COMFAST CF-WU785AC (MT7612UN) - 4* 6dBi antennas (not sure about it)

In your list of 4 adapters, this is the one I would pick but then I gave you two more options at the start of this message.

@morrownr

10muhtesem, 2024-03-19:
> @morrownr

Thank you so, so much. I appreciate your efforts. Have a great day!

fakemanhk, 2024-03-20:
But Chrome OS doesn't support USB mode switching, and I am able to use the
WU-785AC (not 782AC V2)

morrownr, 2024-03-20:
@fakemanhk 

> But Chrome OS doesn't support USB mode switching, and I am able to use the
WU-785AC...

This means one of two things:

- Chrome OS does support mode switching

or 

- Comfast is now making a version of the 785AC that is single state

Please let me know once you return home with a full test but disabling usb_modeswitch. If indeed your 785AC is single state, I will add the adapter back to the Plug and Play List. My 785AC is multi-state but I've had it for 3 years or so.

fakemanhk, 2024-03-22:
> The best way to tell for sure that an adapter is single-state is for you to turn usb\_modeswitch off. Go to:
>
> https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md
>
> then go to the [...]

OK you are right, I just get back home and used my PC with Zorin 17.1 to test, the 785AC was shown as the same as the 782AC v2 which presents itself as a drive, a bit disappointed here, but looks like modern Linux based OS already tackles this well and I don't really need to do anything.

BTW have you ever tried to use USB mode switching on OpenWrt?

morrownr, 2024-03-24:
> but looks like modern Linux based OS already tackles this well and I don't really need to do anything.

This is a correct statement. Modern Linux distros using usb_modeswitch handle most adapters, most of the time that are mullti-state. However, I have seen enough problems over the years that I cannot recommend that users wanting to buy a new adapter, buy one that is multi-state. The Plug and Play List is about adapters that present the fewest problems possible. 

The 785AC is a very solid adapter and I have never seen a problem related to it being multi-state.

> BTW have you ever tried to use USB mode switching on OpenWrt?

I have not but I have seen others talk about it. It works. There is a package, I think the name is usb_modeswitch, that you have to install and then ssh into OpenWRT and set it up. I do not have the exact details on the setup handy but I can look if you cannot find them.

fakemanhk, 2024-03-31:
> I have not but I have seen others talk about it. It works. There is a package, I think the name is usb\_modeswitch, that you have to install and then ssh into OpenWRT and set it up. I do not have the exact details on the setup handy but I can look if you cannot find them.

Just tested with this, looks like I don't have to do anything on OpenWrt after installing this package, my 782AC v2 was recognized automatically, using it as AP, with a phone close to it I can get 540Mbps download + 480Mbps, will test the 785AC very soon.

morrownr, 2024-04-01:
Sounds good. I really like it when things just work. The mt7612u chipset plus driver is really solid on Linux.

10muhtesem, 2024-04-02:
> Fenvi FU-AX1800 https://www.aliexpress.us/item/3256805749323751.html?gatewayAdapt=glo2usa4itemAdapt
 
> > COMFAST CF-WU785AC (MT7612UN) - 4* 6dBi antennas (not sure about it)
> 
> In your list of 4 adapters, this is the one I would pick but then I gave you two more options at the start of this message.

@morrownr Hi again.
I purchased both the adapter COMFAST CF-WU785AC from my own list and the Fenvi FU-AX1800 adapter from your recommendations. I haven't received the Fenvi FU-AX1800 yet, but I had the opportunity to test the COMFAST CF-WU785AC adapter that I received today. I'm just going to ask a simple question. I didn't know before that a single adapter could be used for both Wi-Fi connection and hotspot at the same time, and I thought that 2 adapters were needed for this process. However, thanks to COMFAST CF-WU785AC, I can both get internet connection from Wi-Fi and open a hotspot like a Wi-Fi repeater at the same time. My question is, would using a single adapter to both connect to Wi-Fi and share the connected Wi-Fi internet strain/overheat the device too much? In other words, would there be any problem?

morrownr, 2024-04-02:
Hi @10muhtesem 

>  I didn't know before that a single adapter could be used for both Wi-Fi connection and hotspot at the same time, and I thought that 2 adapters were needed for this process.

Two adapters are required for AP mode to support dual band (this is a common question) but there are situations that work fine with a single adapter performing more than one function. A single adapter, that supports this, can be in managed mode and AP mode at the same time. It could also be in managed mode and P2P-client mode at the same time. This would be handy for people that have a printer that is not connected to the local network but does support P2P. This can be very handy. There are other possibilities as well.

To tell what the capabilities of your adapters are, run the following command:

$ iw list

Look for the `valid interface combinations:` section. You should find that yours looks like the following:

```
	valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2,
		   total <= 2, #channels <= 1, STA/AP BI must match

```

Here is an example from my PCIe WiFi card that uses a mt7922 chipset:

```
	valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 2

```

I can explain what the above means if you want.

So that you know, the Mediatek drivers are feature rich in my opinion. Realtek drivers and chipsets are missing some features... both the in-kernel and out-of-kernel varieties.

Would you be interested in writing a simple guide for how you setup your hotspot with one adapter? I can help edit it and then it can be on the Main Menu for others.

@morrownr

10muhtesem, 2024-04-02:
> @morrownr

First of all, I sincerely apologize for providing incorrect and incomplete information. I made a mistake by not stating from the beginning that it was Windows, and then I made another mistake by stating that I use the same adapter both for receiving signals and as a hotspot. I am not very experienced in these matters, so I expressed myself incorrectly. Actually, what I did was to receive signals with a USB WiFi adapter and share this connection with the PC's built-in adapter. Unfortunately, I conveyed this incorrectly and may have given the impression that I found a way to do something that is currently not possible. I apologize again for the time you've spent. Have a good day.

fakemanhk, 2024-06-11:
@morrownr Well you might think I'm crazy, but I just did something to change both CF-WU782ACv2 & CF-WU785AC back to **single state**.

Accidentally I came across this [GitHub comment on CF-926AC](https://github.com/openwrt/mt76/issues/393#issuecomment-1092754717), this dongle is also a MT7612U based WiFi however also multi-state one (why does COMFAST like to implement this??), someone managed to remove 3 resistors from the dongle and then it becomes single-state. OK since mine are also from COMFAST which I believe they should have similar layout (in fact they are), so I opened up and de-soldered those 3 resistors, now I can confirmed that my 2 COMFAST USB WiFi are single-state! (One RPi without USB modeswitch package installed, the other laptop I disabled the modeswitching and rebooted, both detected MT7612U directly)

morrownr, 2025-09-06:
Hi @Delitants 

Go to the site Main Menu:

https://github.com/morrownr/USB-WiFi

Under menu item 12:

USB Ports can be a source of problems with USB WiFi adapters

Give that document a read and see if it helps.

Menu items 1 and 2 are also good reads in that they can help people pick adapters that are likely to be trouble free.