morrownr/USB-WiFi #406: Two APs with the same SSID

alkisg, 2024-03-20:
My goal is to be able to move around the house without disrupting my phone's Wi-Fi connection.
For simplicity, let's assume that the house has two floors, with one AP on each floor,
and that the APs are connected via wired Ethernet (bridged) to the house router / DHCP server.

If the two APs have different SSIDs, then when I move between floors, my phone's Wi-Fi connection is disrupted (e.g. watching a video).
On the plus side, it's very clear to me which AP I'm connected to.

If the two APs have the same SSID, then if I move e.g. from the first to the second floor, it's possible that...

- Sometimes my phone will remain connected to the first floor AP (=weak signal), with no easy way to force it to switch to the second floor
- I think that sometimes my phone will get confused and not know which AP it's talking to, especially when I'm in a place with equal signal strength to both the APs
- And sometimes it will automatically move to the second floor AP, i.e. everything will work fine

So my question is, what's the best way to configure a home Wi-Fi network with two access points with the same SSID?
And, are there dedicated solutions (e.g. TP-LINK Wi-Fi mesh) that would result in a better user experience than two OpenWRT or Linux based APs?
(in all cases, the APs will have a wired connection to the Internet)
Thank you!

turboproc, 2024-03-20:
Same SSID on all APs (even for 2.4 and 5 GHz) I would say is still considered best practice. Your phone won't become confused by it. Without any help switching from one to the other AP depends on your phone. To control this is a more active way you could consider https://github.com/berlin-open-wireless-lab/DAWN. This is a de-centralized WiFi controller where the APs collaborate to connect each station (your phone) to the optimal AP.

morrownr, 2024-03-21:
> Sometimes my phone will remain connected to the first floor AP (=weak signal), with no easy way to force it to switch to the second floor

> I think that sometimes my phone will get confused and not know which AP it's talking to, especially when I'm in a place with equal signal strength to both the APs

> And sometimes it will automatically move to the second floor AP, i.e. everything will work fine.

I'm not sure that there is a perfect solution. There are standards but then there is a lot of variability in the code that implements those standards. Your experience likely closer to the norm than not. The situation where this is mostly likely come up is with voip. We make overseas calls with WhatsApp to family and friends. The strategy I have used at our last 3 homes has worked well.

- Use the least congested 2.4 GHz band channel for the phones and only for the phones (2.4 GHz has longest range)

- IoT can be on 2.4 GHz as it is stuff that is low bandwidth but I use the second best channel for this (my ap guide and a RasPi can help with this)

- Use a single WiFi router that has excellent range on the 2.4 Ghz band

- Everything that works well on the 5 GHz band is pushed to use 5 GHz. I use DFS channels, 116 currently, and set to 160 MHz width. With the clients that support WiFi 6, 160 MHz width, I see over 1 Gbps speeds.

- The 3 main keys for this to work is Location, Location and Location (of the router/ap)

- Use separate SSID's for 2.4 GHz and 5 GHz (gotta lock things or something will make a bad decision for you)

- Test to make sure your phones are on the best of the 3 available non-overlapping 2.4 GHz channels (1, 6, 11)

Gotta test to find that optimal location for the ap for the phones. The secondary 2.4 GHz ap can be located at an alternate location that works best for what it will serve.

Yes, I have tried other solutions. 80211r works in theory but not so well in reality.

alkisg, 2024-03-21:
Thank you @turboproc and @morrownr, much appreciated!

closed 2024-03-21 by alkisg