morrownr/USB-WiFi #407: I got a fenvi ax1800 but I could not activate hotspot on it

TeddyShmosby, 2024-03-20:
Hey y’all ! 
so i bought a fenvi wifi and for the life of me, I couldn’t get the hotspot working.
Tried every TS I could find bug I’m guessing this adapter doesn’t support hotspot? I tried with a janky ah wifi I had laying around and hotspot worked just fine. 

I’m trying to connect it to my oculus quest 2 for virtual desktop (yes a router is better but I’m just looking to try)

morrownr, 2024-03-20:
Fenvi makes more than one adapter that is ax1800 class. We need to know the chipset. Run and post the results of the following:

$ lsusb

TeddyShmosby, 2024-03-21:
The model is FU-AX1800, I'm not sure how to run $ lsusb (I know nothing of this :( ), as for the chipset which I believe refers to my motherboard? I have a B450m Pro Max

morrownr, 2024-03-21:
I'll back up a little: What is the Linux distro that you are using?

lilkidsuave, 2024-03-22:
> The model is FU-AX1800, I'm not sure how to run $ lsusb (I know nothing of this :( ), as for the chipset which I believe refers to my motherboard? I have a B450m Pro Max

open a terminal
type
lsusb
push enter key
share the results

teslazap, 2024-09-19:
I have the same problem with a FENVI FU-AX1800

lsusb:
Bus 001 Device 002: ID 0e8d:7961 MediaTek Inc. Wireless_Device

lsmod:
mt7921u

The following conf works with wlan0 internal rpi5 card, if I replace wlan0 with wlan1 (the fenvi) clients cannot authenticate anymore. the SSID is visible and the interface up.
auth.log doesn's show any connection attempt, but I'm not sure it's the right file to check

I tried with or without
proto=rsn
pairwise=ccmp

wifi 2.4 or 5:
band=bg
channel=6
or
band=a
channel=36

working with wlan0 /etc/NetworkManager/system-connections/Hotspot.nmconnection
[connection]
id=Hotspot
uuid=6b931918-3da1-4b1c-8e09-6343645b3fde
type=wifi
interface-name=wlan0
timestamp=1725948930

[wifi]
band=a
channel=36
mode=ap
ssid=MYWIFI

[wifi-security]
group=ccmp;
key-mgmt=wpa-psk
pairwise=ccmp;
proto=rsn;
psk=password123456

[ipv4]
address1=172.22.2.1/24
ignore-auto-dns=true
ignore-auto-routes=true
method=manual
never-default=true

[ipv6]
addr-gen-mode=default
method=ignore

I'm assuming it's phy1 because the interface is listed as wlan1, the other one is the internal raspberry 5 card, there is no mac address reference to double check.

iw phy:
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 4
        max scan IEs length: 482 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO

morrownr, 2024-09-19:
@teslazap 

Maybe I can help.

This is going to be messy because I don't use NetworkManager to setup AP's/Hotstops. I use `hostapd`. Yes, I have tried NM but I like the ability to get max performance. You may find some interesting information in my setup or you mind decide to give it a try. My guide shows how to setup a bridged AP/HS:

Here is the site Main Menu:

https://github.com/morrownr/USB-WiFi/blob/main/README.md

Go down to item 8. AP Mode. This is how I set up on a RasPi4B using RasPiOS. I add a setting to NM to tell it to ignore the interface that I want to take into AP Mode. I think the setup would be nearly the same on Debian 12. The guide gives examples details for many WiFi 4 and WiFi 5 adapters but ignore them and look at the next item under 8. AP Mode as it is an example for the chipset in your WiFi 6 adapter. I still call this `Example for WiFi 6...` rc8 but it is pretty darn solid and many people use it.

This way of doing things is different than what you are doing so study my guide and example hostapd.conf file before you jump in. I am here for any questions you may have.

Regards,

@morrownr

teslazap, 2024-09-19:
hey @morrownr thank you for the reply!

It works! I can successfully setup the card in AP mode, I skipped the bridge part because I don't want it configured as a bridge.

The only problem I encountered is the dual configuration file:
Environment=DAEMON_CONF="/etc/hostapd/hostapd-WiFi5.conf /etc/hostapd/hostapd-WiFi4.conf"

It only works if I specify one conf at the time, if I write it as it is it gives me an error:
Configuration file: /etc/hostapd/hostapd-WiFi5.conf /etc/hostapd/hostapd-WiFi4.conf
Could not open configuration file '/etc/hostapd/hostapd-WiFi5.conf /etc/hostapd/hostapd-WiFi4.conf' for reading.
Failed to set up interface with /etc/hostapd/hostapd-WiFi5.conf /etc/hostapd/hostapd-WiFi4.conf
hostapd_init: free iface 0x5555d25da620
Failed to initialize interface

these two lines both work, of course I only get either 2.4 or 5ghz:
Environment=DAEMON_CONF="/etc/hostapd/hostapd-WiFi4.conf"
Environment=DAEMON_CONF="/etc/hostapd/hostapd-WiFi5.conf"

workaround I found from https://forums.raspberrypi.com/viewtopic.php?t=253160
#Environment=DAEMON_CONF="/etc/hostapd/hostapd_5Ghz.conf /etc/hostapd/hostapd_2_4Ghz.conf"
#ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF}
ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B /etc/hostapd/hostapd_5Ghz.conf /etc/hostapd/hostapd_2_4Ghz.conf

it doesn't work either, config files are read but the AP doesn't go up. logfile has a lot of lines now, I think this is the error part
nl80211: Failed to set interface 10 to mode 3: -16 (Device or resource busy)
nl80211 driver initialization failed.

teslazap, 2024-09-19:
could it be this? https://github.com/morrownr/USB-WiFi/issues/107

Bug: (feature request) mt7921u driver does not support 2 interfaces of AP mode on one adapter
Status: open

morrownr, 2024-09-20:
@teslazap 

> The only problem I encountered is the dual configuration file:...

Let me show you my `/etc/systemd/system/hostapd.service` :

```
[Unit]
Description=Access point and authentication server for Wi-Fi and Ethernet
Documentation=man:hostapd(8)
After=network.target
#ConditionFileNotEmpty=/etc/hostapd/hostapd.conf

[Service]
Type=forking
PIDFile=/run/hostapd.pid
Restart=on-failure
RestartSec=3
#Environment=DAEMON_CONF=/etc/hostapd/hostapd.conf
Environment=DAEMON_CONF="/etc/hostapd/hostapd-WiFi6.conf"
#Environment=DAEMON_OPTS="-d -K -f /home/morrow/hostapd.log"
#EnvironmentFile=-/etc/default/hostapd
ExecStartPre=/bin/sleep 6
#ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid $DAEMON_OPTS ${DAEMON_CONF}
ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS $DAEMON_CONF

[Install]
WantedBy=multi-user.target

```

You only need two configuration files if you have two wifi adapters/cards... radios. It is rather common for some people to attach two adapters, one for WiFi 4 and another for WiFi5/6.

My example above is for a single radio. The .conf file you can use is:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.conf

teslazap, 2024-09-20:
> You only need two configuration files if you have two wifi adapters/cards... radios. It is rather common for some people to attach two adapters, one for WiFi 4 and another for WiFi5/6.

What I was trying to do was to have a wifi4 ssid AND a wifi5 SSID on the same card. I didn't notice you had wlan0 in one config and wlan1 in the other.

Meanwhile the wifi4-only conf is running well since yesterday, thanks again.

morrownr, 2024-09-20:
@teslazap 

> What I was trying to do was to have a wifi4 ssid AND a wifi5 SSID on the same card. 

You can't do that. These wifi adapters and cards only have one radio.