morrownr/USB-WiFi #408: 6GHZ Not Working On Realtek 8832cu.

lilkidsuave, 2024-03-21:
I recently bought a fenvi 5400mbps adapter <details><summary>Adapter</summary><p>
https://www.aliexpress.us/item/3256805963591602.html?src=google&src=google&albch=shopping&acnt=548-301-0399&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=en3256805963591602&ds_e_product_merchant_id=109161313&ds_e_product_country=US&ds_e_product_language=en&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=20779689769&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&aff_fcid=0912e6fd199d472c84a44d42d98f6f00-1711001594960-09093-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=0912e6fd199d472c84a44d42d98f6f00-1711001594960-09093-UneMJZVf&terminal_id=d29ec1093adf4cf4adf4a83815d532a4&afSmartRedirect=y&gatewayAdapt=glo2usa
</p></details> and gone through the trouble of installing the drivers and setting the country code. The adapter works with AP mode well in 5GHZ, but gives me the following when I try 6GHZ. 
<details><summary>Error Code</summary><p>

wlx90de80f34d8d: interface state UNINITIALIZED->COUNTRY_UPDATE
The driver does not support the configured HE 6 GHz Max MPDU length
wlx90de80f34d8d: interface state COUNTRY_UPDATE->DISABLED
wlx90de80f34d8d: AP-DISABLED 
wlx90de80f34d8d: Unable to setup interface.
wlx90de80f34d8d: interface state DISABLED->DISABLED
wlx90de80f34d8d: AP-DISABLED 
wlx90de80f34d8d: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlx90de80f34d8d wasn't started
nl80211: deinit ifname=wlx90de80f34d8d disabled_11b_rates=0
</p></details> 
This is my wifi 6E config
<details><summary>hostapd-WiFi6E.conf</summary>
<p>
# /etc/hostapd/hostapd-WiFi6E.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2024-02-28
# Status: rc3, please test and report.
# Note: This hostapd.conf file is for WiFi 6, not WiFi 6e (6 GHz).

# Purpose: hostapd.conf example for WiFi 6 configuration
# - specific to mt7921au chipset based USB WiFi adapters
# - tested on RasPi4B with 64 bit RasPiOS, 2023-12-05
# - see https://github.com/morrownr/USB-WiFi for...
#   -- make sure mt7921 firmware files are installed/updated in RasPiOS
#   -- you need hostap 2.10 or later for WiFi 6 support: $ hostapd -v
# - improvements are welcome
# SSID
ssid=redacted
# PASSPHRASE
wpa_passphrase=redacted
# Band: a = 5GHz & 6GHz (a/n/ac/ax), g = 2Ghz (b/g/n)
hw_mode=a
# Channel availability varies by country
# US:
# 2.4 GHz (1-11)
# 5 GHz ((36(42) and 149(155))
# 5 GHz DFS ((52(58), 100(106), 116(122), 132(138))
# 6 GHz (see op_class below)
# Channel
channel=37
chanlist=37
# The operating class is an indication of the 6 GHz channel size. An
# operating class of 134 indicates a 160 MHz channel. Likewise, 133
# denotes an 80 MHz channel, 132 indicates a 40 MHz channel, and 131
# indicates a channel size of 20 MHz.
# 133 signifies channel width: 80 MHz
#   channels 7, 23, 39, 55, 71, 87, 103, 119, 135, 151, 167, 183, 199, 215
# 134 signifies channel width: 160 MHz
#   channels 15, 47, 79, 111, 143, 175, 207
op_class=134
# Country code
country_code=US
# All environments
country3=0x20
# Outdoor
#country3=0x4f
# Indoor environment only
#country3=0x49
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels if supported (requires ieee80211d=1)
ieee80211h=1

# Bridge interface
[... 1057 lines cut ...]
#he_mu_edca_ac_be_aifsn=8
#he_mu_edca_ac_be_aci=0
#he_mu_edca_ac_be_ecwmin=9
#he_mu_edca_ac_be_ecwmax=10
#he_mu_edca_ac_be_timer=255
#he_mu_edca_ac_bk_aifsn=15
#he_mu_edca_ac_bk_aci=1
#he_mu_edca_ac_bk_ecwmin=9
#he_mu_edca_ac_bk_ecwmax=10
#he_mu_edca_ac_bk_timer=255
#he_mu_edca_ac_vi_ecwmin=5
#he_mu_edca_ac_vi_ecwmax=7
#he_mu_edca_ac_vi_aifsn=5
#he_mu_edca_ac_vi_aci=2
#he_mu_edca_ac_vi_timer=255
#he_mu_edca_ac_vo_aifsn=5
#he_mu_edca_ac_vo_aci=3
#he_mu_edca_ac_vo_ecwmin=5
#he_mu_edca_ac_vo_ecwmax=7
#he_mu_edca_ac_vo_timer=255

# End of hostapd-WiFi6.conf

</p>
</details>

morrownr, 2024-03-21:
Hi @lilkidsuave 

I would like to take a look at the driver you are using. Can you point me to it?

As you probably know, the example hostapd.conf that I have up for WiFi 6 is pretty solid. The one I have been working on for WiFi 6e is not solid and I have been busy on other things. WiFi 6e and 6 GHz gets a lot more complicated so this may take some time... or maybe someone a lot smarter than me has it all figured out and will stop by and help.

@morrownr

lilkidsuave, 2024-03-22:
> Hi @lilkidsuave
>
> I would like to take a look at the driver you are using. Can you point me to it?
>
> As you probably know, the example hostapd.conf that I have up for WiFi 6 is pretty solid. The one I [...]

Its here
https://github.com/lwfinger/rtw8852cu
Iw set country code doesnt work, so this is the way i made it US
https://github.com/lwfinger/rtw8852cu/issues/2#issuecomment-1565658625
According to this forum post, the 8832cu is very similar to the 8852cu and the driver is the same.
https://www.spinics.net/lists/linux-wireless/msg236319.html

lilkidsuave, 2024-03-23:
@lwfinger

lwfinger, 2024-03-23:
As far as I know, setting the country code to US ensures that the 6 GHz band is disabled.

lilkidsuave, 2024-03-23:
> As far as I know, setting the country code to US ensures that the 6 GHz band is disabled.
The iw reg get says otherwise, and also I live in the us, so setting it to somewhere else could be illegal.

lwfinger, 2024-03-23:
Your iw list shows the following:

* 5660 MHz [132] (20.0 dBm)
* 5680 MHz [136] (20.0 dBm)
* 5700 MHz [140] (20.0 dBm)
* 5720 MHz [144] (20.0 dBm)
* 5745 MHz [149] (20.0 dBm)
* 5765 MHz [153] (20.0 dBm)
* 5785 MHz [157] (20.0 dBm)
* 5805 MHz [161] (20.0 dBm)
* 5825 MHz [165] (20.0 dBm)
* 5845 MHz [169] (20.0 dBm)
* 5865 MHz [173] (20.0 dBm)
* 5885 MHz [177] (20.0 dBm)

There are no 6 GHz channels listed.

What kernel are you using? 6 GHz was not added until kernel 5.8.

lilkidsuave, 2024-03-23:
> Your iw list shows the following:
>
> * 5660 MHz [132] (20.0 dBm)
> * 5680 MHz [136] (20.0 dBm)
> * 5700 MHz [140] (20.0 dBm)
> * 5720 MHz [144] (20.0 dBm)
> * 5745 MHz [149] (20.0 dBm)
> * 5765 MHz [153] (20.0 [...]

Scroll down

<details><summary>Details</summary>
<p>
[Code](https://github.com/morrownr/USB-WiFi)
Issues
200
[Pull requests](https://github.com/morrownr/USB-WiFi/pulls)
[Jump to bottom](https://github.com/morrownr/USB-WiFi/issues/408#issue-comment-box)
6GHZ Not Working On Realtek 8832cu. #408
Open
[lilkidsuave](https://github.com/lilkidsuave) opened this issue 3 days ago · 6 comments
Open
[6GHZ Not Working On Realtek 8832cu.](https://github.com/morrownr/USB-WiFi/issues/408#top)
#408
[lilkidsuave](https://github.com/lilkidsuave) opened this issue 3 days ago · 6 comments
Comments
[lilkidsuave](https://github.com/lilkidsuave) commented [3 days ago](https://github.com/morrownr/USB-WiFi/issues/408#issue-2199294120) • 
I recently bought a fenvi 5400mbps adapter

Adapter
and gone through the trouble of installing the drivers and setting the country code. The adapter works with AP mode well in 5GHZ, but gives me the following when I try 6GHZ.
Error Code
This is my wifi 6E config
hostapd-WiFi6E.conf
and here is my sudo iw list
sudo iw list
Wiphy phy1
wiphy index: 1
max # scan SSIDs: 9
max scan IEs length: 2304 bytes
max # sched scan SSIDs: 0
max # match sets: 0
Retry short limit: 7
Retry long limit: 4
Coverage class: 0 (up to 0m)
Device supports AP-side u-APSD.
Supported Ciphers:
* WEP40 (00-0f-ac:1)
* WEP104 (00-0f-ac:5)
* TKIP (00-0f-ac:2)
* CCMP-128 (00-0f-ac:4)
* CMAC (00-0f-ac:6)
* GCMP-128 (00-0f-ac:8)
* GCMP-256 (00-0f-ac:9)
* CCMP-256 (00-0f-ac:10)
* GMAC-128 (00-0f-ac:11)
* GMAC-256 (00-0f-ac:12)
* CMAC-256 (00-0f-ac:13)
Available Antennas: TX 0 RX 0
Supported interface modes:
* IBSS
* managed
* AP
* P2P-client
* P2P-GO
Band 1:
Capabilities: 0x19e3
RX LDPC
HT20/HT40
Static SM Power Save
[... 687 lines cut ...]
interface combinations are not supported
Device supports SAE with AUTHENTICATE command
Device supports scan flush.
Device supports randomizing MAC-addr in scans.
max # scan plans: 1
max scan plan interval: -1
max scan plan iterations: 0
Supported TX frame types:
* IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
* managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
* AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
* AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
* P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
* P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
Supported RX frame types:
* IBSS: 0xd0
* managed: 0x40 0xb0 0xd0
* AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
* AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
* P2P-client: 0x40 0xd0
* P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
Supported extended features:

</p>
</details>

lilkidsuave, 2024-03-23:
Scroll down

<details><summary>Details</summary>
<p>

Frequencies:
* 5955 MHz [1] (20.0 dBm)
* 5975 MHz [5] (20.0 dBm)
* 5995 MHz [9] (20.0 dBm)
* 6015 MHz [13] (20.0 dBm)
* 6035 MHz [17] (20.0 dBm)
* 6055 MHz [21] (20.0 dBm)
* 6075 MHz [25] (20.0 dBm)
* 6095 MHz [29] (20.0 dBm)
* 6115 MHz [33] (20.0 dBm)
* 6135 MHz [37] (20.0 dBm)
* 6155 MHz [41] (20.0 dBm)
* 6175 MHz [45] (20.0 dBm)
* 6195 MHz [49] (20.0 dBm)
* 6215 MHz [53] (20.0 dBm)
* 6235 MHz [57] (20.0 dBm)
* 6255 MHz [61] (20.0 dBm)
* 6275 MHz [65] (20.0 dBm)
* 6295 MHz [69] (20.0 dBm)
* 6315 MHz [73] (20.0 dBm)
* 6335 MHz [77] (20.0 dBm)
* 6355 MHz [81] (20.0 dBm)
* 6375 MHz [85] (20.0 dBm)
* 6395 MHz [89] (20.0 dBm)
* 6415 MHz [93] (20.0 dBm)
* 6435 MHz [97] (20.0 dBm)
* 6455 MHz [101] (20.0 dBm)
* 6475 MHz [105] (20.0 dBm)
* 6495 MHz [109] (20.0 dBm)
* 6515 MHz [113] (20.0 dBm)
* 6535 MHz [117] (20.0 dBm)
* 6555 MHz [121] (20.0 dBm)
* 6575 MHz [125] (20.0 dBm)
* 6595 MHz [129] (20.0 dBm)
* 6615 MHz [133] (20.0 dBm)
* 6635 MHz [137] (20.0 dBm)
* 6655 MHz [141] (20.0 dBm)
* 6675 MHz [145] (20.0 dBm)
* 6695 MHz [149] (20.0 dBm)
* 6715 MHz [153] (20.0 dBm)
* 6735 MHz [157] (20.0 dBm)
* 6755 MHz [161] (20.0 dBm)
* 6775 MHz [165] (20.0 dBm)
* 6795 MHz [169] (20.0 dBm)
* 6815 MHz [173] (20.0 dBm)
* 6835 MHz [177] (20.0 dBm)
* 6855 MHz [181] (20.0 dBm)
* 6875 MHz [185] (20.0 dBm)
* 6895 MHz [189] (20.0 dBm)
* 6915 MHz [193] (20.0 dBm)
* 6935 MHz [197] (20.0 dBm)
* 6955 MHz [201] (20.0 dBm)
* 6975 MHz [205] (20.0 dBm)
* 6995 MHz [209] (20.0 dBm)
* 7015 MHz [213] (20.0 dBm)
* 7035 MHz [217] (20.0 dBm)
* 7055 MHz [221] (20.0 dBm)
* 7075 MHz [225] (20.0 dBm)
* 7095 MHz [229] (20.0 dBm)
* 7115 MHz [233] (20.0 dBm)
* 7135 MHz [237] (disabled)
* 7155 MHz [241] (disabled)
* 7175 MHz [245] (disabled)
* 7195 MHz [249] (disabled)
* 7215 MHz [253] (disabled)

</p>
</details>

lilkidsuave, 2024-03-23:
I am using kernel 6.1
Debian bookworm

lwfinger, 2024-03-23:
I have no explanation as to why your 6 GHz band is blocked other than something is wrong with your regulatory info. Is the package regd installed? That is what works with iw. The regualtory code in the vendor driver is difficult to follow. The standard Makefile sets the country code to XX and the channel plans to 0xFFFF. I have no idea what should be in the channel plans, but the country code should be changed there.

lilkidsuave, 2024-03-24:
Did you even read the first comment.
The issue isn't that its blocked(its not).
The driver does not support the configured HE 6 GHz Max MPDU length
wlx90de80f34d8d: interface state COUNTRY_UPDATE->DISABLED
wlx90de80f34d8d: AP-DISABLED
wlx90de80f34d8d: Unable to setup interface.
wlx90de80f34d8d: interface state DISABLED->DISABLED
wlx90de80f34d8d: AP-DISABLED
wlx90de80f34d8d: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlx90de80f34d8d wasn't started
nl80211: deinit ifname=wlx90de80f34d8d disabled_11b_rates=0

lilkidsuave, 2024-03-24:
Thats the error code

lilkidsuave, 2024-03-24:
before responding, please read the entire thread. Thats twice now in which miscommunication happened

morrownr, 2024-03-25:
I don't have a rtl8852/32cu based adapter so my ability to add anything to the conversation is limited. What I do know is that my experience working on the rtl8852/32bu driver left me feeling that Realtek does not seem to have a good dev working on their WiFi 6 drivers.

lwfinger, 2024-03-25:
That statement is too general to be fair. The open-source Realtek developers work on PCI-based devices, which have good performance. None of that group works on USB devices. The only Realtek people that work on USB are the people that write the vendor drivers, which are mostly developed as a manufacturing aid. As long as the device works, that is sufficient.

morrownr, 2024-03-25:
You are correct. I should have been specific. Realtek's USB WiFi support is terrible.

lilkidsuave, 2024-03-26:
> You are correct. I should have been specific. Realtek's USB WiFi support is terrible.

It exists? A lot of sellers have you install their company drivers for realtek chipsets that aren't natively supported ootb.

ntzb, 2024-04-14:
@lilkidsuave does your adapter support 4addr? can you pleas check?

lilkidsuave, 2024-04-14:
> @lilkidsuave does your adapter support 4addr? can you please check?

How do i check that

ntzb, 2024-04-14:
@lilkidsuave 
not that I'm an expert, but this is what I tried. you have 2 options.
let's assume your device is `wlan5`.
1. work directly on the device (this will break existing connectivity):
`iw dev wlan5 set 4addr on`
if there aren't any errors (such as `Operation not supported (-95)`), then it have worked
finally, disable `4addr`: `iw dev wlan5 set 4addr off`
2. create a vritual device with the property set to `on`:
first you have to take your device down
`sudo ip link set dev wlan5 down`
then you have to create a new virtual interface from it
`sudo iw dev wlan5interface add myNewIface type managed 4addr on`
then if you don't see any errors, bring it up
`sudo ip link set dev myNewIface up`
and finally delete it
`sudo ip link delete myNewIface`

lilkidsuave, 2024-04-15:
I got operation not supported -95

lwfinger, 2024-04-16:
You got an error of -95 because your device does not have a 6 GHz radio, i.e. you are asking for hardware that does not exist on your system.

lilkidsuave, 2024-04-16:
> You got an error of -95 because your device does not have a 6 GHz radio, i.e. you are asking for hardware that does not exist on your system.

It does have a 6ghz radio, did you forget what thread this is.

lwfinger, 2024-04-17:
Apparently.

sj7trunks, 2024-05-08:
First off - much appreciation to @morrownr for putting this together. There's not enough thanks for putting this site together and I've spent the last 4 days going deep into Realtek USB drivers for both the 2.5Gb LAN and Wireless. I always came back to your site to reference findings and try a different card. Ultimately I would like a 160mhz USB dongle for a 6ghz band that I have dedicated right over my head. Everything is at 40mhz across all other bands to keep reliability high for WiFI calling.

Then thank you @lwfinger for your dedication to the Linux kernel for the past 20 years. The open source space tests all of our patience to continue giving for free. I hope you're getting much needed vacation time in.

I have a variant of the card above; [MSI AXE5400](https://www.amazon.com/dp/B0CT3Y72Q5?psc=1&ref=ppx_yo2ov_dt_b_product_details), and experience the same issues. I had to go one step further to add this to @lwfinger  code in os_dep/linux/usb_intf.c from [lwfinger/rtw8852cu](https://github.com/lwfinger/rtw8852cu)

`(USB_DEVICE_AND_INTERFACE_INFO(0x0db0, 0x991d, 0xff, 0xff, 0xff), driver_info = RTL8852C}, `

I had to also do a usb_modswitch and reseat after running the command:

`usb_modeswitch -v 0bda -p 1a2b -V 0bda -P 1a2b -M 555342430860d9a9c0000000800006e0000000000000000000000000000000`

After that I can connect to any 2.4ghz and 5ghz SSID but not any 6ghz SSIDs.

I know this device can do WiFi 6E @ 160mhz in Windows as I can see the full bandwidth after connecting to a 6E band. 

Also I bought a [AX3000M](https://www.amazon.com/dp/B0CQ8F38S3?psc=1&ref=ppx_yo2ov_dt_b_product_details) card that could not do 160mhz but could connect to WiFI 6E band at 80mhz in Linux.

I would surmise that @lwfinger 's driver has WiFi 6 support as I can see IEEE 802.11AX in my iwconfig list; however, I'm not sure if it supports channels and channel widths that the card can provide.

sj7trunks, 2024-05-08:
I have this working with @lwfinger suggestion of hardcoding the following in Makefile of the [rtw8852cu](https://github.com/lwfinger/rtw8852cu) driver

`CONFIG_RTW_COUNTRY_CODE = US
`

[image]

This device kept switching back to 2.4/5ghz but I was able to lock this to my 6ghz radio with the wpa_supplicant having the following:

`freq_list=5935 5955 5975 5995 6015 6035 6055 6075 6095 6115 6135 6155 6175 6195 6215 6235 6255 6275 6295 6315 6335 6355 6375 6395 6415 6435 6455 6475 6495 6515 6535 6555 6575 6595 6615 6635 6655 6675 6695 6715 6735 6755 6775 6795 6815 6835 6855 6875 6895 6915 6935 6955 6975 6995`

morrownr, 2024-05-09:
@sj7trunks 

I don't have an adapter with a rtl8852/32cu chipset yet but may have one in a few months based on the needs of someone I support. That would include me bringing the Realtek out-of-kernel driver up to speed for their needs. I generally also make the drivers public when I do that... so, my point is that I have little add right now for this specific chipset but my experience with the driver for the rtl8852/32au and bu has caused me to ask if Realtek even tests the drivers before shipping so...

...my advice, if you want something that is tri-band and you don't want to pull your hair out getting it to work, take a journey over to the Plug and Play List:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Go down to the section on the mt7921au chipset. Read, then look at the listed adapters. I have the Alfa AXML and the Edup adapter. If you are looking for something that is low cost, the Edup adapter is very good. Fenvi has an adapter that I have listed also. Even Netgear has the A8000 that works well but it is not cheap. All you have to do is plug the adapter in and go to work unless you are using a fairly old distro... needs kernel 5.19 or later.

If you want WiFi 7, I noticed this morning that cards with the new mt7925/27 chips are shipping which means usb adapters will be along soon enough. The driver has been in the kernel since kernel 6.7. The difference between Mediatek and Realtek usb driver support these days is astounding.

@morrownr

lwfinger, 2024-05-09:
The situation is that the driver requires tweaking before 6 GHz will work correctly. I complete solution would be to interrogate regd to get the wireless rules, but I do not have a device, thus testing would be a problem. One could also create a module parameter file containing

options 8852cu rtw_country_code=US (or whatever country is appropriate)

The second problem of switching back to lower frequencies is because of roaming. Unfortunately, there is no switch to turn off roaming. If possible, you could give the 6 GHz band a different SSID.

Okoken, 2024-09-14:
@lilkidsuave 
Please try to add below rows in your hostapd.conf.
`he_6ghz_max_mpdu=0`
`he_6ghz_max_ampdu_len_exp=2`
`he_6ghz_rx_ant_pat=0`
[Here](https://github.com/vanhoefm/hostap-wpa3/blob/master/hostapd/hostapd.conf) is the reference.
Since your error code is "The driver does not support the configured HE 6 GHz Max MPDU length", what you need to do is to configure the MPDU to a lower value for hostapd.
`#he_6ghz_max_mpdu: Maximum MPDU Length of HE 6 GHz band capabilities.`
`# Indicates maximum MPDU length`
`# 0 = 3895 octets`
`# 1 = 7991 octets`
`# 2 = 11454 octets (default)`
`#he_6ghz_max_mpdu=2`

Also would like to point out that it is not an issue of the driver but a question about how to use the hostapd properly. And the issue is not as what you described in the title that "6GHZ Not Working On Realtek 8832cu", but actually 6Ghz is working just you want to setup a 6Ghz access point and failed.
Hope new readers of this threat could understand this point first.

lilkidsuave, 2024-09-14:
> @lilkidsuave Please try to add below rows in your hostapd.conf. `he_6ghz_max_mpdu=0` `he_6ghz_max_ampdu_len_exp=2` `he_6ghz_rx_ant_pat=0` [...]

This is the solution for it being operational, however the 6ghz is slower than what I used to get on 5GHZ. However, things have changed since months ago. i have gotten a used router instead of using the usb, and my nuc that runs the usb is now running arch with lxqt and clear linux kernel.

The updated config that works with 6ghz:<details><summary>hostapd-WiFi6E.conf</summary>
<p>

```
# /etc/hostapd/hostapd-WiFi6E.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2024-02-28
# Status: rc3, please test and report.
# Note: This hostapd.conf file is for WiFi 6, not WiFi 6e (6 GHz).

# Purpose: hostapd.conf example for WiFi 6 configuration
# - specific to mt7921au chipset based USB WiFi adapters
# - tested on RasPi4B with 64 bit RasPiOS, 2023-12-05
# - see https://github.com/morrownr/USB-WiFi for...
#   -- make sure mt7921 firmware files are installed/updated in RasPiOS
#   -- you need hostap 2.10 or later for WiFi 6 support: $ hostapd -v
# - improvements are welcome
# SSID
ssid=redacted
# PASSPHRASE
wpa_passphrase=redacted
# Band: a = 5GHz & 6GHz (a/n/ac/ax), g = 2Ghz (b/g/n)
hw_mode=a
# Channel availability varies by country
# US:
# 2.4 GHz (1-11)
# 5 GHz ((36(42) and 149(155))
# 5 GHz DFS ((52(58), 100(106), 116(122), 132(138))
# 6 GHz (see op_class below)
# Channel
channel=69
chanlist=69
# The operating class is an indication of the 6 GHz channel size. An
# operating class of 134 indicates a 160 MHz channel. Likewise, 133
# denotes an 80 MHz channel, 132 indicates a 40 MHz channel, and 131
# indicates a channel size of 20 MHz.
# 133 signifies channel width: 80 MHz
#   channels 7, 23, 39, 55, 71, 87, 103, 119, 135, 151, 167, 183, 199, 215
# 134 signifies channel width: 160 MHz
#   channels 15, 47, 79, 111, 143, 175, 207
op_class=134
# Country code
country_code=US
# All environments
country3=0x20
# Outdoor
#country3=0x4f
# Indoor environment only
#country3=0x49
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels if supported (requires ieee80211d=1)
ieee80211h=1

# Bridge interface
bridge=br0
# WiFi interface (wlan0 should be changed to your interface name)
interface=wlan0

# Set hostapd driver (nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers)
driver=nl80211

# hostapd event logger configuration
#
# Module bitfield (ORed bitfield of modules that will be logged; -1 = all
# modules):
# bit 0 (1) = IEEE 802.11
# bit 1 (2) = IEEE 802.1X
# bit 2 (4) = RADIUS
# bit 3 (8) = WPA
# bit 4 (16) = driver interface
# bit 6 (64) = MLME
#
# Levels (minimum value for logged events):
#  0 = verbose debugging
#  1 = debugging
#  2 = informational messages
#  3 = notification
#  4 = warning
#
logger_syslog=1
logger_syslog_level=2

# Control interface
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

# Various settings
#beacon_int=100
#dtim_period=2
#multicast_to_unicast=0
#tx_queue_data2_aifs=1
#tx_queue_data2_cwmin=7
#tx_queue_data2_cwmax=15
#tx_queue_data2_burst=3.0
tx_queue_data2_burst=2.0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=1
preamble=1

# Security
# authentication algorithms (3 = both are allowed)
auth_algs=3
# MAC address -based authentication (0 = accept unless in deny list)
macaddr_acl=0
# hidden SSID (1 = hidden)
ignore_broadcast_ssid=0
# Opportunistic Key Caching (1 = enabled)
okc=1
# Enable WPA. (2 is required for WPA2, mixed and WPA3)
wpa=2
# Pairwise cipher
wpa_pairwise=CCMP CCMP-256
rsn_pairwise=CCMP CCMP-256
#
# Activate only one of the following 3 sections
#
# 1. WPA2 Personal
#wpa_key_mgmt=WPA-PSK
#
# 2. WPA2/WPA3 Transitional (mixed)
#wpa_key_mgmt=SAE WPA-PSK
#ieee80211w=1
#sae_require_mfp=1
#
# 3. WPA3 Personal (required for WiFi 6e - 6 GHz)
wpa_key_mgmt=SAE
ieee80211w=2
# WiFi 4 (band 2.4 Ghz)
#ieee80211n=1
#wmm_enabled=1
#
# generic setting - 20 MHz channel width
#ht_capab=[SHORT-GI-20]
#
# generic setting - 40 MHz channel width (if allowed)
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
#
# mt7921au  (HT capabilities 0x9ff)
#ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
# WiFi 5 (5 GHz)
ieee80211ac=1
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=47
#
# generic setting
#vht_capab=[SHORT-GI-80]
#
# mt7921au (VHT capabilities 339071b2)
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
#
# WiFi 6 (bands 2.4 Ghz, 5 GHz and 6 GHz)
ieee80211ax=1
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
he_oper_chwidth=2
#HE center channel (chan + 6)
he_oper_centr_freq_seg0_idx=79
# he_oper_centr_freq_seg1_idx=64
# used by clients to discern the source of interference
# each AP in your area needs to use a different number
# allowed: 1-63
he_mu_edca_qos_info_param_count=0
he_mu_edca_qos_info_q_ack=0
he_mu_edca_qos_info_queue_request=0
he_mu_edca_qos_info_txop_request=0
he_mu_edca_ac_be_aifsn=8
he_mu_edca_ac_be_aci=0
he_mu_edca_ac_vo_ecwmin=9
he_mu_edca_ac_vo_ecwmax=10
#he_mu_
he_6ghz_max_mpdu=0
he_6ghz_max_ampdu_len_exp=2
he_6ghz_rx_ant_pat=0
#he_mu_
# End of hostapd-WiFi6E.conf
```

</p>
</details> 
[image]
[image]

morrownr, 2024-10-20:
@lilkidsuave and anyone else that is interested...

I was recently provided with an adapter that uses the rtl8832cu chip. I was also provided with a very recent copy of the Realtek in kernel driver. I have taken the last couple of days to evaluate the situation. I have decided to try to stand up a repo for this driver. I will keep the repo private until it appears to be ready to open to the public. I could use some testers. I need testers of managed and AP mode. Anyone interested?

I figured out the answer to the original question in this issue.

lilkidsuave, 2024-10-20:
I'm up to test stuff.

morrownr, 2024-10-20:
@lilkidsuave 

> however the 6ghz is slower than what I used to get on 5GHZ.

I still do not have experience with band 4 (6 GHz) but what you are saying is not a surprise as I have listened to a lot of people say the same thing and it makes sense. The reduction in max txpwr allowed as well as the higher frequency will make band 4 capable of limited range. I think band 4 is best for short range high throughput situations. That should have the benefit of taking some of the load off of bands 1 and 2 so if you know wifi, you can really optimize setups to make the most of what is available.

> I'm up to test stuff.

I'll create the repo and do an initial load of clean code. It will be private for now so expect an invitation. I have no set timeline but the things I have looked at so far seem encouraging. I would like to get a band 4 AP going.

morrownr, 2024-10-22:
@lilkidsuave 

Did you get your invitation?

lilkidsuave, 2024-10-22:
Yes

morrownr, 2024-11-01:
@lilkidsuave 

A company sent me an adapter based on the rtl8832cu chip and a copy of the latest source, I have decided to bring up a new repo. The repo is private for now while we test. Would you like to help test?

Thanks for posting the below hostapd.conf. 

@morrownr 

> The updated config that works with 6ghz:
> 
> hostapd-WiFi6E.conf
> 
> ```
> # /etc/hostapd/hostapd-WiFi6E.conf
> # Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
> # 2024-02-28
> # Status: rc3, please test and report.
> # Note: This hostapd.conf file is for WiFi 6, not WiFi 6e (6 GHz).
> 
> # Purpose: hostapd.conf example for WiFi 6 configuration
> # - specific to mt7921au chipset based USB WiFi adapters
> # - tested on RasPi4B with 64 bit RasPiOS, 2023-12-05
> # - see https://github.com/morrownr/USB-WiFi for...
> #   -- make sure mt7921 firmware files are installed/updated in RasPiOS
> #   -- you need hostap 2.10 or later for WiFi 6 support: $ hostapd -v
> # - improvements are welcome
> # SSID
> ssid=redacted
> # PASSPHRASE
> wpa_passphrase=redacted
> # Band: a = 5GHz & 6GHz (a/n/ac/ax), g = 2Ghz (b/g/n)
> hw_mode=a
> # Channel availability varies by country
> # US:
> # 2.4 GHz (1-11)
> # 5 GHz ((36(42) and 149(155))
> # 5 GHz DFS ((52(58), 100(106), 116(122), 132(138))
> # 6 GHz (see op_class below)
> # Channel
> channel=69
> chanlist=69
> # The operating class is an indication of the 6 GHz channel size. An
> # operating class of 134 indicates a 160 MHz channel. Likewise, 133
> # denotes an 80 MHz channel, 132 indicates a 40 MHz channel, and 131
> # indicates a channel size of 20 MHz.
> # 133 signifies channel width: 80 MHz
> #   channels 7, 23, 39, 55, 71, 87, 103, 119, 135, 151, 167, 183, 199, 215
> # 134 signifies channel width: 160 MHz
> #   channels 15, 47, 79, 111, 143, 175, 207
> op_class=134
> # Country code
> country_code=US
> # All environments
> country3=0x20
> # Outdoor
> #country3=0x4f
> # Indoor environment only
> #country3=0x49
> # Advertises the country_code and the set of allowed channels and transmit power levels
> ieee80211d=1
> # Enables support for 5GHz DFS channels if supported (requires ieee80211d=1)
> ieee80211h=1
> 
> # Bridge interface
> bridge=br0
> # WiFi interface (wlan0 should be changed to your interface name)
> interface=wlan0
> 
> # Set hostapd driver (nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers)
> driver=nl80211
> 
> # hostapd event logger configuration
> #
> # Module bitfield (ORed bitfield of modules that will be logged; -1 = all
> # modules):
> # bit 0 (1) = IEEE 802.11
> # bit 1 (2) = IEEE 802.1X
> # bit 2 (4) = RADIUS
> # bit 3 (8) = WPA
> # bit 4 (16) = driver interface
> # bit 6 (64) = MLME
> #
> # Levels (minimum value for logged events):
> #  0 = verbose debugging
> #  1 = debugging
> #  2 = informational messages
> #  3 = notification
> #  4 = warning
> #
> logger_syslog=1
> logger_syslog_level=2
> 
> # Control interface
> ctrl_interface=/var/run/hostapd
> ctrl_interface_group=0
> 
> # Various settings
> #beacon_int=100
> #dtim_period=2
> #multicast_to_unicast=0
> #tx_queue_data2_aifs=1
> #tx_queue_data2_cwmin=7
> #tx_queue_data2_cwmax=15
> #tx_queue_data2_burst=3.0
> tx_queue_data2_burst=2.0
> uapsd_advertisement_enabled=1
> utf8_ssid=1
> multi_ap=0
> bss_load_update_period=60
> chan_util_avg_period=600
> disassoc_low_ack=0
> skip_inactivity_poll=1
> preamble=1
> 
> # Security
> # authentication algorithms (3 = both are allowed)
> auth_algs=3
> # MAC address -based authentication (0 = accept unless in deny list)
> macaddr_acl=0
> # hidden SSID (1 = hidden)
> ignore_broadcast_ssid=0
> # Opportunistic Key Caching (1 = enabled)
> okc=1
> # Enable WPA. (2 is required for WPA2, mixed and WPA3)
> wpa=2
> # Pairwise cipher
> wpa_pairwise=CCMP CCMP-256
> rsn_pairwise=CCMP CCMP-256
> #
> # Activate only one of the following 3 sections
> #
> # 1. WPA2 Personal
> #wpa_key_mgmt=WPA-PSK
> #
> # 2. WPA2/WPA3 Transitional (mixed)
> #wpa_key_mgmt=SAE WPA-PSK
> #ieee80211w=1
> #sae_require_mfp=1
> #
> # 3. WPA3 Personal (required for WiFi 6e - 6 GHz)
> wpa_key_mgmt=SAE
> ieee80211w=2
> # WiFi 4 (band 2.4 Ghz)
> #ieee80211n=1
> #wmm_enabled=1
> #
> # generic setting - 20 MHz channel width
> #ht_capab=[SHORT-GI-20]
> #
> # generic setting - 40 MHz channel width (if allowed)
> #ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
> #
> # mt7921au  (HT capabilities 0x9ff)
> #ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
> # WiFi 5 (5 GHz)
> ieee80211ac=1
> # Channel width (0 = 40 MHz. 1 = 80 Mhz)
> vht_oper_chwidth=1
> # VHT center channel (chan + 6)
> vht_oper_centr_freq_seg0_idx=47
> #
> # generic setting
> #vht_capab=[SHORT-GI-80]
> #
> # mt7921au (VHT capabilities 339071b2)
> vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
> #
> # WiFi 6 (bands 2.4 Ghz, 5 GHz and 6 GHz)
> ieee80211ax=1
> # Channel width (0 = 40 MHz. 1 = 80 Mhz)
> he_oper_chwidth=2
> #HE center channel (chan + 6)
> he_oper_centr_freq_seg0_idx=79
> # he_oper_centr_freq_seg1_idx=64
> # used by clients to discern the source of interference
> # each AP in your area needs to use a different number
> # allowed: 1-63
> he_mu_edca_qos_info_param_count=0
> he_mu_edca_qos_info_q_ack=0
> he_mu_edca_qos_info_queue_request=0
> he_mu_edca_qos_info_txop_request=0
> he_mu_edca_ac_be_aifsn=8
> he_mu_edca_ac_be_aci=0
> he_mu_edca_ac_vo_ecwmin=9
> he_mu_edca_ac_vo_ecwmax=10
> #he_mu_
> he_6ghz_max_mpdu=0
> he_6ghz_max_ampdu_len_exp=2
> he_6ghz_rx_ant_pat=0
> #he_mu_
> # End of hostapd-WiFi6E.conf
> ```
>

lilkidsuave, 2024-11-01:
Did the band 4 config work for you? It was made with the repo before the last two.
Also, whats new between old source and new source.

Been busy with work, haven't done a deep dive into it.

lilkidsuave, 2024-11-01:
I would like to help test.

morrownr, 2024-11-03:
@lilkidsuave 

> Also, whats new between old source and new source.

I have not compared them.

> I would like to help test.

I went to put you in and noticed that you are already in:

https://github.com/morrownr/rtl8852cu-20240510

> Did the band 4 config work for you?

It really helped. Thanks. I am including a hostaps-rtl8852cu.conf file in the repo.