morrownr/USB-WiFi #410: mt7921u pending tx

Simon566, 2024-03-22:
- kernel 6.6.22
-  x86-64 , intel
- Odroid H3
- ALFA USB stick

[Do Mär 21 22:48:23 2024] mt7921u 1-1:1.3: Message 00020003 (seq 6) timeout
[Do Mär 21 22:48:24 2024] mt7921u 1-1:1.3: timed out waiting for pending tx
[Do Mär 21 22:48:24 2024] mt7921u 1-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a

[Do Mär 21 22:48:24 2024] mt7921u 1-1:1.3: WM Firmware Version: ____010000, Build Time: 20240219111038
[Fr Mär 22 06:17:31 2024] mt7921u 1-1:1.3: vendor request req:63 off:d02c failed:-110
[Fr Mär 22 06:17:34 2024] mt7921u 1-1:1.3: vendor request req:63 off:d054 failed:-110
[Fr Mär 22 06:17:37 2024] mt7921u 1-1:1.3: vendor request req:63 off:d058 failed:-110
[Fr Mär 22 06:17:41 2024] mt7921u 1-1:1.3: vendor request req:63 off:53b8 failed:-110

.....alot more of the -110

[Fr Mär 22 06:42:20 2024] task:kworker/u8:3    state:D stack:0     pid:1179348 ppid:2      flags:0x00004000
[Fr Mär 22 06:42:20 2024] Workqueue: phy0 mt792x_mac_work [mt792x_lib]
[Fr Mär 22 06:42:20 2024] Call Trace:
[Fr Mär 22 06:42:20 2024]  <TASK>
[Fr Mär 22 06:42:20 2024]  __schedule+0x3c4/0xb50
[Fr Mär 22 06:42:20 2024]  schedule+0x61/0xe0
[Fr Mär 22 06:42:20 2024]  schedule_preempt_disabled+0x18/0x30
[Fr Mär 22 06:42:20 2024]  __mutex_lock.constprop.0+0x3b4/0x700
[Fr Mär 22 06:42:20 2024]  mt792x_mac_work+0x28/0xb0 [mt792x_lib]
[Fr Mär 22 06:42:20 2024]  process_one_work+0x171/0x340
[Fr Mär 22 06:42:20 2024]  worker_thread+0x27b/0x3a0
[Fr Mär 22 06:42:20 2024]  ? __pfx_worker_thread+0x10/0x10
[Fr Mär 22 06:42:20 2024]  kthread+0xf4/0x130
[Fr Mär 22 06:42:20 2024]  ? __pfx_kthread+0x10/0x10
[Fr Mär 22 06:42:20 2024]  ret_from_fork+0x31/0x50
[Fr Mär 22 06:42:20 2024]  ? __pfx_kthread+0x10/0x10
[Fr Mär 22 06:42:20 2024]  ret_from_fork_asm+0x1b/0x30
[Fr Mär 22 06:42:20 2024]  </TASK>
[Fr Mär 22 06:42:20 2024] Future hung task reports are suppressed, see sysctl kernel.hung_task_warnings
[Fr Mär 22 06:42:22 2024] mt7921u 1-1:1.3: vendor request req:63 off:4230 failed:-110
[Fr Mär 22 06:42:25 2024] mt7921u 1-1:1.3: vendor request req:63 off:4230 failed:-110

Simon566, 2024-03-23:
here is the next:
[Sa Mär 23 16:32:33 2024] mt7921u 1-1:1.3: Message 00020003 (seq 7) timeout
[Sa Mär 23 16:32:34 2024] mt7921u 1-1:1.3: timed out waiting for pending tx
[Sa Mär 23 16:32:34 2024] ------------[ cut here ]------------
[Sa Mär 23 16:32:34 2024] WARNING: CPU: 3 PID: 544288 at kernel/kthread.c:659 kthread_park+0x85/0xa0
[Sa Mär 23 16:32:34 2024] Modules linked in: nf_conntrack_netlink cls_matchall sch_ingress sch_cake act_mirred ifb nft_masq nft_nat nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 wireguard veth libchacha20poly1305 chacha_x86_64 poly1305_x86_64 curve25519_x86_64 libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel ccm nf_tables pppoe pppox crc32c_generic nfnetlink ppp_generic slhc nvme_fabrics msr bridge binfmt_misc snd_sof_pci_intel_icl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation snd_sof_intel_hda_mlink soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp mt7921u snd_sof snd_hda_codec_hdmi mt792x_usb snd_sof_utils snd_soc_hdac_hda mt7921_common snd_hda_codec_realtek snd_hda_codec_generic snd_hda_ext_core ledtrig_audio snd_soc_acpi_intel_match mt792x_lib x86_pkg_temp_thermal snd_soc_acpi intel_powerclamp snd_soc_core mt76_connac_lib kvm_intel snd_compress soundwire_bus mt76_usb i915 snd_hda_intel btusb mt76 kvm nls_ascii btrtl nls_cp437 btintel snd_intel_dspcfg
[Sa Mär 23 16:32:34 2024]  vfat btbcm mac80211 snd_intel_sdw_acpi btmtk irqbypass bluetooth fat snd_hda_codec ghash_clmulni_intel 88XXau(OE) sha256_ssse3 libarc4 sha1_ssse3 snd_hda_core drm_buddy drm_display_helper snd_hwdep intel_rapl_msr mei_pxp cfg80211 processor_thermal_device_pci_legacy mei_hdcp aesni_intel sha3_generic jitterentropy_rng cec processor_thermal_device mei_me crypto_simd snd_pcm sha512_ssse3 sha512_generic processor_thermal_rfim cryptd ctr drbg iTCO_wdt snd_timer ansi_cprng processor_thermal_mbox ftdi_sio rc_core ecdh_generic mei intel_pmc_bxt processor_thermal_rapl ecc ttm iTCO_vendor_support usbserial snd nfsd intel_rapl_common rfkill watchdog pcspkr drm_kms_helper wmi_bmof crc16 intel_cstate ee1004 soundcore int340x_thermal_zone i2c_algo_bit intel_soc_dts_iosf intel_pmc_core evdev acpi_pad joydev sg button acpi_tad auth_rpcgss nfs_acl lockd it87(OE) hwmon_vid grace emc2103 coretemp sunrpc 8021q drm garp stp mrp llc dm_mod fuse configfs loop efi_pstore efivarfs ip_tables x_tables autofs4 btrfs xor raid6_pq
[Sa Mär 23 16:32:34 2024]  libcrc32c hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid sd_mod nvme ahci libahci nvme_core libata xhci_pci xhci_hcd t10_pi r8169 crc64_rocksoft crc64 crc_t10dif realtek i2c_i801 scsi_mod crct10dif_generic mdio_devres usbcore i2c_smbus libphy crct10dif_pclmul crct10dif_common crc32_pclmul crc32c_intel intel_lpss_pci scsi_common intel_lpss usb_common idma64 video wmi
[Sa Mär 23 16:32:34 2024] CPU: 3 PID: 544288 Comm: kworker/u8:11 Tainted: G     U     OE      6.6.22 #5
[Sa Mär 23 16:32:34 2024] Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
[Sa Mär 23 16:32:34 2024] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
[Sa Mär 23 16:32:34 2024] RIP: 0010:kthread_park+0x85/0xa0
[Sa Mär 23 16:32:34 2024] Code: 00 48 85 c0 74 2d 31 c0 5b 5d c3 cc cc cc cc 0f 0b 48 8b ab 40 0a 00 00 a8 04 74 ac 0f 0b b8 da ff ff ff 5b 5d c3 cc cc cc cc <0f> 0b b8 f0 ff ff ff eb d5 0f 0b eb cf 66 66 2e 0f 1f 84 00 00 00
[Sa Mär 23 16:32:34 2024] RSP: 0018:ffffb24346c5bd58 EFLAGS: 00010202
[Sa Mär 23 16:32:34 2024] RAX: 0000000000000004 RBX: ffff9c39cc2cb100 RCX: 0000000000000100
[Sa Mär 23 16:32:34 2024] RDX: 0000000080000000 RSI: 0000000000000283 RDI: ffff9c39cc2cb100
[Sa Mär 23 16:32:34 2024] RBP: ffff9c3a16415200 R08: ffffffffc051e368 R09: 0000000000000000
[Sa Mär 23 16:32:34 2024] R10: 0000000000000001 R11: 0000000000000000 R12: ffff9c39cd8c20a8
[Sa Mär 23 16:32:34 2024] R13: ffff9c3a168743a8 R14: ffff9c39cd8c20a8 R15: 0000000000000100
[Sa Mär 23 16:32:34 2024] FS:  0000000000000000(0000) GS:ffff9c3d2ff80000(0000) knlGS:0000000000000000
[Sa Mär 23 16:32:34 2024] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Sa Mär 23 16:32:34 2024] CR2: 00007f8e9c554000 CR3: 0000000442dae000 CR4: 0000000000350ee0
[Sa Mär 23 16:32:34 2024] Call Trace:
[Sa Mär 23 16:32:34 2024]  <TASK>
[Sa Mär 23 16:32:34 2024]  ? kthread_park+0x85/0xa0
[Sa Mär 23 16:32:34 2024]  ? __warn+0x81/0x130
[Sa Mär 23 16:32:34 2024]  ? kthread_park+0x85/0xa0
[Sa Mär 23 16:32:34 2024]  ? report_bug+0x191/0x1c0
[Sa Mär 23 16:32:34 2024]  ? handle_bug+0x41/0x70
[Sa Mär 23 16:32:34 2024]  ? exc_invalid_op+0x17/0x70
[Sa Mär 23 16:32:34 2024]  ? asm_exc_invalid_op+0x1a/0x20
[Sa Mär 23 16:32:34 2024]  ? kthread_park+0x85/0xa0
[Sa Mär 23 16:32:34 2024]  mt76u_stop_tx+0x216/0x2f0 [mt76_usb]
[Sa Mär 23 16:32:34 2024]  ? __pfx_autoremove_wake_function+0x10/0x10
[Sa Mär 23 16:32:34 2024]  mt7921u_mac_reset+0x72/0x1b0 [mt7921u]
[Sa Mär 23 16:32:34 2024]  mt7921_mac_reset_work+0x97/0x180 [mt7921_common]
[Sa Mär 23 16:32:34 2024]  ? __schedule+0x3cc/0xb50
[Sa Mär 23 16:32:34 2024]  process_one_work+0x171/0x340
[Sa Mär 23 16:32:34 2024]  worker_thread+0x27b/0x3a0
[Sa Mär 23 16:32:34 2024]  ? __pfx_worker_thread+0x10/0x10
[Sa Mär 23 16:32:34 2024]  kthread+0xf4/0x130
[Sa Mär 23 16:32:34 2024]  ? __pfx_kthread+0x10/0x10
[Sa Mär 23 16:32:34 2024]  ret_from_fork+0x31/0x50
[Sa Mär 23 16:32:34 2024]  ? __pfx_kthread+0x10/0x10
[Sa Mär 23 16:32:34 2024]  ret_from_fork_asm+0x1b/0x30
[Sa Mär 23 16:32:34 2024]  </TASK>
[Sa Mär 23 16:32:34 2024] ---[ end trace 0000000000000000 ]---

fhteagle, 2024-04-01:
Also seeing that in journalctl as well:

Mar 28 22:22:52 ***hostname here*** kernel: mt7921u 1-1.4:1.3: timed out waiting for pending tx

Seems to be associated with the random failure to forward packets issue I have seen over the past few weeks on mt7921u and mt7921k. Fortunately, this is mitigated by restarting hostapd most of the time, but very rarely I will need to full reboot the machine the mt7921 is attached to, too. I will try to come up with concrete steps to replicate the issue.

morrownr, 2024-04-01:
@Simon566 @fhteagle 

After seeing the first message I have been keeping an eye open. It is clear that @fhteagle is in AP mode but it is not clear what mode @Simon566 is in. That would be good info to know.

I have seen some errors like this when inadequate or failing power supplies were involved.

@fhteagle 

I have a mt7921au based adapter in AP Mode on a Pi4B. It is running RasPiOS with kernel 6.6. I'm not seeing this. Can you go into more detail about:

> Seems to be associated with the random failure to forward packets issue I have seen over the past few weeks ...

Simon566, 2024-04-03:
Hi ,

its the same here. Im running AP mode on X86-64 hardware, running Debian Bookworm with latest kernel 6.6.23 and hostapd from git mainline. Having a single client doesnt seem to trigger the fault, even when using things like openspeedtest or similar sites. But when i put the AP into my real network with 10 clients , it fails within a day mostly.

Using a rtl8812au AP runs for weeks at least before having some issues, also on USB hardware.

regards,
Simon

fhteagle, 2024-04-03:
I cannot give a cause of the issue, nor how to reproduce reliably. Best pattern I can describe now is three general classes of failure, that may or may not be related:

1. hostapd itself soft bricks, which will lead to even root privilege user cannot get any information from hostapd_cli tool
2. hostapd process still live, STAs correctly listed in ```hostapd_cli list_sta```, pings get through to STAs, but no other packet type or port makes it to that STA from any other device on LAN
3. hostapd process still live, STAs correctly listed in ```hostapd_cli list_sta```, pings and other port traffic gets through to STAs from IPs on LAN, but no pings or other packet whatsoever are passed between two STAs connected to the same AP

All 3 failure modes can be cleared with ~95%+ success rate by a simple systemd restart of the hostapd service for that adapter. ~5% of the time I have to reboot to clear the failure mode.

The pending tx journal message is most strongly associated with failure modes 2 and 3, but not perfectly correlated. I can have a mode 2 or mode 3 failure with or without the journal message. With the journal message, there will always be either mode 2 or mode 3 failure. 

Again, I'm running bleeding edge hostapd-git , so entirely possible this is the root cause. Behavior seen on arch x86_64 and aarch64 (raspberry pi 3b and 4b) hosts, using MT7921K (pci) and MT7921AU (usb) cards. Adapters are bridged, either via systemd/networkd config, or the bridge = option in hostapd.conf, with one or more ethernet NIC and other adapter device on each host.

whitslack, 2024-08-26:
I've spent the past several hours trying to get to the bottom of this issue. The `timed out waiting for pending tx` message comes from the chip reset code path, which only gets invoked following a command timeout. Even the preceding `Message 00020003 (seq __) timeout` is only a symptom of the underlying fault condition.

I added some debug log messages to trace the flow of command messages to/from the mt7921u chip. Setting up the chip takes several dozen command messages, as one might expect:
```
<6>usb 2-7: new high-speed USB device number 4 using ehci-pci
<6>usb 2-7: reset high-speed USB device number 4 using ehci-pci
<6>mt7921u 2-7:1.3: sent cmd=00000010 seq=1
<6>mt7921u 2-7:1.3: rcvd seq=1
<6>mt7921u 2-7:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240716150944a
<6>
<6>mt7921u 2-7:1.3: sent cmd=00000005 seq=2
<6>mt7921u 2-7:1.3: rcvd seq=2
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=3
⋮
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=10
<6>mt7921u 2-7:1.3: sent cmd=00000007 seq=11
<6>mt7921u 2-7:1.3: rcvd seq=11
<6>mt7921u 2-7:1.3: sent cmd=00000010 seq=12
<6>mt7921u 2-7:1.3: rcvd seq=12
<6>mt7921u 2-7:1.3: WM Firmware Version: ____010000, Build Time: 20240716151027
<6>mt7921u 2-7:1.3: sent cmd=00000001 seq=13
<6>mt7921u 2-7:1.3: rcvd seq=13
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=14
⋮
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=12
<6>mt7921u 2-7:1.3: sent cmd=00000001 seq=13
<6>mt7921u 2-7:1.3: rcvd seq=13
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=14
⋮
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=5
<6>mt7921u 2-7:1.3: sent cmd=00000001 seq=6
<6>mt7921u 2-7:1.3: rcvd seq=6
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=7
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=8
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=9
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=10
<6>mt7921u 2-7:1.3: sent cmd=00000001 seq=11
<6>mt7921u 2-7:1.3: rcvd seq=11
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=12
⋮
<6>mt7921u 2-7:1.3: sent cmd=000000ee seq=9
<6>mt7921u 2-7:1.3: sent cmd=00000002 seq=10
<6>mt7921u 2-7:1.3: rcvd seq=10
<6>mt7921u 2-7:1.3: sent cmd=0004008a seq=11
<6>mt7921u 2-7:1.3: rcvd seq=11
<6>mt7921u 2-7:1.3: sent cmd=000400c5 seq=12
<6>mt7921u 2-7:1.3: sent cmd=000021ed seq=13
<6>mt7921u 2-7:1.3: rcvd seq=13
<6>mt7921u 2-7:1.3: sent cmd=00003eed seq=14
<6>mt7921u 2-7:1.3: rcvd seq=14
<6>mt7921u 2-7:1.3: sent cmd=0004000f seq=15
<6>mt7921u 2-7:1.3: sent cmd=0004005d seq=1
⋮
<6>mt7921u 2-7:1.3: sent cmd=0004005d seq=7
<6>mt7921u 2-7:1.3: sent cmd=000400ca seq=8
<6>mt7921u 2-7:1.3: sent cmd=000046ed seq=9
<6>mt7921u 2-7:1.3: rcvd seq=9
<6>mt7921u 2-7:1.3: sent cmd=0004000f seq=10
<6>mt7921u 2-7:1.3: sent cmd=00004eed seq=11
<6>mt7921u 2-7:1.3: rcvd seq=11
<6>mt7921u 2-7:1.3: sent cmd=0004005d seq=12
⋮
<6>mt7921u 2-7:1.3: sent cmd=0004005d seq=3
<6>mt7921u 2-7:1.3: sent cmd=00020001 seq=4
<6>mt7921u 2-7:1.3: rcvd seq=4
<6>mt7921u 2-7:1.3: sent cmd=00020002 seq=5
<6>mt7921u 2-7:1.3: rcvd seq=5
<6>mt7921u 2-7:1.3: sent cmd=0004000a seq=6
<6>mt7921u 2-7:1.3: sent cmd=0004001d seq=7
<6>mt7921u 2-7:1.3: sent cmd=0004000a seq=8
[... 107 lines cut ...]
```
<6>mt7921u 2-7:1.3: sent cmd=00020003 seq=9
<3>mt7921u 2-7:1.3: Message 00020003 (seq 9) timeout
<3>mt7921u 2-7:1.3: timed out waiting for pending tx
```
In the case that, immediately following the hang, the driver next sends `cmd=00020003`, a command that _is_ expected to elicit a reply from the MCU, then that command gets sent successfully, but the driver times out while waiting to receive the reply to the command. This causes it immediately to attempt to reset the chip, which ends up in the `timed out waiting for pending tx` code path as above.

----

For whatever it is worth, it does appear that the chip can be resurrected from a crash without rebooting the system by unbinding the `mt7921u` driver from the USB device and rebinding it. This action causes a reset of the USB interface, which does seem to allow the chip to recover.
```
# echo 2-7:1.3 >/sys/bus/usb/drivers/mt7921u/unbind
# echo 2-7:1.3 >/sys/bus/usb/drivers/mt7921u/bind
```
(Adjust the `2-7:1.3` to the actual USB bus address of your mt7921u device.)
```
<6>usb 2-7: reset high-speed USB device number 4 using ehci-pci
<6>mt7921u 2-7:1.3: sent cmd=00000010 seq=1
<6>mt7921u 2-7:1.3: rcvd seq=1
<6>mt7921u 2-7:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240716150944a
```

----

I have to put this (and myself) to bed for now, but I may eventually circle back around to attempt to trace the (much more complex) data path to try to find what conditions cause the chip to go out to lunch and never return.

gdluca, 2024-08-27:
Hi @whitslack 

great troubleshooting. I learned a few things thanks to your comment but don't have a fix myself. I recently bought EDUP EP-AX1672 that use the same driver and I am getting the same pending tx errors which make the adapter unusable till hostapd restart. I wanted to add that until yesterday, I used it in AP mode on my desktop x86_64 using the latest archlinux kernel 6.10.6 without any issues. Then I tried my new raspberrypi 5 and that's when pending errors started. I suspect that it is something to do with the usb 3.0 root hub that can't handle both ports at high utilization( the other port is used for a usb<-> 1gbps ethernet). So this morning, I moved the ethernet adapter to one of the usb 2.0 ports and so far it seems fine ( speedtests, heavy transfer files).

whitslack, 2024-08-27:
@gdluca: The problem is definitely not caused by USB 3. I can readily reproduce the hang with an mt7921u adapter plugged into a system that does not have any USB 3 controllers.

gdluca, 2024-08-27:
@whitslack I see. Then mine will probably get errors again at some point.. Will try to do more tests in the weekend.

gdluca, 2024-09-19:
Since I cannot swap and test the adapter from the desktop (which works fine - full story https://github.com/morrownr/USB-WiFi/issues/410 )back to the rpi5, I just purchased a second EDUP EP-AX1672. Who knows, maybe with a different hw it works fine too eheh. 

CC: @morrownr

morrownr, 2024-09-19:
@gdluca 

The search continues.

gdluca, 2024-09-25:
And here it is my  second EDUP. Currently I am testing like this:
・SSID:labcrash - NEW EDUP on rpi5 with usb power hub
  - ax/40Mhz/channel 36

・SSD:nXXXXX - OLD EDUP on desktop
  - ax/40Mhz/channel 44
[image]

I was able to reproduce it again with the new EDUP afater ~322seconds of iperf3. Good at least now there is no impact on my remote job and I can test at the same time hehe.

@morrownr

whitslack, 2024-09-25:
@gdluca: When your AP hangs, see if you can dump the queue states from the `mt76` driver:
```
# cat /sys/kernel/debug/ieee80211/*/mt76/*-queues
```
(This assumes you have `debugfs` mounted at `/sys/kernel/debug`.)

I spent close to 12 hours this past weekend going through the mt76 USB code with a fine-toothed comb. I did find some places where the code makes some assumptions that don't strike me as obviously valid, and I also found an area where I believe there may be a race condition caused by the omission of needed memory barriers. It would be helpful to see a dump of those debug files taken while the driver is hung, as it could confirm my suspicion that the driver may be failing to wake up its kernel worker task despite a queue not being empty. I will also be dumping those files the next time my AP hangs, but you might beat me to it if I get lucky.

gdluca, 2024-09-25:
@whitslack 

For below queue results, I redid an iperf3, waited and the AP hung. There are no dmesg messages yet and I have not restarted hostapd. It is still in this situation so if you want more logs, I should be able to get them.

```
[root@alarm ~]# cat /sys/kernel/debug/ieee80211/*/mt76/*-queues
     queue | hw-queued |      head |      tail |
         0 |       128 |        44 |        44 |
         1 |       128 |         0 |         0 |
     queue | hw-queued |      head |      tail |
         0 |         0 |       132 |       132 |
         1 |         0 |         0 |         0 |
         2 |       224 |       150 |       182 |
         3 |         0 |         0 |         0 |
         4 |         0 |       132 |       132 |
```

gdluca, 2024-09-25:
@whitslack !!

I waited I think 20 min (I am away from the pc but more or less) and the AP is back automatically!! Now I have dmesg errors

```
[Wed Sep 25 03:17:56 2024] mt7921u 2-1.2:1.0: Message 00020003 (seq 2) timeout
[Wed Sep 25 03:17:57 2024] mt7921u 2-1.2:1.0: timed out waiting for pending tx
[Wed Sep 25 03:17:57 2024] ------------[ cut here ]------------
[Wed Sep 25 03:17:57 2024] WARNING: CPU: 0 PID: 45567 at kernel/kthread.c:659 kthread_park+0xc0/0xe0
[Wed Sep 25 03:17:57 2024] Modules linked in: aes_ce_ccm mt7921u mt7921_common mt792x_lib mt76_connac_lib mt792x_usb mt76_usb mt76 mac80211 libarc4 xt_connmark xt_mark xt_comment xt_addrtype nft_compat wireguard libchacha20poly1305 chacha_neon poly1305_neon ip6_udp_tunnel udp_tunnel libcurve25519_generic libchacha veth vhost_vsock vmw_vsock_virtio_transport_common vhost vhost_iotlb vsock binfmt_misc sch_cake ifb sch_htb cls_u32 cls_flow cls_fw act_mirred sch_ingress 8021q garp bridge stp llc nft_nat nft_masq nft_chain_nat nf_nat nft_ct nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables rpivid_hevc(C) brcmfmac_wcc pisp_be brcmfmac v4l2_mem2mem hci_uart videobuf2_dma_contig btbcm brcmutil videobuf2_memops videobuf2_v4l2 bluetooth aes_ce_blk cfg80211 aes_ce_cipher videodev ghash_ce gf128mul sha2_ce videobuf2_common ecdh_generic sha256_arm64 ecc rfkill sha1_ce libaes ax88179_178a raspberrypi_hwmon rp1_adc raspberrypi_gpiomem i2c_brcmstb mc nvmem_rmem pwm_fan uio_pdrv_genirq uio sch_fq_codel tcp_bbr fuse dm_mod nfnetlink ip_tables
[Wed Sep 25 03:17:57 2024]  x_tables ipv6 vc4 snd_soc_hdmi_codec snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd drm_display_helper drm_dma_helper drm_kms_helper v3d drm_shmem_helper cec gpu_sched drm drm_panel_orientation_quirks backlight spidev spi_bcm2835 gpio_keys
[Wed Sep 25 03:17:57 2024] CPU: 0 PID: 45567 Comm: kworker/u8:2 Tainted: G         C         6.6.51-3-rpi-16k #1
[Wed Sep 25 03:17:57 2024] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[Wed Sep 25 03:17:57 2024] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
[Wed Sep 25 03:17:57 2024] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[Wed Sep 25 03:17:57 2024] pc : kthread_park+0xc0/0xe0
[Wed Sep 25 03:17:57 2024] lr : mt76u_stop_tx+0x25c/0x360 [mt76_usb]
[Wed Sep 25 03:17:57 2024] sp : ffffc000829a3c40
[Wed Sep 25 03:17:57 2024] x29: ffffc000829a3c40 x28: 0000000000000000 x27: ffff8001160e9d30
[Wed Sep 25 03:17:57 2024] x26: ffff8001160e2078 x25: ffff8001160e4820 x24: ffff8001160e2078
[Wed Sep 25 03:17:57 2024] x23: ffff8001160e2058 x22: ffffc000829a3c88 x21: ffff8001160e7420
[Wed Sep 25 03:17:57 2024] x20: ffff8001015cde00 x19: ffff800108c85b80 x18: 0000000000000000
[Wed Sep 25 03:17:57 2024] x17: 0000000000000000 x16: ffffd000840be418 x15: 0000000000000000
[Wed Sep 25 03:17:57 2024] x14: 0000000000000000 x13: 000000000000035d x12: 0000000000000001
[Wed Sep 25 03:17:57 2024] x11: ffffd000858308c8 x10: 0000000000001a20 x9 : ffffd00014a2e96c
[Wed Sep 25 03:17:57 2024] x8 : ffff800100372d00 x7 : 0000000000000000 x6 : ffffd000859b2650
[Wed Sep 25 03:17:57 2024] x5 : ffffd000859b2650 x4 : ffffd000859b2650 x3 : 0000000000002800
[Wed Sep 25 03:17:57 2024] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000004
[Wed Sep 25 03:17:57 2024] Call trace:
[Wed Sep 25 03:17:57 2024]  kthread_park+0xc0/0xe0
[Wed Sep 25 03:17:57 2024]  mt76u_stop_tx+0x25c/0x360 [mt76_usb]
[Wed Sep 25 03:17:57 2024]  mt7921u_mac_reset+0x8c/0x290 [mt7921u]
[Wed Sep 25 03:17:57 2024]  mt7921_mac_reset_work+0xc0/0x220 [mt7921_common]
[Wed Sep 25 03:17:57 2024]  process_one_work+0x144/0x3c0
[Wed Sep 25 03:17:57 2024]  worker_thread+0x328/0x460
[Wed Sep 25 03:17:57 2024]  kthread+0x11c/0x128
[Wed Sep 25 03:17:57 2024]  ret_from_fork+0x10/0x20
[Wed Sep 25 03:17:57 2024] ---[ end trace 0000000000000000 ]---
[Wed Sep 25 03:17:57 2024] mt7921u 2-1.2:1.0: HW/SW Version: 0x8a108a10, Build Time: 20240826150948a

[Wed Sep 25 03:17:57 2024] mt7921u 2-1.2:1.0: WM Firmware Version: ____010000, Build Time: 20240826151030
```

whitslack, 2024-09-25:
@gdluca: Thanks. I'll have to dig back into the driver code to remember what those numbers mean. No promises on when I'll get around to that. 

Can you also check your kernel log to see if the mt76 driver has complained about any URB mismatches? (It would appear before the message about pending tx.)

gdluca, 2024-09-25:
@whitslack 

before the pending tx, I have

```
[Wed Sep 25 02:23:02 2024] mt7921u 2-1.2:1.0 wlu1u2: entered allmulticast mode
[Wed Sep 25 02:23:02 2024] mt7921u 2-1.2:1.0 wlu1u2: entered promiscuous mode
[Wed Sep 25 02:23:02 2024] br0: port 5(wlu1u2) entered blocking state
[Wed Sep 25 02:23:02 2024] br0: port 5(wlu1u2) entered forwarding state
```

which is when I plugged in the adapter.

whitslack, 2024-09-25:
Alright. There won't necessarily be a mismatched URB. I suspect that occurs due to an entirely separate problem in the code: namely, that the driver assumes that URB completion callbacks will execute serially despite that there is no guarantee (to my knowledge) that the kernel executes URB completion callbacks in a single tasklet thread. The mt76 driver takes care to synchronize access to its queue structures with a spinlock, but there is no guarantee about the order in which racing tasklet threads would enter the spinlock, so there is no guarantee that the driver will observe its submitted URBs to have completed in the same order as it submitted them. That the mt76 driver would implement its own URB queue from scratch strikes me as odd since the kernel already offers a standard `struct usb_anchor` and associated utility functions for keeping track of a collection of URBs.

Independent of the URB mismatch issue, I suspect that the mt76 driver code that manages the worker tasks that process the queues of completed URBs may have inadequate memory synchronization to ensure that a queue will always be serviced. It looks like there may be the possibility, with very low probability (but in multithreaded coding, anything that _can_ happen _will_ happen eventually), that a completed URB may be enqueued after the worker has already decided to go to sleep but before it has actually done so (and so the URB completion callback's attempt to awaken the worker does nothing since the worker is not yet asleep). The mt76 code does try to guard against this, but I believe it may be making invalid assumptions about CPU cache coherency. Again, it strikes me as odd that the mt76 driver attempts to roll its own queue worker implementation from scratch since the kernel already provides standard mechanisms for work queues. It kind of smells like whoever wrote the mt76 driver is a fairly decent programmer but one who isn't too experienced with coding within the Linux kernel and one who might not fully appreciate the need for memory barriers when writing thread synchronization code (which is why the kernel provides standard, battle-hardened implementations, because getting it right is tricky).

whitslack, 2024-09-25:
I've been perusing the mt76 code some more…

> ```
>      queue | hw-queued |      head |      tail |
>          0 |       128 |        44 |        44 |
>          1 |       128 |         0 |         0 |
> ```

These^ are the USB receive queues. They each show 128 URBs enqueued, which means that the driver has handed off 128 buffers to the USB subsystem to be filled with packets coming from the USB device, and none of those URBs have been completed, or in other words, the driver has received no packets from the hardware that the driver has not yet processed. This is good.

> ```
>      queue | hw-queued |      head |      tail |
>          0 |         0 |       132 |       132 |
>          1 |         0 |         0 |         0 |
>          2 |       224 |       150 |       182 [...]

These^ are the USB transmit queues. The driver doesn't actually appear to use queue 3, but it uses queue 0 for transmitting `BK` (background) packets, queue 1 for transmitting `BE` (best effort) packets, queue 2 for transmitting `VI` (video) packets, and queue 4 for transmitting `VO` (voice) packets. Your dump indicates that the driver has enqueued 224 `BE` packets for transmission but has not yet called back to the networking subsystem to confirm transmission of those packets. Unfortunately, the `debugfs` dump does not show the `first` field of those queues, which is used on transmit queues to indicate which packet in the queue is the first that still needs to be sent out over USB, so we can't discern whether the breakdown occurred because the driver failed to kick packets out to the hardware or because the driver failed to process the completions of those packets after the hardware confirmed transmitting them.

While examining the transmit code paths, I found something that looks very bad and like it could potentially be causing nasty issues. In [`drivers/net/wireless/mediatek/mt76/usb.c`](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/mediatek/mt76/usb.c), there are three functions that manipulate the transmit queues: `mt76u_status_worker`, `mt76u_tx_queue_skb`, and `mt76u_tx_kick`. The latter two are called from multiple sites in [`drivers/net/wireless/mediatek/mt76/tx.c`](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/mediatek/mt76/tx.c), always while holding the queue's spinlock, so no problem there. However, the first, `mt76u_status_worker`, is called from `__mt76_worker_fn` in [`drivers/net/wireless/mediatek/mt76/util.c`](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/mediatek/mt76/util.c), which is a generic threaded worker implementation that does not know anything about the packet queues or their spinlocks. And the bombshell is that `mt76u_status_worker` does not hold the queues' spinlocks while manipulating the queues!!! This is catastrophic since the status worker runs in its own thread, likely concurrent with user threads that would be enqueuing packets into the transmit queues. Multiple threads performing concurrent, non-atomic, read-modify-write cycles on the structural pointers of the transmit queues. That's a recipe for disaster. I'm going to see if locking the spinlocks while accessing the queues in `mt76u_status_worker` fixes the hangs.

gdluca, 2024-09-25:
I can only appreciate your knowledge and dedication, @whitslack . I cannot read that code so thanks!

whitslack, 2024-09-26:
Another interesting observation: the MT7921AUN hardware advertises two USB inbound (device-to-host) bulk endpoints, ostensibly with the first being for normal network traffic and the second being for command responses from the microcontroller (MCU), yet the firmware actually only ever sends command responses on the first inbound bulk endpoint, mixed in with all the network traffic. The mt7921u driver is prepared to receive command responses on the second endpoint, but it never actually receives any, as all command responses from the MCU arrive on the first endpoint. I conjecture that command responses could be getting dropped if the first endpoint is overly busy with network traffic. This would manifest in exactly the scenario we are observing, where the driver times out while waiting to receive a response to a command that it has sent to the MCU. I posit that the MCU did receive the command, process it, and try to respond, but the response could not be enqueued to the first endpoint due to all the other traffic clogging it up.

It may be telling that the mt7915 driver did recently receive a [patch](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=3688c18b65aeb2a1f2fde108400afbab129a8cc1) for the mt7915 to retry sending commands to the MCU whenever a wait to receive a response times out. I am going to try porting that new functionality over to the mt7921u driver to see if it can work around the hang we're experiencing. Note, it will only be a workaround; the proper fix would be for the firmware to respond to commands on the second endpoint so its responses can't be drowned out by network traffic.

whitslack, 2024-09-26:
> It may be telling that the mt7915 driver did recently receive a [patch](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=3688c18b65aeb2a1f2fde108400afbab129a8cc1) to retry sending commands to the MCU whenever a wait to receive a response times out. I am going to try porting that new functionality over to the mt7921u driver to see if it can work around the hang we're experiencing.

Nope. The retry logic does activate, but by the time that happens, the chip is already out to lunch. In fact, an Ookla Speedtest on my phone can hang the AP without there being any command messages pending (and so none to timeout and hit the retry logic). After a while, HostAPd will time out the station association and try to remove the station from the AP, and that action does cause a command message to be sent to the MCU. Curiously, the USB subsystem confirms completion of that outbound URB, but then the mt7921u driver times out while awaiting a response from the MCU, at which point the new retry logic kicks in and attempts to retransmit the command, but then the USB subsystem times out the retry URB. That's a different kind of timeout: in the former case, the command URB completes, and the mt7921u driver subsequently times out while awaiting a command response message from the MCU, but in the latter case, the command URB does not complete, and so the mt7921u driver does not progress to the point of awaiting a command response. It is as though the the MCU accepts the first command for processing (and acknowledges receipt of that command) but then hangs while processing the command and so is unavailable to acknowledge any subsequent commands. Note, it is **not** the command that is causing the problem, as the chip is already buggered before the command is even sent to it. The MCU's failure to respond to the command is a _symptom_ of the hung state that the chip is already in.

> Unfortunately, the `debugfs` dump does not show the `first` field of those queues

I augmented the `xmit-queues` file in `debugfs` to show the `first` field of the transmit queues. During a hang, there are URBs in the queue (`hw-queued` > 0) and `first` == `head`, which indicates that the mt7921u driver did submit those URBs to the USB subsystem. However, the `done` flag is not set on any of the URBs in the queue, meaning that the USB subsystem has not yet indicated completion of those URBs. Everything points to a problem in the firmware, as it appears that the Linux kernel driver is doing what it should.

Thinking that the high bandwidth of the speed test might be overrunning a queue or buffer in the chip and crashing the firmware, I tried reducing the number of transmit URBs that can be outstanding concurrently (from 224 to 96), but that had no effect on the hangs. I guess maybe I'll try sending out all traffic on the first outbound endpoint rather than distributing it among the endpoints according to WMM service class. Now I am really just stabbing at the air.

gdluca, 2024-09-26:
@whitslack 

thanks for trying!

> Thinking that the high bandwidth of the speed test might be overrunning a queue or buffer in the chip and crashing the firmware

Indeed and I still have to make it crash with an iperf test limited to 100Mbps... however! Once (and only once so who knows) I made it hang with just opening a video file from a sshfs mounted (basically streaming via ssh) and the video was not that high quality to go over 100Mbps, not event 50Mbps.

gdluca, 2024-09-26:
@whitslack 

fantastic news! I will try it in the weekend.

gdluca, 2024-09-26:
It didn't work for me. Hope it stays stable for you so I know it is something on my side.

`Linux alarm 6.6.51-3-rpi-16k-mt76usbfix #1 SMP PREEMPT Thu Sep 26 12:18:55 UTC 2024 aarch64 GNU/Linux`

```
[root@alarm ~]# cat /sys/kernel/debug/ieee80211/*/mt76/*-queues
     queue | hw-queued |      head |      tail |
         0 |       128 |       113 |       113 |
         1 |       128 |         0 |         0 |
     queue | hw-queued |      head |      tail |
         0 |         0 |        39 |        39 |
         1 |         0 |         0 |         0 |
         2 |       224 |       104 |       136 |
         3 |         0 |         0 |         0 |
         4 |         0 |        39 |        39 |
```

whitslack, 2024-09-26:
> It didn't work for me.

You might also need the status worker spinlock fix, and you could try putting _all_ outgoing traffic on the same endpoint.

```diff
--- a/drivers/net/wireless/mediatek/mt76/usb.c
+++ b/drivers/net/wireless/mediatek/mt76/usb.c
@@ -772,6 +772,7 @@
 		if (!q)
 			continue;
 
+		spin_lock(&q->lock);
 		while (q->queued > 0) {
 			if (!q->entry[q->tail].done)
 				break;
@@ -779,11 +780,13 @@
 			entry = q->entry[q->tail];
 			q->entry[q->tail].done = false;
 
+			spin_unlock(&q->lock);
 			mt76_queue_tx_complete(dev, q, &entry);
+			spin_lock(&q->lock);
 		}
-
 		if (!q->queued)
 			wake_up(&dev->tx_wait);
+		spin_unlock(&q->lock);
 
 		mt76_worker_schedule(&dev->tx_worker);
 	}
@@ -923,7 +926,7 @@
 		return lmac_queue_map[ac];
 	}
 
-	return mt76_ac_to_hwq(ac);
+	return mt76_chip(dev) == 0x7961 ? 0 : mt76_ac_to_hwq(ac);
 }
 
 static int mt76u_alloc_tx(struct mt76_dev *dev)
```

> Hope it stays stable for you so I know it is something on my side.

I suspect I'm not entirely avoiding the problem in the firmware, merely reducing the probability of hitting it. My wild guess is that they have a reëntrancy problem in their code, such that processing of a packet arriving on one endpoint gets interrupted by the arrival of a packet on a second endpoint, and the processing of the second packet corrupts the internal state of the processing of the first packet. They would need to disable interrupts on _all_ endpoints while processing a packet received on any endpoint. (This is an absolutely wild guess made with no knowledge of their architecture, so I could be way off base.)

You might also like to enhance your `debugfs` output with additional details to help elucidate this issue:
```diff
--- a/drivers/net/wireless/mediatek/mt76/debugfs.c
+++ b/drivers/net/wireless/mediatek/mt76/debugfs.c
@@ -56,15 +56,20 @@
 	struct mt76_dev *dev = dev_get_drvdata(s->private);
 	int i;
 
-	seq_puts(s, "     queue | hw-queued |      head |      tail |\n");
+	seq_puts(s, "     queue | hw-idx | hw-queued |     first |      head |      tail |\n");
 	for (i = 0; i < ARRAY_SIZE(dev->phy.q_tx); i++) {
 		struct mt76_queue *q = dev->phy.q_tx[i];
 
 		if (!q)
 			continue;
 
-		seq_printf(s, " %9d | %9d | %9d | %9d |\n",
-			   i, q->queued, q->head, q->tail);
+		seq_printf(s, " %9d | %6d | %9d | %9d | %9d | %9d | ",
+			   i, q->hw_idx, q->queued, q->first, q->head, q->tail);
+		if (q->queued)
+			for (int j = q->tail; j != q->head; j = (j + 1) % q->ndesc)
+				seq_printf(s, "%c", q->entry[j].done ? 'd' : '.');
+		seq_printf(s, "%s%s\n",
+			   q->stopped ? " stopped" : "", q->blocked ? " blocked" : "");
 	}
 
 	return 0;
```

gdluca, 2024-09-27:
This is going to take some time on my side. I have to change ISP as soon as the end of this month  so I won't have easy home internet access until mid October. I still have internet using a 4g mobile router borrowed from the company but I need it for work. Will keep an eye on this and try somehow to use internet.

gdluca, 2024-09-27:
brief update:the adapter on my desktop just hung and I was just browsing. Wow, it finally hung on the desktop too. It took a few weeks (meanwhile I updated a few times the 6.10.y kernel so there is also that).

gdluca, 2024-10-08:
@whitslack It's been a while.. how are things for you?

(I am still without home internet and they postponed to the 25th Oct.. lets see)

Simon566, 2024-10-08:
Hi @whitslack , @gdluca .

 i have been following the patch and the idea , shall i give it a shot ? im on 6.10.13 currently. gdluca , do you have any dmesg or so from your crash ? is whitslacks changes available as patch ?

regards,
Simon

whitslack, 2024-10-08:
> @whitslack It's been a while.. how are things for you?

@gdluca: I have not needed to reboot or restart my AP since I posted "I think I have a workaround." I wish it had worked for you as well as it works for me. Did you try with the spinlock fix I posted? 

@morrownr: It appears that the problem is exacerbated when multiple hardware transmit queues are in use simultaneously. Maybe your iperf3 testing exercises only one queue (probably the `BE` one). If there is a way in iperf3 to set the QoS/DiffServ class of the transmitted packets, then you might try testing with two simultaneous streams of differing classes. When the bug was triggering for me, I had packets stuck in both the `BE` and `BK` queues. That _should_ translate, respectively, from the default DiffServ class (i.e., DSCP 0) and the Lower-Effort (LE) class defined by [IETF RFC 8622](https://datatracker.ietf.org/doc/html/rfc8622) to use DSCP 1, but I'll admit I don't know how Linux _actually_ maps from DiffServ classes to WMM queues. You might also try the older (obsolete) Class Selector 1 (CS1), which has a DSCP of 8, as that was sometimes (inconsistently) used to denote traffic of lower-than-default priority. You can monitor the `xmit-queues` in debugfs to see whether you are exercising multiple transmit queues simultaneously.

gdluca, 2024-10-08:
@whitslack No, I have not tried yet. Good news for you :)

gdluca, 2024-10-08:
@Simon566 No, I don't have a patch file for the workaround as I manually edited the files. I cannot test till the end of this month. If you have time, do some tests :)

whitslack, 2024-10-08:
@Simon566: I posted my changes in patch format above. Just hit the floating Copy button in the code snippets, paste into a text editor, and save to a file. Then you can feed that file to the `patch` program with your kernel sources root as the working directory.

morrownr, 2024-10-08:
@whitslack 

> It appears that the problem is exacerbated when multiple hardware transmit queues are in use simultaneously.

Thank you for this information. Now I have a rabbit hole to follow. Just this last weekend I started reconfiguring my AP setup to get some things figured out about 6 GHz as more people are trying to do 6 Ghz APs and it works in more countries as the regdb stuff is updated. I have WiFi 6 instructions and example hostapd.conf in good shape but WiFi 6e is a challenge as it is far more complicated. I need to stay on that for a few days but have queued your note for work. Thanks again.

Simon566, 2024-10-09:
i instantly get a tx urb failed . i do always delete the bluetooth firmware to prevent it from loading:

[Mi Okt  9 08:40:52 2024] usb 2-2: reset SuperSpeed USB device number 3 using xhci_hcd
[Mi Okt  9 08:40:52 2024] bluetooth hci0: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin failed with error -2
[Mi Okt  9 08:40:52 2024] Bluetooth: hci0: Failed to load firmware file (-2)
[Mi Okt  9 08:40:52 2024] Bluetooth: hci0: Failed to set up firmware (-2)
[Mi Okt  9 08:40:52 2024] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[Mi Okt  9 08:40:53 2024] usbcore: registered new interface driver mt7921u
[Mi Okt  9 08:40:53 2024] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240826150948a

[Mi Okt  9 08:40:53 2024] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20240826151030
[Mi Okt  9 08:40:54 2024] mt7921u 2-2:1.3 wlx00c0cab38118: renamed from wlan0
[Mi Okt  9 08:41:12 2024] br-iot: port 2(wlx00c0cab38118) entered blocking state
[Mi Okt  9 08:41:12 2024] br-iot: port 2(wlx00c0cab38118) entered disabled state
[Mi Okt  9 08:41:12 2024] mt7921u 2-2:1.3 wlx00c0cab38118: entered allmulticast mode
[Mi Okt  9 08:41:12 2024] mt7921u 2-2:1.3 wlx00c0cab38118: entered promiscuous mode
[Mi Okt  9 08:41:13 2024] br-iot: port 2(wlx00c0cab38118) entered blocking state
[Mi Okt  9 08:41:13 2024] br-iot: port 2(wlx00c0cab38118) entered forwarding state
[Mi Okt  9 08:42:19 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:19 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:19 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:19 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:20 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:20 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:21 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:42 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:42 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:42 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Mi Okt  9 08:42:42 2024] mt7921u 2-2:1.3: tx urb failed: -71

Simon566, 2024-10-09:
kernel is 6.10.13 with your patch, i needed to modify slightly so it applied

whitslack, 2024-10-09:
> mt7921u 2-2:1.3: tx urb failed: -71

@Simon566: 71 is `EPROTO`. That's something outside of what my changes could effect. I haven't worked with kernels past the 6.6 LTS series. In particular I saw somewhere that 6.10 really screwed up mt76. Any chance you can try 6.6.54?

Simon566, 2024-10-10:
- so far the patch on 6.10.13 looks good , one day runtime .... lets wait
- im just switching to 6.11.3, will report after some reasonable runtime

whitslack, 2024-10-10:
> * so far the patch on 6.10.13 looks good

@Simon566: I'm glad you got it working. For our benefit: how did you resolve the `EPROTO` errors et al.?

Simon566, 2024-10-11:
- i recognized that i rebooted the system but maybe usb hasnt got a powercycle, i replugged the adapter and did some rmmod/modprobe commands to reload the firmware
- i turned off XHCI handoff in the bios
- used your patch, without the last few lines here
- -	return mt76_ac_to_hwq(ac);
+	return mt76_chip(dev) == 0x7961 ? 0 : mt76_ac_to_hwq(ac);

so far its running for 3 days . i switched to 6.11.3 now and also there it seems to be running. lets wait for some days

whitslack, 2024-10-11:
> without the last few lines here

@Simon566: Interesting. That change is the one that actually avoids the problem for me. I have to wonder if you need the patch at all. Maybe all the other things you did actually solved your original problem.

If not, then I guess the spinlock fix really is relevant and important. 🤔

Simon566, 2024-10-11:
to me it looks like these last lines arent really applicable in 6.11.3, because the logic is identical . can you check ?

Simon566, 2024-10-11:
static void
mt76u_ac_to_hwq(struct mt76_dev *dev, struct mt76_queue *q, u8 qid)
{
	u8 ac = qid < IEEE80211_NUM_ACS ? qid : IEEE80211_AC_BE;

	switch (mt76_chip(dev)) {
	case 0x7663: {
		static const u8 lmac_queue_map[] = {
			/* ac to lmac mapping */
			[IEEE80211_AC_BK] = 0,
			[IEEE80211_AC_BE] = 1,
			[IEEE80211_AC_VI] = 2,
			[IEEE80211_AC_VO] = 4,
		};

		q->hw_idx = lmac_queue_map[ac];
		q->ep = q->hw_idx + 1;
		break;
	}
	case 0x7961:
	case 0x7925:
		q->hw_idx = mt76_ac_to_hwq(ac);
		q->ep = qid == MT_TXQ_PSD ? MT_EP_OUT_HCCA : q->hw_idx + 1;
		break;
	default:
		q->hw_idx = mt76_ac_to_hwq(ac);
		q->ep = q->hw_idx + 1;
		break;
	}
}

whitslack, 2024-10-11:
> to me it looks like these last lines arent really applicable in 6.11.3, because the logic is identical . can you check ?

@Simon566: The upstream logic was changed significantly in [`375c5eebbfa2b354f689e971c4d001798b8da4c7`](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=375c5eebbfa2b354f689e971c4d001798b8da4c7) and [`5304bf3b9915ba35aff3c9064fe37165d5471542`](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=5304bf3b9915ba35aff3c9064fe37165d5471542), and it is not equivalent. Unfortunately, the MediaTek devs never seem to care much for writing descriptive commit log messages, so we are given no visibility into the problem they were trying to solve. It is possible they have fixed the issue.

Simon566, 2024-10-12:
sadly i have to report a common crash with 6.11.3:

Okt 11 20:23:58 odroid-h3 root[181896]: Firewall fertig
Okt 11 20:46:40 odroid-h3 kernel: mt7921u 2-2:1.3: tx urb failed: -71
Okt 11 20:46:40 odroid-h3 kernel: mt7921u 2-2:1.3: tx urb failed: -71
Okt 11 20:46:40 odroid-h3 kernel: mt7921u 2-2:1.3: tx urb failed: -71
Okt 11 20:46:40 odroid-h3 kernel: mt7921u 2-2:1.3: tx urb failed: -71
Okt 11 20:46:40 odroid-h3 kernel: mt7921u 2-2:1.3: tx urb failed: -71
Okt 11 20:46:40 odroid-h3 kernel: mt7921u 2-2:1.3: tx urb failed: -71
Okt 11 20:46:40 odroid-h3 kernel: mt7921u 2-2:1.3: tx urb failed: -71
Okt 11 20:46:40 odroid-h3 kernel: mt7921u 2-2:1.3: tx urb failed: -71

i cant get the full dmesg , buffer is full
[77171.619682] mt7921u 2-2:1.3: Failed to send patch
[77171.821653] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240826150948a

[77171.829829] mt7921u 2-2:1.3: Failed to send patch

[77172.026104] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240826150948a

[77172.034377] mt7921u 2-2:1.3: Failed to send patch
[77172.042506] mt7921u 2-2:1.3: chip reset failed
[77175.061392] mt7921u 2-2:1.3: Message 00020001 (seq 5) timeout
[77178.261345] mt7921u 2-2:1.3: Message 00020002 (seq 8) timeout
[77181.457907] mt7921u 2-2:1.3: Message 00020003 (seq 9) timeout
[77184.657817] mt7921u 2-2:1.3: Message 00020002 (seq 10) timeout
[77184.850702] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240826150948a

Simon566, 2024-10-12:
after rmmod / modprobe i get this !! NULL ptr

[Sa Okt 12 06:15:25 2024] mt7921u 2-2:1.3: chip reset failed
[Sa Okt 12 08:26:15 2024] r8169 0000:01:00.0 enp1s0: Link is Down
[Sa Okt 12 08:26:17 2024] r8169 0000:01:00.0 enp1s0: Link is Up - 100Mbps/Full - flow control off
[Sa Okt 12 08:26:17 2024] br-lan: port 2(enp1s0) entered blocking state
[Sa Okt 12 08:26:17 2024] br-lan: port 2(enp1s0) entered forwarding state
[Sa Okt 12 08:26:18 2024] r8169 0000:01:00.0 enp1s0: Link is Down
[Sa Okt 12 08:26:18 2024] br-lan: port 2(enp1s0) entered disabled state
[Sa Okt 12 08:26:20 2024] r8169 0000:01:00.0 enp1s0: Link is Up - 100Mbps/Full - flow control off
[Sa Okt 12 08:26:20 2024] br-lan: port 2(enp1s0) entered blocking state
[Sa Okt 12 08:26:20 2024] br-lan: port 2(enp1s0) entered forwarding state
[Sa Okt 12 08:26:30 2024] r8169 0000:01:00.0 enp1s0: Link is Down
[Sa Okt 12 08:26:30 2024] br-lan: port 2(enp1s0) entered disabled state
[Sa Okt 12 08:26:33 2024] r8169 0000:01:00.0 enp1s0: Link is Up - 2.5Gbps/Full - flow control off
[Sa Okt 12 08:26:33 2024] br-lan: port 2(enp1s0) entered blocking state
[Sa Okt 12 08:26:33 2024] br-lan: port 2(enp1s0) entered forwarding state
[Sa Okt 12 08:27:04 2024] r8169 0000:01:00.0 enp1s0: Link is Down
[Sa Okt 12 08:27:04 2024] br-lan: port 2(enp1s0) entered disabled state
[Sa Okt 12 08:27:07 2024] r8169 0000:01:00.0 enp1s0: Link is Up - 2.5Gbps/Full - flow control off
[Sa Okt 12 08:27:07 2024] br-lan: port 2(enp1s0) entered blocking state
[Sa Okt 12 08:27:07 2024] br-lan: port 2(enp1s0) entered forwarding state
[Sa Okt 12 08:27:16 2024] r8169 0000:01:00.0 enp1s0: Link is Down
[Sa Okt 12 08:27:16 2024] br-lan: port 2(enp1s0) entered disabled state
[Sa Okt 12 08:27:19 2024] r8169 0000:01:00.0 enp1s0: Link is Up - 2.5Gbps/Full - flow control off
[Sa Okt 12 08:27:19 2024] br-lan: port 2(enp1s0) entered blocking state
[Sa Okt 12 08:27:19 2024] br-lan: port 2(enp1s0) entered forwarding state
[Sa Okt 12 08:31:09 2024] mt7921u 2-2:1.3 wlx00c0cab38118: left allmulticast mode
[Sa Okt 12 08:31:09 2024] mt7921u 2-2:1.3 wlx00c0cab38118: left promiscuous mode
[Sa Okt 12 08:31:09 2024] br-iot: port 2(wlx00c0cab38118) entered disabled state
[Sa Okt 12 08:31:09 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Okt 12 08:31:19 2024] usbcore: deregistering interface driver mt7921u
[Sa Okt 12 08:31:19 2024] usb 2-2: Disable of device-initiated U1 failed.
[Sa Okt 12 08:31:27 2024] BUG: kernel NULL pointer dereference, address: 00000000000000c8
[Sa Okt 12 08:31:27 2024] #PF: supervisor write access in kernel mode
[Sa Okt 12 08:31:27 2024] #PF: error_code(0x0002) - not-present page
[Sa Okt 12 08:31:27 2024] PGD 0 P4D 0
[Sa Okt 12 08:31:27 2024] Oops: Oops: 0002 [#1] PREEMPT SMP NOPTI
[Sa Okt 12 08:31:27 2024] CPU: 2 UID: 0 PID: 312365 Comm: modprobe Tainted: G     U  W  OE      6.11.0-dirty #1
[Sa Okt 12 08:31:27 2024] Tainted: [U]=USER, [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[Sa Okt 12 08:31:27 2024] Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
[Sa Okt 12 08:31:27 2024] RIP: 0010:btusb_disconnect+0x2c/0x170 [btusb]
[Sa Okt 12 08:31:27 2024] Code: 1e fa 0f 1f 44 00 00 41 54 55 48 89 fd 53 48 8b 9f c8 00 00 00 0f 1f 44 00 00 48 85 db 0f 84 ac 00 00 00 48 8b 43 10 4c 8b 23 <48> c7 80 c8 00 00 00 00 00 00 00 48 8b 43 18 48 85 c0 74 0b 48 c7
[Sa Okt 12 08:31:27 2024] RSP: 0018:ffffc06ec52e79c0 EFLAGS: 00010286
[Sa Okt 12 08:31:27 2024] RAX: 0000000000000000 RBX: ffff9e5cc2ff5428 RCX: 0000000000000000
[Sa Okt 12 08:31:27 2024] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff9e5e74c3d800
[Sa Okt 12 08:31:27 2024] RBP: ffff9e5e74c3d800 R08: 0000000000000000 R09: 0000000000000000
[Sa Okt 12 08:31:27 2024] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[Sa Okt 12 08:31:27 2024] R13: ffffffffc10bc278 R14: 0000000000000080 R15: ffff9e5e74c3d850
[Sa Okt 12 08:31:27 2024] FS:  00007fb1c34e1040(0000) GS:ffff9e602ff00000(0000) knlGS:0000000000000000
[Sa Okt 12 08:31:27 2024] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Sa Okt 12 08:31:27 2024] CR2: 00000000000000c8 CR3: 000000012921e000 CR4: 0000000000350ef0
[Sa Okt 12 08:31:28 2024] Call Trace:
[Sa Okt 12 08:31:28 2024]  <TASK>
[Sa Okt 12 08:31:28 2024]  ? __die+0x23/0x80
[Sa Okt 12 08:31:28 2024]  ? page_fault_oops+0x173/0x5d0
[Sa Okt 12 08:31:28 2024]  ? __kmalloc_cache_node_noprof+0x151/0x310
[Sa Okt 12 08:31:28 2024]  ? exc_page_fault+0x76/0x1b0
[Sa Okt 12 08:31:28 2024]  ? asm_exc_page_fault+0x26/0x30
[Sa Okt 12 08:31:28 2024]  ? btusb_disconnect+0x2c/0x170 [btusb]
[Sa Okt 12 08:31:28 2024]  usb_unbind_interface+0x95/0x2d0 [usbcore]
[Sa Okt 12 08:31:28 2024]  ? preempt_count_add+0x53/0xc0
[Sa Okt 12 08:31:28 2024]  device_release_driver_internal+0x1a7/0x210
[Sa Okt 12 08:31:28 2024]  usb_forced_unbind_intf+0x5d/0xc0 [usbcore]
[Sa Okt 12 08:31:28 2024]  usb_reset_device+0xec/0x2a0 [usbcore]
[Sa Okt 12 08:31:28 2024]  mt7921u_probe+0xab/0x240 [mt7921u]
[Sa Okt 12 08:31:28 2024]  usb_probe_interface+0xe5/0x310 [usbcore]
[Sa Okt 12 08:31:28 2024]  really_probe+0xd3/0x390
[Sa Okt 12 08:31:28 2024]  ? __pfx___driver_attach+0x10/0x10
[Sa Okt 12 08:31:28 2024]  __driver_probe_device+0x78/0x160
[... 21 lines cut ...]
[Sa Okt 12 08:31:28 2024] Code: 08 89 e8 5b 5d c3 66 2e 0f 1f 84 00 00 00 00 00 90 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d b7 06 0d 00 f7 d8 64 89 01 48
[... cut ...]
[Sa Okt 12 08:31:28 2024] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[Sa Okt 12 08:31:28 2024] R13: ffffffffc10bc278 R14: 0000000000000080 R15: ffff9e5e74c3d850
[Sa Okt 12 08:31:28 2024] FS:  00007fb1c34e1040(0000) GS:ffff9e602ff00000(0000) knlGS:0000000000000000
[Sa Okt 12 08:31:28 2024] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Sa Okt 12 08:31:28 2024] CR2: 00000000000000c8 CR3: 000000012921e000 CR4: 0000000000350ef0
[Sa Okt 12 08:31:28 2024] note: modprobe[312365] exited with irqs disabled

Simon566, 2024-10-12:
im not loading bluetooth firmware and recognize this here:

[Sa Okt 12 08:33:15 2024] ---[ end trace 0000000000000000 ]---
[Sa Okt 12 08:33:15 2024] RIP: 0010:btusb_disconnect+0x2c/0x170 [btusb]
[Sa Okt 12 08:33:15 2024] Code: 1e fa 0f 1f 44 00 00 41 54 55 48 89 fd 53 48 8b 9f c8 00 00 00 0f 1f 44 00 00 48 85 db 0f 84 ac 00 00 00 48 8b 43 10 4c 8b 23 <48> c7 80 c8 00 00 00 00 00 00 00 48 8b 43 18 48 85 c0 74 0b 48 c7
[Sa Okt 12 08:33:15 2024] RSP: 0018:ffffc06ec52e79c0 EFLAGS: 00010286
[Sa Okt 12 08:33:15 2024] RAX: 0000000000000000 RBX: ffff9e5cc2ff5428 RCX: 00000000000000

Simon566, 2024-10-12:
and the next crash after the reboot, bluetooth firmware is present now , i just replugged the device because it wasnt reacting

[Sa Okt 12 09:41:25 2024] br-lan: port 2(enp1s0) entered forwarding state
[Sa Okt 12 09:42:44 2024] usbcore: deregistering interface driver mt7921u
[Sa Okt 12 09:42:49 2024] usbcore: registered new interface driver mt7921u
[Sa Okt 12 09:43:38 2024] usbcore: deregistering interface driver mt7921u
[Sa Okt 12 09:44:25 2024] usb 2-2: new SuperSpeed USB device number 7 using xhci_hcd
[Sa Okt 12 09:44:25 2024] usb 2-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[Sa Okt 12 09:44:25 2024] usb 2-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[Sa Okt 12 09:44:25 2024] usb 2-2: Product: Wireless_Device
[Sa Okt 12 09:44:25 2024] usb 2-2: Manufacturer: MediaTek Inc.
[Sa Okt 12 09:44:25 2024] usb 2-2: SerialNumber: 000000000
[Sa Okt 12 09:44:30 2024] Bluetooth: hci0: Failed to write uhw reg(-110)
[Sa Okt 12 09:44:30 2024] usb 2-2: Disable of device-initiated U1 failed.
[Sa Okt 12 09:44:30 2024] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20240826151221
[Sa Okt 12 09:44:35 2024] usb 2-2: Disable of device-initiated U2 failed.
[Sa Okt 12 09:44:35 2024] Bluetooth: hci0: urb 00000000535beec5 failed to resubmit (2)
[Sa Okt 12 09:44:40 2024] Bluetooth: hci0: Execution of wmt command timed out
[Sa Okt 12 09:44:40 2024] Bluetooth: hci0: Failed to send wmt patch dwnld (-110)
[Sa Okt 12 09:44:40 2024] Bluetooth: hci0: Failed to set up firmware (-110)
[Sa Okt 12 09:44:40 2024] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[Sa Okt 12 09:44:40 2024] BUG: kernel NULL pointer dereference, address: 00000000000000c8
[Sa Okt 12 09:44:40 2024] #PF: supervisor write access in kernel mode
[Sa Okt 12 09:44:40 2024] #PF: error_code(0x0002) - not-present page
[Sa Okt 12 09:44:40 2024] PGD 0 P4D 0
[Sa Okt 12 09:44:40 2024] Oops: Oops: 0002 [#1] PREEMPT SMP NOPTI
[Sa Okt 12 09:44:40 2024] CPU: 3 UID: 0 PID: 37771 Comm: (udev-worker) Tainted: G     U  W  OE      6.11.0-dirty #1
[Sa Okt 12 09:44:40 2024] Tainted: [U]=USER, [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[Sa Okt 12 09:44:40 2024] Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
[Sa Okt 12 09:44:40 2024] RIP: 0010:btusb_disconnect+0x2c/0x170 [btusb]
[Sa Okt 12 09:44:40 2024] Code: 1e fa 0f 1f 44 00 00 41 54 55 48 89 fd 53 48 8b 9f c8 00 00 00 0f 1f 44 00 00 48 85 db 0f 84 ac 00 00 00 48 8b 43 10 4c 8b 23 <48> c7 80 c8 00 00 00 00 00 00 00 48 8b 43 18 48 85 c0 74 0b 48 c7
[Sa Okt 12 09:44:40 2024] RSP: 0018:ffffaa9b016c3880 EFLAGS: 00010286
[Sa Okt 12 09:44:40 2024] RAX: 0000000000000000 RBX: ffff88a51bca5828 RCX: 0000000000000000
[Sa Okt 12 09:44:40 2024] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff88a4aec1bc00
[Sa Okt 12 09:44:40 2024] RBP: ffff88a4aec1bc00 R08: 0000000000000000 R09: 0000000000000000
[Sa Okt 12 09:44:40 2024] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[Sa Okt 12 09:44:40 2024] R13: ffffffffc114b278 R14: 0000000000000080 R15: ffff88a4aec1bc50
[Sa Okt 12 09:44:40 2024] FS:  00007ff3f193a8c0(0000) GS:ffff88a7eff80000(0000) knlGS:0000000000000000
[Sa Okt 12 09:44:40 2024] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Sa Okt 12 09:44:40 2024] CR2: 00000000000000c8 CR3: 0000000195bd0000 CR4: 0000000000350ef0
[Sa Okt 12 09:44:40 2024] Call Trace:
[Sa Okt 12 09:44:40 2024]  <TASK>
[Sa Okt 12 09:44:40 2024]  ? __die+0x23/0x80
[Sa Okt 12 09:44:40 2024]  ? page_fault_oops+0x173/0x5d0
[Sa Okt 12 09:44:40 2024]  ? exc_page_fault+0x76/0x1b0
[Sa Okt 12 09:44:40 2024]  ? asm_exc_page_fault+0x26/0x30
[Sa Okt 12 09:44:40 2024]  ? btusb_disconnect+0x2c/0x170 [btusb]
[Sa Okt 12 09:44:40 2024]  usb_unbind_interface+0x95/0x2d0 [usbcore]
[Sa Okt 12 09:44:40 2024]  ? preempt_count_add+0x53/0xc0
[Sa Okt 12 09:44:40 2024]  device_release_driver_internal+0x1a7/0x210
[Sa Okt 12 09:44:40 2024]  usb_forced_unbind_intf+0x5d/0xc0 [usbcore]
[Sa Okt 12 09:44:40 2024]  usb_reset_device+0xec/0x2a0 [usbcore]
[Sa Okt 12 09:44:40 2024]  mt7921u_probe+0xab/0x240 [mt7921u]
[Sa Okt 12 09:44:40 2024]  usb_probe_interface+0xe5/0x310 [usbcore]
[Sa Okt 12 09:44:40 2024]  really_probe+0xd3/0x390
[Sa Okt 12 09:44:40 2024]  ? __pfx___driver_attach+0x10/0x10
[Sa Okt 12 09:44:40 2024]  __driver_probe_device+0x78/0x160
[Sa Okt 12 09:44:40 2024]  driver_probe_device+0x1f/0xa0
[Sa Okt 12 09:44:40 2024]  __driver_attach+0xea/0x1e0
[Sa Okt 12 09:44:40 2024]  bus_for_each_dev+0x89/0xe0
[Sa Okt 12 09:44:40 2024]  bus_add_driver+0x14d/0x280
[Sa Okt 12 09:44:40 2024]  driver_register+0x5d/0x120
[Sa Okt 12 09:44:40 2024]  usb_register_driver+0x90/0x170 [usbcore]
[Sa Okt 12 09:44:40 2024]  ? __pfx_mt7921u_driver_init+0x10/0x10 [mt7921u]
[Sa Okt 12 09:44:40 2024]  do_one_initcall+0x58/0x330
[Sa Okt 12 09:44:40 2024]  do_init_module+0x90/0x290
[Sa Okt 12 09:44:40 2024]  init_module_from_file+0x86/0xd0
[Sa Okt 12 09:44:40 2024]  idempotent_init_module+0x120/0x320
[Sa Okt 12 09:44:40 2024]  __x64_sys_finit_module+0x5e/0xc0
[Sa Okt 12 09:44:40 2024]  do_syscall_64+0x84/0x1b0
[... 11 lines cut ...]
[Sa Okt 12 09:44:40 2024] Code: 08 89 e8 5b 5d c3 66 2e 0f 1f 84 00 00 00 00 00 90 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d b7 06 0d 00 f7 d8 64 89 01 48
[Sa Okt 12 09:44:40 2024] RSP: 002b:00007ffe3ce2a7b8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[... cut ...]
[Sa Okt 12 09:44:40 2024] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[Sa Okt 12 09:44:40 2024] R13: ffffffffc114b278 R14: 0000000000000080 R15: ffff88a4aec1bc50
[Sa Okt 12 09:44:40 2024] FS:  00007ff3f193a8c0(0000) GS:ffff88a7eff80000(0000) knlGS:0000000000000000
[Sa Okt 12 09:44:40 2024] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Sa Okt 12 09:44:40 2024] CR2: 00000000000000c8 CR3: 0000000195bd0000 CR4: 0000000000350ef0
[Sa Okt 12 09:44:40 2024] note: (udev-worker)[37771] exited with irqs disabled

whitslack, 2024-10-12:
@Simon566: I think you need to open a new issue because what you're seeing is unrelated to the hung transmit queues. Personally, I would advise downgrading to an LTS kernel version because clearly 6.10 and 6.11 have major stability problems in their mt76 drivers.

Krobix, 2024-10-16:
Hello,

Sorry, this question may be slightly unrelated. I am using A mt7921u USB wifi adapter as an AP and it has the issue described. I was wondering if there was any consistent way, preferably using bash commands, to check if this crash has happened, as even when it does happen pinging and every other method I am aware of returns like normal. I would like to know this because I was hoping to set up a script that can check if the crash has happened every few minutes, and if t has, can automatically reset the USB and restart hostapd.

whitslack, 2024-10-16:
@Krobix: If pinging is working for you, then you aren't currently experiencing the hang. When the driver hangs, no packets are transmitted (including beacon frames), as the hardware transmit queues are stalled.

You can detect that the hang is occurring by examining the `xmit-queues` file for the interface in debugfs. If you see a queue with a large (100+) backlog that never diminishes over successive checks of that file, then you can conclude that the queues are stalled.

Theoretically, although I haven't tried this, you could also check for a hang by performing any action that requires a command message to be sent to the MCU in the chip. If the action hangs (and eventually times out), then the chip is hanging. For example, you might try setting the transmit power, as that's an action that requires sending a command to the MCU but otherwise wouldn't affect the operation of the interface. However, it is unknown whether the transmit power of these chips can currently be set via the mac80211 interface since the driver returns a nonsensical value for the current txpower.

Krobix, 2024-10-16:
Hello,

Here are some Dmesg logs: (not entire logs, just some pertaining to the USB adapter) : https://pastebin.com/swEQx8Sv
Here is the hostapd conf i use for the USB adapter : https://pastebin.com/BbZHER8s
Here is the fastfetch output on the AP: https://pastebin.com/ZNrjis59

The particular USB adapter that I am using is An EDUP Wifi 6E AX3000.

Some notes about my particular error:
I notice the error usually because the AP disappears. Before I used a service to run this AP I was running it manually, and attempting to stop the AP via an interrupt (CTRL+C) would hang for a long while (like a minute or two?) before hostapd would stop and output some vague message about being unable to reach the device and possibly getting stuck in a loop (I do not have this message anymore). I have used two separate USB ports with my adapter, the one on the back of the machine (USB 2) and the one on the front (USB 3). It can last longer in the USB 3 port, running for about two days before crashing, where as in the usb 2 port it would crash maybe twice a day. I am almost certain that the USB port is the difference here because even after functioning on the USB 3 port for a long time, I tried plugging it back into the USB 2 out of curiosity and it crashed again within an hour. 

Apart from that, the crashes seemed almost totally random, and AFAIK did not correlate to the amount of data being transferred (i.e. i was able to download a Steam game at 100 mbps without it crashing, however some time after that download with seemingly very little traffic it did crash and I had to unplug the device plug it back in, and restart.) Just restarting hostapd has never worked for me, and I have always had to unplug the adapter and plug it back in in order for it to work again. Even after it has crashed, though, I am able to ping the adapter from any LAN device just fine.

morrownr, 2024-10-16:
@Krobix 

Good info. Let's get some things up to date before continuing to investigate:

1. Build Time: 20230117170942 - the firmware for your adapter is about 3-4 versions behind. Recommend you up the files:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

2. Kernel: Linux 6.1.0-25-amd64 - recommend you run it up to kernel 6.6. Do you know how to upgrade the kernel in Debian?

3. I see some things in hostapd.conf that cause me to scratch my head but first tell me what you are trying to do:

What channel width do you want? 20, 40, 80
Do you see more congestion on the low channels (36) or high channels (149)?
I see that WiFi 6 is turned on, is that what you want?

I see some missing lines but I can't really see what you are trying to do so we need to discuss this. hostapd can be a complicated thing.

Krobix, 2024-10-16:
Hello,

I did install the new firmware files as recommended, and will reboot in a bit.

As for the  kernel version, as far as I can tell there is no prebuilt kernel version > 6.1.0 and <= 6.6. I am very hesitant to attempt any sort of manual kernel compilation because this is a machine that i use as a headless server, and I do not have any monitor right now that I could use in case I somehow mess up my install. Admittedly I am more accustomed to using Arch then I am Debian so I could be missing something.

As for my configuration, it is a heavily modified version of the example configuration on [the gentoo wiki](https://wiki.gentoo.org/wiki/Hostapd). I actually have two hostapd instances- one running at 2.4ghz on the machine's built-in wifi and one running at 5ghz on the usb adapter. The configuration for the internal wifi is very similar with a few changes, and it functions completely fine, though as a result of the hardware being cheap (and not supporting 5ghz in AP mode) it is quite slow, which is the main reason for me using the USB wifi adapter.

As for the channels being used, I will admit I am not knowledgeable enough to check which channels have more congestion or make good decisions about which channel to use or make decisions about channel width. I essentially searched which channels were allowed for 5ghz and arbitrarily picked one. I originally wanted to use the ACS survey function to automatically select a channel, but as far as I could tell it simply is not supported on either wifi hardware.

On the wifi 6- I did enable this intentionally, mainly because my desire is to maximize speed. As I understand it this trades some of the range of the wifi for higher speeds, though if there are any other negative consequences then I am unaware of them.

morrownr, 2024-10-16:
> as far as I can tell there is no prebuilt kernel version

Yes there is. Start synaptic and search for `linux-image`. Debian compiles kernels for nearly everything. probably one in there for an IBM XT (kidding), if you need to ask which one to install, then ask.

> As for my configuration, it is a heavily modified version of the example configuration on...

The one of the gentoo site is probably not a good example for your mt7921au based adapter. Try this:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.conf

Are you running a bridged or routed setup? If routed, you might need to change one line.

> As for the channels being used, I will admit I am not knowledgeable enough to check which channels have more congestion...

We can work on that. Do you have an Android phone?

> On the wifi 6- I did enable this intentionally, mainly because my desire is to maximize speed. As I understand it this trades some of the range of the wifi for higher speeds.

You are still on band 2 (5 GHz) so likely little difference in range. WiFi 6 is a little faster than WiFi 5 so we will work at fine tuning WiFi 6 on 5 GHz. The example hostapd.conf that I posted the link for above does exactly that.

morrownr, 2024-10-17:
How about giving this one a try: (I assume secure boot is off)

linux-image-6.9.10+bpo-amd64-unsigned - Linux 6.9 for 64-bit PCs

> (My setup is routed)

Then the following line is not what you want:

bridge=br0

Somebody else that is used to setting up routed APs needs to jump in and help here.

> I do have an Android phone

I use an Android app called `WiFi Analyzer`. You can see what is going on. It won't measure exact congestion as that would take something expensive but it will give you an idea of the APs operating on the various channels which can give you and idea how much traffic there is. Failing that, once we get a solid setup, simply test various channels to see which is fastest.

whitslack, 2024-10-17:
> > (My setup is routed)
> 
> Then the following line is not what you want:
> 
> bridge=br0

That's not necessarily true. You might have a router with one upstream interface (say, WAN) and two downstream interfaces (say, LAN and WLAN, with the LAN interface being a wired Ethernet adapter connected to an Ethernet switch). The two downstream interfaces would be bridged together to form one Ethernet broadcast domain.

morrownr, 2024-10-18:
> I will let you know if/when I experience the crash again.

Please do so.

Keep in mind that the WiFi Gods can be rather difficult at time so you may have to sacrifice a lamb to appease them.

Simon566, 2024-10-19:
i have very strange crashes with 6.11.4, my HW is  stable , its running for years already.

any idea about this No 1:

[   14.484619] ------------[ cut here ]------------
[   14.484623] Unpatched return thunk in use. This should not happen!
[   14.484627] WARNING: CPU: 0 PID: 358 at arch/x86/kernel/cpu/bugs.c:3025 __warn_thunk+0x2c/0x40
[   14.484635] Modules linked in: snd_hda_intel(+) 88XXau(OE+) intel_rapl_common vfat snd_intel_dspcfg crct10dif_pclmul drm_display_helper processor_thermal_wt_req ghash_clmulni_intel sha512_ssse3 snd_intel_sdw_acpi cfg80211 snd_hda_codec sha256_ssse3 fat snd_hda_core ttm processor_thermal_power_floor sha1_ssse3 snd_hwdep snd_pcm aesni_intel drm_kms_helper processor_thermal_mbox mei_pxp mei_hdcp gf128mul crypto_simd snd_timer intel_pmc_core int340x_thermal_zone mei_me snd cryptd evdev mei i2c_algo_bit pcspkr intel_soc_dts_iosf joydev rfkill intel_vsec soundcore nfsd iTCO_wdt intel_pmc_bxt ee1004 iTCO_vendor_support watchdog pmt_telemetry intel_cstate wmi_bmof ftdi_sio usbserial acpi_tad pmt_class sg button acpi_pad it87(OE) hwmon_vid emc2103 coretemp 8021q auth_rpcgss drm garp nfs_acl lockd stp grace sunrpc mrp fuse dm_mod llc efi_pstore loop configfs efivarfs ip_tables x_tables autofs4 btrfs xor raid6_pq libcrc32c hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid sd_mod xhci_pci xhci_hcd ahci libahci libata
[   14.484702]  usbcore nvme r8169 scsi_mod crc32_pclmul intel_lpss_pci intel_lpss idma64 realtek mdio_devres usb_common nvme_core libphy crc32c_intel i2c_i801 scsi_common i2c_smbus video wmi
[   14.484716] CPU: 0 UID: 0 PID: 358 Comm: (udev-worker) Tainted: G           OE      6.11.4-dirty #2
[   14.484720] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[   14.484721] Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
[   14.484723] RIP: 0010:__warn_thunk+0x2c/0x40
[   14.484727] Code: 1f 00 0f 1f 44 00 00 80 3d a7 2c db 01 00 74 07 31 ff c3 cc cc cc cc 48 c7 c7 b0 0b cd 8a c6 05 90 2c db 01 01 e8 34 bd 06 00 <0f> 0b 31 ff c3 cc cc cc cc 66 2e 0f 1f 84 00 00 00 00 00 90 90 90
[   14.484729] RSP: 0018:ffffa5ff406ffa78 EFLAGS: 00010246
[   14.484731] RAX: 0000000000000000 RBX: ffffffffc1101010 RCX: 0000000000000000
[   14.484733] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[   14.484734] RBP: ffffa5ff406ffac8 R08: 0000000000000000 R09: 0000000000000000
[   14.484736] R10: 0000000000000000 R11: 0000000000000000 R12: ffffa5ff406ffb80
[   14.484737] R13: ffffa5ff406ffae8 R14: ffff8ea654232700 R15: ffff8ea5f89bdd88
[   14.484739] FS:  00007f6e3831a8c0(0000) GS:ffff8ea92fe00000(0000) knlGS:0000000000000000
[   14.484741] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   14.484743] CR2: 0000557354f75f30 CR3: 000000012d38a000 CR4: 0000000000350ef0
[   14.484745] Call Trace:
[   14.484747]  <TASK>
[   14.484750]  ? __warn+0x80/0x120
[   14.484754]  ? __warn_thunk+0x2c/0x40
[   14.484757]  ? report_bug+0x1a2/0x1d0
[   14.484760]  ? prb_read_valid+0x1b/0x30
[   14.484764]  ? handle_bug+0x42/0x80
[   14.484768]  ? exc_invalid_op+0x17/0x80
[   14.484772]  ? asm_exc_invalid_op+0x1a/0x20
[   14.484777]  ? ____versions+0x5a0080277f50/0x5a0080277f50 [88XXau]
[   14.484840]  ? __warn_thunk+0x2c/0x40
[   14.484843]  ? __warn_thunk+0x2c/0x40
[   14.484845]  warn_thunk_thunk+0x1a/0x30
[   14.484852]  init_module+0x14/0xff0 [88XXau]
[   14.484887]  ? do_one_initcall+0x58/0x330
[   14.484893]  ? do_init_module+0x90/0x290
[   14.484897]  ? init_module_from_file+0x86/0xd0
[   14.484900]  ? idempotent_init_module+0x120/0x320
[   14.484903]  ? __x64_sys_finit_module+0x5e/0xc0
[   14.484905]  ? do_syscall_64+0x84/0x1b0
[   14.484908]  ? vm_mmap_pgoff+0x13f/0x1d0
[   14.484912]  ? fpregs_assert_state_consistent+0x2b/0x70
[   14.484917]  ? syscall_exit_to_user_mode+0x4d/0x210
[   14.484920]  ? do_syscall_64+0x90/0x1b0
[   14.484923]  ? kmem_cache_free+0x34f/0x490
[   14.484927]  ? do_sys_openat2+0x9d/0xe0
[   14.484930]  ? __seccomp_filter+0x331/0x500
[   14.484935]  ? fpregs_assert_state_consistent+0x2b/0x70
[   14.484938]  ? syscall_exit_to_user_mode+0x4d/0x210
[   14.484941]  ? do_syscall_64+0x90/0x1b0
[   14.484944]  ? clear_bhb_loop+0x25/0x80
[   14.484946]  ? clear_bhb_loop+0x25/0x80
[   14.484949]  ? entry_SYSCALL_64_after_hwframe+0x76/0x7e
[   14.484952]  </TASK>
[   14.484954] ---[ end trace 0000000000000000 ]---

i have a security hardenend kernel , but no modifications / patches except mt7921u stuff.....

[56 more replies]