morrownr/USB-WiFi #411: Comfast

InfiniteBSOD, 2024-03-22:
Hello,

First off. Great job with this site.
Really a godsend to know which adapters are supported under Linux.
Looking for a Wifi 6E (2.4GHz+5GHz+6GHz) adapter to use with Kali for mapping wifi hotspots in any of the three (2.4GHz + 5GHz + 6GHz) mentioned frequencies.

However:

Tried to buy a Comcast CF-953AX listed under the category of [tri-band-usb-wifi-adapters-that-are-supported-with-linux-in-kernel-drivers](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md#tri-band-usb-wifi-adapters-that-are-supported-with-linux-in-kernel-drivers) from AliExpress a couple of days ago (2024-03-18-ish) and received the response:
"CF-953AX is discontinued, I shipped CF-952AX instead"

Now CF-952AX isn't listed in the list of [usb-wifi-adapters-that-are-supported-with-linux-in-kernel-drivers](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md#usb-wifi-adapters-that-are-supported-with-linux-in-kernel-drivers) but alluded to having the same mt7921au chipset as the CF-953AX even though it is stated to but the CF-953AX (instead of the CF-952AX - implicit).

I went to Alibaba instead and ordered a Comfast CF-953AX and received the same reply there:
"Out of stock, replaced with CF-952AX"
and I asked:
"so it is still wifi 6e (2.4GHz + 5GHz + 6GHz) and has the mt7921au chipset?"
and received the reply:
"MT7921AU is normal WiFi6 1800Mbps, not WiFi6E"
and was referred to the CF-972AX which has the RTL8832CU chipset which according to this might in the future (not now?)  [out-of-kernel_drivers](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapter_out-of-kernel_drivers_for_Linux.md#chipset---rtl8832cu--rtl8852cu---axe3000---usb-3).

The product spec's the seller linked to (see attached) states that the CF-972AX has a 6GHz band while the CF-953AX and CF-952AX states having a 2.4GHz and 5GHz frequency not mentioning 6GHz.
[image]
[image]

The driver on Comfast's [website ](https://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=30&id=653) for Linux seem to indicate that the CF-952AX V2 is using a Realtek chipset.

TL;DR:
I am very confused.
I got a Comfast CF-952AX on the way; will it have the MT7921AU chipset?
Is it Wifi 6E (6GHz)?

Thank you in advance - InfiniteBSOD

lilkidsuave, 2024-03-22:
> Hello,
> 
> First off. Great job with this site. Really a godsend to know which adapters are supported under Linux. Looking for a Wifi 6E (2.4GHz+5GHz+6GHz) adapter to use with Kali for mapping wifi hotspots in any of the three (2.4GHz + 5GHz + 6GHz) mentioned frequencies.
> 
> However:
> 
> Tried to buy a Comcast CF-953AX listed under the category of [tri-band-usb-wifi-adapters-that-are-supported-with-linux-in-kernel-drivers](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md#tri-band-usb-wifi-adapters-that-are-supported-with-linux-in-kernel-drivers) from AliExpress a couple of days ago (2024-03-18-ish) and received the response: "CF-953AX is discontinued, I shipped CF-952AX instead"
> 
> Now CF-952AX isn't listed in the list of [usb-wifi-adapters-that-are-supported-with-linux-in-kernel-drivers](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md#usb-wifi-adapters-that-are-supported-with-linux-in-kernel-drivers) but alluded to having the same mt7921au chipset as the CF-953AX even though it is stated to but the CF-953AX (instead of the CF-952AX - implicit).
> 
> I went to Alibaba instead and ordered a Comfast CF-953AX and received the same reply there: "Out of stock, replaced with CF-952AX" and I asked: "so it is still wifi 6e (2.4GHz + 5GHz + 6GHz) and has the mt7921au chipset?" and received the reply: "MT7921AU is normal WiFi6 1800Mbps, not WiFi6E" and was referred to the CF-972AX which has the RTL8832CU chipset which according to this might in the future (not now?) [out-of-kernel_drivers](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapter_out-of-kernel_drivers_for_Linux.md#chipset---rtl8832cu--rtl8852cu---axe3000---usb-3).
> 
> The product spec's the seller linked to (see attached) states that the CF-972AX has a 6GHz band while the CF-953AX and CF-952AX states having a 2.4GHz and 5GHz frequency not mentioning 6GHz. ![9e681f3fe8c90145301db82f30a0cec1](https://private-user-images.githubusercontent.com/126592660/316185289-a8e53a6a-f34b-49f8-ada9-16541dcd8893.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTExNTA1MDEsIm5iZiI6MTcxMTE1MDIwMSwicGF0aCI6Ii8xMjY1OTI2NjAvMzE2MTg1Mjg5LWE4ZTUzYTZhLWYzNGItNDlmOC1hZGE5LTE2NTQxZGNkODg5My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyMlQyMzMwMDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NzBhMDZlNDYyMmU1OTFiYzI5YjZhMjgzMGVjZmE3OWIzZGM1ODNhMGJhNDliMWZlM2VmN2Y5ZmQ4OTBkOTlhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.pSWrp8JmdFJCm6_scA_ICY-4xIeDUQ0onMv75hs-7GU) ![20220623112048427](https://private-user-images.githubusercontent.com/126592660/316185297-bf833b1b-8a11-403a-9c75-195d63ab2110.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTExNTA1MDEsIm5iZiI6MTcxMTE1MDIwMSwicGF0aCI6Ii8xMjY1OTI2NjAvMzE2MTg1Mjk3LWJmODMzYjFiLThhMTEtNDAzYS05Yzc1LTE5NWQ2M2FiMjExMC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyMlQyMzMwMDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NTlhNjhiZjE0Yzk4MGY3YmUzYjBhNGM2NWU2NDRhZWRkN2ZiN2Y1ZDE0MzZlMzc2MGNjNzdiMjU2NTMzNGMwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.uQfUlXJ2S1qmBbbuquV7wV4wXHpSxlfNnUyZ7lSLZUQ)
> 
> The driver on Comfast's [website ](https://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=30&id=653) for Linux seem to indicate that the CF-952AX V2 is using a Realtek chipset.
> 
> TL;DR: I am very confused. I got a Comfast CF-952AX on the way; will it have the MT7921AU chipset? Is it Wifi 6E (6GHz)?
> 
> Thank you in advance - InfiniteBSOD

Best case it's the mediatek. Worst case you need drivers for the 8832cu, which does support wifi 6E technically, but I'm having trouble getting it to work in AP 6ghz.

lilkidsuave, 2024-03-22:
https://github.com/morrownr/USB-WiFi/issues/408#issue-2199294120
The sudo iw list shows 6ghz support.

morrownr, 2024-03-23:
Hi @InfiniteBSOD 

> First off. Great job with this site. Really a godsend to know which adapters are supported under Linux.

Thanks for the kind words. The single biggest problem for Linux users that want to use USB WiFi adapters has been a location that aggregates good quality information. That is job number 1 around here. I get a lot help. This a hard job to do well.

 > Looking for a Wifi 6E (2.4GHz+5GHz+6GHz) adapter to use with Kali for mapping wifi hotspots in any of the three (2.4GHz + 5GHz + 6GHz) mentioned frequencies.

Understand.

> "CF-953AX is discontinued...

Good information to know. I will likely need to remove it from the Plug and Play List very soon.

> "MT7921AU is normal WiFi6 1800Mbps, not WiFi6E"

Welcome to the world of retailers that have no idea what they are talking about. The mt7921au chipset was first used by Samsung in Smart TV's and then it was made available for adapter makers. While adapter makers can turn off some capabilities in the chipsets they use, I do not have confirmation, either with my testing or from users reporting results, that any maker has turned off 6 GHz support in the chipset. Some of the early adapters with this chipset only advertised  AX1800 dual band capability but users have reported that 6 GHz works on Linux. The common theory for this is when the early adapters were made, Windows 10, and I think even Windows 11, did not support 6 GHz so the early Windows drivers may not have had support for 6 GHz. Linux, since the driver went into Linux kernel 5.18, has had 6 GHz support. Therefore, the above statement is wrong. The mt7921au chipset has always been WiFi 6e capable.

> Now CF-952AX isn't listed in the list of...

It is not listed in the Plug and Play List for a very good reason. From the beginning there has been confusion about what chipset it contains. Evidence points to Confast acting like TP-Link and putting out some adapters with mt7921au chips and some with rtl8852/32bu chips... while using the same model number. I just can't list situations like that.

> ...and was referred to the CF-972AX which has the RTL8832CU chipset which according to this might in the future (not now?) out-of-kernel_drivers.

I have an out-of-kernel driver for the rtl8852/32bu chipset up at this site. I had previously tried to get a rtl8852/32au out-of-kernel driver up but finally gave up. These are BAD drivers. The one I do have up is basically managed mode only and for many, it can work but it is a terrible driver. I recommend strongly that Linux users avoid USB WiFi adapters that use Realtek WiFi6(e) chipsets.

> The driver on Comfast's website for Linux seem to indicate that the CF-952AX V2 is using a Realtek chipset.

As mentioned above, who knows?

> I am very confused.

We are here to help. Take a deep breathe. This is not a social media site so some users only stop by every month or two but a lot of very informed users to do stop by. This site gets over 20,000 hits per week so there is an information flow.

A couple of days ago I posted about an EDUP adapter that seems to have potential:

https://github.com/morrownr/USB-WiFi/issues/409

There are excellent adapters for Kali and monitor mode work. I cannot recommend any adapters with Realtek chipsets for use with Kali. The best chipsets/drivers for Kali, in my opinion, are the mt7610u, mt7612u and the mt7921au. The driver the mt7925 is in the Linux kernel but we have no adapters to buy yet. Status of each driver:

mt7610u - supports active monitor mode and no reported problems with VM's. The Alfa ACHM is a star adapter with this chipset and has very long range but it is dual band.

mt7612u - supports active monitor mode but can be challenge with VM's.

mt7921au - the driver currently has a bug with active monitor mode that hopefully is fixed soon. We don't have what I would call a long range adapter with this chipset yet but works well with Kali, just use regular monitor mode and not active monitor mode for now.

What you could do while you learn about the options for 6 GHz and even WiFi 7, once we have adapters, is get a solid dual band adapter like the Alfa ACHM and practice.

Regards,

@morrownr

morrownr, 2024-03-23:
@lilkidsuave 

> The sudo iw list shows 6ghz support.

But that does not mean that it works.

I was going to continue helping you but a couple days ago I got sick. I was also pondering the issue and I wonder how much help I can be without a rtl8852/32cu based adapter.

InfiniteBSOD, 2024-03-23:
> 
> A couple of days ago I posted about an EDUP adapter that seems to have potential:
> 
> #409
> 

Thank you for your reply.

I actually was hunting for this adapter (EP-AX1672) but asked "EDUP Official Store" on AliExpress and received the following reply:
"sorry friend, we don't sell it on AliExpress " (2024-03-14).

It seems to be available on [Amazon DE](https://www.amazon.de/-/en/Wireless-802-11AX-Tri-Band-Antennas-Compatible-black/dp/B0CVVWNSH2) but does not ship to my country (Sweden).

The link to [Amazon US](https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CVVWNSH2) allows me to buy it without any VAT but with a shipping cost that is around 50% of the price of the device itself.

Device: €27.75
Shipping: €12.74
Total: €40.49 which is around 462 SEK.

Thinking it might be better to wait for a compatible WiFi 7 USB-adapter instead since it is the latest standard and routers having WiFi 7 are starting to come out in the wild here in Sweden....

fakemanhk, 2024-04-03:
Oh discontinued??? A couple weeks ago when I went to HK and I saw a shop selling CF-953AX (apparently they have ~100pcs around) at around USD20, probably I should get more.

Badger101, 2024-05-08:
First off thanks to @morrownr and all the people who have put this site together - it has been a great help and source of information!

I've ordered two CF-953AX in a space of a couple of months, and the last one I had the same issue and they sent me a CF-952AX because the CF-953AX was out of stock - I promptly sent it back. I don't think they are discontinued, they just want the sale.
I ordered another one from a different listing and received the correct one.

You might just need to look around.

morrownr, 2024-05-08:
@Badger101 

Thanks to you and the others for providing up to date information. For now I have removed the CF-953AX from the Plug and Play List due to concerns as outlined here. I did replace it with an Edup adapter so that there are some adapters to choose from. I can put the CF-953AX back in the list if I can get more confirmations that it should not be problematic for users to order and get the right product.

The Plug and Play List is about more than just adapters, it is also about providing links the are dependable for users. Can you provide the link where you had success? Sellers that mess around and change out products without talking to customers do not sit well with me.

Badger101, 2024-05-13:
>   Can you provide the link where you had success?

I don't think there is a reliable source for these adapters that I have found which is a shame - they have worked for me really well as long as you remove the Bluetooth driver.
Even where I purchased them don't have them anymore..

Badger101, 2024-05-17:
Here looks like a stable (for now) source:

https://www.sunsky-online.com/p/EDA003280301A/COMFAST-CF-953AX-1800Mbps-USB-3.0-WiFi6-Wireless-Network-Card-with-Antenna-Black-.htm

I've purchased one from here and can confirm it was correct.

morrownr, 2026-02-08:
Hi @alx-xlx 

> I am just wondering if this wifi usb adapter will work on raspberrypi 5 OS arm64, has anyone tried it ?

Comfast USB WiFi adapters tend to be multi-state (Windows driver onboard which is problematic) and this is particularly true when Realtek chips are used.

> However it says Testing has shown that adapter with the rtl8952/32cu chips are particularly problematic.

It is true. You do not want an adapter with this chip unless you first confirm it is single-state.

The Plug and Play List shows several adapters that work really well with Linux. Here is a link to one of them so that you can read the reviews:

https://www.amazon.com/dp/B0F6MY7H62

Note that the ad says it is for Linux and it is. Brostrend actually has a Linux support tech. I have one of these Brostrend adapters and use it with the RasPiOS on my RasPi4B. The Plug and Play List also has adapters by other makers like Panda, Edup, Alfa and more.

alx-xlx, 2026-02-10:
That looks like a very good adapter, but the issue is its regional availability. I can't get it and most of the other adapters
I think I will have to wait for more driver support