morrownr/USB-WiFi #412: MT7612u (Netgear A6210) 5ghz Issues

localboast, 2024-03-24:
Hello,

I'm at a loss here, scoured the internet and cannot find a solution. I'm on Arch linux (6.8.1-arch1-1). For the life of me, I cannot get this adapter to connect to 5ghz. My network is Ubiquiti AP > DFS 149 > 80mhz. Any ideas at all would be helpful, I have installed wireless-regdb and enabled only US in /etc/conf.d/wireless-regdom as I am in the US.

<details>
<summary>dmesg | grep -i firmware</summary>
mt76x2u 1-5:1.0: Firmware Version: 0.0.00
</details>

<details>
<summary>dmesg</summary>
[   28.580004] wlan0: 80 MHz not supported, disabling VHT
[   29.446140] wlan0: authenticate with e0:63:da:23:f3:9e (local address=b0:39:56:98:b5:fa)
[   29.446147] wlan0: send auth to e0:63:da:23:f3:9e (try 1/3)
[   29.449037] wlan0: authenticated
[   29.450251] wlan0: associate with e0:63:da:23:f3:9e (try 1/3)
[   29.454556] wlan0: RX AssocResp from e0:63:da:23:f3:9e (capab=0x1411 status=0 aid=11)
[   29.647086] wlan0: associated
[   29.647125] wlan0: Limiting TX power to 27 (27 - 0) dBm as advertised by e0:63:da:23:f3:9e
[   31.983366] warning: `ThreadPoolForeg' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[   69.379554] wlan0: deauthenticating from e0:63:da:23:f3:9e by local choice (Reason: 3=DEAUTH_LEAVING)
[   97.485482] wlan0: Trigger new scan to find an IBSS to join
[  111.401932] wlan0: Creating new IBSS network, BSSID 46:49:cb:35:c0:ad
[  176.986299] wlan0: authenticate with e2:63:da:33:f3:9f (local address=b0:39:56:98:b5:fa)
[  176.986306] wlan0: send auth to e2:63:da:33:f3:9f (try 1/3)
[  176.988530] wlan0: authenticated
[  176.990780] wlan0: associate with e2:63:da:33:f3:9f (try 1/3)
[  176.993726] wlan0: RX AssocResp from e2:63:da:33:f3:9f (capab=0x1111 status=0 aid=17)
[  177.187262] wlan0: associated
[  246.818630] wlan0: deauthenticating from e2:63:da:33:f3:9f by local choice (Reason: 3=DEAUTH_LEAVING)
[  359.692008] wlan0: 80 MHz not supported, disabling VHT
[  360.555686] wlan0: authenticate with e0:63:da:23:f3:9e (local address=b0:39:56:98:b5:fa)
[  360.555692] wlan0: send auth to e0:63:da:23:f3:9e (try 1/3)
[  360.558595] wlan0: authenticated
[  360.560752] wlan0: associate with e0:63:da:23:f3:9e (try 1/3)
[  360.573104] wlan0: RX AssocResp from e0:63:da:23:f3:9e (capab=0x1411 status=0 aid=11)
[  360.762652] wlan0: associated
[  360.875611] wlan0: Limiting TX power to 27 (27 - 0) dBm as advertised by e0:63:da:23:f3:9e
</details>

<details>
<summary>iw reg get</summary>
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
</details>

<details>
<summary>iw phy phy0 channels</summary>
Band 1:
        * 2412 MHz [1] 
          Maximum TX power: 18.0 dBm
          Channel widths: 20MHz HT40+
        * 2417 MHz [2] 
          Maximum TX power: 18.0 dBm
          Channel widths: 20MHz HT40+
        * 2422 MHz [3] 
          Maximum TX power: 18.0 dBm
          Channel widths: 20MHz HT40+
        * 2427 MHz [4] 
[... 398 lines cut ...]
                 * IBSS: 0x40 0xb0 0xc0 0xd0
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ VHT_IBSS ]: VHT-IBSS
                * [ RRM ]: RRM
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ POWERED_ADDR_CHANGE ]: can change MAC address while up
</details>

bjlockie, 2024-03-24:
Try a different channel.

morrownr, 2024-03-24:
Hi @localboast 

> [   28.580004] wlan0: 80 MHz not supported, disabling VHT

Interesting. The A6210 supports 80 MHz channel width and VHT. Not sure what is going on with that.

> My network is Ubiquiti AP > DFS 149 > 80mhz. 

Channel 149 is not a DFS channel in the US.

52 - 144 are the DFS channels in the US. I use DFS channel 116, 160 MHz width on my primary router (I am in the US also).

> Try a different channel.

I agree with @bjlockie . Try changing your channel to 36 for testing, then report.

FYI: I have an A6210 and can test it if needed.

lilkidsuave, 2024-03-24:
> Hi @localboast
>
>
> Interesting. The A6210 supports 80 MHz channel width and VHT. Not sure what is going on with that.
>
>
> Channel 149 is not a DFS channel in the US.
>
> 52 - 144 are the DFS channels in [...]

Using channel 36 for debugging is good advice. However channel 149 should work oob though, the extended 5ghz doesn't start until channel 166. If a usb adapter is 802.11ac it should support it.There is the possibility that 80mhz is not enabled in the driver properties, so you may have to enable it.

lilkidsuave, 2024-03-24:
Usually driver files for editing properties are in 
/usr/lib/modprobe.d/
If its not there, you may need to create the file and figure out what varable enables 80mhz.

morrownr, 2024-03-25:
@lilkidsuave 

>  However channel 149 should work oob though, the extended 5ghz doesn't start until channel 166. If a usb adapter is 802.11ac it should support it.

You are correct, I had to chuckle a little as you used the word `should` a couple of times there. WiFi is complicated and should does not always work out. 

> Usually driver files for editing properties are in /usr/lib/modprobe.d/

The mt7612u driver is a modern mac80211, standards compliant, in-kernel driver. It only has one module parameter that is unlikely to be the cause of the problem in this case. Older Realtek out-of-kernel, non standards compliant drivers did use a lot of module parameters.

The mt7612u driver is very stable and time tested. I suspect the cause could be a setting or bug in the AP/router. I think testing other channels is a good place to start. I have multiple adapters that use the mt7921u driver. It should just work but the variability in the code to support wifi can cause issues to come up.

lilkidsuave, 2024-03-25:
> @lilkidsuave
>
>
> You are correct, I had to chuckle a little as you used the word `should` a couple of times there. WiFi is complicated and should does not always work out.
>
>
> The mt7612u driver is a [...]

That's a good point about the AP settings.
@localboast 2 things:
Can it connect to 2.4?
If not, is the AP WPA3? The chipset doesn't seem to be able to support WPA3, so it may not connect to a access point that uses it.

localboast, 2024-03-25:
Wow, thanks for the immediate responses! Apologize about stating 149 was DFS, my mistake. I'm hoping to not have to switch channels as I guess out of principal I shouldn't have to for a single malfunctioning device haha. I'll try taking a look at the properties as mentioned. My network has tons of different wireless and IoT clients all without issue for literal years, so I don't think it's anything on my Unifi network's end. Anywho I'll experiment with what everyone has suggested here tomorrow and report back. I have seen whispers in other threads that folks only got 5ghz working on channels in the 30s and lower with this adapter.

morrownr, 2024-03-25:
> I have seen whispers in other threads that folks only got 5ghz working on channels in the 30s and lower with this adapter.

I had time so I connected my A6210. This message is brought to you by my A6210 connected to channel 116 DFS, 80 MHz channel width and... using WPA3. My distro is Debian 12. I guess maybe tomorrow I can change my primary wifi router to channel 149 and see what happens.

@lilkidsuave 

I've done a lot of testing on WPA3. My testing has shown that the chipset has little to do with WPA3 support. The driver and some parts of the distro do have a lot to do with it. I have very old adapters with WiFi 4 chipsets that support WPA3. All in-kernel drivers that I have tested support WPA3 but it would not hurt to turn it off in the wifi router just as a test..

morrownr, 2024-03-25:
Couldn't help myself:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.236 port 53584 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  35.2 MBytes   296 Mbits/sec    0    650 KBytes       
[  5]   1.00-2.00   sec  32.5 MBytes   273 Mbits/sec    0    762 KBytes       
[  5]   2.00-3.00   sec  35.0 MBytes   294 Mbits/sec    0    840 KBytes       
[  5]   3.00-4.00   sec  31.2 MBytes   262 Mbits/sec    0    884 KBytes       
[  5]   4.00-5.00   sec  32.5 MBytes   273 Mbits/sec    0    884 KBytes       
[  5]   5.00-6.00   sec  31.2 MBytes   262 Mbits/sec    0    884 KBytes       
[  5]   6.00-7.00   sec  35.0 MBytes   294 Mbits/sec    0    884 KBytes       
[  5]   7.00-8.00   sec  33.8 MBytes   283 Mbits/sec    0    884 KBytes       
[  5]   8.00-9.00   sec  33.8 MBytes   283 Mbits/sec    0    884 KBytes       
[  5]   9.00-10.00  sec  33.8 MBytes   283 Mbits/sec    0    884 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   334 MBytes   280 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   331 MBytes   278 Mbits/sec                  receiver

iperf Done.
```

My Alfa ACM with the same chipset can do over 400 Mbps but it has a tx power of 20 whereas Netgear decided the A6210 only needed tx power of 18 plus the ACM probably has antennas that help. I have 2 walls to the AP so this is not bad. The A6210 is a cool little adapter, especially for going on the road.

lilkidsuave, 2024-03-25:
> I had time so I connected my A6210. This message is brought to you by my A6210 connected to channel 116 DFS, 80 MHz channel width and... using WPA3. My distro is Debian 12. I guess maybe tomorrow I [...]

I've haven't heard of drivers backporting WPA3 to older/less capable devices, thats pretty cool.

localboast, 2024-03-25:
Alright I'm back. Moved to channel 36 on 40mhz and it's good! To be honest should have looked at this (in Ubiquiti Wifiman app) a while ago as I'm better here anyway, and do not need 80mhz throughput. Less congested here. @morrownr thank you for testing this along side me, verdict is still out on this adapter supporting 149+. One thing I'll mention is that in Arch Network settings, I am in Infastructure mode and it's grabbing 5ghz now. However if I change to Adhoc, my only Band option is A for 5ghz. If i select this and channel 36 (which I'm currently on), it will not connect. I have to remain in Infastructure and let it "auto" grab 5ghz. Your results may vary here I imagine, based on distro, no idea.

Looking at the image in the first comment on this post, perhaps this adapter doesn't like UNII-3? I'll wait until someone else tries their luck inthat range, but it did not work for me. At least we know 80mhz and WPA3 is supported as well.

https://community.ui.com/questions/No-Line-of-Sight-and-channel-widths/8d296fa7-77a7-4bc4-ae97-2944daea1a6e

morrownr, 2024-03-25:
@localboast 

>  verdict is still out on this adapter supporting 149+

I changed my main wifi router to channel 149 this morning. It worked fine and with 80 MHz range. It is highly likely that your A6210 is identical to mine. There is little difference in capability between different adapters if they use the same chipset. Makers can select and use their own antennas, amps and various materials to make the physical adapter but the chipset is what gives the capability so not much variance. In this case, given the same maker and model, the adapters are likely identical. 

So, while your jury may still be deliberating, mine is back and their verdict was that the A6210 is innocent of all charges related to its failure to connect to channel 149. If you had another wifi router to test with, that would be great. If you don't, do you have a spare RasPi and another usb wifi adapter?

What you could do is try channel 157 with both 40 and 80 widths in the router to see what you get. I've seen a lot of variety in the coding of routers . If your router supports DFS, you could try channels 100 or 116 to see if either of both is good to use in your area and if all of your 5 GHz clients work with DFS, you might as well use it as there will likely be no congestion.

localboast, 2024-03-25:
That's great news! You can close this issue, as it sounds like it's something to do with my setup specifically. I may experiment some more in the future, but for now I'm happy. Many thanks for everyone's input!

dubhater, 2024-03-26:
> I've haven't heard of drivers backporting WPA3 to older/less capable devices, thats pretty cool. Advertising for the 7612u mentions WPA2 and before, but not WPA3, my bad.

If the driver uses mac80211, which is the case here, it gets WPA3 support automatically, even if the chip itself is not capable.

lilkidsuave, 2024-03-26:
__

> If the driver uses mac80211, which is the case here, it gets WPA3 support automatically, even if the chip itself is not capable.

How does the driver work around the previous incapability of using WPA3?

dubhater, 2024-03-26:
If the chip doesn't support WPA3, the driver tells this to mac80211, which will handle the encryption in software.

The advertising may be based on what the manufacturer's driver can do. I think Mediatek produces out-of-kernel drivers too, like Realtek, in addition to developing the in-kernel drivers.