morrownr/USB-WiFi #414: rtl8821cu AP mode traces in phydm_ccx.c with ht_capab [HT40-][HT40+] enabled

meskilla, 2024-03-29:
rtl8821cu AP mode traces in phydm_ccx.c with ht_capab [HT40-][HT40+] enabled

This is happening with a 4 antenna bearing unbranded RTL8811 usb adapter I bought on aliexpress [[1](https://aliexpress.com/item/1005006449878467.html)]

> [   12.329125] ================================================================================
[   12.339091] UBSAN: invalid-load in /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/phydm/phydm_ccx.c:1375:2
[   12.350236] load of value 255 is not a valid value for type '_Bool'
[   12.358116] CPU: 2 PID: 802 Comm: RTW_CMD_THREAD Tainted: G           OE     5.15.0-1049-raspi #52-Ubuntu
[   12.358135] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   12.358140] Call trace:
[   12.358143]  dump_backtrace+0x0/0x200
[   12.358161]  show_stack+0x20/0x30
[   12.358168]  dump_stack_lvl+0x8c/0xb8
[   12.358181]  dump_stack+0x18/0x34
[   12.358189]  ubsan_epilogue+0x10/0x44
[   12.358196]  __ubsan_handle_load_invalid_value+0x80/0x90
[   12.358209]  phydm_nhm_mntr_set+0x168/0x16c [8821cu]
[   12.358522]  phydm_env_mntr_trigger+0x54/0x174 [8821cu]
[   12.358740]  rtw_acs_trigger+0x98/0x120 [8821cu]
[   12.358961]  sitesurvey_cmd_hdl+0xe6c/0x10e0 [8821cu]
[   12.359171]  rtw_cmd_thread+0x4b4/0x690 [8821cu]
[   12.359383]  kthread+0x128/0x134
[   12.359394]  ret_from_fork+0x10/0x20
[   12.359405] ================================================================================

other than this trace the device has run stable in managed mode using my laptop (Ubuntu 22.04 + Windows 11)

The embeded (so called driverless) Windows 11 driver did run stable in AP mode, but I didn't know how to add [HT40-][HT40+] improvments, so the speed I benched with Windows 11 and a Lenovo Tablet was saturated at  90Mbps and not 200Mbps as I'm able to bench in a linux AP mode enviroment. (Benched using [speedtest.net](https://www.speedtest.net/))

My linux AP mode setup is a Raspberry Pi 4B (4GB) running Ubuntu Jammy arm64 kernel 5.15, the USB 3.0 port is
offerning 5.2V to the wifi adaper plugged into a usb 3.0 cable.

my config setup in ubuntu is the following:

```
grep -Ev ^# /etc/modprobe.d/8821cu.conf 
blacklist rtw88_8821cu
options 8821cu rtw_drv_log_level=1 rtw_wireless_mode=95 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_dfs_region_domain=1 rtw_country_code=CH

grep -Ev "^#|^$" /etc/hostapd/hostapd.conf
interface=wlan0
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
ssid=MYWIFISSID
channel=36
hw_mode=a
ieee80211n=1
ieee80211ac=1
require_vht=1
ieee80211h=1
wpa=2
wpa_passphrase=MYWIFIPASSWORD
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
rsn_pairwise=CCMP
beacon_int=100
auth_algs=1
wmm_enabled=1
ieee80211d=1
country_code=CH
ht_capab=[HT40-][HT40+][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT][MAX-A-MPDU-LEN-EXP7]

grep -Ev "^#|^$" /etc/modprobe.d/cfg80211.conf 
options cfg80211 ieee80211_regdom=CH

grep -Ev "^#|^$" /etc/default/crda 
REGDOMAIN=CH

#lsusb after it switched from CD-ROM mode to NIC mode
Bus 001 Device 007: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

cat /etc/udev/rules.d/40-usb_modeswitch.rules
ACTION!="add|change", GOTO="modeswitch_rules_end"
ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="usb_modeswitch '/%k'"
LABEL="modeswitch_rules_end"

cat /etc/usb_modeswitch.d/0bda\:1a2b 
DefaultVendor="0x0bda"
DefaultProduct="0x1a2b"
TargetVendor="0x0bda"
TargetProduct="0xc811"
StandardEject=1

you might have to add "usb-storage.quirks=0bda:1a2b:i" to cmdline.txt so the switching works with newer kernels (>5.xx)
```

I'd like to test dual band setup, this is why  rtw_wireless_mode=95 is set and ready, but I'll have to figure out how to add a matching hostapd.conf to test this.

attached is a picture of this beauty. Without the kernel trace this should be fun to use even in a 5GHz only setup.
Price was 5$ + 2$ of shipping to Europe.
[image]

bjlockie, 2024-03-29:
Can you upgrade the OS to Ubuntu 23.10?

meskilla, 2024-03-29:
@bjlockie: I have Ubuntu Noble 24.04 on standby, that will ship kernel 6.8, is this kernel worth testing against rtw88_8821cu ? Another option is kali-rolling. On Kali I did prefer using 8821cu because rtw88_8821cu dropped the connection far to often. I did test managed mode only on these two systems only. I haven't installed ubuntu noble on the pi4 to test ap mode.

morrownr, 2024-03-29:
@meskilla 

> This is happening with a 4 antenna bearing unbranded RTL8811 usb adapter...

This is a new one on me. The ad seems to indicate it is an AC1200 adapter supporting USB3.

Are you using the driver we have up here at this site in the following repo?

https://github.com/morrownr/8821cu-20210916

> is this kernel worth testing against rtw88_8821cu ?

I'd recommend not using rtw88's rtw88_8821cu driver yet. I am working with a team that is improving that driver and patches are going in.

Now, the more I think about this, the more this seems wrong. Can I see the results of the following commands:

$ lsmod
$ lsusb

This is driving a Corvette with a 1.2 liter engine. I really need to confirm exactly what we have here.

meskilla, 2024-03-30:
hello nick,

lsusb's with the device plugged directly into a usb 3.0 port (no hub in between, no cable)

lsusb before modeswitch (CD-Mode)
> Bus 008 Device 003: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)

lsusb after modeswitch ( NIC Mode)
> Bus 008 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

Why does lsusb in CD Mode indicate this being a RTL8188**G**U when after modeswitch it works like RTL8188**C**U ?

Should I try one of the few RTL8188GU repositories and see if one of those drivers work?

The plastic casing might be tricky to open but it might be worth to examine the actual chip I bought ... :D 

lsub -vv again after modeswitching

> Bus 008 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               2.00
>   bDeviceClass            0 
>   bDeviceSubClass         0 
>   bDeviceProtocol         0 
>   bMaxPacketSize0        64
>   idVendor           0x0bda Realtek Semiconductor Corp.
>   idProduct          0xc811 
>   bcdDevice            2.00
>   iManufacturer           1 Realtek
>   iProduct                2 802.11ac NIC
>   iSerial                 3 123456
>   bNumConfigurations      1
>   Configuration Descriptor:
>     bLength                 9
>     bDescriptorType         2
>     wTotalLength       0x0035
>     bNumInterfaces          1
>     bConfigurationValue     1
>     iConfiguration          0 
>     bmAttributes         0xa0
>       (Bus Powered)
>       Remote Wakeup
>     MaxPower              500mA
>     Interface Descriptor:
>       bLength                 9
>       bDescriptorType         4
>       bInterfaceNumber        0
>       bAlternateSetting       0
>       bNumEndpoints           5
>       bInterfaceClass       255 Vendor Specific Class
>       bInterfaceSubClass    255 Vendor Specific Subclass
>       bInterfaceProtocol    255 Vendor Specific Protocol
>       iInterface              2 (error)
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x84  EP 4 IN
>         bmAttributes            2
>           Transfer Type            Bulk
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0200  1x 512 bytes
>         bInterval               0
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x05  EP 5 OUT
>         bmAttributes            2
>           Transfer Type            Bulk
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0200  1x 512 bytes
>         bInterval               0
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x06  EP 6 OUT
>         bmAttributes            2
>           Transfer Type            Bulk
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0200  1x 512 bytes
>         bInterval               0
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x87  EP 7 IN
>         bmAttributes            3
>           Transfer Type            Interrupt
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0040  1x 64 bytes
>         bInterval               3
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x08  EP 8 OUT
>         bmAttributes            2
>           Transfer Type            Bulk
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0200  1x 512 bytes
>         bInterval               0
> 
> 

I did use your 8821cu-20210916 repository, but if you can recommend other github drivers to try I will test them against ubuntu 22.04 or 24.04 (I'm about to install this to my rpi 4 anyway, it will be released in short time)

Feel free to move this ticket to the 8821cu-20210916 bugtracker. I somehow ended up opening this on the wrong githup site, because I keep reading up on your usb-wifi documentation.

meskilla, 2024-03-30:
Update: device was easier to open than I thought, so to confirm this is in fact RTL8811C**U** I took a few fotos:

[image]
[image]
[image]
[image]
[image]
[image]

Puya seems to be a flash manufacturer in Taiwan, so this should be the chip containing the windows driver.

And to confirm one more issue you had with this being usb 3.0 I think you are correct: according to [Realteks rtl8811cu publifications](https://www.realtek.com/en/products/communications-network-ics/item/rtl8811cu) the host interface is a  "USB2.0 for WLAN and BT controller". So wifi speed will never reach more than 480MBps, I reached about half of it with 200MBps. Don't see the point why to advertise this as USB 3.0 when in fact it is not ... it must be a marketing thing these days ... ;-)

morrownr, 2024-03-31:
> Why does lsusb in CD Mode indicate this being a RTL8188GU when after modeswitch it works like RTL8188CU ?

I guess that is just what was assigned at some point for some unknown reason. The vid/pid you are seeing is common for multi-state adapters.

> Should I try one of the few RTL8188GU repositories and see if one of those drivers work?

No. Won't work.

> I did use your 8821cu-20210916 repository, but if you can recommend other github drivers to try I will test them...

You are welcome to try others but this is probably the most modern, well maintained driver for this chipset that is avalable.

> bcdUSB 2.00

It looks like the adapter is in usb2 mode. My opinion is that the makers of this adapter are trying to fool us into thinking it is usb3 capable when it is not. Even if it was, it would not help as usb2 is capable of more speed than this AC600 chipset can take advantage of. I also think 2 of those antennas are fake. Can you see if the wires only go to 2 of the antennas?

> rtw_wireless_mode=95

I would take that out of the `options` line in 8821cu.conf.

The 8821cu driver works well in AP mode. This is an interesting adapter. You might as well use a usb2 port. We may have to do some further testing if the above does not get you the results you are looking for.

meskilla, 2024-04-01:
> I also think 2 of those antennas are fake. Can you see if the wires only go to 2 of the antennas?
>
It is difficult to spot in the photo but each of them go into one of the 4 antennas.
> >
> > rtw_wireless_mode=95
> 
> I would take that out of the `options` line in 8821cu.conf.
> 
Will do, any other options I should try instead to max out performance?  I'm glad you added comment suggestions in the conf files, otherwise people will be literally lost by the number off switches you can set to tweak most of these realtek drivers.

> The 8821cu driver works well in AP mode. This is an interesting adapter. You might as well use a usb2 port. We may have to do some further testing if the above does not get you the results you are looking for.
>
I read your iperf benches, but you are benching these devices in managed mode against a Zyxel AP aren' t you? Can you tell me how fast your AC600 8821cu devices rate in AP mode when you add HT40 and VHT optimizations to hostapd?

Can you get past the 200MBit and does your kernel not trace? If not what kernel/os did you use?

dubhater, 2024-10-01:
Saw another one, also "RTL8811" inside:

[image]

(It's a scam, RTL8811 is AC600 no matter how many antennas you attach.)

morrownr, 2024-10-05:
@dubhater 

There is a lot of bullshit going on in the wifi adapter market. There are a lot of cheap white box products that are designed to appeal to what they think will appeal to consumers but the products are a scam. If there are problems then the seller disappears.

I do my best to ensure the Plug and Play List contains good quality adapters for Linux users.